emsesp/EMS-ESP #252: telnet : info : outside temperature wrongly reported

moustic999, 2019-12-05:
in version 1.9.4b21
the temperature reported by telnet is wrong, the negative sign is missing.
however the temperature reported by MQTT (boiler_data) is correct.

info

EMS-ESP system stats:
  System logging set to None
  LED: on, Listen mode: off
  Boiler: enabled, Thermostat: enabled, Solar Module: disabled, Mixing Module: disabled
  Shower Timer: disabled, Shower Alert: disabled

EMS Bus stats:
  Bus is connected, protocol: Buderus
  Rx: # successful read requests=4765, # CRC errors=0
  Tx: Last poll=2.400 seconds ago, # successful write requests=1

Boiler stats:
  Boiler: Nefit Topline/Buderus GB162 (DeviceID:0x08 ProductID:115 Version:05.09)
  Hot tap water: off
  Central heating: active
  Warm Water activated: on
  Warm Water circulation pump available: off
  Warm Water comfort setting: Hot
  Warm Water selected temperature: 60 C
  Warm Water desired temperature: 70 C
  Warm Water current temperature: 59.9 C
  Warm Water current tap water flow: 0.0 l/min
  Warm Water # starts: 2693 times
  Warm Water active time: 33 days 18 hours 31 minutes
  Warm Water 3-way valve: off
  Selected flow temperature: 65 C
  Current flow temperature: 65.0 C
  Return temperature: 57.2 C
  Gas: on
  Boiler pump: on
  Fan: on
  Ignition: off
  Circulation pump: off
  Burner selected max power: 100 %
  Burner current power: 24 %
  Flame current: 16.4 uA
  System pressure: 1.5 bar
  System service code: -H (200)
  Heating temperature setting on the boiler: 90 C
  Boiler circuit pump modulation max power: 100 %
  Boiler circuit pump modulation min power: 30 %
  Outside temperature: 3.5 C
  Boiler temperature: 59.9 C
  Pump modulation: 72 %
  Burner # starts: 72752 times
  Total burner operating time: 310 days 21 hours 39 minutes
  Total heat operating time: 277 days 3 hours 8 minutes
  Total UBA working time: 1452 days 6 hours 11 minutes

Thermostat stats:
  Thermostat: RC300/RC310/Moduline 3000/Bosch CW400/W-B Sense II (DeviceID:0x10 ProductID:158 Version:11.08)
  Thermostat time is 08:39:04 05/12/2019
  Heating Circuit 1
   Current room temperature: ? C
   Setpoint room temperature: 21.0 C
   Mode is set to auto
   Day Mode is set to night

MQTT  : 
{"wWComfort":"Hot","wWSelTemp":60,"wWDesiredTemp":70,"selFlowTemp":65,"selBurnPow":100,"curBurnPow":24,"pumpMod":72,"wWCircPump":0,"outdoorTemp":-3.5,"wWCurTmp":59.9,"wWCurFlow":0,"curFlowTemp":65,"retTemp":57.2,"switchTemp":0,"sysPress":1.5,"boilTemp":59.9,"wWActivated":"on","wWOnetime":"off","burnGas":"on","flameCurr":15.9,"heatPmp":"on","fanWork":"on","ignWork":"off","wWCirc":"off","heating_temp":90,"pump_mod_max":100,"pump_mod_min":30,"wWHeat":"off","wWStarts":2693,"wWWorkM":48631,"UBAuptime":2091255,"burnStarts":72752,"burnWorkMin":447702,"heatWorkMin":399071,"ServiceCode":"-H","ServiceCodeNumber":200}

proddy, 2019-12-05:
thanks for reporting. I found the bug and do a blind fix, but can't check until I get home tonight.

moustic999, 2019-12-05:
seems not good, got the latest dev code :

  Outside temperature: ?21.00.01 C

on mqtt : "outdoorTemp":1.4

moustic999, 2019-12-05:
and now there is also another strange stuff :  

   Setpoint room temperature: ?21.0 C

there is a ? which were not there before ( see log in first message)

moustic999, 2019-12-07:
and  Flame current: ?21.015.6 uA has also the same formatting issue

proddy, 2019-12-07:
Ill look into it. Thanks for pointing it out

proddy, 2019-12-08:
do you get the flame current issue with the latest 1.9.4b22? I can't reproduce.

moustic999, 2019-12-08:
seems ok on 1.9.4b23 :

just need to wait for a negative value on the outdoor temp ;-) 

Boiler stats:
  Boiler: Nefit Topline/Buderus GB162 (DeviceID:0x08 ProductID:115 Version:05.09)
  Hot tap water: off
  Central heating: off
  Warm Water activated: on
  Warm Water circulation pump available: off
  Warm Water comfort setting: Hot
  Warm Water selected temperature: 60 C
  Warm Water desired temperature: 70 C
  Warm Water current temperature: 60.6 C
  Warm Water current tap water flow: 0.0 l/min
  Warm Water # starts: 2697 times
  Warm Water active time: 33 days 20 hours 41 minutes
  Warm Water 3-way valve: off
  Selected flow temperature: 47 C
  Current flow temperature: 34.7 C
  Return temperature: 34.7 C
  Gas: off
  Boiler pump: on
  Fan: off
  Ignition: off
  Circulation pump: off
  Burner selected max power: 100 %
  Burner current power: 0 %
  Flame current: 0.0 uA
  System pressure: 1.3 bar
  System service code: 0A (202)
  Heating temperature setting on the boiler: 90 C
  Boiler circuit pump modulation max power: 100 %
  Boiler circuit pump modulation min power: 30 %
  Outside temperature: 9.1 C
  Boiler temperature: 60.6 C
  Pump modulation: 45 %
  Burner # starts: 73032 times
  Total burner operating time: 312 days 6 hours 40 minutes
  Total heat operating time: 278 days 9 hours 59 minutes
  Total UBA working time: 1455 days 7 hours 18 minutes

Thermostat stats:
  Thermostat: RC300/RC310/Moduline 3000/Bosch CW400/W-B Sense II (DeviceID:0x10 ProductID:158 Version:11.08)
  Thermostat time is 09:46:40 08/12/2019
  Heating Circuit 1
   Current room temperature: ? C
   Setpoint room temperature: 21.0 C
   Mode is set to auto
   Day Mode is set to day

proddy, 2019-12-08:
ok, if it still happens please reopen.

closed 2019-12-08 by proddy