emsesp/EMS-ESP #253: too many MQTT packet sent

moustic999, 2019-12-05:
version 1.9.4b21

when setting mqtt publish_time to 0, there are too many packet issued. 

when the thermostat data is received from cyclic data, everything is sent, even boiler and water/heating data. also as the packet is long, it is splitted into 2 packet by EMS, triggering twice all mqtt .

when boiler data is sent, all mqtt are also sent, where only boiler should be expected.

see log for thermostat data :  

(08:02:45.184) Thermostat -> All, type 0x0167, telegram: 10 00 FF 00 01 67 00 00 (CRC=AB) #data=2
(08:02:45.420) Thermostat -> All, type 0x01A5, telegram: 10 00 FF 00 01 A5 80 00 03 2A 41 00 2A 2A 02 0C 05 01 03 02 0C 00 D3 00 00 11 01 03 FF FF 00 (CRC=FE) #data=25
Publishing boiler data via MQTT
Publishing hot water and heating states via MQTT
Publishing thermostat data via MQTT
(08:02:45.829) Thermostat -> All, type 0x01A5, telegram: 10 00 FF 19 01 A5 08 04 00 00 00 00 FF 64 4B 00 3C 01 FF 01 02 (CRC=57) #data=15
Publishing boiler data via MQTT
Publishing hot water and heating states via MQTT
Publishing thermostat data via MQTT
(08:02:46.027) Thermostat -> All, type 0x021D, telegram: 10 00 FF 00 02 1D 00 00 08 07 (CRC=06) #data=4

8 MQTT  published where only 1 should be expected ( thermostat_data)

proddy, 2019-12-05:
this is by design. Default behaviour is to only send when data has changed (boiler, thermostat, solar module, heating on/off etc). Then someone wanted the data to be always sent which I why I implemented 'publish_time 0' which is exactly what its doing. Indeed is flooding MQTT and shouldn't be used and not advised.

proddy, 2019-12-05:
what is the bug exactly? If you can suggest what the logic should be I can take a look, otherwise I'm closing this as "not fixing".

moustic999, 2019-12-05:
Data should be sent when ems bus publish it, that's the idea ;-)
today when 01A5 comes mqtt send boiler/thermostat/hotwater/heating. What I would expect is only a mqtt thermostat data. then when boiler data comes, I would expect to have only mqtt boiler data; 

Not sure it is correctly explained ?

proddy, 2019-12-05:
Yes I understand. I’ll think of how to program this.

proddy, 2019-12-06:
made a change in b22. MQTT will be sent when receiving new telegrams, split by module (boiler, thermostat, solar, heatpump etc). The publish_time value is used to force a publish at an interval (which is 2 minutes). publish_time of 0 will turn this off. So in theory the best results are to have publish_time set to 120 and you should be seeing MQTT come in after each telegram. If its too much let me know.

closed 2020-01-26 by proddy