emsesp/EMS-ESP #254: unknown? (DeviceID:0x09 ProductID:95 Version:23.12)

ekivel, 2019-12-07:
Hello,

Looks like onboard controller not added yet:

These device IDs are on the EMS Bus: 0x08 0x09 0x10
and 3 were recognized by EMS-ESP as:
 Bosch Condens 2500/Buderus Logamax GB062/Junkers  (DeviceID:0x08 ProductID:95 Version:23.12)
 unknown? (DeviceID:0x09 ProductID:95 Version:23.12)
 RC300/RC310/Moduline 3000/Bosch CW400 (DeviceID:0x10 ProductID:158 Version:33.04)

The boiler is Bosch Condens 3000 W with thermostat CW400

Let me know if you need more info.
thx

proddy, 2019-12-07:
I'll add, but it shouldn't effect any of the readings. The controller does really do anything. Do you get the correct boiler and thermostat details coming in?

ekivel, 2019-12-08:
Few values are missing: boiler and return temp, system pressure.
Also I don’t have mqtt Topic:home/ems-esp/thermostat_cmd in ems-esp device so I am not able to control the thermostat. 
Rest of the values and functions are OK.

proddy, 2019-12-08:
some of these values are tricky to find - its been reported before in a few issues here on this GitHub project. Can you see them on the boiler's display?

proddy, 2019-12-08:
and that missing controller was already added to release 1.9.4. Which version are you using?

closed 2019-12-08 by proddy

ekivel, 2019-12-08:
Thanks. Upgraded to latest version, it looks ok now. Also I am able to set room temperature now via mqtt.
I am afraid the missing values are not available on this boiler, I don’t see them in the thermostat menu.

proddy, 2019-12-08:
ok, thanks for checking. in future versions I'll add support to prevent those missing values from being shown.