emsesp/EMS-ESP #255: New feature - nightly restart?

Gert05, 2019-12-08:
Thanks again for a great piece of software.

I noticed that ems-esp sometimes "freezes". MQTT seems disconnected (web page shoows disconnect), Domoticz cannot read status nor can I set the themostat's temperature. A simple reboot does the trick until the next freeze.

A nightly restart (eg at 01.00 every 2nd or 3rd night) may resolve these issues. Can this be implemented?

proddy, 2019-12-08:
Which version are you on? I was hoping I had fixed the majority of the connectivity problems in the last builds.

Gert05, 2019-12-08:
That would be 1.9.2 running on a wemos  d1 mini pro.

proddy, 2019-12-09:
Try 1.9.4.

Gert05, 2019-12-09:
That opens up a whole new can of worms. I have had problems with dev branch in the past so I stayed away from it until now...

I have recently upgraded my PC and started from scratch. Did as suggested in the wiki (Win 10, PlatformIO).
I cannot get PlatformIO to install the libraries for me, so I have manually copied libraries into the lib subdirectory.

"Build" ends with a few errors, starting with the following:
Compiling .pio\build\debug\libdde\ESP8266WiFi\BearSSLHelpers.cpp.o
src\TelnetSpy.cpp: In member function 'void TelnetSpy::setDebugOutput(bool)':
src\TelnetSpy.cpp:442:48: error: invalid conversion from 'void*' to 'fp_putc_t {aka void (*)(char)}' [-fpermissive]
         os_install_putc1((void *)TelnetSpy_putc); // Set system printing (os_printf) to TelnetSpy

proddy, 2019-12-09:
Those build errors should be gone if you use the dev branch (in pio click bottom left where it says master). If not let me know!

If you prefer, just take the latest [firmware](https://github.com/proddy/EMS-ESP/releases/tag/travis-dev-build) and follow the instructions on the wiki to upload the binary.

glitter-ball, 2019-12-15:
I, too, have been having problems recently. I'm using 1.9.3 feeding Home Assistant on a RPi3. Three times, now, I've found the HA web server unresponsive. When I restart HA, it shows EMS-ESP as offline and there is no incoming MQTT data when I listen.

When I telnet into ems-esp it shows MQTT as disconnected, despite saying that the data is being published. I end up doing a 'restart' on ems-esp and it all recovers.

proddy, 2019-12-15:
1.9.4 fixes the stability issues (hopefully). I've been testing and fine-tuning for a month now. I'll push it to master soon.

closed 2019-12-23 by Gert05