emsesp/EMS-ESP #256: Setpoint temp is 0.0 C when day mode is set to night

ekivel, 2019-12-09:
Hello,

Here is the info from the console of the device:
Thermostat stats:
  Thermostat: RC300/RC310/Moduline 3000/Bosch CW400/W-B Sense II (DeviceID:0x10 ProductID:158 Version:33.04)
  Thermostat time is 13:23:14 09/12/2019
  Heating Circuit 1
   Current room temperature: 21.9 C
   Setpoint room temperature: 0.0 C
   Mode is set to auto
   Day Mode is set to night

I am ready to help with the debug, just let me know pls which telegrams to capture.

Here are the current logs with log t:

(13:40:07) Thermostat -> UBAMaster, type 0x35, telegram: 90 08 35 00 11 01 (CRC=A9) #data=2
(13:40:07) Thermostat -> UBAMaster, type 0x35, telegram: 90 08 35 03 37 (CRC=74) #data=1
(13:40:10) Thermostat -> All, type 0x01A5, telegram: 90 00 FF 0F 01 A5 01 91 (CRC=FD) #data=2
Publishing thermostat data via MQTT
(13:40:10) Thermostat -> All, type 0x01A5, telegram: 90 00 FF 0D 01 A5 00 4F (CRC=01) #data=2
Publishing thermostat data via MQTT
(13:40:10) Thermostat -> All, type 0x01A5, telegram: 90 00 FF 08 01 A5 00 4F (CRC=51) #data=2
Publishing thermostat data via MQTT
Fetching data from EMS devices
Publishing boiler data via MQTT
Publishing hot water and heating states via MQTT
Publishing thermostat data via MQTT
(13:40:13) Thermostat -> Me, type 0x01A5, telegram: 90 0B FF 00 01 A5 00 DB 23 00 00 00 00 2E 00 4F 01 01 03 00 4F 01 91 00 00 11 01 00 08 8E 00 (CRC=A8) #data=25
Publishing thermostat data via MQTT
(13:40:14) Thermostat -> Me, telegram: 90 0B FF 00 01 A6 (CRC=18)
(13:40:14) Thermostat -> Me, telegram: 90 0B FF 00 01 A7 (CRC=19)
(13:40:14) Thermostat -> Me, telegram: 90 0B FF 00 01 A8 (CRC=16)
(13:40:14) Thermostat -> Me, type 0x06, telegram: 90 0B 06 00 13 0C 0D 09 29 07 00 00 10 FF 00 (CRC=95) #data=11

thx

proddy, 2019-12-09:
ah, not this issue again! It'll be great if you can help. What we need is the data from the 0x01A5 telegram. You can track these using `log w 1a5`. Then using this [table](https://github.com/proddy/EMS-ESP/wiki/RC3xx-Thermostats#type-01a5) we're looking for the temperature which will be stored is a single hex value. Check your thermostat for the real setpoint temp and then see if you can find it somewhere in that block of 25 bytes.

For example I see you received `90 0B FF 00 01 A5 00 DB 23 00 00 00 00 2E 00 4F 01 01 03 00 4F 01 91 00 00 11 01 00 08 8E 0`

The data block is therefore  ` 00 DB 23 00 00 00 00 2E 00 4F 01 01 03 00 4F 01 91 00 00 11 01 00 08 8E 0` and my code reads the setpoint temperature value from positions 4 and/or 7 depending. I don't look at the modes which I think is the problem. So in your example this would be values 0 and 0, hence why you are seeing 0 degrees. The real values could be 0x23 =  17.5 (night temp) or 0x2E = 23.0 (auto mode temp) - can you check and I'll modify the code?

ekivel, 2019-12-09:
Looks like thermostat is sending  00 in byte 4 and 6 when predefined night and day temps are used in auto mode (it is called temperature levels in the manual). I don't see the required data in any of the telegrams.
Now I configured the thermostat in another way: not to use the temp levels but freely adjustable temp values it is solved this issue.
Thanks anyway.

proddy, 2019-12-10:
the correct method I think is for EMS-ESP to detect which mode its in and then extract the data from the telegram depending on the mode type. So in your example when day mode is night it should take the data at position 3 (like 0x23 in the example above). Can you check so I can implement it so that it works for everyone?

ekivel, 2019-12-10:
I played with settings yesterday, changed night temp multiple times, but 0x23 did not change.

moustic999, 2019-12-11:
@ekivel  Can you tell me which brand/model is your thermostat ? Also which modes are really available on it ( I expect auto and manual ).
Regarding the auto mode, how did you configure your calendar ? does it use absolute or relative temp ? 
do you switch between mode or between temp ? What are the mode you have ( eco, comfort1, comfort2, comfort3 , .. ) not all should be there...

What is the behavior when being in Eco mode ? in the heater setting you can choose that heating is stopped when mode is eco ? is it the case in your heater ? that could explain ...

which heater is it ? heat pump , gas boiler ? 

the 01A5 message does not provide the mode configured and their temperature setting.
only current mode and next mode can be retrieved, and the current setpoint temp.

in you case the message tells that you are in eco, that next setpoint will be confort2. but no temperature setpoint and the byte 2 ( value 23 ) is quite strange, each time it is 2x... we get nothing in the current setpoint ...

moustic999, 2019-12-11:
Can you try to send this : 
`send 0B 90 F7 00 0F FF 01 A5`
and provide me the output returned by your thermostat.

This is a byte mapping representing which value are really used by your thermostat for the message 01A5 .

ekivel, 2019-12-11:
It is a Bosch CW400. Two modes are manual and auto. In auto mode you can specify heating temp (day) and setback temp (night) it is so called temperature levels and it is changes automatically based on schedule. I need to enter absolute temp values.
During setback period no set temperature is reported via the bus (see the logs sent earlier). During heating all parameters are there as in the wiki.
The boiler is a gas boiler (Bosch Condens 3000W).

moustic999, 2019-12-11:
in this document (sorry french) which is the profesional installation guide, at page 22 they explain what are the different mode for "type d'abaissement" (lowering type) 
depending on what is set for you, you can have or not the value filled in the message. So can you check how is it configured for you

moustic999, 2019-12-11:
the explanation extracted from installation guide in french : 

> Si le chauffage doit être arrêté pendant les phases d’abaissement (la protection antigel reste active), régler dans le menu

principal Chauff. -- Réglages de la température -- Abaissement -- Arrêt (mode arrêt, le réglage du type d’abaissement n’est plus pris en compte dans le comportement du
régulateur).

glitter-ball, 2019-12-15:
I have a W-B Sense II and have also seen this happen. I'm running my system with discrete temperature settings for each period rather than three fixed temperatures : frost, eco and comfort. I assumed this was why the system was reporting zero overnight, when the temperature is actually set at 15ºC.

Happy to send some logs if that helps.

tbr, 2020-01-03:
On my Buderus RC300, I also have this issue.

proddy, 2020-01-03:
@moustic999 any idea how to fix this?

glitter-ball, 2020-01-03:
Have looked a bit more at this today. I set the heating down to 15C for a couple of hours while we went out. HA tracked the setpoint change and showed the system in Manual.

Overnight the system stays in Auto (which is reasonable) but HA reports 0C as the setpoint. I’ve a feeling the Sense II controller also reports zero overnight - I’ll try and check it if I remember. This is odd because I’d expect it to report the frost protection setpoint at least. That said, there are a couple of threshold temperature settings which might be causing this - I never fully got my head around them. I’ll read the manual again.

I guess there may be another locations to get this data from. On my system, the full 0x15A telegram from the Sense II spans two messages so there’s a lot of other info. One of the bytes shows the next setpoint temp so I may look and see what it’s transmitting in that.

moustic999, 2020-01-03:
Yes, there is an easy solution. On rc300, there is no day/night temp, just hide the line when rc300 is detected...

-------- Message d'origine --------

glitter-ball, 2020-01-05:
I played with my Sense II with this. There's a setting for setback threshold on the heating menu. If you set a temperature below this, the system appears to switch off and reports a set temperature of zero and status as 'Off'. If you set a temperature above this, the system then reports the set temperature and status as 'Setback'. I remember struggling to get my head around the purpose of this when I set the system up and an e-mail from W-B tech support helped a bit.

proddy, 2020-01-31:
so fix is when using an RC300 and its in day or night mode, just don't show the setpoint temp???

grappa24, 2020-02-01:
same problem with my RC300 type thermostat: 
When setting mode to auto then seltemp goes to "0" (using **1.9.5b23**)

topic "thermostat_data" only has the values "seltemp" and "mode"

proddy, 2020-02-02:
@grappa24 did you ever see a setpoint temp shown in EMS-ESP with any other mode setting?

The problem is that we're reading the values from the 0x1A5 telegram and for some reason they all have 0 values. So depending on the mode I'll need to find the temperatures from some other telegram. If you can provide some example telegrams (like 0x1A5) that would be helpful.

grappa24, 2020-02-03:
with peasure I will support finding a solution for this issue. But since I'm not really experienced in sniffing and telegrams could you please give me a hint how to produce the telegrams you need?

proddy, 2020-02-03:
ok in telnet, `log w 1a5`. This should show all the 0x1A5 traffic. Then change modes on the thermostat and capture the logging making note on when the changes were made.

grappa24, 2020-02-03:
my boiler has two **internal** temp levels, called daytemp and nightemp (Heizen and Absenken in german). In my case daytemp is set to 23, nightemp set to 20. As I see it, daytemp and nighttemp **can't bei set via EMS**.

When the boilers internal clock is on "daytemp", the EMS command "set auto" works fine and sets the seltemp to 23 (in my case):
[telnet log 1a5 set auto daytemp (23).log](https://github.com/proddy/EMS-ESP/files/4148790/telnet.log.1a5.set.auto.daytemp.23.log)

But when the boilers internal clock is on "nighttemp", the EMS command "set auto" **sets seltemp to 0**:
[telnet log 1a5 set auto nighttemp (20).log](https://github.com/proddy/EMS-ESP/files/4148793/telnet.log.1a5.set.auto.nighttemp.20.log)

proddy, 2020-02-03:
I see in "auto nighttemp" file the line `(43:17:02.544) 90 0B FF 00 01 A5 80 00 01 00 3D 00 00 2E 02 6B 01 01 03 02 6B 04 25 00 00 11 01 00 FF FF 00` which has 0 for the seltemp. The value of 20 degrees should be a hex value of 0x28 hidden in telegram but its not there. What I can do is just ignore any temps that are 0 - maybe that'll work?

grappa24, 2020-02-03:
yes, please try this, I will test again, but on wednesday first ...
but beware: the log above was produced with 1.9.4 - sorry
will test with 1.9.5b then ...

proddy, 2020-02-04:
should be the same as 1.9.5

sq4bja, 2020-02-07:
I had the same problem with RC310. I noticed that EMS-ESP changes the way the heating temperature is switched. Before connecting the EMS-ESP, the daytime temperature was 21, the night temperature 20. After connecting, the daytime temperature remained 21, the night temperature 0. After entering the RC310 controller settings, I noticed that the way of selecting the circulation temperature was changed. Earlier indoor temperature was taken into account. Currently outside temperature. This causes the RC310 itself to switch to 0C at night. After changing the operating mode with reduced temperature, EMS-ESP shows the correct temperature set day and night.

proddy, 2020-02-07:
This needs to looking into by someone who has an RC300/RC310 and willing to invest some time looking into the telegrams and adjusting the code. There's not much I can do here, unless someone wants to loan me a thermostat. Just saying.

tbr, 2020-02-07:
I'll try to monitor the values at my place (RC310). Just as a side-note, I scripted HA to not accept 0 as valid temp and just show 17 (or whatever value I use at night time). It's a hack, but for me it's good enough.

grappa24, 2020-02-07:
if someone please could give me a concrete to-do I will monitor my RC300 telegrams, just have time enough :)

proddy, 2020-02-08:
@grappa24 in your examples, does EMS-ESP show the correct mode (e.g. day or night)? If so I think a solution is to not read the setpoint temperature when its in night mode.

proddy, 2020-02-08:
made a change as proposed to ignore the setpoint temp when in night mode

grappa24, 2020-02-08:
> 
> 
> made a change as proposed to ignore the setpoint temp when in night mode

being in nightmode the "set mode auto" command worked and seltemp **stays as set** before (should go to the boilers internal nighttemp :( ), but selflowtemp goes to a strange value of 5 (not correct)

grappa24, 2020-02-08:
> 
> 
> @grappa24 in your examples, does EMS-ESP show the correct mode (e.g. day or night)? If so I think a solution is to not read the setpoint temperature when its in night mode.

as i see it. ems-esp does not show mode day or mode night.
thermostat_data "mode" only shows the values "auto" or "manual" but never "daytemp" or "nightemp"

proddy, 2020-02-08:
> > made a change as proposed to ignore the setpoint temp when in night mode
> 
> being in nightmode the "set mode auto" command worked and seltemp **stays as set** before (but should go to the boilers internal nighttemp) and selflowtemp goes to a strange value of 5 (not correct)

I'm not sure why setting the mode on the thermostat (via EMS-ESP) would change the Boiler's selected flow temperature (selflowtemp). This may be a new bug

proddy, 2020-02-08:
> i think, that the data-model of ems-esp does not have /misses /show the internal daytemp and nightemp of the boiler ???? in my opinion this is necessary for setting the right temp when using the auto-command ...

If you can provide some 0x1A5 example telegrams when the thermostat is in "night mode" and also let me know what the temperature should be, I can see if it's hidden someone in the telegram. Should be possible. See https://emsesp.github.io/docs/#/RC3xx-Thermostats?id=type-01a5

moustic999, 2020-02-08:
There is no day night in Rc300, only auto or manual. Manual replace day and night.

-------- Message d'origine --------

grappa24, 2020-02-08:
> If you can provide some 0x1A5 example telegrams when the thermostat is in "night mode" and also let me know what the temperature should be, I can see if it's hidden someone in the telegram. Should be possible. See https://emsesp.github.io/docs/#/RC3xx-Thermostats?id=type-01a5

proddy, 2020-02-08:
> There is no day night in Rc300, only auto or manual.

@moustic999 I was looking through the RC300 manual earlier and saw it does mention a "night" setting and that you can set two time periods (day & night). Perhaps calling it 'mode' is not the correct naming. When I see 01a5 examples from @grappa24 there is clearly 4 modes: manual, auto, night and comfort. This is also described in my wiki https://emsesp.github.io/docs/#/RC3xx-Thermostats?id=type-01a5.

So guys, I don't have a RC300 and want to make EMS-ESP work for everyone so what is it??? :-)

tbr, 2020-02-08:
Here is some log data of the 0x1A5 telegram.
My home CV switches two circuits to 17C at 23:00 hours.

The MQTT data is ok when the day program is active, and the seltemp goes to 0 (while it should be 17?) at night. I added that in the log.

[log.txt](https://github.com/proddy/EMS-ESP/files/4175389/log.txt)

proddy, 2020-02-08:
thanks @tbr . This is helpful. Hopefully the seltemp 0 problem is gone in the latest dev build. So apart from the 0 does everything else look ok?

Could you also provide a couple of telegrams from 0x01B9 too? This is where some of the other temps and mode settings are stored.

tbr, 2020-02-09:
I will log the 1B9 telegrams too. Now, I'm running 1.9.5b14 (just so that you know). I didn't upgrade a lot lately, because the EMS-ESP is working good enough for me atm.

moustic999, 2020-02-09:
> @moustic999 I was looking through the RC300 manual earlier and saw it does mention a "night" setting and that you can set two time periods (day & night). Perhaps calling it 'mode' is not the correct [...]

ok, that's not exactly working like that.
RC300 has 2 modes : Auto and Manual. You can change between these modes via buttons on the thermostat itself.
When in Auto, the heating is Calendar controlled and can automatically switch between 2 temperature levels : eco and comfort.

RC300 is the evolution of RC35. on RC35, you can choose between 3 modes via buttons : auto, night, day.
when you are in auto, the heating is calendar controlled and can switch automatically between 2  temperature levels : day and night. these are same than via buttons.
The side effect of the rc35 is that when you switch to Day via buttons, and override the temp via the rotary button, the temperature level will also be changed in the calendar. this is bad.
In order to avoid this, RC300 has no night / day button but manual. in Manual mode, if you override the temp, you don't change the temps level for eco/confort.

What do you want to display excatly  ? Most probably I can tell you where to retrieve it.

proddy, 2020-02-09:
Thanks @moustic999 for explaining that. What I need is someone to test EMS-ESP (the latest dev build) on a RC300 and confirm that everything works
 - setting temp
 - setting mode
 - correctly showing the data in both telnet and MQTT

moustic999, 2020-02-09:
OK @proddy 
I can confirm that 
via MQTT setting temp and setting mode works perfectly
via telnet setting temp and setting mode works too
however, when setting temp, I get this : `Setting new thermostat temperature to 22.0 for heating circuit 1 type 0 (0=auto,1=night,2=day,3=holiday)`
whatever the mode is ( auto or manual for rc300) type is always 0. not sure what it means

Regarding showing the data 
in MQTT I have : ` Topic:thermostat_data Payload:{"hc1":{"seltemp":22,"mode":"manual"}}`
this is OK ( I don't have currTemp, it is normal) 
 
in Telnet, as already explained behavior is strange : especially the last line : 
`Thermostat data:
  Thermostat: RC300/RC310/Moduline 3000/Bosch CW400/W-B Sense II (DeviceID:0x10 ProductID:158 Version:11.08)
  Thermostat time is 20:34:16 09/02/2020
  Heating Circuit 1
   Current room temperature: ? C
   Setpoint room temperature: 22.0 C
   Mode is set to manual
   Day Mode is set to day
`
There is no Day Mode in RC300. Not sure what should be shown there ? What is expected there ?
if it is the actual temp level running in auto, it should be either "comfort" or "eco"
comfort is bit equal to day, eco is bit equal to night.

apart from that, I use ems-esp daily since many months so change temp and mode and it works perfectly.
>

moustic999, 2020-02-09:
> I had the same problem with RC310. I noticed that EMS-ESP changes the way the heating temperature is switched. Before connecting the EMS-ESP, the daytime temperature was 21, the night temperature [...]

You are probably right, my thermostat is configured to take only outdoor temp into account and operating mode with reduced temp. I don't have any issue with ems-esp.
I basically decrypted 0x1a5 messages, but always for my thermostat, probably others settings on the thermostat make changes on the message. Could you share you settings ( thermostat internals ) and some log of 0x1a5

sq4bja, 2020-02-09:
After upgrade to Beta30 i can`t see current boiler temperature:
Boiler temperature: ? C
On telnet and MQTT too. I`m using RC310.

grappa24, 2020-02-09:
@sq4bja : where did you get Beta 30?

proddy, 2020-02-09:
@sq4bja thanks for reporting. I thought the boiler temp in that telegram was only for Bosch, but it seems Junkers HT3 too. I'll make a quick fix now and release as b30.

@grappa24 I did a quiet dev release today, under https://github.com/proddy/EMS-ESP/releases/download/travis-dev-build/EMS-ESP-1_9_5b30-d1_mini-noMM.bin

proddy, 2020-02-09:
@moustic999 thanks for checking. I'll make some changes to telnet and remove the day mode confusion. I'm busy rewriting the whole of EMS-ESP logic for v2 in C++17 so not spending too much time on improving things unless they are severe bugs.

martinSezman, 2020-02-09:
Hi. Please find below few commands and debug for RC300/RC310 thermostat.

For HC1 (01B9):
>send 0B 10 FF 00 01 B9 FF - set Auto thermostat mode   
send 0B 10 FF 02 01 B9 24 - set temp 18 for Day (comfort2) in Auto termostat mode 
send 0B 10 FF 04 01 B9 20 - set temp 16 for Night (eco) in Auto termostat mode
send 0B 10 FF 08 01 BA 26 - set temp 19 - temporary temp since next setpoint in calendar ( for Auto termostat mode)

>send 0B 10 FF 00 01 B9 00 - set  Manual thermostat mode
send 0B 10 FF 0A 01 B9 26 - set temp 19 for Manual thermostat mode

It works in the same way for another HCs - 01BA, 01BB, 01BC.

 
Send message to set Auto on HC2: 0B 10 FF 00 01 BA FF  responds:
>(21:54:41) 10 00 FF 15 01 A6 03    - ???
(21:54:44) 10 00 FF 08 01 A6 02 5E - remaining time in minutes until next setpoint temp change 025E=606 (it will be at 8am)
(21:54:45) 10 00 FF 0A 01 A6 01  - 1=mode auto
(21:54:46) 10 00 FF 03 01 A6 1C 14 00 1C   - current target temperature 1C=28/2=14
(21:54:47) 10 00 FF 07 01 A6 1D            - next setpoint temperature 1D=29/2=14.5
(21:54:48) 10 00 FF 15 01 A6 04 06 88 00 04  -???

And set Manual on HC2:
>(22:14:50) Sending raw: 0B 10 FF 00 01 BA 00
(22:14:53) 10 00 FF 08 01 A6 05 A0
(22:14:54) 10 00 FF 0A 01 A6 02             - 2=manual mode
(22:14:55) 10 00 FF 03 01 A6 20 11 00 20    - current target temperature
(22:14:56) 10 00 FF 07 01 A6 20             - next setpoint temperature
(22:14:56) 10 00 FF 15 01 A6 02
(22:14:57) 10 00 FF 19 01 A6 01
(22:15:20) 10 00 FF 0D 01 A6 02 49 01 77
(22:15:22) 10 00 FF 00 01 A6 00 A7 21 20 11 00 20 20 05 A0 02 01 03 02 49 01 77 00 00 11 01 02 06 84 00
(22:15:23) 10 00 FF 19 01 A6 01 04 FF E2 00 00 FF 64 2B 00 3C 01 FF 01 02
(22:15:31) 10 00 FF 04 01 A6 14
(22:15:35) 10 00 FF 1B 01 A6 FF EB
(22:15:36) 10 0B FF 00 01 A6 00 A7 21 20 14 00 20 20 05 A0 02 01 03 02 49 01 77 00 00 11 01 02 06 84 00
(22:16:22) 10 00 FF 0D 01 A6 02 48 01 78

Set Auto again.
>(22:18:47) Sending raw: 0B 10 FF 00 01 BA FF

>(22:18:49) 10 00 FF 0A 01 A6 01 - 1=mode auto
(22:18:50) 10 00 FF 03 01 A6 1C - current target temperature
(22:18:51) 10 00 FF 06 01 A6 1C - current target temperature
(22:18:51) 10 00 FF 07 01 A6 1D - next setpoint temperature
(22:18:52) 10 00 FF 15 01 A6 04 06 84 00 04
(22:18:54) 10 00 FF 08 01 A6 02 46
(22:18:54) 10 00 FF 04 01 A6 11  - taget flow temp 17 ?
(22:18:59) 10 00 FF 1B 01 A6 FF E2

martinSezman, 2020-02-09:
@proddy 1.9.5b30 do not show Setpoint room temperature if thermostat is in Auto mode. there is question mark.
After switch to Manual mode  is OK. But after switch again to Auto, temperature value stay on Manual setpoint.

sq4bja, 2020-02-09:
I found the same in 1.9.5b29 - no setpoint room after autoswitch heating into nighttime/eco (working with lower temperature).
Currently b30 still doesn`t show boiler temperature. 
I every beta version i have a big issues with webcontrol. I can login after log time pressing F5 on and on.

sq4bja, 2020-02-09:
I don`t know, when Hot tap water indicator is working, but now i`m running warm wather heating manualy and indicator still shows off::
Hot tap water: off

martinSezman, 2020-02-09:
I think EMS-ESP should get  Setpoint room temperature from telegram 01A5,01A6..
send 0B 90 FF 03 01 01 A6 - get current Setpoint room temperature
send 0B 90 FF 07 01 01 A6 - get next Setpoint room temperature ( should be hide for manual mode)

I noticed also some  new value in offset 10 of 01A6 telegram if is set temporary temperature (works if thermostat is in Auto mode only) by
>send 0B 10 FF 08 01 BA 27   ( set temporary 19.5)

Request send 0B 90 FF 0A 01 01 A6
>10 00 FF 0A 01 A6 05

bit3: 1-temporary temp ?

moustic999, 2020-02-10:
> @proddy 1.9.5b30 do not show Setpoint room temperature if thermostat is in Auto mode. there is question mark.
> After switch to Manual mode is OK. But after switch again to Auto, temperature value stay on Manual setpoint.

I don't have the same behavior on my system, The setpoint temp is always shown And I use it a lot.
I remain always in auto mode and I increase the setpoint when I need

martinSezman, 2020-02-10:
> I don't have the same behavior on my system, The setpoint temp is always shown And I use it a lot.
I remain always in auto mode and I increase the setpoint when I need

It worked for me when I had b28 or earlier, but for b29 and b30 it don't

moustic999, 2020-02-10:
> It worked for me when I had b28 or earlier, but for b29 and b30 it don't
everything works fine for me in b29, not yet tried b30

proddy, 2020-02-11:
I made a change in b30 that only the setpoint would be read (offset 4) from 0x01A5 when its in day mode. So this is probably why it shows as ? when in Auto. I'm not sure where to find the setpoint when in Auto - could this offset 7?

Can someone help?

e.g.

```
10 00 FF 00 01 A5 80 00 01 30 28 00 30 28 01 54 03 03 01 01 54 02 A8 00 00 11 01 03 FF FF 00
                                    ^
                                    |
                                 offset 7

```

moustic999, 2020-02-11:
01a5 does not contains temp levels. Only current temp is there.... whatever the mode ...
I will check my notes to find the command to get temp levels...

-------- Message d'origine --------

proddy, 2020-02-11:
in 01A5 there's also the next setpoint temp when in auto mode, at offset 8 (data[7] in my code). Maybe this is the one?

https://emsesp.github.io/docs/#/RC3xx-Thermostats?id=type-01a5

moustic999, 2020-02-11:
No, there is a specific way to get calendar and configured temps for each levels, when I will have access to my gateway and my notes I will recheck

-------- Message d'origine --------

https://emsesp.github.io/docs/#/RC3xx-Thermostats?id=type-01a5<https://eur04.safelinks.protection.outlook.com/?url=https%3A%2F%2Femsesp.github.io%2Fdocs%2F%23%2FRC3xx-Thermostats%3Fid%3Dtype-01a5&data=02%7C01%7C%7C190a6c60a2af4bfef10808d7af3149c7%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637170498525403019&sdata=6xbsu%2FDWmiwHiJrZ%2Bkro3d5vEMDZ2pZe1neS08i4%2FTU%3D&reserved=0>

[17 more replies]

closed 2020-02-15 by proddy