emsesp/EMS-ESP #257: Monitorig burner starts…

pulmol, 2019-12-10:
...with node-red (or other systems) is a good idea to get information about the health of your heating system.
One burner start within an hour should be acceptable, less is not bad at all. If using a modulating heating system the burner starts should be minimized. Sometimes you will notice more starts than necessary. This could be caused by a badly adjusted heating curve, an improper pump performance or sometimes even a wrong gas flow adjustment. To observe the number of burner starts, I calculated the daily burner starts, moved this value when the day changed (at midnight) to "StartCntYesterday" and  displayed both daily and yesterday values on my node-red dashboard.
This is accomplished easily by adding a few lines of JavaScript code to a node-red function node like this example:
Name of the function node is „Neuer Tag“ (New day)
`var NewDay = global.get("NewDay");
var StartCntYesterday = global.get("StartCntYesterday");
var d = new Date();
var n = d.getDay();

if (NewDay===n){
    //Same day
}
else {
// New day
    global.set("StartCntYesterday", msg.payload - global.get("NewDayStartCnt"))
    global.set("NewDayStartCnt", msg.payload );
    global.set("NewDay",n);
}
var DayCnt = msg.payload - global.get("NewDayStartCnt");
var msg1 = {payload:DayCnt};
var msg2 = {payload:StartCntYesterday};
return [msg1,msg2];`

[image]

The contents of the function node „burnStarts“ simply is: 

_msg.payload = msg.payload.burnStarts;
return msg;_

glitter-ball, 2019-12-15:
The other poential cause of frequent burner starts is low heating demand vs. boiler output.

I have a 24kW Worcester Bosch Greenstar ErP boiler with weather compensation and the minimum output from this boiler is 30% (7.2kW). In mild weather conditions, the heat demand of my home is less than this, with the result that the boiler has to cycle to match the heating load.

There is no solution to this and, hence, numerous burner starts are to be expected. But it's worth ruling out other issues before accepting this.

higgers, 2019-12-19:
This is a very interesting post.  I have also noticed that my boiler frequently stops and starts or "short cycles", as my plumber friend describes it.

I have a WB 550CDi which has a heating output of 30.6Kw and a minimum burner level of 32% so the minimum heat output is around 10Kw.  It would have to be very cold outside and we'd need to be heating every room in the house for the boiler to stay on constantly.

I've been graphing the heating setpoint, the heating water temperature (flow temp) and the burner output (in percent) in grafana and the graph typically looks like this:

[image]

From the shape of the flow temp graph (in green) it looks to me like the thermometer for the flow temp is in or very close to the heat exchanger.  After the burner has ignited the flow temp increases very rapidly and decreases very rapidly after the burner switches off.  This is with the central heating pump set to it's fastest setting.

We chose a big boiler due to its ability to provide a high flow rate of hot water for filling baths, running showers, etc.  However, without the ability to reduce to the heating output to a sufficiently low proportion of it's maximum it seems it will always short cycle.

glitter-ball, 2019-12-19:
My flow temp looks very similar. The burner always fires at 100% to get it lit reliably. It then quickly throttles back to 30% and then slowly rises to match demand until the flow temp exceeds the setpoint by a margin, when it shuts down.

With hindsight, an 18kW boiler would probably have met the demand OK. My hot water cylinder primary coil won’t do more than about 18kW anyway.

The other annoyance is that the boiler only starts the pump a second before the burner. If it got the water moving 20-30 secs before ignition, there’d be less chance of the flow temp rocketing up.

I could disable the pump energy saving so it runs constantly during heating demand but that has an energy cost...

higgers, 2019-12-19:
A similar thought occurred to me too.  One larger boiler for high hot water flow and a small boiler for heating.  Or maybe a smaller boiler plus a very well insulated hot water tank.  Or maybe I'm just overthinking it. :)

What I didn't mention in my last post is that I found the service manual for my boiler and used it to set the maximum heating burner output to the minimum possible: 32%.  I also extended the anticycle time from the default 3 minutes to 5 minutes.  Together these have reduced the number of starts to a certain degree.

lsw271, 2020-03-17:
@higgers, @glitter-ball , @pulmol did you think about to try change the minimum or max power by telnet? I would like to try this, my issue about that: [https://github.com/proddy/EMS-ESP/issues/349](url). My minimum power is 6,6kw by default and i can't set the lower value in service menu. 
I would't to drastically reduce it, but maybe to 4 kW. I know of a case where despite the 6-24 kW range, the maximum furnace power has been reduced to 15% (3.6 kW), but this is not ems-esp (BUS) but Opentherm.

proddy, 2020-10-26:
closing this.

closed 2020-10-26 by proddy