emsesp/EMS-ESP #258: old bug re-introduced?

LineF, 2019-12-10:
Hi,

since rev. f96e2efe I can't again save values from the Web-UI...
Seems that correction from rev. b8a55a1 has been lost?
Cherry-picking commit b8a55a1 and merging it to latest rev. 1.9.4b24 solves the problem.

Kind regards,
Martin

proddy, 2019-12-10:
I'll take a look. Does it happen with all the settings pages in the web?

LineF, 2019-12-10:
Hmm, only quick tested with pull down menu of time zones and enabling/disabling serial port.
I noticed the error when I wanted to initial setup the wifi settings - these didn't work too.

proddy, 2019-12-10:
ok, should be easy for me to reproduce then.

proddy, 2019-12-11:
looked like the changes were somehow reversed, I re-committed them.

closed 2019-12-11 by proddy