emsesp/EMS-ESP #259: "Write failed" during "boiler flowtemp" command

arturasi, 2019-12-15:
Hi there, I always get "write failed" message in log when I try to change flow temperature:

> Setting boiler flow temperature to 40 C
(17:19:40) Sending write of type 0x1A to 0x09, telegram: 8B 09 1A 00 28 (CRC=70)
(17:19:40) Sending validate of type 0x1A to 0x09, telegram: 8B 89 1A 00 01 (CRC=3D)
-> Write failed. Compared set value 0x28 with received value of 0x34
-> Write didn't work, retrying (1/2)...
(17:19:40) UBAMaster -> All, type 0x07, telegram: 88 00 07 00 0B 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=56) #data=15
(17:19:40) Sending write of type 0x1A to 0x09, telegram: 8B 09 1A 00 28 (CRC=70)
(17:19:41) Sending validate of type 0x1A to 0x09, telegram: 8B 89 1A 00 01 (CRC=3D)
-> Write failed. Compared set value 0x28 with received value of 0x34
-> Write didn't work, retrying (2/2)...
(17:19:41) Sending write of type 0x1A to 0x09, telegram: 8B 09 1A 00 28 (CRC=70)
(17:19:41) Sending validate of type 0x1A to 0x09, telegram: 8B 89 1A 00 01 (CRC=3D)
-> Write failed. Compared set value 0x28 with received value of 0x34
-> Write failed. Giving up, removing from queue

Despite of this error, inside "info" output I see that flow temp is changed ("Selected flow temperature" field):

> EMS-ESP system stats:
  System logging set to Verbose
  LED: on, Listen mode: off
  Boiler: enabled, Thermostat: disabled, Solar Module: disabled, Mixing Module: disabled
  Shower Timer: disabled, Shower Alert: disabled

>EMS Bus stats:
  Bus is connected, protocol: HT3
  Rx: # successful read requests=33, # CRC errors=0
  Tx: Last poll=0.785 seconds ago, # successful write requests=3

>Boiler stats:
  Boiler: Bosch Condens 2500/Buderus Logamax GB062/Junkers Cerapur Top/Worcester Greenstar i/Generic HT3 (DeviceID:0x09 ProductID:95 Vers
  Hot tap water: off
  Central heating: off
  Warm Water activated: on
  Warm Water circulation pump available: off
  Warm Water comfort setting: Hot
  Warm Water selected temperature: 50 C
  Warm Water desired temperature: 75 C
  Warm Water current temperature: 31.6 C
  Warm Water current tap water flow: 0.0 l/min
  Warm Water # starts: 55549 times
  Warm Water active time: 56 days 12 hours 31 minutes
  Warm Water 3-way valve: off
  Selected flow temperature: 40 C
  Current flow temperature: 30.4 C
  Return temperature: ? C
  Gas: off
  Boiler pump: on
  Fan: off
  Ignition: off
  Circulation pump: off
  Burner selected max power: 40 %
  Burner current power: 0 %
  Flame current: -0.1 uA
  System pressure: ? bar
  System service code:  (0)
  Heating temperature setting on the boiler: 47 C
  Boiler circuit pump modulation max power: 100 %
  Boiler circuit pump modulation min power: 10 %
  Boiler temperature: ? C
  Pump modulation: 100 %
  Burner # starts: 392001 times
  Total burner operating time: 372 days 0 hours 4 minutes
  Total heat operating time: 315 days 11 hours 33 minutes
  Total UBA working time: 1804 days 4 hours 28 minutes

But after few minutes this field changes back to value set on boiler itself ("Heating temperature setting on the boiler") value:

> EMS-ESP system stats:
  System logging set to Verbose
  LED: on, Listen mode: off
  Boiler: enabled, Thermostat: disabled, Solar Module: disabled, Mixing Module: disabled
  Shower Timer: disabled, Shower Alert: disabled

>EMS Bus stats:
  Bus is connected, protocol: HT3
  Rx: # successful read requests=58, # CRC errors=0
  Tx: Last poll=1.216 seconds ago, # successful write requests=3

>Boiler stats:
  Boiler: Bosch Condens 2500/Buderus Logamax GB062/Junkers Cerapur Top/Worcester Greenstar i/Generic HT3 (DeviceID:0x09 ProductID:95 Vers
  Hot tap water: off
  Central heating: off
  Warm Water activated: on
  Warm Water circulation pump available: off
  Warm Water comfort setting: Hot
  Warm Water selected temperature: 50 C
  Warm Water desired temperature: 75 C
  Warm Water current temperature: 31.2 C
  Warm Water current tap water flow: 0.0 l/min
  Warm Water # starts: 55549 times
  Warm Water active time: 56 days 12 hours 31 minutes
  Warm Water 3-way valve: off
  Selected flow temperature: 47 C
  Current flow temperature: 29.3 C
  Return temperature: ? C
  Gas: off
  Boiler pump: on
  Fan: off
  Ignition: off
  Circulation pump: off
  Burner selected max power: 40 %
  Burner current power: 0 %
  Flame current: -0.1 uA
  System pressure: ? bar
  System service code:  (0)
  Heating temperature setting on the boiler: 47 C
  Boiler circuit pump modulation max power: 100 %
  Boiler circuit pump modulation min power: 10 %
  Boiler temperature: ? C
  Pump modulation: 100 %
  Burner # starts: 392002 times
  Total burner operating time: 372 days 0 hours 5 minutes
  Total heat operating time: 315 days 11 hours 34 minutes
  Total UBA working time: 1804 days 4 hours 33 minutes

Could anybody explain such strange behaviour?

glitter-ball, 2019-12-15:
How does your system work? Is it weather compensated (usually have outdoor sensor)? If so, the thermostat will periodically re-calculate the appropriate boiler flow temperature based on the heating curve, outside temperature and, possibly, room temperature.

So, although you've set the flow temperature to 40ºC, the thermostat's calculation cycle will eventually run and produce a new target flow temperature.

What happens if you change the 47ºC setting on the boiler? Does it change there too?

arturasi, 2019-12-15:
My system is a standalone boiler Junkers CerapurSmart ZWB series (with built-in water heater), no thermostats/external temperature sensor etc connected, just a boiler with simplest manual flow temperature regulators (one for water another for heating), that's it.

If I manually change heating temperature on the boiler - I see same temperature in "info" in "Heating temperature setting on the boiler" field, so information from the boiler is read out ok.

proddy, 2019-12-15:
I think the problem is in the validate process. In the early days I wasn't confident the writes were working so built a 3-stage workflow to write first, wait and then read, and finally validate. Validate compares the values and if it would fail it would retry 2 more times. In the flowtemp function after the value is set I issue a read command to the same telegram type 0x1A to pull back the new flowtemp and compare the values. Perhaps I'm reading the wrong value. You can see it in your log

```
(17:19:40) Sending write of type 0x1A to 0x09, telegram: 8B 09 1A 00 28 (CRC=70)
(17:19:40) Sending validate of type 0x1A to 0x09, telegram: 8B 89 1A 00 01 (CRC=3D)
-> Write failed. Compared set value 0x28 with received value of 0x34
-> Write didn't work, retrying (1/2)...
```

Could you get more details of what to compare with `log v`.

arturasi, 2019-12-16:
Hm I don't see reading type 0x1A in the log, here is what I have:

> info

>EMS-ESP system stats:
  System logging set to None
  LED: on, Listen mode: off
  Boiler: enabled, Thermostat: disabled, Solar Module: disabled, Mixing Module: disabled
  Shower Timer: disabled, Shower Alert: disabled

>EMS Bus stats:
  Bus is connected, protocol: HT3
  Rx: # successful read requests=6658, # CRC errors=0
  Tx: Last poll=0.784 seconds ago, # successful write requests=0

>Boiler stats:
  Boiler: Bosch Condens 2500/Buderus Logamax GB062/Junkers Cerapur Top/Worcester Greenstar i/Generic HT3 (DeviceID:0x09 ProductID:95 Vers
  Hot tap water: off
  Central heating: off
  Warm Water activated: on
  Warm Water circulation pump available: off
  Warm Water comfort setting: Hot
  Warm Water selected temperature: 50 C
  Warm Water desired temperature: 75 C
  Warm Water current temperature: 37.6 C
  Warm Water current tap water flow: 0.0 l/min
  Warm Water # starts: 55602 times
  Warm Water active time: 56 days 13 hours 51 minutes
  Warm Water 3-way valve: off
  Selected flow temperature: 43 C
  Current flow temperature: 39.0 C
  Return temperature: ? C
  Gas: off
  Boiler pump: on
  Fan: on
  Ignition: off
  Circulation pump: off
  Burner selected max power: 40 %
  Burner current power: 0 %
  Flame current: -0.1 uA
  System pressure: ? bar
  System service code:  (0)
  Heating temperature setting on the boiler: 43 C
  Boiler circuit pump modulation max power: 100 %
  Boiler circuit pump modulation min power: 10 %
  Boiler temperature: ? C
  Pump modulation: 100 %
  Burner # starts: 392306 times
  Total burner operating time: 372 days 12 hours 44 minutes
  Total heat operating time: 315 days 22 hours 53 minutes
  Total UBA working time: 1805 days 6 hours 44 minutes

>log v

>System Logging set to Verbose
(19:33:14) Sending read of type 0x18 to 0x09, telegram: 8B 89 18 00 20 (CRC=14)
(19:33:15) Controller -> Me, type 0x18, telegram: 89 0B 18 00 2B 01 6E 28 00 01 02 20 C0 01 79 80 00 80 00 FF FF FF 00 00 00 00 00 00 00 (CRC=26) #data=25
(19:33:15) Sending read of type 0x19 to 0x09, telegram: 8B 89 19 00 20 (CRC=10)
(19:33:15) Controller -> Me, type 0x19, telegram: 89 0B 19 00 80 00 80 00 80 00 FF FF 03 64 05 FC 72 08 2F 7C 00 00 00 06 F1 3D 05 23 40 80 00 (CRC=F4) #data=27
(19:33:15) UBAMaster -> All, type 0x07, telegram: 88 00 07 00 0B 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=56) #data=15
(19:33:15) Sending read of type 0x33 to 0x09, telegram: 8B 89 33 00 20 (CRC=B8)
(19:33:15) Controller -> Me, type 0x33, telegram: 89 0B 33 00 08 FF 32 00 00 00 00 02 4B 00 FF FF (CRC=15) #data=12
(19:33:15) Sending read of type 0x16 to 0x09, telegram: 8B 89 16 00 20 (CRC=2C)
(19:33:15) Controller -> Me, type 0x16, telegram: 89 0B 16 00 FF 2B 28 00 00 FB 03 01 03 64 0A 02 (CRC=54) #data=12
(19:33:15) Sending read of type 0x14 to 0x09, telegram: 8B 89 14 00 20 (CRC=24)
(19:33:15) Controller -> Me, type 0x14, telegram: 89 0B 14 00 27 AA B5 (CRC=3D) #data=3
(19:33:15) UBAMaster -> All, type 0x18, telegram: 88 00 18 00 2B 01 6C 28 00 01 02 20 C0 01 79 80 00 80 00 FF FF FF 00 00 00 00 00 00 00 (CRC=3F) #data=25
(19:33:16) UBAMaster -> All, type 0x07, telegram: 88 00 07 00 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=E6) #data=15
(19:33:23) UBAMaster -> All, type 0x18, telegram: 88 00 18 00 2B 01 63 28 00 01 02 20 C0 01 79 80 00 80 00 FF FF FF 00 00 00 00 00 00 00 (CRC=BE) #data=25
(19:33:23) UBAMaster -> All, type 0x34, telegram: 88 00 34 00 32 01 7A 80 00 81 00 00 01 00 01 3E 3F 00 D9 32 00 (CRC=AA) #data=17
[... 65 lines cut ...]
Requesting type UBAParameterWW(0x33) from dest 0x09
Requesting type UBAParametersMessage(0x16) from dest 0x09
Requesting type UBATotalUptimeMessage(0x14) from dest 0x09
(19:35:14) Sending read of type 0x18 to 0x09, telegram: 8B 89 18 00 20 (CRC=14)
(19:35:14) Controller -> Me, type 0x18, telegram: 89 0B 18 00 28 01 41 28 00 01 03 20 C0 01 83 80 00 80 00 FF FF FF 00 00 00 00 00 00 00 (CRC=A4) #data=25
(19:35:14) Sending read of type 0x19 to 0x09, telegram: 8B 89 19 00 20 (CRC=10)
(19:35:14) Controller -> Me, type 0x19, telegram: 89 0B 19 00 80 00 80 00 80 00 FF FF 01 64 05 FC 72 08 2F 7C 00 00 00 06 F1 3D 05 23 40 80 00 (CRC=06) #data=27
(19:35:14) UBAMaster -> All, type 0x07, telegram: 88 00 07 00 0B 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=56) #data=15
(19:35:14) Sending read of type 0x33 to 0x09, telegram: 8B 89 33 00 20 (CRC=B8)
(19:35:14) Controller -> Me, type 0x33, telegram: 89 0B 33 00 08 FF 32 00 00 00 00 02 4B 00 FF FF (CRC=15) #data=12
(19:35:14) Sending read of type 0x16 to 0x09, telegram: 8B 89 16 00 20 (CRC=2C)
(19:35:15) Controller -> Me, type 0x16, telegram: 89 0B 16 00 FF 2B 28 00 00 FB 03 01 03 64 0A 02 (CRC=54) #data=12
(19:35:15) Sending read of type 0x14 to 0x09, telegram: 8B 89 14 00 20 (CRC=24)
(19:35:15) Controller -> Me, type 0x14, telegram: 89 0B 14 00 27 AA B7 (CRC=3F) #data=3
(19:35:15) UBAMaster -> All, type 0x07, telegram: 88 00 07 00 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=E6) #data=15
(19:35:17) UBAMaster -> All, type 0x18, telegram: 88 00 18 00 28 01 3F 28 00 01 03 20 C0 01 84 80 00 80 00 FF FF FF 00 00 00 00 00 00 00 (CRC=E4) #data=25
(19:35:23) UBAMaster -> All, type 0x18, telegram: 88 00 18 00 28 01 3F 28 00 01 03 20 C0 01 83 80 00 80 00 FF FF FF 00 00 00 00 00 00 00 (CRC=86) #data=25
(19:35:23) UBAMaster -> All, type 0x34, telegram: 88 00 34 00 32 01 84 80 00 81 00 00 01 00 01 3E 3F 00 D9 32 00 (CRC=FB) #data=17
[... cut ...]
(19:35:43) UBAMaster -> All, type 0x34, telegram: 88 00 34 00 32 01 83 80 00 81 00 00 01 00 01 3E 3F 00 D9 32 00 (CRC=99) #data=17
(19:35:44) UBAMaster -> All, type 0x18, telegram: 88 00 18 00 28 01 3D 28 00 01 03 20 C0 01 83 80 00 80 00 FF FF FF 00 00 00 00 00 02 18 (CRC=3D) #data=25
(19:35:45) UBAMaster -> All, type 0x18, telegram: 88 00 18 00 28 01 3D 28 00 01 03 24 C0 01 83 80 00 80 00 FF FF FF 00 00 00 00 00 02 18 (CRC=CF) #data=25
(19:35:46) UBAMaster -> All, type 0x18, telegram: 88 00 18 00 28 01 3D 28 00 01 03 2D C0 01 83 80 00 80 00 FF FF FF 00 00 00 00 00 02 18 (CRC=82) #data=25

proddy, 2019-12-17:
and `boiler read 1a` also comes back with nothing?

arturasi, 2019-12-18:
Ok here the log

> EMS-ESP system stats:
  System logging set to Verbose
  LED: on, Listen mode: off
  Boiler: enabled, Thermostat: disabled, Solar Module: disabled, Mixing Module: disabled
  Shower Timer: disabled, Shower Alert: disabled

>EMS Bus stats:
  Bus is connected, protocol: HT3
  Rx: # successful read requests=29, # CRC errors=0
  Tx: Last poll=0.779 seconds ago, # successful write requests=0

>Boiler stats:
  Boiler: Bosch Condens 2500/Buderus Logamax GB062/Junkers Cerapur Top/Worcester Greenstar i/Generic HT3 (DeviceID:0x09 ProductID:95 Vers
  Hot tap water: off
  Central heating: off
  Warm Water activated: on
  Warm Water circulation pump available: off
  Warm Water comfort setting: Hot
  Warm Water selected temperature: 50 C
  Warm Water desired temperature: 75 C
  Warm Water current temperature: 31.1 C
  Warm Water current tap water flow: 0.0 l/min
  Warm Water # starts: 55694 times
  Warm Water active time: 56 days 16 hours 17 minutes
  Warm Water 3-way valve: off
  Selected flow temperature: 35 C
  Current flow temperature: 28.4 C
  Return temperature: ? C
  Gas: off
  Boiler pump: on
  Fan: off
  Ignition: off
  Circulation pump: off
  Burner selected max power: 40 %
  Burner current power: 0 %
  Flame current: -0.1 uA
  System pressure: ? bar
  System service code:  (0)
  Heating temperature setting on the boiler: 35 C
  Boiler circuit pump modulation max power: 100 %
  Boiler circuit pump modulation min power: 10 %
  Boiler temperature: ? C
  Pump modulation: 100 %
  Burner # starts: 393022 times
  Total burner operating time: 373 days 2 hours 41 minutes
  Total heat operating time: 316 days 10 hours 24 minutes
  Total UBA working time: 1807 days 5 hours 19 minutes

>boiler flowtemp 40

>Setting boiler flow temperature to 40 C
(21:13:11) Sending write of type 0x1A to 0x09, telegram: 8B 09 1A 00 28 (CRC=70)
(21:13:11) Sending validate of type 0x1A to 0x09, telegram: 8B 89 1A 00 01 (CRC=3D)
-> Write failed. Compared set value 0x28 with received value of 0x34
-> Write didn't work, retrying (1/2)...
(21:13:11) UBAMaster -> All, type 0x07, telegram: 88 00 07 00 0B 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=56) #data=15
(21:13:12) Sending write of type 0x1A to 0x09, telegram: 8B 09 1A 00 28 (CRC=70)
(21:13:12) Sending validate of type 0x1A to 0x09, telegram: 8B 89 1A 00 01 (CRC=3D)
-> Write failed. Compared set value 0x28 with received value of 0x34
-> Write didn't work, retrying (2/2)...
(21:13:12) Sending write of type 0x1A to 0x09, telegram: 8B 09 1A 00 28 (CRC=70)
(21:13:12) Sending validate of type 0x1A to 0x09, telegram: 8B 89 1A 00 01 (CRC=3D)
-> Write failed. Compared set value 0x28 with received value of 0x34
-> Write failed. Giving up, removing from queue 

>boiler read 1a

>Requesting type UBASetPoints(0x1A) from dest 0x09
(21:14:22) Sending read of type 0x1A to 0x09, telegram: 8B 89 1A 00 20 (CRC=1C)
(21:14:22) Controller -> Me, telegram: 89 0B 1A 00 (CRC=34)

proddy, 2019-12-20:
I've disabled the validate in 1.9.5b3

closed 2019-12-20 by proddy

arturasi, 2019-12-27:
Ok, on 1.9.5b4 I see no validation after "boiler flowtemp". 

However, same error ("validation failed") occurs after "boiler wwtemp" command:

> Setting boiler warm water temperature to 60 C
(17:40:25) Sending write of type 0x33 to 0x09, telegram: 8B 09 33 02 3C (CRC=C4)
(17:40:25) Sending validate of type 0x33 to 0x09, telegram: 8B 89 33 02 01 (CRC=9D)
-> Write failed. Compared set value 0x3C with received value of 0x32

"boiler read 33" returns this:

>Requesting type UBAParameterWW(0x33) from dest 0x09
(17:40:39) Sending read of type 0x33 to 0x09, telegram: 8B 89 33 00 20 (CRC=B8)
(17:40:39) Controller -> Me, type 0x33, telegram: 89 0B 33 00 08 FF 32 00 00 00 00 02 4B 00 FF FF (CRC=15) #data=12
 
so it seems that value is left unchanged (0x32).

I think such behaviour (boiler settings cannot be changed) is because in my system there is no thermostat connected. Embedded boiler controller see that and therefore does not allow to change the settings.

Also I've noticed that "boiler flowtemp" can be changed only to value below "Heating temperature setting on the boiler", and new flowtemp is working for few minutes only, after that it is goes back to the value set on boiler panel (=Heating temperature setting on the boiler).

I will create a new issue about how to change boiler settings without thermostat connected.