emsesp/EMS-ESP #262: Thermostat RC35 (Moduline 400) not recognized.

GLD-NL, 2019-12-19:
Is there a problem with the moduline 400 thermostat? After the upgrade from 1.8.3 to 1.9.4 I experiencing a problem with this thermostat. The thermostat is no longer recognized.
What I see is a:
- Nefit Topline / Buderus GB162 (Version: 03.06 ProductID: 115 DeviceID: 0x08)
- MM400 Mixer Module (Version: 03.03 ProductID: 78 DeviceID: 0x10)
- BC10 Base Controller (Version: 01.03 ProductID: 114 DeviceID: 0x09)
But no thermostat. In version 1.8.3 this went well . With a downgrade from 1.9.4 to 1.8.3. the thermostat becomes visible again.

Is there anyone who has an idea where this goes wrong?

bbqkees, 2019-12-19:
You mean 1.8.3 (Telnet only) or 1.9.3 (Telnet+webinterface)?

GLD-NL, 2019-12-19:
Hi Kees, I don't quite understand what you mean. With ROM version 1.8.3 I see the thermostat with a telnet session and can use it with MQTT. But in version 1.9.4 I see no thermostat. Not in a telnet session, web interface or by MQTT

bbqkees, 2019-12-19:
What I meant was if you updated from the very old firmware 1.8.3 directly to 1.9.4 or that it was a typo and you actually updated from the recent 1.9.3 to 1.9.4.
This is important for pinpointing the problem as 1.8.3 is >10 releases ago.

GLD-NL, 2019-12-19:
Oke now i get it. It was not a typo. It was indeed from version 1.8.3 directly to 1.9.4 on the same hardware. A lot of changes in intermediate versions. I hope we can find something ;-) If I can help. Let me know

bbqkees, 2019-12-19:
What you try do yourself is to downgrade to 1.9.3, 19.2 etc and see if its detected or not and so on.
I have a test Moduline 400 at home and with at least 1.9.3 it was detected.

GLD-NL, 2019-12-19:
I'll give this a try. This will take a while. 1.9.3 currently running. At least after a few minutes "Tx: No signal" and no EMS device detected yet. I will let you know.

GLD-NL, 2019-12-19:
After de-energizing the wemos-d1 and the boiler, I see the termostat in version 1.9.3. The following devices have now been detected:
- Nefit Topline / Buderus GB162 (Version: 03.06 ProductID: 115 DeviceID: 0x08)
- RC30 / Moduline 400 (Version: 03.03 ProductID: 78 DeviceID: 0x10)
- BC10 Base Controller (Version: 01.03 ProductID: 114 DeviceID: 0x09)

Gert05, 2019-12-19:
Same here. Upgraded tonight to 194 master. Moduline 400 no longer detected...

Gert05, 2019-12-23:
Done a bit of testing. Upgrading to 195b3 didn't solve the problem: no thermostat detected.
Downgrading to 192 finally gave me the RC30/Moduline 400 (ProductID:78 Version:02.20) back...
193 also works.

proddy, 2019-12-23:
I think it's getting mixed up with the MM400 mixer which has the same product ID. I'll make a change in the next dev build.

Gert05, 2019-12-23:
That seems to work just fine. I uploaded the new 195b3 (b5eebea) and I still have a thermostat RC30/Moduline 400 (Version:02.20 ProductID:78 DeviceID:0x10).

Thanks!

GLD-NL, 2019-12-24:
I'm running now on version 1.9.5b4. I can say that the thermostat is also working in this version. Well done!

proddy, 2019-12-24:
thanks @GLD-NL , finally some good news. Nice to know I didn't break it. Merry Christmas :-)

closed 2019-12-30 by proddy