emsesp/EMS-ESP #263: Losing boiler data due to second UBAMonitorFast (18) telegram

logiclabs, 2019-12-20:
**Bug description**
My boiler is a WB Greenstar i30. Each UBAMonitorFast telegram seems to be delivered in 2 parts. Since v1.9.4, this second telegram overwrites the first telegram with zeros. This leaves some of the boiler data values set to zero.  I think the code needs to check if an offset is specified in the 4th byte.

**Telegrams**
(00:45:09) UBAMaster -> All, type 0x18, telegram: 88 00 18 00 35 01 9C 64 00 01 03 20 D0 01 6C 80 00 80 00 00 00 FF 00 00 00 CA 00 00 00 00 00 (CRC=14) #data=27
Publishing boiler data via MQTT
(00:45:09) UBAMaster -> All, type 0x18, telegram: 88 00 18 1B 00 00 00 00 00 00 00 00 00 00 00 (CRC=5C) #data=11

**Device information**
Boiler stats:
  Boiler: Buderus Logamax plus/GB192/Bosch Condens GC9000 (DeviceID:0x08 ProductID:208 Version:01.04)
  Hot tap water: off
  Central heating: off
  Warm Water activated: on
  Warm Water circulation pump available: off
  Warm Water comfort setting: Hot
  Warm Water selected temperature: 60 C
  Warm Water desired temperature: 70 C
  Warm Water current temperature: 35.6 C
  Warm Water current tap water flow: 0.0 l/min
  Warm Water # starts: 11595 times
  Warm Water active time: 5 days 9 hours 0 minutes
  Warm Water 3-way valve: off
  **Selected flow temperature: 0 C
  Current flow temperature: 0.0 C**
  Return temperature: ? C
  Gas: off
  Boiler pump: off
  Fan: off
  Ignition: off
  Circulation pump: off
  **Burner selected max power: 0 %
  Burner current power: 0 %
  Flame current: 0.0 uA**
  System pressure: ? bar
  System service code:  (204)
  Heating temperature setting on the boiler: 56 C
  Boiler circuit pump modulation max power: 100 %
  Boiler circuit pump modulation min power: 60 %
  Boiler temperature: ? C
  Pump modulation: 53 %
  Burner # starts: 42124 times

proddy, 2019-12-20:
Thanks for reporting. I think your suggestion is the correct solution and will look into it

proddy, 2019-12-20:
Can you try the latest dev branch 1.9.5b3

logiclabs, 2019-12-20:
Thanks for the quick fix!  That seems to work great. 

I've also set publish_time to zero and all the data is updating fine in HASS as soon as it's available.

PhillyGilly, 2019-12-26:
Sorry, I have the same problem with 1.9.4 but can't find a bin file for 1.9.5b3 in the dev branch. Can you post a link?

proddy, 2019-12-26:
The automatic build is failing because the platformio.org site crashed.
They’re working on it.

PhillyGilly, 2019-12-26:
There were four bin files in the branch but I couldn't work out which one to use from the file names. Back on 1.9.3 now :-)

PhillyGilly, 2019-12-29:
Upgraded from 1.9.3 to 1.9.5b5 (EMS-ESP-dev-d1_mini.bin).
Boiler values are updating fine now.  Thanks.

closed 2020-01-26 by proddy