emsesp/EMS-ESP #265: Heat request from mqtt

nz51ww, 2019-12-23:
Hi there, 
            I'm building the heating system for my new house and I'm planning to use Home Assistant to control the heating valves from the underfloor heating. My question is if there is any way to send a "heat request" command to the boiler from the ems-esp interface via mqtt that I can trigger from Home Assistant.  Currently the only way I've to perform the heat request is with a Sonoff SV switch that opens/closes the contact for heat request.  My boiler is a Bosch Condens GC9000iW

Thanks,

proddy, 2019-12-30:
If we can figure out the EMS commands to send then yes it should be possible, and extending the MQTT messages is easy. It will take some trial and error though on your part to find the correct values to use.

proddy, 2020-01-31:
I would need some more specific examples of what you're trying to do. There is a one-time function on most boilers which is worth looking into.

closed 2020-03-08 by proddy