emsesp/EMS-ESP #266: Thermostat Sieger ES73 not recognized anymore

Chaot1, 2019-12-26:
Hello.
Since Firmware 1.9.2 i my Sieger Thermostat isn't recongnized anymore.
I can see:

-Sieger BK13,BK15/Nefit Smartline/Buderus GB1x2 (Version:03.06 ProductID:64 DeviceID:0x08)BC10/RFM20 
-Receiver (Version:02.03 ProductID:68 
DeviceID:0x09)
-unknown? (Version:02.08 ProductID:76 DeviceID:0x10)

Actually i tried Firmware 1.9.5.b4 because the ES73 is smilar to RC35. But it is still not recognized.

proddy, 2019-12-26:
oops, sorry. Was a mistake in the product id in ems_devices.h. Should be ok now.

Chaot1, 2019-12-26:
Thank you! Great Work!
so i can wait for 1.9.5b5. Correct?

proddy, 2019-12-26:
It’s done but build is not on Travis as the platformIo.org is down. You’ll
need to make the build yourself.

Chaot1, 2019-12-27:
Thank you so much. Great work!

closed 2019-12-27 by Chaot1