emsesp/EMS-ESP #267: Solar Module SM200

PhillyGilly, 2019-12-26:
Hello Proddy,
Is it possible to extend the supported Solar Modules to include the Bosch/Bruderus SM200
Here is the output of my logging.

[SM200.txt](https://github.com/proddy/EMS-ESP/files/4002973/SM200.txt)

proddy, 2019-12-26:
Yes, should be possible. I'll take a look.

proddy, 2019-12-30:
I extended the code to support the SM200 in 1.9.5b6

But what I find worrying for your logs is the message "Error! Unable to read the EMS bus.". Do you always get that?

PhillyGilly, 2019-12-30:
Hi Proddy.  Thanks. I will check later today but I need to eliminate errors
due to the fact that I have two EMS buses and only connected the EMS-ESP to
them in turn. This is because the Sense II, which is acting as a passive
display for the SM200, tries to sieze control of the boiler if it's on the
same bus. Now that I have 1.9.5b installed I can remove the Sense II and
connect the SM200 to the boiler bus and use my PC to monitor it. That could
be interesting as the Sense II displays the 14 days data. I can't use the
Sense II as a controller until I've got the MM200 in and working, so I'll
add that next and see how it works for monitoring temperatures.

PhillyGilly, 2019-12-30:
Here is the info from 1.9.5b with only the Sense II and the MS200 attached to the EMS-ESP. Looks good. I haven't tested it through to HA yet.
I tried running with Boiler, MS200 and EMS-ESP but there was an EMS issue.

[FS] System config loaded (423 bytes)                                           
[FS] Custom config loaded (222 bytes)                                           
[WEB] Web server started                                                       
[SYSTEM] Free Heap: 24232 bytes initially | 4168 bytes used (17%) | 20064 bytes free (82%)                                                                      [UART] Rx/Tx connection established                                             
[WIFI] Connecting to SSID: Static@7E...                                         
[WIFI] Connected to SSID Static@7E (hostname: ems-esp, IP: 192.168.2.101)       
[OTA] Listening to firmware updates on ems-esp.local:8266                       
[SYSTEM] Disabling serial port communication                                    
[NTP] NTP internet time enabled via server pool.ntp.org with timezone 3         
[MQTT] Connecting to MQTT...                                                    
[NTP] Internet time: 16:53:25 UTC on 30/12. Local time: 16:53:25 GMT            
[MQTT] MQTT connected                                                           
Shower alert has been set to disabled                                           
Shower timer has been set to disabled                                           
New EMS device recognized as a Thermostat (DeviceID:0x10 ProductID:207 Version:17.01)                                                                           New EMS device recognized as a Solar Module (DeviceID:0x30 ProductID:164 Version:25.04)                                                                         Shower alert has been set to disabled                                           
disabledimer has been se                                                        
[TELNET] Connected to EMS-ESP version 1.9.5b5. Type ? for commands.             
System Logging set to None                                                      
mqttlog                                                                                                                                                         MQTT publish log:                                                                 
Timestamp:16:53:26 Topic:status Payload:online                                  Timestamp:16:53:26 Topic:start Payload:start                                    Timestamp:16:55:20 Topic:shower_data Payload:{"timer":"0","alert":"0"}          Timestamp:16:55:20 Topic:heartbeat Payload: {"version":"1.9.5b5","IP":"192.168.2.101","rssid":62,"load":0,"uptime":120,"freemem":57,"MQTTdisconnects":0}                                                                                       

MQTT subscriptions:                                                              
Topic:home/ems-esp/restart                                                      
Topic:home/ems-esp/start                                                        
Topic:home/ems-esp/thermostat_cmd_temp1                                         
Topic:home/ems-esp/thermostat_cmd_mode1                                         
Topic:home/ems-esp/thermostat_cmd_temp2                                         
Topic:home/ems-esp/thermostat_cmd_mode2                                         
Topic:home/ems-esp/thermostat_cmd_temp3                                         
Topic:home/ems-esp/thermostat_cmd_mode3                                         
Topic:home/ems-esp/thermostat_cmd_temp4                                         
Topic:home/ems-esp/thermostat_cmd_mode4                                         
Topic:home/ems-esp/thermostat_cmd                                               
Topic:home/ems-esp/boiler_cmd                                                   
Topic:home/ems-esp/boiler_cmd_wwactivated                                       
Topic:home/ems-esp/boiler_cmd_wwonetime                                         
Topic:home/ems-esp/boiler_cmd_wwtemp                                            
Topic:home/ems-esp/generic_cmd                                                  
Topic:home/ems-esp/shower_data                                                                                                                                

log v                                                                                                                                                           System Logging set to Verbose                                                   (16:56:17) Solar Module -> All, type 0x0263, telegram: B0 00 FF 00 02 63 80 00 80 00 00 00 80 00 80 00 80 00 00 80 00 (CRC=5A) #data=15                         (16:56:17) Solar Module -> All, type 0x0264, telegram: B0 00 FF 00 02 64 64 00 00 00 00 00 FF 00 00 00 00 01 64 00 00 00 00 00 (CRC=E4) #data=18                (16:56:17) Solar Module -> All, type 0x0266, telegram: B0 00 FF 00 02 66 0D 6F 00 03 08 (CRC=D4) #data=5                                                        (16:56:17) Solar Module -> All, type 0x0268, telegram: B0 00 FF 00 02 68 0C 00 (CRC=08) #data=2                                                                 (16:56:18) Solar Module -> All, type 0x0269, telegram: B0 00 FF 00 02 69 00 (CRC=0A) #data=1                                                                    (16:56:18) Solar Module -> All, type 0x026A [... cut ...]
(17:00:18) Solar Module -> All, type 0x0266, telegram: B0 00 FF 00 02 66 0D 6F 00 03 08 (CRC=D4) #data=5                                                        (17:00:18) Solar Module -> All, type 0x0268, telegram: B0 00 FF 00 02 68 0C 00 (CRC=08) #data=2                                                                 (17:00:18) Solar Module -> All, type 0x0269, telegram: B0 00 FF 00 02 69 00 (CRC=0A) #data=1                                                                    (17:00:18) Solar Module -> All, type 0x026A, telegram: B0 00 FF 00 02 6A 03 03 03 03 03 03 03 03 03 03 03 03 (CRC=95) #data=12                                  Fetching data from EMS devices

PhillyGilly, 2019-12-31:
I ran 1.9.5b7 overnight and found that I had lost connection between the EMS-ESP and web interface this morning. I rebooted the EMS-ESP and connection was re-established. Initially things looked fine and I confirmed this via a telnet session and left it running with log v.
After running for about 90 minutes, I checked the HA interface and I have "flat lining" where data isn't arriving
[image]
Here are the last few entries from the log.
(10:37:25) Solar Module -> All, type 0x026A, telegram: B0 00 FF 00 02 6A 03 03 03 03 03 03 03 03 03 03 03 03 (CRC=95) #data=12                                  
(10:37:30) Solar Module -> All, type 0x07, telegram: B0 00 07 00 00 01 00 00 00 01 00 00 00 00 00 00 00 00 00 (CRC=7E) #data=15                                 
(10:37:53) Solar Module -> All, type 0x0266, telegram: B0 00 FF 00 02 66 0D 6F 00 03 08 (CRC=D4) #data=5                                                        
Error! Unable to read the EMS bus.                                              
(10:38:23) Solar Module -> All, type 0x0262, telegram: B0 00 FF 00 02 62 00 DC 00 F8 80 00 80 00 80 00 80 00 80 00 80 00 80 00 80 00 80 00 80 00 (CRC=A8) #data=24                                                                              (10:38:23) Solar Module -> All, type 0x0262, telegram: B0 00 FF 18 02 62 80 00 (CRC=B8) #data=2                                                                 (10:38:24) Solar Module -> All, type 0x0263, telegram: B0 00 FF 00 02 63 80 00 80 00 00 00 80 00 80 00 80 00 00 80 00 (CRC=5A) #data=15                         
(10:38:24) Solar Module -> All, type 0x0264, telegram: B0 00 FF 00 02 64 64 00 00 04 00 00 FF 00 00 00 00 03 64 00 00 00 00 00 (CRC=3C) #data=18                
(10:38:24) Solar Module -> All, type 0x0266, telegram: B0 00 FF 00 02 66 0D 6F 00 03 08 (CRC=D4) #data=5                                                        
(10:38:25) Solar Module -> All, type 0x0268, telegram: B0 00 FF 00 02 68 0C 00 (CRC=08) #data=2                                                                 (10:38:25) Solar Module -> All, type 0x0269, telegram: B0 00 FF 00 02 69 00 (CRC=0A) #data=1                                                                    (10:38:25) Solar Module -> All, type 0x026A, telegram: B0 00 FF 00 02 6A 03 03 03 03 03 03 03 03 03 03 03 03 (CRC=95) #data=12                                  
(10:38:25) Thermostat -> All, type 0x06, telegram: 90 00 06 00 13 0C 0A 1F 25 2D 01 00 10 FF 00 (CRC=7A) #data=11                                               
(10:38:30) Solar Module -> All, type 0x07, telegram: B0 00 07 00 00 01 00 00 00 01 00 00 00 00 00 00 00 00 00 (CRC=7E) #data=15                                 
(10:38:43) Thermostat -> Solar Module, type 0x1802, telegram: 90 B0 FF 00 18 02 91 (CRC=32) #data=1                                                             
(10:38:43) Solar Module -> Thermostat, type 0x0291, telegram: B0 10 FF 00 02 91 00 02 FB F4 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=4A) #data=24                                                                       (10:38:43) Thermostat -> Solar Module, type 0x1802, telegram: 90 B0 FF 18 18 02 91 (CRC=F2) #data=1                                                             
(10:38:43) Solar Module -> Thermostat, type 0x0291, telegram: B0 10 FF 18 02 91 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=2D) #data=24                                                                       (10:38:43) Thermostat -> Solar Module, type 0x1802, telegram: 90 B0 FF 30 18 02 91 (CRC=AB) #data=1                                                             
(10:38:43) Solar Module -> Thermostat, type 0x0291, telegram: B0 10 FF 30 02 91 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=5F) #data=24                                                                       (10:38:43) Thermostat -> Solar Module, type 0x0402, telegram: 90 B0 FF 48 04 02 91 (CRC=30) #data=1                                                             
(10:38:43) Solar Module -> Thermostat, telegram: B0 10 FF 48 02 91 (CRC=DD)     
(10:38:48) Thermostat -> All, type 0xBF, telegram: 90 00 BF 00 10 CF 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=8D) #data=24        
(10:38:53) Solar Module -> All, type 0x0266, telegram: B0 00 FF 00 02 66 0D 6F 00 03 08 (CRC=D4) #data=5                                                        
Starting scheduled MQTT publish...                                              
Error! Unable to read the EMS bus.                                              
(10:39:24) Solar Module -> All, type 0x0262, telegram: B0 00 FF 00 02 62 00 DA 00 F8 80 00 80 00 80 00 80 00 80 00 80 00 80 00 80 00 80 00 80 00 (CRC=58) #data=24                                                                              (10:39:24) Solar Module -> All, type 0x0262, telegram: B0 00 FF 18 02 62 80 00 (CRC=B8) #data=2                                                                 (10:39:24) Solar Module -> All, type 0x0263, telegram: B0 00 FF 00 02 63 80 00 80 00 00 00 80 00 80 00 80 00 00 80 00 (CRC=5A) #data=15                         
(10:39:24) Solar Module -> All, type 0x0264, telegram: B0 00 FF 00 02 64 64 00 00 04 00 00 FF 00 00 00 00 03 64 00 00 00 00 00 (CRC=3C) #data=18                
[... cut ...]
(10:40:25) Solar Module -> All, type 0x0268, telegram: B0 00 FF 00 02 68 0C 00 (CRC=08) #data=2                                                                 (10:40:25) Solar Module -> All, type 0x0269, telegram: B0 00 FF 00 02 69 00 (CRC=0A) #data=1                                                                    (10:40:26) Solar Module -> All, type 0x026A, telegram: B0 00 FF 00 02 6A 03 03 03 03 03 03 03 03 03 03 03 03 (CRC=95) #data=12                                  
(10:40:27) Thermostat -> All, type 0x06, telegram: 90 00 06 00 13 0C 0A 1F 27 2F 01 00 10 FF 00 (CRC=BA) #data=11

proddy, 2019-12-31:
I wouldn't bother with the web interface - it's problematic and I'm busy rewriting the whole thing anyway.

If you think you've lost a connection, the first thing is to start the Telnet and look for errors like 'TCP disconnect'. Then type 'system' and check the uptime and look down for the "# disconnects". It could be that the MQTT was flooding again and not sending packages out. I haven't personally tested with a SM (don't have one) but could expect a lot of information is being sent to the broker. Use something like MQTT Explorer to see. Then also check settings like MQTT QOS and increase the publish_time to 5 or 10 seconds.

I will continue to run stress tests here.

PhillyGilly, 2019-12-31:
Here is the content of my system.
Not sure I like the fatal errors!!

ESP8266 System stats:                                                                                                                                            [APP] EMS-ESP version: 1.9.5b7                                                  
[APP] MyESP version: 1.2.22                                                     
[APP] Build timestamp: 2019-12-30 14:54:37                                      
[APP] Uptime: 0 days 3 hours 38 minutes 51 seconds                              
[APP] System Load: 1%                                                           
[WIFI] WiFi Hostname: ems-esp                                                  
[WIFI] WiFi IP: 192.168.2.101                                                   
[WIFI] WiFi signal strength: 66%                                                
[WIFI] WiFi MAC: 4C:11:AE:10:8F:E6                                              
[MQTT] is connected (heartbeat enabled)                                         
[NTP] Local Time is 16:40:49 GMT (1577810449)                                   
[SYSTEM] System is Stable                                                      
[SYSTEM] Board: PLATFORMIO_D1_MINI                                              
[SYSTEM] CPU frequency: 80 MHz                                                 
[SYSTEM] SDK version: 2.2.2-dev(38a443e)                                        
[SYSTEM] CPU chip ID: 0x108FE6                                                  
[SYSTEM] Core version: 2_6_3                                                   
[SYSTEM] Boot version: 6                                                       
 [SYSTEM] Boot mode: 1                                                           
[SYSTEM] Last reset reason: Hardware Watchdog                                  
[SYSTEM] Last reset info: Fatal exception:4 flag:1 (WDT) epc1:0x40000f68 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000                    
[SYSTEM] Restart count: 0                                                       
[SYSTEM] # TCP disconnects: 0                                                   
[SYSTEM] rtcmem status: blocks:2 addr:0x60001280                                
[SYSTEM] rtcmem 00: 1163087990                                                  
[SYSTEM] rtcmem 01: 65536                                                       
[FLASH] Flash chip ID: 0x1640EF                                                 
[FLASH] Flash speed: 40000000 Hz                                                
[FLASH] Flash mode: DIO                                                         
[FLASH] Flash size (CHIP): 4194304                                              
[FLASH] Flash size (SDK): 4194304                                               
[FLASH] Flash Reserved: 4096                                                    
[MEM] Firmware size: 616640                                                     
[MEM] Max OTA size: 2523136                                                     
[MEM] OTA Reserved: 16384                                                       
[MEM] Free Heap: 23992 bytes initially | 10408 bytes used (43%) | 13584 bytes free (56%)                                                                                                                                                       Error! Unable to read the EMS bus.

proddy, 2020-01-01:
looks fine. the fatal exception is normal after a reset (cold-start) of the ESP. no need to panic there.

proddy, 2020-01-06:
I forgot about this one, implementing SM200. I'll take a look this week.

PhillyGilly, 2020-01-06:
I will try to reconfigure and run the trace tonight.  It would be neat if you could produce JSONs with sections for each Solar Circuit as sc1, sc2 in the same way that multiple heating circuits are hc1, hc2 etc but I don't know what that means for the current SM100 implementations.
    
FWIW I'm still on 1.9.5b7 (as I still can't find any newer bin files on the dev branch)  and living with the web interface EMS bus error as it doesn't seen to impact telnet logging or HA.

proddy, 2020-01-06:
wow, didn't notice my automatic TravisCI builds were breaking. Last one was
6 days ago!

PhillyGilly, 2020-01-06:
Here is the trace log:
[TELNET] Connected to EMS-ESP version 1.9.5b7. Type ? for commands.             
System Logging set to None                                                      
devices
These device IDs are on the EMS Bus: 0x10 0x30
and 2 were recognized by EMS-ESP as:
unknown? (DeviceID:0x10 ProductID:207 Version:17.01)
SM200 Solar Module (DeviceID:0x30 ProductID:164 Version:25.04)
You have a device is that is not yet known by EMS-ESP.
Please report this as a GitHub issue so we can expand the EMS device library.                                                                                                    
info EMS-ESP system stats:                
System logging set to None                                                      
LED: on, Listen mode: off                                                       
Boiler: disabled, Thermostat: disabled, Solar Module: enabled, Mixing Module: disabled                                                                          
Shower Timer: disabled, Shower Alert: disabled                                                                                                                
EMS Bus stats:
No connection can be made to the EMS bus                                                                                                                      
Boiler stats:                                                                     
Boiler: <not enabled>                                                           
Warm Water activated: ?                                                         
Warm Water circulation pump available: ?                                        
Warm Water selected temperature: ? C                                            
Warm Water desinfection temperature: ? C                                        
Warm Water current temperature: ? C                                             
Warm Water current tap water flow: ? l/min                                      
Warm Water # starts: ? times                                                    
Warm Water 3-way valve: on                                                      
Selected flow temperature: ? C                                                  
Current flow temperature: ? C                                                   
Return temperature: ? C                                                         
Gas: ?                                                                          
Boiler pump: ?                                                                  
Fan: ?                                                                          
Ignition: ?                                                                     
Circulation pump: ?                                                             
Burner selected max power: ? %                                                  
Burner current power: ? %                                                       
Flame current: ? uA                                                             
System pressure: ? bar                                                          
System service code: ??                                                         
Heating temperature setting on the boiler: ? C                                  
Boiler circuit pump modulation max power: ? %                                   
Boiler circuit pump modulation min power: ? %                                   
Boiler temperature: ? C                                                         
Exhaust temperature: ? C                                                        
Pump modulation: ? %                                                            
Burner # starts: ? times                                                                                                                                      
Solar Module stats:                                                               
Solar module: SM200 Solar Module (DeviceID:0x30 ProductID:164 Version:25.04)    
Collector temperature: 9.2 C                                                    
Bottom temperature: 18.1 C                                                      
Pump modulation: 0 %                                                            
Pump active: off                                                                
Energy last hour: 0.0 Wh                                                        
Energy today: 0 Wh                                                              
Energy total: 1624.6 kWh                                                                                                                                      
log v                                                                                                                                                           
System Logging set to Verbose                                                   
Error! Unable to read the EMS bus.                                              
(16:15:20) Solar Module -> All, type 0x0266, telegram: B0 00 FF 00 02 66 0F 7B 00 4B 09 (CRC=C5) #data=5                                                        
Error! Unable to read the EMS bus.                                              
(16:15:48) Solar Module -> All, type 0x07, telegram: B0 00 07 00 00 01 00 00 00 01 00 00 00 00 00 00 00 00 00 (CRC=7E) #data=15                                 
[... cut ...]
(16:17:53) Solar Module -> All, type 0x026A, telegram: B0 00 FF 00 02 6A 03 03 03 03 03 03 03 03 03 03 03 03 (CRC=95) #data=12                                  
Error! Unable to read the EMS bus.                                              
(16:18:21) Solar Module -> All, type 0x0266, telegram: B0 00 FF 00 02 66 0F 7B 00 4B 09 (CRC=C5) #data=5

There is new data in the 0x0262 telegram at positions 16+17 which looks like the bottom temperature on the second circuit.  I will test that using ice/hot water. But I can't see anything obvious relating the three way valve postion or status.

proddy, 2020-01-07:
a few observations
- you don't have a boiler so I should probably suppress showing the Boiler Stats in `info`, right?
- I see the warning "No connection can be made to the EMS bus" and "Error! Unable to read the EMS bus." yet you are receiving data so that is something else I need to fix
- did you see you had multiple circuits on the SM200? Can you identify them in the incoming telegrams?

PhillyGilly, 2020-01-07:
Hi @proddy 
_You don't have a boiler so I should probably suppress showing the Boiler Stats in info, right?_
I do have a boiler (WB Greenstar see #226 but it isn't on the EMS bus yet. However suppressing unnecessary info would be better but this is a low priority as most user have a boiler.)
_I see the warning "No connection can be made to the EMS bus" and "Error! Unable to read the EMS bus." yet you are receiving data so that is something else I need to fix_
Agreed.
_did you see you had multiple circuits on the SM200? Can you identify them in the incoming telegrams?_
Yes (and no).  It turns out that the SM200 does not have multiple circuits like I suggested. However all the "missing" data is contained in the 0x0262 temperature message or in the 0x026A status message. 

[image]

I can't definitively test my findings in January but I've described my process here:
 
[MS200 analysis.pdf](https://github.com/proddy/EMS-ESP/files/4031178/MS200.analysis.pdf)

PhillyGilly, 2020-01-07:
Bin file for 1.9.5b16 found and uploaded.
New devices added (note: it is a CS200 )
[image]
Thanks :-)

proddy, 2020-01-08:
What should we consider adding to the project
a) Including the 6 other TSs (TS3-TS8) using 0x26A to see which are active, and having their data included in a single MQTT payload ?
b) or as above but just the the additional cylinder as in your setup

and how much of a priority is this. There's a lot on the to-do list and I'm thinking of moving them to a backlog

PhillyGilly, 2020-01-08:
Hi @proddy 
Bearing in mind everything that is going on I would be grateful if you could just add TS5 and VS2 into the existing SM100 set up as below.
This would work for 100% of known SM200 users and also inconvenience 0% of current SM100 users. 
[image]
I agree that using separate or nested payloads would be a tidier way of proceeding, but wouldn't we need a fully configured installation to test and map the physical connections to telegram data? As there aren't any users with this configuration (AFAIK) I suggest you put this development into the backlog.
Thanks.

PhillyGilly, 2020-01-18:
Hello @proddy. In issue #299 @MichaelDvP has kindly created a test version for the MM200 problem. This made me think that I could help get the missing SM200 into a future release by making a version that brings back TS5 and VS2.  I have looked at how the SM100 code works in ems.cpp and it seems fairly straight forward to add in those two points. So, if I were to do this which file version should I use (i.e. last stable 1.9.4 of latest version 1.95b17) ?

proddy, 2020-01-18:
@PhillyGilly sure, give it a shot. Add the two variables to ems.h, make sure you init them in ems_init(), add the handler to EMS_Types[] and create the new functions. I can help you out if you get stuck. Actually looking at ems.cpp is monstrously long. It's in need of a refactor to separate classes and files which I'll do after the new web UI is functional.

Also thinking about it, I could create a new branch for the SM200/MM200 work so we can all work collaboratively without PRs. Let me know if that's easier for y'all.

proddy, 2020-02-02:
The SM200 telegrams look to be documented now in Norbert's hometop (https://github.com/norberts1/hometop_HT3/blob/master/HT3/docu/HT3-Bus_Telegramme.pdf) which may give some insights in where the rest of the data is hiding. I'm busy refactoring the code to make it much easier to extend. The solar module will be a separate class library for example so I would before adding code to 1.9,x

PhillyGilly, 2020-02-02:
Hello @proddy. I started writing code to bring it line with the Wiki  https://emsesp.github.io/docs/#/SM100.
The various declarations and functions are here:
[SM200 code.txt](https://github.com/proddy/EMS-ESP/files/4144155/SM200.code.txt)
I'm not going to get a chance to compile it for a bit but if anybody can yet give me a bin file, I will be able to test it.

PhillyGilly, 2020-02-02:
I have had a quick look at Norbert's documentation of x0262, x0264, and x028E telegrams. Everything appears to be the same in these. In x026A he identifies Solar Pump on/off although his values are 0/1 for on/off where we have 3/4 and he does not identify VS2 at all.  As I have seen this change at byte 4, I am happy with the changes proposed above

proddy, 2020-02-02:
ok, let's stick to the original design. I'll incorporate them into the new code. Just did heatpumps. Solar is next

PhillyGilly, 2020-02-02:
I agree. Is there any sensible way fir me to feedback my thoughts on x0262 to Norbert?

proddy, 2020-02-02:
> I agree. Is there any sensible way fir me to feedback my thoughts on x0262 to Norbert?

just create an issue on his github and refer to the document.

proddy, 2020-02-24:
I forgot about this one. What are the next steps?

PhillyGilly, 2020-02-25:
Regarding Norbert: I created an issue on his Github and have had a small dialogue with him. In summary he is on the case for the 0x0262 Temperature but I can contribute further on the statuses in 0x026A. I don't need anything else from you on this one.

Regarding the SM200 module:  I followed your suggestions above and "wrote" the lines of code for the ems.h and ems.cpp as below (and in txt file due to format) and posted them above at the beginning of the Feb.  As you can see I have simply added a few lines to your code, and although there may be more elegant solutions than multiple nested elseif, I don't think there's a good enough reason to rewrite the working SM100 code.  I was hoping that you would be able to add this into into a new release as its a small change. I will test the firmware if somebody can compile it for me.

[SM200 code.txt](https://github.com/proddy/EMS-ESP/files/4249065/SM200.code.txt)

**for ems.h**
at line~40
// Device Flags
#define EMS_DEVICE_FLAG_NONE 0   // no flags set
#define EMS_DEVICE_FLAG_SM10 10  // solar module1
#define EMS_DEVICE_FLAG_SM100 11 // solar module2
#define EMS_DEVICE_FLAG_SM200 12  // solar module3

at line~350
// Solar Module - SM10/SM100/SM200/ISM1
typedef struct {
    uint8_t      device_id;    // the device ID of the Solar Module
    uint8_t      device_flags; // Solar Module flags
    const char * device_desc_p;
    uint8_t      product_id;
    char         version[10];
    int16_t      collectorTemp;          // collector temp (TS1)
    int16_t      bottomTemp;             // bottom temp (TS2)
    int16_t      bottomTemp2;             // bottom temp cylinder 2 (TS5)
    uint8_t      pumpModulation;         // modulation solar pump
    uint8_t      pump;                   // pump active
    uint8_t      valveStatus;            // valve status (VS2)
    int16_t      setpoint_maxBottomTemp; // setpoint for maximum collector temp
    uint16_t     EnergyLastHour;
    uint16_t     EnergyToday;
    uint16_t     EnergyTotal;
    uint32_t     pumpWorkMin; // Total solar pump operating time
} _EMS_SolarModule;

**for ems.cpp**
at line~180
    // Solar Module values
    EMS_SolarModule.collectorTemp          = EMS_VALUE_SHORT_NOTSET; // collector temp from SM10/SM100/SM200
    EMS_SolarModule.bottomTemp             = EMS_VALUE_SHORT_NOTSET; // bottom temp from SM10/SM100/SM200
    EMS_SolarModule.bottomTemp2            = EMS_VALUE_SHORT_NOTSET; // bottom temp 2 from SM200
    EMS_SolarModule.pumpModulation         = EMS_VALUE_INT_NOTSET;   // modulation solar pump SM10/SM100/SM200
    EMS_SolarModule.pump                   = EMS_VALUE_BOOL_NOTSET;  // pump active
    EMS_SolarModule.valveStatus            = EMS_VALUE_BOOL_NOTSET;  // valve status from SM200
    EMS_SolarModule.EnergyLastHour         = EMS_VALUE_USHORT_NOTSET;
    EMS_SolarModule.EnergyToday            = EMS_VALUE_USHORT_NOTSET;
    EMS_SolarModule.EnergyTotal            = EMS_VALUE_USHORT_NOTSET;
    EMS_SolarModule.device_id              = EMS_ID_NONE;
    EMS_SolarModule.product_id             = EMS_ID_NONE;
    EMS_SolarModule.pumpWorkMin            = EMS_VALUE_LONG_NOTSET;
    EMS_SolarModule.setpoint_maxBottomTemp = EMS_VALUE_SHORT_NOTSET;

at line~1470
/*
 * SM200Monitor - type 0x0262 EMS+
 * B0 0B FF 00 02 62 00 44 02 7A 80 00 80 00 80 00 80 00 80 00 80 00 00 7C 80 00 80 00 80 00 80
 * bytes 0+1 = TS1 Temperature sensor for collector
 * bytes 2+3 = TS2 Temperature sensor bottom cylinder 1
 * bytes 16+17 = TS5 Temperature sensor bottom cylinder 2
 */
void _process_SM200Monitor(_EMS_RxTelegram * EMS_RxTelegram) {
    _setValue(EMS_RxTelegram, &EMS_SolarModule.collectorTemp, 0); // is *10
    _setValue(EMS_RxTelegram, &EMS_SolarModule.bottomTemp, 2);    // is *10
    _setValue(EMS_RxTelegram, &EMS_SolarModule.bottomTemp2, 16);    // is *10
}

/*
 * SM200Status - type 0x0264 EMS+ for pump modulation
 * e.g. 30 00 FF 09 02 64 64 = 100%
 *      30 00 FF 09 02 64 1E = 30%
[... 25 lines cut ...]
at line~1950
void ems_getSolarModuleValues() {
    if (ems_getSolarModuleEnabled()) {
        if (EMS_SolarModule.device_flags == EMS_DEVICE_FLAG_SM10) {
            ems_doReadCommand(EMS_TYPE_SM10Monitor, EMS_ID_SM); // fetch all from SM10Monitor
        } else if (EMS_SolarModule.device_flags == EMS_DEVICE_FLAG_SM100) {
            ems_doReadCommand(EMS_TYPE_SM100Monitor, EMS_ID_SM); // fetch all from SM100Monitor
        } else if (EMS_SolarModule.device_flags == EMS_DEVICE_FLAG_SM200) {
            ems_doReadCommand(EMS_TYPE_SM200Monitor, EMS_ID_SM); // fetch all from SM200Monitor
        }
    }

at line~2785
    // Solar Module devices
    {EMS_DEVICE_UPDATE_FLAG_SOLAR, EMS_TYPE_SM10Monitor, "SM10Monitor", _process_SM10Monitor},
    {EMS_DEVICE_UPDATE_FLAG_SOLAR, EMS_TYPE_SM100Monitor, "SM100Monitor", _process_SM100Monitor},
    {EMS_DEVICE_UPDATE_FLAG_SOLAR, EMS_TYPE_SM100Status, "SM100Status", _process_SM100Status},
    {EMS_DEVICE_UPDATE_FLAG_SOLAR, EMS_TYPE_SM100Status2, "SM100Status2", _process_SM100Status2},
    {EMS_DEVICE_UPDATE_FLAG_SOLAR, EMS_TYPE_SM100Energy, "SM100Energy", _process_SM100Energy},
    {EMS_DEVICE_UPDATE_FLAG_SOLAR, EMS_TYPE_SM200Monitor, "SM200Monitor", _process_SM100Monitor},
    {EMS_DEVICE_UPDATE_FLAG_SOLAR, EMS_TYPE_SM200Status, "SM200Status", _process_SM100Status},
    {EMS_DEVICE_UPDATE_FLAG_SOLAR, EMS_TYPE_SM200Status2, "SM200Status2", _process_SM100Status2},
    {EMS_DEVICE_UPDATE_FLAG_SOLAR, EMS_TYPE_SM200Energy, "SM200Energy", _process_SM100Energy},
    {EMS_DEVICE_UPDATE_FLAG_SOLAR, EMS_TYPE_ISM1StatusMessage, "ISM1StatusMessage", _process_ISM1StatusMessage},
    {EMS_DEVICE_UPDATE_FLAG_SOLAR, EMS_TYPE_ISM1Set, "ISM1Set", _process_ISM1Set},a

proddy, 2020-02-25:
sorry, must have missed that. I'll add the code tonight.

PhillyGilly, 2020-02-25:
I'm not getting bottomtemp2 still in 1.9.5b43
[image]

proddy, 2020-02-26:
you code hasn't gone in yet, hence no bottomtemp2. Also MQTT log has been removed entirely from telnet console and web. It was too much memory overhead and not really aligned with the goals of EMS-ESP

proddy, 2020-02-26:
btw this is all documented in the chnagelog.md file

PhillyGilly, 2020-02-26:
Sorry. I jumped the gun. I didn't look at the code/change log before
commenting
Agree there's no need for mqttlog when external app can be used.
Also mqttlog still appears as option after telnet ? command.
Cheers.

proddy, 2020-02-26:
mqttlog shows the publish queue. It has a limit of 50 messages and is sent sequentially every 750ms. A better name for this would probably be `mqttqueue` I agree.

proddy, 2020-02-26:
@PhillyGilly I merged in your changes and tried to simplify it a bit. It's in 1.9.5b44. Can you check? Note only telnet will show the boiler temp2, not the web UI.

PhillyGilly, 2020-02-26:
Hello @proddy . Thank you. Yes the SM100/SM200 code was a bit repetitive.
I've got all the data in telenet info but not MQTT explorer.
(However can you note that the IP in my Explorer is the HA Raspberry Pi and my Telnet IP is the ESP. Is that OK?)
[image]

proddy, 2020-02-26:
didn't do the MQTT. If the data looks ok I'll do that next

PhillyGilly, 2020-02-26:
Yes it looks right. Cylinder 1 has been heating today. Cylinder 2 is colder at the bottom because that is where it fills.  
[image]

proddy, 2020-02-26:
ok. Added to 1.9.5b45 just now

PhillyGilly, 2020-02-26:
@Proddy. BottomTemperature2 and ValveStatus both there! Thank you very much. 
[image]

proddy, 2020-02-26:
great. BTW that HA dashboard is sweet. If at some point you want to share the config let me know and I'll add it to the wiki

PhillyGilly, 2020-02-27:
Sure. Here it is. I've included code for the Lovelace tab, sensor.yaml and binary_sensor.yaml 
[HomeAssistantSM200.txt](https://github.com/proddy/EMS-ESP/files/4260449/HomeAssistantSM200.txt)
I can do a screenshot too, once my system is fixed and the sun is shining.
(BTW I wouldn't have found out that the valve is plumbed and wired incorrectly if I hadn't been able to extract the data and be confident that it was right and the physical world was wrong).

PhillyGilly, 2020-03-03:
Correction to above:
#04 sensor.hw_solar_valve_status  
- platform: mqtt
  state_topic: 'home/ems-esp/sm_data'
  name: 'Solar Valve Status'
  icon: 'mdi:valve'
  value_template: "{% if value_json.valvestatus == 'on' %} Cyl#2 active {% else %} Cyl#1 active {% endif %}"

proddy, 2020-03-03:
ok, I'll adjust the files on the wiki too

PhillyGilly, 2020-03-03:
The installation team came and reorientated the three way valve in my
system today, so I should have a good screenshot in a week and be able to
run the digital tests for @Norbert too.
When I have posted the picture I will close the Issue.

joanwa, 2020-06-27:
> Hello Proddy,
> Is it possible to extend the supported Solar Modules to include the Bosch/Bruderus SM200.

stupid question: is the Bosch/Buderus SM200 solar module the same as MS200? Or is SM200 the Buderus name, and MS200 the Bosch/Junkers name?

PhillyGilly, 2020-06-27:
There is only one device which is badged variously as SM200 and MM200. So which ever you've got should work already. Obviously @proddy has only written code that can be tested against available configurations. So if you have a really "interesting"  installation it might need a bit of tweaking. Referring to the installation manual look at some of the earlier posts - (or the device set up) do you know what configuration uou have?

proddy, 2020-06-27:
correct, the Buderus SM100/SM200 and Junkers MS100/MS200 are all EMS plus and use the same range of telegram types (0x262, 0x264, 0x26A, 0x28E). I haven't tested it though but in theory....

joanwa, 2020-06-27:
ok danke! In dem MS200 manual steht in kapitel 2.8 dass das MS200 eine EMS2/EMS2plus Bus schnittstelle hat. D.h. EMS2 und EMSplus oder EMS2 plus ist das gleiche?

Mir ist ein EMS 1 gerät kaputt gegangen und ersetze dann gleich alles mit EMS2 geräten. Hatte FW200, ISM2, IPM1 und ersetze die mit CW400, MS200, MM100. Das CW400 und MM100 habe ich schon, das Solarmodul muss ich noch besorgen und mich hat die ganze EMS2 vs EMSplus SM vs. MS200 Geschichte verwirrt. EMS-ESP hat mir schon gute dienste bei der fehlersuche geleistet :)

proddy, 2020-06-27:
I haven't heard of EMS2plus. I think its all the same, i.e. EMS 2.0 = EMS+ = EMS Plus. It's just telegrams of bytes after all. There is however a difference between some telegrams that Buderus devices and HT3 (Junkers/Bosch/Worcester) use, so a lot of EMS-ESP was built by trial and error with input from the good folks on the forum (like PhillyGilly!). If you run into strange telegrams then let us know and we'll figure them out.  Those MB LAN modules are seriously overpriced.

joanwa, 2020-06-27:
yeah, sorry just EMSplus, not EMS2plus, corrected above. I'll definitely post here what issues I run into with my configuration. Would also gladly assist in getting the CW400 supported. I'll keep you posted.

PhillyGilly, 2020-06-28:
I think we are all pretty much mehrsprachig here as we are dealing with deutsche Technik anything works. I remain interested in the physical plumbing of your system as you will be the only the second known MS200 installation. Depending on what cylinders (kessel), pumps and valves we may be able to see new data in status telegram.  The guru for telegram content is Norbert and I wanted to discuss some results with him regarding this 0x26A telegram (see my post 25th February above).  As you are going to have to configure your MS200 usw you will need to know and it would be good to be able to test/confirm the software.

joanwa, 2020-06-28:
> There is however a difference between some telegrams that Buderus devices and HT3 (Junkers/Bosch/Worcester) use

That makes me wonder if my planned combination of Bosch branded CW400 and MM100 will talk to Buderus branded SM200.

Apart from that, off-topic here, is there still some need for testing on an FW200, IPM1 & ISM2 combination?

PhillyGilly, 2020-06-28:
> That makes me wonder if my planned combination of Bosch branded CW400 and MM100 will talk to Buderus branded SM200?
> 
I have Junkers MM200 and Bosch MS200 coexisting with my Bosch Sense II controller.
All of the information relating to the Junkers solar module device shows it is identical -
[pu_ems_plus_2016-03_6720818016_01.pdf](https://github.com/proddy/EMS-ESP/files/4842747/pu_ems_plus_2016-03_6720818016_01.pdf)
- my opinion is that Bosch don't make many of these devices, they simply rebadge other suppliers' products.
If you think about it the devices don't actually talk to one another, they just pub/sub on the bus. As solar modules are going to have a pretty much defined set of data needs, there really isn't any good reason why they should be different.
(FWIW my MM200 and MS200 also come in identical plastic enclosures, so I have changed the covers and they are all labelled "Bosch" now!).

joanwa, 2020-07-20:
I now have the Junkers MS200 which is recognized as
```
Solar Module: SM200 (DeviceID:0x30, ProductID:164, Version:25.05)
 This Solar Module will respond to telegram type IDs: 0x362 0x363 0x366 0x364 0x36A 0x38E
```
but have further telegram types coming from 0x30:
```
000+15:39:53.012 D 64: [emsesp] No telegram type handler found for ID 0x368 (src 0x30, dest 0x00)
000+15:39:53.474 D 65: [emsesp] No telegram type handler found for ID 0x369 (src 0x30, dest 0x00)
000+15:39:54.131 D 67: [emsesp] No telegram type handler found for ID 0x3A9 (src 0x30, dest 0x00)
000+15:39:54.321 D 68: [emsesp] No telegram type handler found for ID 0x3AA (src 0x30, dest 0x00)
```
missing reported values are pump status and pump modulation of a second circuit from heat exchanger to buffer reservoir and also the temperature sensor of the second circuits return. How can I help decoding them?

joanwa, 2020-07-20:
0x369 already would be helpful. I already have the solar yield posted with the existing SM200 config, but that yields 0. I think that has sth to do with my 'external' heat exchanger and how it measures the yield there.

attached around 100 lines of Rx telegrams:
```
ems-esp:/$ watch raw 30
Watching incoming telegrams, displayed as raw bytes
Filtering only telegrams that match a device ID or telegram type of 0x30
000+01:25:33.549 N 47: [telegram] Rx: B0 08 35 02 00 73
000+01:25:33.635 N 48: [telegram] Rx: B0 00 FF 00 02 62 02 5D 02 10 80 00 80 00 80 00 80 00 80 00 80 00 80 00 80 00 02 62 80 00 C6
000+01:25:33.818 N 49: [telegram] Rx: B0 00 FF 18 02 62 80 00 B8
000+01:25:34.332 N 50: [telegram] Rx: B0 00 FF 00 02 63 80 00 80 00 00 00 80 00 80 00 80 00 00 80 00 5A
000+01:25:34.551 N 51: [telegram] Rx: B0 00 FF 00 02 64 64 00 00 04 00 00 FF 00 64 5E 00 00 64 00 00 00 00 00 0F
000+01:25:35.058 N 52: [telegram] Rx: B0 00 FF 00 02 66 0D 2D 00 12 08 D4
000+01:25:35.242 N 53: [telegram] Rx: B0 00 FF 00 02 68 0C 00 08
000+01:25:35.735 N 54: [telegram] Rx: B0 00 FF 00 02 69 00 0A
000+01:25:35.941 N 55: [telegram] Rx: B0 00 FF 00 02 6A 03 03 03 03 03 04 03 03 03 03 04 03 42
000+01:25:36.485 N 56: [telegram] Rx: B0 00 FF 00 02 A9 80 00 26
000+01:25:36.676 N 57: [telegram] Rx: B0 00 FF 00 02 AA 00 00 03 64 00 45
000+01:25:51.195 N 58: [telegram] Rx: B0 00 FF 09 02 64 5C 04
000+01:26:03.954 N 59: [telegram] Rx: B0 0B FF 00 02 62 02 5D 02 10 80 00 80 00 80 00 80 00 80 00 80 00 80 00 80 00 02 62 80 00 80 EB
000+01:26:04.337 N 60: [telegram] Rx: B0 0B FF 00 02 63 80 00 80 00 00 00 80 00 80 00 80 00 00 80 00 8A
000+01:26:04.483 N 61: [telegram] Rx: B0 00 FF 00 02 66 0D 2D 00 12 08 D4
000+01:26:04.907 N 62: [telegram] Rx: B0 0B FF 00 02 66 0D 2D 00 12 08 0B
000+01:26:05.328 N 63: [telegram] Rx: B0 0B FF 00 02 8E 00 00 00 00 00 00 00 00 00 00 00 00 2A
000+01:26:19.093 N 64: [telegram] Rx: B0 00 FF 09 02 64 5A 02
000+01:26:34.604 N 65: [telegram] Rx: B0 00 FF 00 02 62 02 5C 02 11 80 00 80 00 80 00 80 00 80 00 80 00 80 00 80 00 02 62 80 00 E4
000+01:26:34.788 N 66: [telegram] Rx: B0 00 FF 18 02 62 80 00 B8
000+01:26:35.270 N 67: [telegram] Rx: B0 00 FF 00 02 63 80 00 80 00 00 00 80 00 80 00 80 00 00 80 00 5A
000+01:26:35.490 N 68: [telegram] Rx: B0 00 FF 00 02 64 64 00 00 04 00 00 FF 00 64 5A 00 00 64 00 00 00 00 00 6B
000+01:26:35.965 N 69: [telegram] Rx: B0 00 FF 00 02 66 0D 2D 00 12 08 D4
000+01:26:36.149 N 70: [telegram] Rx: B0 00 FF 00 02 68 0C 00 08
000+01:26:36.642 N 71: [telegram] Rx: B0 00 FF 00 02 69 00 0A
000+01:26:36.847 N 72: [telegram] Rx: B0 00 FF 00 02 6A 03 03 03 03 03 04 03 03 03 03 04 03 42
000+01:26:37.517 N 73: [telegram] Rx: B0 00 FF 00 02 A9 80 00 26
000+01:26:37.707 N 74: [telegram] Rx: B0 00 FF 00 02 AA 00 00 03 64 00 45
000+01:26:38.247 N 75: [telegram] Rx: B0 00 BF 00 30 A4 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01
000+01:26:49.611 N 76: [telegram] Rx: B0 00 FF 00 02 62 02 5B 67
000+01:26:50.388 N 77: [telegram] Rx: B0 00 FF 09 02 64 58 00
000+01:27:03.707 N 78: [telegram] Rx: B0 0B FF 00 02 62 02 5B 02 11 80 00 80 00 80 00 80 00 80 00 80 00 80 00 80 00 02 61 80 00 80 E9
000+01:27:04.096 N 79: [telegram] Rx: B0 0B FF 00 02 63 80 00 80 00 00 00 80 00 80 00 80 00 00 80 00 8A
000+01:27:04.411 N 80: [telegram] Rx: B0 0B FF 00 02 66 0D 2D 00 12 08 0B
000+01:27:04.738 N 81: [telegram] Rx: B0 0B FF 00 02 8E 00 00 00 00 00 00 00 00 00 00 00 00 2A
000+01:27:05.483 N 82: [telegram] Rx: B0 00 FF 00 02 66 0D 2D 00 12 08 D4
000+01:27:21.185 N 83: [telegram] Rx: B0 00 FF 09 02 64 56 0E
000+01:27:35.300 N 84: [telegram] Rx: B0 08 35 02 00 73
000+01:27:35.386 N 85: [telegram] Rx: B0 00 FF 00 02 62 02 58 02 12 80 00 80 00 80 00 80 00 80 00 80 00 80 00 80 00 02 60 80 00 A5
000+01:27:35.570 N 86: [telegram] Rx: B0 00 FF 18 02 62 80 00 B8
000+01:27:36.084 N 87: [telegram] Rx: B0 00 FF 00 02 63 80 00 80 00 00 00 80 00 80 00 80 00 00 80 00 5A
000+01:27:36.303 N 88: [telegram] Rx: B0 00 FF 00 02 64 64 00 00 04 00 00 FF 00 64 56 00 00 64 00 00 00 00 00 C7
000+01:27:36.810 N 89: [telegram] Rx: B0 00 FF 00 02 66 0D 2D 00 12 08 D4
000+01:27:36.994 N 90: [telegram] Rx: B0 00 FF 00 02 68 0C 00 08
000+01:27:37.674 N 91: [telegram] Rx: B0 00 FF 00 02 69 00 0A
000+01:27:37.880 N 92: [telegram] Rx: B0 00 FF 00 02 6A 03 03 03 03 03 04 03 03 03 03 04 03 42
000+01:27:38.393 N 93: [telegram] Rx: B0 00 FF 00 02 A9 80 00 26
000+01:27:38.584 N 94: [telegram] Rx: B0 00 FF 00 02 AA 00 00 03 64 00 45
000+01:27:49.705 N 95: [telegram] Rx: B0 00 FF 09 02 64 54 0C
000+01:28:03.716 N 96: [telegram] Rx: B0 0B FF 00 02 62 02 55 02 12 80 00 80 00 80 00 80 00 80 00 80 00 80 00 80 00 02 5D 80 00 80 90
000+01:28:04.094 N 97: [telegram] Rx: B0 0B FF 00 02 63 80 00 80 00 00 00 80 00 80 00 80 00 00 80 00 8A
000+01:28:04.420 N 98: [telegram] Rx: B0 0B FF 00 02 66 0D 2D 00 12 08 0B
000+01:28:04.773 N 99: [telegram] Rx: B0 0B FF 00 02 8E 00 00 00 00 00 00 00 00 00 00 00 00 2A
000+01:28:06.515 N 100: [telegram] Rx: B0 00 FF 00 02 66 0D 2D 00 12 08 D4
000+01:28:20.814 N 101: [telegram] Rx: B0 00 FF 09 02 64 52 0A
000+01:28:21.626 N 102: [telegram] Rx: B0 00 FF 14 02 62 02 5A 3F
000+01:28:33.253 N 103: [telegram] Rx: B0 00 FF 00 02 62 02 51 6D
000+01:28:36.861 N 104: [telegram] Rx: B0 00 FF 00 02 62 02 51 02 12 80 00 80 00 80 00 80 00 80 00 80 00 80 00 80 00 02 57 80 00 08
000+01:28:37.044 N 105: [telegram] Rx: B0 00 FF 18 02 62 80 00 B8
[... cut ...]
000+01:30:39.461 N 144: [telegram] Rx: B0 00 FF 00 02 64 64 00 00 04 00 00 FF 00 64 4A 00 00 64 00 00 00 00 00 E2
000+01:30:41.183 N 145: [telegram] Rx: B0 00 FF 00 02 66 0D 2D 00 12 08 D4
000+01:30:41.366 N 146: [telegram] Rx: B0 00 FF 00 02 68 0C 00 08
000+01:30:42.452 N 147: [telegram] Rx: B0 00 FF 00 02 69 00 0A
000+01:30:42.658 N 148: [telegram] Rx: B0 00 FF 00 02 6A 03 03 03 03 03 04 03 03 03 03 04 03 42
000+01:30:43.202 N 149: [telegram] Rx: B0 00 FF 00 02 A9 80 00 26
000+01:30:43.392 N 150: [telegram] Rx: B0 00 FF 00 02 AA 00 00 03 64 00 45
```

proddy, 2020-07-20:
quick scan for 0x368, 0x3A9, 0x3AA

0x368 = B0 00 FF 00 02 68    0C 00 08 (that number 12 could mean something?)
0x3A9 = B0 00 FF 00 02 A9    80 00 26 (single value of either 38 or 19 degrees for something?
0x3AA = B0 00 FF 00 02 AA    00 00 03 64 00 45 (0x64 is most likely 100%, the 0x45 is 69 or 34.5 degrees?)

joanwa, 2020-07-21:
ok, thanks. I need to watch another batch and write down what the offline heat exchange temperature gauge says (this is where the sensor also sits) with pump of second circuit on, and another watch batch when pump is off.

joanwa, 2020-07-27:
> I have Junkers MM200 and Bosch MS200 coexisting with my Bosch Sense II controller.
> All of the information relating to the Junkers solar module device shows it is identical - [...]

So turns out Buderus SM200 is different from Bosch/Junkers MS200. Tried and tested myself and Norbert provided some more details on the gitter channel https://gitter.im/EMS-ESP/community#:

>  that moduls: SMx00 (Buderus'/Bosch-Brand) and MSx00 (Junkers/Bosch-Brand) are not identical and will not work on the same boiler. Only Buderus-moduls will work on Buderus-boiler and Junkers-moduls will work on Junkers-boilers.
> The reason is at least the different 'POLLING' sequences created by the boiler.
> Buderus is using: Bit7 (MSB) is set for the polling like: 8B ... (for the service-key) and
> Junkers is using: Bit7 (MSB) is UNset for the polling like: 0B ...
> Apart from that the telegrams itself and there payload are identical or at least compare able.
> Buderus-Details see also: https://emswiki.thefischer.net/doku.php?id=wiki:ems:ems-telegramme
> So you can't run that SM200 on your Junkers CW400 and boiler-System. This will not work and is independand from that EMS-ESP gateway.

proddy, 2020-07-27:
That's correct. EMS-ESP works out which protocol (HT vs Buderus) is used based on the first telegram it receives. It's is quite easy to change this to detect both. This will enable writing/Tx to any EMS device independent from the brand (junkers vs bosch).

In any case EMS-ESP should be able to read and decode from a Junkers MM200 and Bosch MS200 - does this not work?

PhillyGilly, 2020-07-27:
> That's correct. EMS-ESP works out which protocol (HT vs Buderus) is used based on the first telegram it receives. It's is quite easy to change this to detect both. This will enable writing/Tx to any EMS device independent from the brand (junkers vs bosch).
> 
> In any case EMS-ESP should be able to read and decode from https://github.com/proddy/EMS-ESP/issues/267 - does this not work?

Yes, it does work.

The EMS-ESP is taking to both the MM200 and the MS200 and both are recognised by my controller (RC300/RC310/Moduline 3000/Bosch CW400/W-B Sense II (DeviceID: 0x10, ProductID: 158, Version: 33.03).

My MM200 module is badged as Junkers but internally it says Bosch and EMS2 
[image]

Obviously I haven't seen or tested a Junkers MS200 module and I have to bow to @norberts superior knowledge in that, and I shouldn't have said the modules were the same and I apologise to @joanwa if I misled her. I'm still unclear if my Junkers MM200 module would work with other Junkers kit, but as it works for me, I guess I'm OK.

[19 more replies]

closed 2020-09-05 by proddy