emsesp/EMS-ESP #268: Error write on type 0x1D

mellbo, 2019-12-26:
My boiler: Bosch Condens 2500W

I have one error in the function **ems_setWarmTapWaterActivated(bool activated)**
I try to send manualy to boiler  `8B 08 1D 00 5A 00 64 FF` with no result.
Is any trick to put this boiler into Test mode ?...

proddy, 2019-12-26:
Hmm. Not sure. Does anything come back as a response to 1D?

mellbo, 2019-12-26:
nope. 

```
(00:08:19.640) Sending raw: 8B 88 1D 00 20 (CRC=08)
(00:08:19.657) 8B 88 1D 00 20 (CRC=08) #data=1
(00:08:19.674) 88 0B 1D 00 (CRC=32)

(00:09:11.350) Sending raw: 8B 08 1D 00 5A 00 64 FF (CRC=87)
(00:09:11.371) 8B 08 1D 00 5A 00 64 FF (CRC=87) #data=4
(00:09:15.415) 88 00 18 00 00 01 23 00 00 00 02 00 C0 01 23 80 00 80 00 FF FF FF 00 00 00 00 00 00 00 (CRC=FD) #data=25
```

and one more things at **ems_setWarmWaterActivated(bool activated)**
for **0x33** type, at address 01 "WW ENABLED/DISABLED" is not 0xFF for ON. For this boiler 0x08 means ON and 0x00 is OFF. 

so, for the Bosch Condens 2500W I think this line
 `EMS_TxTelegram.dataValue     = (activated ? 0xFF : 0x00); `
must be replaced with
` EMS_TxTelegram.dataValue     = (activated ? 0x08 : 0x00); `

proddy, 2019-12-30:
@mellbo thanks for reporting the setWarmWater issue. You are correct that with Heatronics the values are x08 for ON. I checked also with https://github.com/norberts1/hometop_HT3/blob/65be31e9265599f198d7d6c7aa9b8e4f5939ded4/HT3/sw/lib/ht_discode.py#L2012

closed 2020-01-26 by proddy