emsesp/EMS-ESP #269: TX queue is blowing up

GLD-NL, 2019-12-27:
It seems that there is a problem with the hardware that I use and all 1.9.x. versions. The TX queue fills itself in short time to the maximum of 50 events. This problem does not exist in the 1.8 (1.8.3) version.

Someone recognize this?

Here some background information.

These device IDs are on the EMS Bus: 0x08 0x09 0x10
and 3 were recognized by EMS-ESP as:
 Nefit Topline/Buderus GB162 (DeviceID:0x08 ProductID:115 Version:03.06)
 BC10 Base Controller (DeviceID:0x09 ProductID:114 Version:01.03)
 RC30/Moduline 400 (DeviceID:0x10 ProductID:78 Version:03.03)

EMS-ESP system stats:
  System logging set to None
  LED: on, Listen mode: off
  Boiler: enabled, Thermostat: enabled, Solar Module: disabled, Mixing Module: disabled
  Shower Timer: disabled, Shower Alert: disabled

EMS Bus stats:
  Bus is connected, protocol: Buderus
  Rx: # successful read requests=3736, # CRC errors=14305
  Tx: Last poll=2.738 seconds ago, # successful write requests=0

log v:
(09:09:04) Thermostat -> All, type 0x06, telegram: 10 00 06 00 13 0C 09 1B 0A 3A 04 00 00 00 00 00 00 (CRC=2A) #data=13
(09:09:04) Thermostat -> All, type 0xA3, telegram: 10 00 A3 00 F6 00 00 (CRC=5D) #data=3
(09:09:06) Sending read of type 0x16 to 0x08, telegram: 0B 88 16 00 20 (CRC=EC)
** error sending buffer: BRK
(09:09:06) Corrupt telegram: 0B 08 10 00 00
(09:09:08) Sending read of type 0x16 to 0x08, telegram: 0B 88 16 00 20 (CRC=EC)
** error sending buffer: BRK
(09:09:09) Corrupt telegram: 0B 08 10 00 00
(09:09:09) UBAMaster -> All, type 0x34, telegram: 08 00 34 00 3C 01 9F 80 00 21 00 00 03 00 00 C1 A2 00 6E 33 00 (CRC=E9) #data=17
(09:09:09) UBAMaster -> All, type 0x18, telegram: 08 00 18 00 41 02 54 0C 00 01 01 20 62 01 9F 80 00 02 45 00 00 11 30 45 01 09 00 00 00 (CRC=34) #data=25
(09:09:11) Sending read of type 0x16 to 0x08, telegram: 0B 88 16 00 20 (CRC=EC)
** error sending buffer: BRK
(09:09:12) Corrupt telegram: 0B 08 10 00 00
Fetching data from EMS devices
Requesting type RC30StatusMessage(0x41) from dest 0x10
Requesting type RC30Set(0xA7) from dest 0x10
Requesting type RCTime(0x06) from dest 0x10
Requesting type UBAMonitorFast(0x18) from dest 0x08
Requesting type UBAMonitorSlow(0x19) from dest 0x08
Requesting type UBAParameterWW(0x33) from dest 0x08
Requesting type UBAParametersMessage(0x16) from dest 0x08
Requesting type UBATotalUptimeMessage(0x14) from dest 0x08
(09:09:14) Thermostat -> UBAMaster, type 0x35, telegram: 10 08 35 00 11 00 (CRC=21) #data=2
(09:09:14) Sending read of type 0x16 to 0x08, telegram: 0B 88 16 00 20 (CRC=EC)
** error sending buffer: BRK
(09:09:14) Corrupt telegram: 0B 08 10 00 00
log(09:09:17) Sending read of type 0x16 to 0x08, telegram: 0B 88 16 00 20 (CRC=EC)
** error sending buffer: BRK
(09:09:17) Corrupt telegram: 0B 08 10 00 00
(09:09:19) UBAMaster -> All, type 0x34, telegram: 08 00 34 00 3C 01 9F 80 00 21 00 00 03 00 00 C1 A2 00 6E 33 00 (CRC=E9) #data=17
(09:09:19) UBAMaster -> All, type 0x18, telegram: 08 00 18 00 41 02 53 0C 00 01 01 20 62 01 9F 80 00 02 45 00 00 11 30 45 01 09 00 00 00 (CRC=1B) #data=25
n(09:09:20) Sending read of type 0x16 to 0x08, telegram: 0B 88 16 00 20 (CRC=EC)
** error sending buffer: BRK
(09:09:20) Corrupt telegram: 0B 08 10 00 00

Tx queue (50/50)
 [1] action=read dest=0x08 type=0x16 offset=0 length=6 dataValue=32 comparisonValue=0 type_validate=0x00 comparisonPostRead=0x00 @ (14:28:00)
 [2] action=read dest=0x08 type=0x14 offset=0 length=6 dataValue=32 comparisonValue=0 type_validate=0x00 comparisonPostRead=0x00 @ (14:28:00)
 [3] action=read dest=0x10 type=0x41 offset=0 length=6 dataValue=32 comparisonValue=0 type_validate=0x00 comparisonPostRead=0x00 @ (14:29:00)
 [4] action=read dest=0x10 type=0xa7 offset=0 length=6 dataValue=32 comparisonValue=0 type_validate=0x00 comparisonPostRead=0x00 @ (14:29:00)
 [5] action=read dest=0x10 type=0x06 offset=0 length=6 dataValue=32 comparisonValue=0 type_validate=0x00 comparisonPostRead=0x00 @ (14:29:00)
 [6] action=read dest=0x08 type=0x18 offset=0 length=6 dataValue=32 comparisonValue=0 type_validate=0x00 comparisonPostRead=0x00 @ (14:29:00)
 [7] action=read dest=0x08 type=0x19 offset=0 length=6 dataValue=32 comparisonValue=0 type_validate=0x00 comparisonPostRead=0x00 @ (14:29:00)
 [8] action=read dest=0x08 type=0x33 offset=0 length=6 dataValue=32 comparisonValue=0 type_validate=0x00 comparisonPostRead=0x00 @ (14:29:00)
 [9] action=read dest=0x08 type=0x16 offset=0 length=6 dataValue=32 comparisonValue=0 type_validate=0x00 comparisonPostRead=0x00 @ (14:29:00)
 [10] action=read dest=0x08 type=0x14 offset=0 length=6 dataValue=32 comparisonValue=0 type_validate=0x00 comparisonPostRead=0x00 @ (14:29:00)
[... 15 lines cut ...]
 [26] action=read dest=0x08 type=0x14 offset=0 length=6 dataValue=32 comparisonValue=0 type_validate=0x00 comparisonPostRead=0x00 @ (14:31:00)
 [27] action=read dest=0x10 type=0x41 offset=0 length=6 dataValue=32 comparisonValue=0 type_validate=0x00 comparisonPostRead=0x00 @ (14:32:00)
 [28] action=read dest=0x10 type=0xa7 offset=0 length=6 dataValue=32 comparisonValue=0 type_validate=0x00 comparisonPostRead=0x00 @ (14:32:00)
 [29] action=read dest=0x10 type=0x06 offset=0 length=6 dataValue=32 comparisonValue=0 type_validate=0x00 comparisonPostRead=0x00 @ (14:32:00)
 [30] action=read dest=0x08 type=0x18 offset=0 length=6 dataValue=32 comparisonValue=0 type_validate=0x00 comparisonPostRead=0x00 @ (14:32:00)
 [31] action=read dest=0x08 type=0x19 offset=0 length=6 dataValue=32 comparisonValue=0 type_validate=0x00 comparisonPostRead=0x00 @ (14:32:00)
 [32] action=read dest=0x08 type=0x33 offset=0 length=6 dataValue=32 comparisonValue=0 type_validate=0x00 comparisonPostRead=0x00 @ (14:32:00)
 [33] action=read dest=0x08 type=0x16 offset=0 length=6 dataValue=32 comparisonValue=0 type_validate=0x00 comparisonPostRead=0x00 @ (14:32:00)
 [34] action=read dest=0x08 type=0x14 offset=0 length=6 dataValue=32 comparisonValue=0 type_validate=0x00 comparisonPostRead=0x00 @ (14:32:00)
 [35] action=read dest=0x10 type=0x41 offset=0 length=6 dataValue=32 comparisonValue=0 type_validate=0x00 comparisonPostRead=0x00 @ (14:33:00)
 [36] action=read dest=0x10 type=0xa7 offset=0 length=6 dataValue=32 comparisonValue=0 type_validate=0x00 comparisonPostRead=0x00 @ (14:33:00)
 [37] action=read dest=0x10 type=0x06 offset=0 length=6 dataValue=32 comparisonValue=0 type_validate=0x00 comparisonPostRead=0x00 @ (14:33:00)
 [38] action=read dest=0x08 type=0x18 offset=0 length=6 dataValue=32 comparisonValue=0 type_validate=0x00 comparisonPostRead=0x00 @ (14:33:00)
 [39] action=read dest=0x08 type=0x19 offset=0 length=6 dataValue=32 comparisonValue=0 type_validate=0x00 comparisonPostRead=0x00 @ (14:33:00)
 [40] action=read dest=0x08 type=0x33 offset=0 length=6 dataValue=32 comparisonValue=0 type_validate=0x00 comparisonPostRead=0x00 @ (14:33:00)
 [41] action=read dest=0x08 type=0x16 offset=0 length=6 dataValue=32 comparisonValue=0 type_validate=0x00 comparisonPostRead=0x00 @ (14:33:00)
 [42] action=read dest=0x08 type=0x14 offset=0 length=6 dataValue=32 comparisonValue=0 type_validate=0x00 comparisonPostRead=0x00 @ (14:33:00)
 [43] action=read dest=0x10 type=0x41 offset=0 length=6 dataValue=32 comparisonValue=0 type_validate=0x00 comparisonPostRead=0x00 @ (14:34:00)
 [44] action=read dest=0x10 type=0xa7 offset=0 length=6 dataValue=32 comparisonValue=0 type_validate=0x00 comparisonPostRead=0x00 @ (14:34:00)
 [45] action=read dest=0x10 type=0x06 offset=0 length=6 dataValue=32 comparisonValue=0 type_validate=0x00 comparisonPostRead=0x00 @ (14:34:00)
 [46] action=read dest=0x08 type=0x18 offset=0 length=6 dataValue=32 comparisonValue=0 type_validate=0x00 comparisonPostRead=0x00 @ (14:34:00)
 [47] action=read dest=0x08 type=0x19 offset=0 length=6 dataValue=32 comparisonValue=0 type_validate=0x00 comparisonPostRead=0x00 @ (14:34:00)
 [48] action=read dest=0x08 type=0x33 offset=0 length=6 dataValue=32 comparisonValue=0 type_validate=0x00 comparisonPostRead=0x00 @ (14:34:00)
 [49] action=read dest=0x08 type=0x16 offset=0 length=6 dataValue=32 comparisonValue=0 type_validate=0x00 comparisonPostRead=0x00 @ (14:34:00)
 [50] action=read dest=0x08 type=0x14 offset=0 length=6 dataValue=32 comparisonValue=0 type_validate=0x00 comparisonPostRead=0x00 @ (14:34:00)

proddy, 2019-12-27:
Have you tried different `tx_mode` settings? https://github.com/proddy/EMS-ESP/wiki/Troubleshooting

GLD-NL, 2019-12-27:
Below the results of testing with the different TX modes. I hope you can something with this information.

*** EMS+:
The strange thing about EMS+ is that TX indicates that there is no signal "Tx: no signal". But settings can be sent to the thermostat. Furthermore, only the thermostat is recognized. The boiler is no longer detected by the EMS+ bus. The TX queue now behaves normally and is most of the time empty.

EMS-ESP system stats:
  System logging set to None
  LED: on, Listen mode: off
  Boiler: disabled, Thermostat: enabled, Solar Module: disabled, Mixing Module: disabled
  Shower Timer: disabled, Shower Alert: disabled

EMS Bus stats:
  Bus is connected, protocol: Buderus
  Rx: # successful read requests=0, # CRC errors=53
  Tx: no signal

These device IDs are on the EMS Bus: 0x08 0x09 0x10
and 1 were recognized by EMS-ESP as:
 RC30/Moduline 400 (DeviceID:0x10 ProductID:78 Version:03.03)

System Logging set to Verbose
(10:53:45) UBAMaster -> All, type 0x34, telegram: 08 00 34 00 3C 01 B0 80 00 21 00 00 03 00 00 C1 98 00 6E 35 00 (CRC=4E) #data=17
(10:53:45) UBAMaster -> All, type 0x18, telegram: 08 00 18 00 41 01 E8 09 15 09 01 25 62 01 B0 80 00 01 74 00 57 10 2D 48 00 C8 00 02 00 (CRC=D7) #data=25
(10:53:53) UBAMaster -> All, type 0x18, telegram: 08 00 18 00 41 01 EA 09 15 09 01 25 62 01 B0 80 00 01 75 00 55 10 2D 48 00 C8 00 02 00 (CRC=26) #data=25
(10:53:55) UBAMaster -> All, type 0x34, telegram: 08 00 34 00 3C 01 B0 80 00 21 00 00 03 00 00 C1 98 00 6E 35 00 (CRC=4E) #data=17
(10:53:55) UBAMaster -> All, type 0x18, telegram: 08 00 18 00 41 01 EA 09 15 09 01 25 62 01 B0 80 00 01 76 00 57 10 2D 48 00 C8 00 02 00 (CRC=B8) #data=25
(10:54:01) Thermostat -> UBAMaster, type 0x1A, telegram: 10 08 1A 00 5A 0B 00 00 (CRC=5F) #data=4
(10:54:05) UBAMaster -> All, type 0x19, telegram: 08 00 19 00 80 00 01 BF 80 00 00 00 00 32 00 CB C6 07 E9 45 00 00 00 07 27 AD 00 5D 91 00 00 (CRC=D0) #data=27
(10:54:05) UBAMaster -> All, type 0x1C, telegram: 08 00 1C 00 8E 01 07 18 28 00 00 00 00 00 00 (CRC=03) #data=11
(10:54:05) UBAMaster -> All, type 0x34, telegram: 08 00 34 00 3C 01 AF 80 00 21 00 00 03 00 00 C1 98 00 6E 35 00 (CRC=E5) #data=17
(10:54:06) UBAMaster -> All, type 0x07, telegram: 08 00 07 00 03 01 00 00 00 00 00 00 00 00 00 00 00 (CRC=DA) #data=13
(10:54:06) UBAMaster -> All, type 0x18, telegram: 08 00 18 00 41 01 EC 0B 15 09 01 25 62 01 AF 80 00 01 77 00 56 10 2D 48 00 C8 00 02 00 (CRC=41) #data=25
(10:54:15) UBAMaster -> All, type 0x34, telegram: 08 00 34 00 3C 01 AF 80 00 21 00 00 03 00 00 C1 98 00 6E 35 00 (CRC=E5) #data=17
(10:54:15) UBAMaster -> All, type 0x18, telegram: 08 00 18 00 41 01 EE 0B 15 09 01 25 62 01 AF 80 00 01 7A 00 56 10 2D 48 00 C8 00 02 00 (CRC=00) #data=25
(10:54:20) Thermostat -> UBAMaster, type 0x35, telegram: 10 08 35 00 11 00 (CRC=21) #data=2
Fetching data from EMS devices
Requesting type RC30StatusMessage(0x41) from dest 0x10
Requesting type RC30Set(0xA7) from dest 0x10
Requesting type RCTime(0x06) from dest 0x10
(10:54:25) Sending read of type 0x41 to 0x10, telegram: 0B 90 41 00 20 (CRC=69)
(10:54:25) Corrupt telegram: 0B 0B 90 90 41 41 00 00 20
(10:54:25) UBAMaster -> All, type 0x34, telegram: 08 00 34 00 3C 01 AF 80 00 21 00 00 03 00 00 C1 98 00 6E 35 00 (CRC=E5) #data=17
(10:54:25) UBAMaster -> All, type 0x18, telegram: 08 00 18 00 41 01 F0 0B 15 09 01 25 62 01 AF 80 00 01 7B 00 52 10 2D 48 00 C8 00 02 00 (CRC=1B) #data=25
(10:54:28) Sending read of type 0xA7 to 0x10, telegram: 0B 90 A7 00 20 (CRC=DA)
(10:54:28) Corrupt telegram: 0B 0B 90 90 A7 A7 00 00 20
(10:54:32) Thermostat -> All, type 0x41, telegram: 10 00 41 00 80 24 00 B1 00 00 FF FF 0D 5A 02 00 B1 00 (CRC=C0) #data=14
(10:54:33) Sending read of type 0x06 to 0x10, telegram: 0B 90 06 00 20 (CRC=6C)
** Warning, we missed the bus - Rx non-idle!
(10:54:33) Corrupt telegram: 02 F0 B1 03 00 10
(10:54:35) UBAMaster -> All, type 0x34, telegram: 08 00 34 00 3C 01 AF 80 00 21 00 00 03 00 00 C1 98 00 6E 35 00 (CRC=E5) #data=17
(10:54:35) UBAMaster -> All, type 0x18, telegram: 08 00 18 00 41 01 F2 0B 15 09 01 25 62 01 AF 80 00 01 7D 00 52 10 2D 48 00 C8 00 02 00 (CRC=FD) #data=25
(10:54:36) Thermostat -> All, type 0x06, telegram: 10 00 06 00 13 0C 0A 1B 38 21 04 00 00 00 00 00 00 (CRC=03) #data=13
(10:54:37) Thermostat -> All, type 0xA3, telegram: 10 00 A3 00 FF 00 00 (CRC=79) #data=3

*** Junkers Heatronic HT3:
This gives the best results. The HT3 protocol detects the base controller and the thermostat. TX now gives that it is connected. The queue is also empty now.

EMS-ESP system stats:
  System logging set to None
  LED: on, Listen mode: off
  Boiler: disabled, Thermostat: enabled, Solar Module: disabled, Mixing Module: disabled
  Shower Timer: disabled, Shower Alert: disabled

EMS Bus stats:
  Bus is connected, protocol: Buderus
  Rx: # successful read requests=14, # CRC errors=70
  Tx: Last poll=2.565 seconds ago, # successful write requests=2

These device IDs are on the EMS Bus: 0x08 0x09 0x10
and 2 were recognized by EMS-ESP as:
[... 16 lines cut ...]
(11:11:41) UBAMaster -> All, type 0x18, telegram: 08 00 18 00 41 02 37 0D 15 09 01 25 62 01 B0 80 00 01 CA 00 59 11 2D 48 00 C8 00 02 00 (CRC=8D) #data=25
(11:11:42) UBAMaster -> All, type 0x34, telegram: 08 00 34 00 3C 01 B0 80 00 21 00 00 03 00 00 C1 98 00 6E 35 00 (CRC=4E) #data=17
(11:11:42) Thermostat -> UBAMaster, type 0x1A, telegram: 10 08 1A 00 5A 0D 00 00 (CRC=47) #data=4
Fetching data from EMS devices
[... cut ...]
(11:11:54) Sending read of type 0x06 to 0x10, telegram: 0B 90 06 00 20 (CRC=6C)
** Warning, we missed the bus - Rx non-idle!
(11:11:54) Corrupt telegram: 02 6C 0C 10 0B 06 00 13 0C 0B 1B 0D 34 04 00 00 00 00 00 00
(11:11:55) Thermostat -> All, type 0x06, telegram: 10 00 06 00 13 0C 0B 1B 0D 34 04 00 00 00 00 00 00 (CRC=E2) #data=13
(11:11:55) UBAMaster -> All, type 0x07, telegram: 08 00 07 00 03 01 00 00 00 00 00 00 00 00 00 00 00 (CRC=DA) #data=13
(11:11:56) Thermostat -> All, type 0xA3, telegram: 10 00 A3 00 00 00 00 (CRC=AE) #data=3

proddy, 2019-12-27:
Did you build the circuit yourself? There was a mistake in the tx circuit
of my original design which was corrected a while ago. And you said all was
working fine with 1.8.x?

GLD-NL, 2019-12-27:
What I use is a board from BBQKees. I bought this at the end of February 2018

proddy, 2019-12-27:
Ok, it should work. Your tx_mode should be set to 1 for the Buderus/Nefit  topline. Can you confirm that everything was ok when you were using version 1.8.x and this only happened in 1.9.x?

GLD-NL, 2019-12-27:
yep! In the 1.8.3 version, the queue was no problem. I tested all versions from 1.9.0 and later. All have the same problem. With the EMS protocol (option 1) The queue is full within a few minutes. If I can help you. Pleace let me know.

proddy, 2019-12-30:
I can't think of what is causing this. The Tx code (with tx_mode set to 1) is the same as in version 1.8.3. Looking at your logs sometimes I see a missing BRK signal, sometimes there are echo's and sometimes just garbage.

GLD-NL, 2019-12-30:
Hi proddy,

Sorry for this! Here some logging with firmware 1.8.3 running on the wemos-d1. Can it be that the problem not is in the protocol but in the way how you manage the events in the queue? I see that 1.8.3 has a maximum of 100 events and in version 1.9.x is the maximum 50.

info
EMS-ESP system stats:
  System logging set to None
  LED is on, Listen mode is off
  Boiler is enabled, Thermostat is enabled, Solar Module is disabled, Shower Timer is disabled, Shower Alert is disabled

EMS Bus stats:
  Bus is connected
  Rx: # successful read requests=375, # CRC errors=141
  Tx: Last poll=2.390 seconds ago, # successful write requests=2

queue
Tx queue is empty

log v
System Logging set to Verbose
(01:06:46.255) Boiler -> all, type 0x18, telegram: 08 00 18 00 41 02 4A 0F 15 09 01 25 62 01 E5 80 00 01 DC 00 53 11 2D 48 00 C8 00 02 00 (CRC=D1) #data=25
<--- UBAMonitorFast(0x18)
(01:06:46.570) Boiler -> all, type 0x34, telegram: 08 00 34 00 3C 01 E5 80 00 21 00 00 03 00 00 C1 BF 00 6E 3B 00 (CRC=DA) #data=17
<--- UBAMonitorWWMessage(0x34)
(01:06:56.317) Boiler -> all, type 0x18, telegram: 08 00 18 00 41 02 4B 0F 15 09 01 25 62 01 E5 80 00 01 DD 00 57 11 2D 48 00 C8 00 02 00 (CRC=0E) #data=25
<--- UBAMonitorFast(0x18)
(01:06:56.533) Boiler -> all, type 0x34, telegram: 08 00 34 00 3C 01 E5 80 00 21 00 00 03 00 00 C1 BF 00 6E 3B 00 (CRC=DA) #data=17
<--- UBAMonitorWWMessage(0x34)
(01:06:59.112) Thermostat -> all, type 0xA2, telegram: 10 00 A2 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=C0) #data=13
(01:07:01.961) Thermostat -> Boiler, type 0x1A, telegram: 10 08 1A 00 5A 0F 00 00 (CRC=4F) #data=4
(01:07:06.254) Boiler -> all, type 0x18, telegram: 08 00 18 00 41 02 4C 0F 15 09 01 25 62 01 E5 80 00 01 DD 00 5E 11 2D 48 00 C8 00 02 00 (CRC=F0) #data=25
<--- UBAMonitorFast(0x18)
(01:07:06.470) Boiler -> all, type 0x34, telegram: 08 00 34 00 3C 01 E5 80 00 21 00 00 03 00 00 C1 BF 00 6E 3B 00 (CRC=DA) #data=17
<--- UBAMonitorWWMessage(0x34)
(01:07:16.192) Boiler -> all, type 0x18, telegram: 08 00 18 00 41 02 4D 0F 15 09 01 25 62 01 E5 80 00 01 DE 00 5D 11 2D 48 00 C8 00 02 00 (CRC=A8) #data=25
<--- UBAMonitorFast(0x18)
(01:07:16.407) Boiler -> all, type 0x34, telegram: 08 00 34 00 3C 01 E5 80 00 21 00 00 03 00 00 C1 BF 00 6E 3B 00 (CRC=DA) #data=17
<--- UBAMonitorWWMessage(0x34)
(01:07:18.135) Boiler -> all, type 0x18, telegram: 08 00 18 00 41 02 4E 0F 15 09 01 25 62 01 E5 80 00 01 DF 00 5C 11 2D 48 00 C8 00 02 00 (CRC=AD) #data=25
<--- UBAMonitorFast(0x18)
Requesting scheduled EMS device data
Requesting type RC30StatusMessage(0x41) from dest 0x10
Requesting type RC30Set(0xA7) from dest 0x10
Requesting type RCTime(0x06) from dest 0x10
Requesting type UBAMonitorFast(0x18) from dest 0x08
Requesting type UBAMonitorSlow(0x19) from dest 0x08
Requesting type UBAParameterWW(0x33) from dest 0x08
Requesting type UBAParametersMessage(0x16) from dest 0x08
Requesting type UBATotalUptimeMessage(0x14) from dest 0x08
(01:07:23.608) Sending read of type 0x41 to 0x10:telegram: 0B 90 41 00 20 (CRC=69) #data=108
(01:07:23.737) Thermostat -> me, type 0x41, telegram: 10 0B 41 00 80 24 00 B5 00 00 FF FF 0F 5A 02 00 B5 00 (CRC=1B) #data=14
<--- RC30StatusMessage(0x41)
(01:07:23.775) Boiler -> all, type 0x07, telegram: 08 00 07 00 0B 01 00 00 00 00 00 00 00 00 00 00 00 (CRC=F6) #data=13
(01:07:23.980) Sending read of type 0xA7 to 0x10:telegram: 0B 90 A7 00 20 (CRC=DA) #data=164
(01:07:24.150) Corrupt telegram: 0F 10 0B A7 00 01 00 FF F6 01 06 00 01 0D 01 00 FF FF 01 02 02 02 00 00 05 1F 05 1F 02 0E 00 00 E7
(01:07:25.120) Boiler -> all, type 0x07, telegram: 08 00 07 00 03 01 00 00 00 00 00 00 00 00 00 00 00 (CRC=DA) #data=13
(01:07:26.105) Boiler -> all, type 0x18, telegram: 08 00 18 00 41 02 4F 0F 15 09 01 25 62 01 E5 80 00 01 DF 00 5B 11 2D 48 00 C8 00 02 00 (CRC=3D) #data=25
<--- UBAMonitorFast(0x18)
(01:07:26.381) Boiler -> all, type 0x19, telegram: 08 00 19 00 80 00 02 21 80 00 00 00 00 32 00 CB F6 07 EF 32 00 00 00 07 2D 73 00 5D BB 00 00 (CRC=83) #data=27
<--- UBAMonitorSlow(0x19)
(01:07:26.587) Boiler -> all, type 0x1C, telegram: 08 00 1C 00 8E 01 07 18 28 00 00 00 00 00 00 (CRC=03) #data=11
(01:07:26.819) Boiler -> all, type 0x34, telegram: 08 00 34 00 3C 01 E5 80 00 21 00 00 03 00 00 C1 BF 00 6E 3B 00 (CRC=DA) #data=17
<--- UBAMonitorWWMessage(0x34)
(01:07:27.112) Boiler -> all, type 0x07, telegram: 08 00 07 00 03 01 00 00 00 00 00 00 00 00 00 00 00 (CRC=DA) #data=13
(01:07:27.326) Boiler -> all, type 0x18, telegram: 08 00 18 00 41 02 4F 0F 15 09 01 25 62 01 E5 80 00 01 DF 00 56 11 2D 48 00 C8 00 02 00 (CRC=88) #data=25
<--- UBAMonitorFast(0x18)
(01:07:27.965) Sending read of type 0x06 to 0x10:telegram: 0B 90 06 00 20 (CRC=6C) #data=108
(01:07:28.110) Corrupt telegram: 1B 10 0B 06 00 13 0C 0D 1E 38 2D 00 00 00 00 00 00 00 9E
(01:07:28.357) Boiler -> all, type 0x07, telegram: 08 00 07 00 0B 01 00 00 00 00 00 00 00 00 00 00 00 (CRC=F6) #data=13
(01:07:28.613) Sending read of type 0x18 to 0x08:telegram: 0B 88 18 00 20 (CRC=D4)
(01:07:28.669) Corrupt telegram: 0F 08 0B 18 00 41 02 4F 0F 15 09 01 25 62 01 E5 80 00 01 DF 00 53 11 2D 48 00 C8 00 02 00 46
(01:07:29.331) Thermostat -> Boiler, type 0x35, telegram: 10 08 35 00 11 00 (CRC=21) #data=2
(01:07:29.534) Sending read of type 0x19 to 0x08:telegram: 0B 88 19 00 20 (CRC=D0)
(01:07:29.593) Corrupt telegram: 0F 08 0B 19 00 80 00 02 22 80 00 00 00 00 32 00 CB F6 07 EF 32 00 00 00 07 2D 73 00 5D BB 00 00 8D
[... cut ...]
(01:07:34.161) Boiler -> all, type 0x07, telegram: 08 00 07 00 03 01 00 00 00 00 00 00 00 00 00 00 00 (CRC=DA) #data=13
(01:07:36.117) Boiler -> all, type 0x18, telegram: 08 00 18 00 41 02 51 0F 15 09 01 25 62 01 E5 80 00 01 E1 00 58 11 2D 48 00 C8 00 02 00 (CRC=7D) #data=25
<--- UBAMonitorFast(0x18)
(01:07:36.332) Boiler -> all, type 0x34, telegram: 08 00 34 00 3C 01 E5 80 00 21 00 00 03 00 00 C1 BF 00 6E 3B 00 (CRC=DA) #data=17
<--- UBAMonitorWWMessage(0x34)
(01:07:38.712) Thermostat -> all, type 0xA3, telegram: 10 00 A3 00 F6 00 00 (CRC=5D) #data=3
<--- RCOutdoorTempMessage(0xA3)

proddy, 2019-12-30:
That's just the queue size. Even 50 is too much. I reduced it in 1.9.x to save on heap memory. As soon as something comes into the queue it only stays there until the next EMS bus poll which is every second, so nothing lingers for long.

If that was the log from 1.8.3 then I see Tx never worked either. You should be seeing any CRC errors at all. So something else is very wrong. Perhaps the wiring?

GLD-NL, 2019-12-31:
Hi proddy,

You suggested to look at the wiring. This is what I have done.

I tested with the boiler directly connected to the EMS bus using the clamps and with a stereo jack plug on the service port on the front of the boiler. Both methods give the same results.

The question is... Why is there a difference in the behavior of the queue when the wiring is not changed?

If I can help you with some testing. Let me know. Perhaps BBQKees kan give some advice. He has designed the board that I'am using.

proddy, 2019-12-31:
If you're in NL we can send you another board, or you can borrow a USB logic analyzer to see why the Tx is all mangled up?  You have the same boiler as me so I know it works.

GLD-NL, 2019-12-31:
I live in the Netherlands. If you can send another board that will be great! Then we can discard that the board is the problem. Can I send my address by email. I don't like to put it here on the site ;-)

proddy, 2019-12-31:
you'll need to contact Kees at https://bbqkees-electronics.nl/contact/ and see what he can do. I have a few spare you can borrow but you'll need to drive down to Den Bosch or Eindhoven.

bbqkees, 2019-12-31:
A Nefit ProLine boiler with a Moduline 400 is not an uncommon combination.
As it seems you are the only one with this problem it makes sense to look at the hardware (board+wemos) to see if we can identify a problem there.

GLD-NL, 2020-01-07:
Hi Proddy, it was a hardware problem! :-( I received another board from BBQKees. With the new board the queue is empty and all devices are detected!
Thanks guys!

proddy, 2020-01-07:
horray!

closed 2020-01-07 by proddy