emsesp/EMS-ESP #270: no data from MM10

MichaelDvP, 2019-12-27:
Hello,

the MM10 module is recognized, but there are no date from the modul.

devices
These device IDs are on the EMS Bus: 0x08 0x09 0x10 0x21
and 4 were recognized by EMS-ESP as:
 MC10 Module (DeviceID:0x08 ProductID:72 Version:03.00)
 BC10/RFM20 Receiver (DeviceID:0x09 ProductID:68 Version:02.03)
 RC35 (DeviceID:0x10 ProductID:86 Version:01.18)
 MM10 Mixer Module (DeviceID:0x21 ProductID:69 Version:02.01)

info
[...]
Mixing module stats:
  Switch temperature: 3200.0 C

mqtt: 
mixing_data {}

I've checked the code from 1.9.5b4 and found in ems_devices.h only EMS+ for MM100
According to wiki_ems_telegrams.pdf MM10 sends EMS-Type 0xAB with data Setpoint, flowtemperature and valvestatus. 
log give:
(16:34:11) Mixing Module -> All, type 0xAB, telegram: 21 00 AB 00 2D 01 BE 64 04 01 00 (CRC=15) #data=7

as q&d-test i added in ems_devices.h in line 198
// MM10
#define EMS_TYPE_MMStatusMessage 0xAB       // mixer status
#define EMS_OFFSET_MMStatusMessage_flow_set 0     // flow setpoint
#define EMS_OFFSET_MMStatusMessage_flow_temp 1    // flow temperature
#define EMS_OFFSET_MMStatusMessage_valve_status 3 // valve in percent

and in ems.cpp in line 1227:
void _process_MMStatusMessage(_EMS_RxTelegram * EMS_RxTelegram) {
    uint8_t hc = (EMS_RxTelegram->type - EMS_TYPE_MMStatusMessage); 
    if (hc!=0) {
        return; // invalid type
    }
    EMS_Mixing.hc[hc].active = true;
    _setValue(EMS_RxTelegram, &EMS_Mixing.hc[hc].flowTemp, EMS_OFFSET_MMStatusMessage_flow_temp);
    _setValue(EMS_RxTelegram, &EMS_Mixing.hc[hc].pumpMod, EMS_OFFSET_MMStatusMessage_flow_set);
    _setValue(EMS_RxTelegram, &EMS_Mixing.hc[hc].valveStatus, EMS_OFFSET_MMStatusMessage_valve_status);
}
and line 2670:
    {EMS_DEVICE_UPDATE_FLAG_MIXING, EMS_TYPE_MMStatusMessage, "MMStatusMessage", _process_MMStatusMessage}
	
This works so far an i get in info:
 
Mixing module stats:
  Switch temperature: 3200.0 C
  Mixing Circuit 1
   Current flow temperature: 45.3 C
   Current pump modulation: 45 %
   Current valve status: 100 %

and mqtt: mixing_data {"hc1":{"flowTemp":47.6,"pumpMod":49,"valveStatus":100}}

with the setpoint temperature in the pump modulation field.
I dont know what pump modulation is in MM100, but i found:
https://github.com/Th3M3/buderus_ems-wiki/blob/master/Quelle_21.md
that MM50 has EMS+ and in message type 0x01d8 at address 5 there is also the temperature setpoint and no pump modulation. 

Proddy can you please add MM10 to official code?

PhillyGilly, 2019-12-27:
Is that MM10 or MM100? I think you meant MM100. I am hoping to install a MM200 next year and my understanding is that the MM200 is essentially two MM100 in a single unit, so working code for your unit should be a positive for me.

MichaelDvP, 2019-12-28:
Don't worry, mine is MM10, an older EMS-device. 
MM50 to MM400 are EMS+ and they are covered by the existing code.

proddy, 2019-12-28:
@MichaelDvP if your code works why not try a PR on the dev branch? I'd be happy to review and add it to the master

MichaelDvP, 2019-12-30:
Ok, i'll try. But i'm new to github and first have to learn how to make a pull request. Give me some time.

proddy, 2019-12-30:
merged, great work Michael!

lmdecroos, 2019-12-31:
I'm testing 1.9.5b7 and this release included additional information about the MM10 mixing module, but the current flow temperature is not correct in the "mixing module stat". the details are correct in the boiler.

EMS-ESP version: 1.9.5b7
New EMS device recognized as a Mixing Module: MM10 Mixer Module (DeviceID:0x21 ProductID:69 Version:02.01)

   Boiler stats:
  Boiler: Nefit Topline/Buderus GB162 (DeviceID:0x08 ProductID:115 Version:05.05)
  ....
  **Selected flow temperature: 60 C**
  **Current flow temperature: 60.4 C**
  Return temperature: 55.5 C
  Gas: on
  Boiler pump: on
  Fan: on
  Ignition: off
  Circulation pump: off
  Burner selected max power: 100 %
  Burner current power: 28 %
  Flame current: 10.7 uA
  System pressure: 1.4 bar
  System service code: -H (200)
  Heating temperature setting on the boiler: 76 C
  Boiler circuit pump modulation max power: 91 %
  Boiler circuit pump modulation min power: 30 %
  Boiler temperature: 58.9 C
  Exhaust temperature: ? C
  **Pump modulation: 37 %**

  Mixing module stats:
  Switch temperature: 58.0 C
  Mixing Circuit 1
   **Current flow temperature: 3353.6 C**
   **Setpoint flow temperature: 0 C**
   **Current pump modulation: 0 %**

proddy, 2019-12-31:
made a small change - see if it fixes the "3353.6 C" problem

lmdecroos, 2019-12-31:
Still the same with b8

EMS-ESP system stats:
  System logging set to None
  LED: on, Listen mode: off
  Boiler: enabled, Thermostat: enabled, Solar Module: disabled, Mixing Module: enabled
  Shower Timer: disabled, Shower Alert: disabled

EMS Bus stats:
  Bus is connected, protocol: Buderus
  Rx: # successful read requests=36, # CRC errors=0
  Tx: Last poll=2.590 seconds ago, # successful write requests=0

Boiler stats:
  Boiler: Nefit Topline/Buderus GB162 (DeviceID:0x08 ProductID:115 Version:05.05)
  Hot tap water: off
  Central heating: active
  Warm Water activated: on
  Warm Water circulation pump available: off
  Warm Water comfort setting: Hot
  Warm Water selected temperature: 56 C
  Warm Water desinfection temperature: 70 C
  Warm Water current temperature: 61.3 C
  Warm Water current tap water flow: 0.0 l/min
  Warm Water # starts: 5658 times
  Warm Water active time: 28 days 6 hours 14 minutes
  Warm Water 3-way valve: off
  Selected flow temperature: 72 C
  Current flow temperature: 44.2 C
  Return temperature: 36.1 C
  Gas: on
  Boiler pump: on
  Fan: on
  Ignition: off
  Circulation pump: off
  Burner selected max power: 100 %
  Burner current power: 52 %
  Flame current: 10.5 uA
  System pressure: 1.4 bar
  System service code: -H (200)
  Heating temperature setting on the boiler: 76 C
  Boiler circuit pump modulation max power: 91 %
  Boiler circuit pump modulation min power: 30 %
  Boiler temperature: 43.4 C
  Exhaust temperature: ? C
  Pump modulation: 43 %
  Burner # starts: 27472 times
  Total burner operating time: 299 days 16 hours 16 minutes
  Total heat operating time: 271 days 10 hours 2 minutes
  Total UBA working time: 2413 days 17 hours 29 minutes

Thermostat stats:
  Thermostat: RC35 (DeviceID:0x10 ProductID:86 Version:21.08)
  Thermostat time is 15:29:41 31/12/2019
  Heating Circuit 1
   Current room temperature: 18.8 C
   Setpoint room temperature: 19.0 C
   Day temperature: 19.0 C
   Night temperature: 15.0 C
   Vacation temperature: 17.0 C
   Mode is set to day
   Day Mode is set to day
  Heating Circuit 2
   Current room temperature: 15.8 C
   Setpoint room temperature: 19.5 C
   Day temperature: 18.0 C
   Night temperature: 13.5 C
   Vacation temperature: 17.0 C
   Mode is set to day
   Day Mode is set to day

Mixing module stats:
  Switch temperature: 39.3 C
  Mixing Circuit 1
   Current flow temperature: 3353.6 C
   Setpoint flow temperature: 0 C
   Current pump modulation: 100 %

And there is also a MQTT error on the first publish only, the following MQTT publishes of the mixing_data are ok  
Error publishing to home/ems-esp/mixing_data with payload {"hc1":{"flowTemp":3353.6,"setflowTemp":0,"pumpMod":100}} [error 0]

MichaelDvP, 2019-12-31:
can you please post the log for MM10? 
In my configuration that looks like:

log w ab
System Logging set to Watch mode
(20:55:44) 21 00 AB 00 38 02 1D 64 1D 01 00
(20:56:44) 21 00 AB 00 38 02 46 64 E5 01 00
(20:57:44) 21 00 AB 00 38 02 18 64 1E 01 00
log w ac
System Logging set to Watch mode
(20:58:13) 10 21 AC 00 38 64 01

lmdecroos, 2020-01-01:
Please find below :

Mixing module stats:
  Switch temperature: 28.1 C
  Mixing Circuit 1
   Current flow temperature: 3353.6 C
   Setpoint flow temperature: 0 C
   Current pump modulation: 0 %

log w ab

System Logging set to Watch mode
(08:13:55) 21 00 AB 00 00 83 00 00 00 01 00 
Fetching data from EMS devices
(08:14:55) 21 00 AB 00 00 83 00 00 00 01 00 
Fetching data from EMS devices
Starting scheduled MQTT publish...
Publishing boiler data via MQTT
Publishing hot water and heating states via MQTT
Publishing thermostat data via MQTT
Publishing mixing device data via MQTT
[MQTT] Error publishing to home/ems-esp/mixing_data with payload {"hc1":{"flowTemp":3353.6,"setflowTemp":0,"pumpMod":0}} [error 0]
(08:15:55) 21 00 AB 00 00 83 00 00 00 01 00 
Fetching data from EMS devices
log w ac

System Logging set to Watch mode
(08:16:28) 10 21 AC 00 05 00 01 
Fetching data from EMS devices
Starting scheduled MQTT publish...
Publishing boiler data via MQTT
Publishing thermostat data via MQTT
Publishing mixing device data via MQTT
(08:17:28) 10 21 AC 00 05 00 01 
Fetching data from EMS devices

log w 1e
(09:00:11) 11 08 1E 00 01 E7 
(09:00:21) 11 08 1E 00 01 E7 
(09:00:31) 11 08 1E 00 01 E7 
(09:00:41) 11 08 1E 00 01 E7 
(09:00:51) 11 08 1E 00 01 E7 
(09:01:01) 11 08 1E 00 01 E8 

I have a MM10 for HC1 and WM10 for HC2

proddy, 2020-01-01:
@lmdecroos the flow temp is in 0xAB. Can you send examples of that telegram? Also for 0xAC which I think is your HC2.

MichaelDvP, 2020-01-01:
@Imdecroos: Are you shure you have MM10 for HC1? 
RC35 manual writes on p.33: Mixer: Can only be set for heating circuit 2 onwards.

Your value for flowTemp is 0x8300, i have the same value for the switch temp in 0x18 telegram.
I think that this a value for "not-set" for signed short (-32000 dez). 
@proddy: in the _setValue-routine in ems.cpp you also check for  0x7D00 (i.e. +32000dez), so -32000 is not unlikely.

proddy, 2020-01-01:
yes @MichaelDvP , I think 0x8300 must be ignored. I'll add it to the code.

proddy, 2020-01-01:
it's in the latest dev now. I didn't bump the version since its a small change.

MichaelDvP, 2020-01-01:
I don't think this is correct. 
0xAB is telegram from MM10 to RC35, 0xAC is from RC35 to MM10 as in the doc/ems references described. 
In my opinion it's only possible to attach one MM10 to HC2 or higher, but we can not know in which HC it is.
The other is, that i think the temperatures switchTemp in UBASlow and flowTemp in Mixer have to be signed int and EMS_VALUE_SHORT_NOTSET has to be -32000

proddy, 2020-01-01:
then I implemented it wrong. I thought AC was for HC2 (like AB was for HC1). But you're correct, AC is sent from the RCxx thermostat.

lmdecroos, 2020-01-01:
@MichaelDvP <https://github.com/MichaelDvP>  you are correct the WM10 is connected to the HC1 pump  and has the temp sensor the MM10 is connected to the HC2 pump
sorry for the confusion

lmdecroos, 2020-01-10:
i’m still not sure witch values should displayed on for the mixing module?

Are these the same once from the boiler data ?
 Boiler stats:
Selected flow temperature: 7 C
Current flow temperature: 46.5 C
Pump modulation: 30 %

Or are they specific values from the MM10 ?

Mixing module stats:
  Switch temperature: 45.0 C
  Mixing Circuit 1
   Current flow temperature: ? C
   Setpoint flow temperature: 0 C
   Current pump modulation: 0 %

the MQTT message has still a total wrong value for the flowTemp  Topic:mixing_data Payload:{"hc1":{"flowTemp":3276.8,"setflowTemp":0,"pumpMod":0}}

MichaelDvP, 2020-01-11:
i'm not sure what MM10 does in your system. The values you get in EMS-ESP are exact what the EMS-telegram gives. Your 0xAB message reads: 21 00 AB 00 00 83 00 00 00 01 00
It's MM10(21) to all(00), message AB, offset 00, Setpoint 00, Flowtemp 8300, pump off (00), mixingvalve  00. 
So MM10 is doing nothing, or maybe switching a pump?

I have a unmodulated oil-boiler (GB125) and in hc2 a mixing circuit HSM25/E with pump, mixingvalve and sensor for flow temperatur, all conected to MM10. RC35 is configured for a mixed hc2 an no other hc.
So the boiler flow temp is much higher e.g. now i read:
  Selected flow temperature: 64 C
  Current flow temperature: 68.3 C
  Boiler temperature: 69.5 C 
and RC35 gives a calculated flow temperature 49 C (only in mqtt)
this temperature is send  via AC-telegram to MM10 and MM10 values published by AB are:
   Current flow temperature: 48.7 C
   Setpoint flow temperature: 49 C
   Current pump modulation: 100 %
this seems ok for MM10 attached to a mixing circuit.

lmdecroos, 2020-01-11:
I have 2 HC’s and 2 Mixer modules ( see attached picture)
The left one (with cover) is for HC1 and the right one ( whereof I detached the cover) is for HC2
Both has their  own dedicated pump
There is one temperature sensor, who is located where the main hot water entry from my boiler is connected.

WM10 connections:
Heat sensor
pump for HC1

MM10 connections
pump for HC2

when looking at the telegrams :

WM10Status
(20:44:59) 11 08 1E 00 00 D0 
(20:48:28) 11 08 1E 00 01 9F 
(20:55:57) 11 08 1E 00 01 63 

The 0x1E detects the temperature of the heat sensor

MM10Status
(20:53:58) 21 00 AB 00 00 83 00 64 00 01 00 
(20:54:58) 21 00 AB 00 00 83 00 64 00 01 00 
(21:02:56) 21 00 AB 00 00 83 00 00 00 01 00 

The 0xAB detects only the pump modulation for HC2

MM10Parameter
(21:17:02) 10 21 AC 00 05 00 01 
Setting thermostat mode to manual for heating circuit 2
(21:17:13) 10 21 AC 00 4B 64 01 
(21:18:13) 10 21 AC 00 4B 64 01 
(21:19:13) 10 21 AC 00 4B 64 01 
Setting thermostat mode to off for heating circuit 2
(21:19:28) 10 21 AC 00 1A 64 01

PhillyGilly, 2020-01-11:
@lmdecroos sorry but I can’t see your pictures

lmdecroos, 2020-01-12:
see at http://users.telenet.be/lmd/photos/ems/Buderus_Pump.JPG <http://users.telenet.be/lmd/photos/ems/Buderus_Pump.JPG>

proddy, 2020-01-12:
@lmdecroos I fixed the problem with the incorrect flowtemp in the MQTT `mixing_data Payload:{"hc1":{"flowTemp":3276.8,"setflowTemp":0,"pumpMod":0}}`

The code only handles a single Mixer module at the moment. It's ready to be extended except I don't know where the values will come from. It seems telegram type `0xAB` MM10Status is only for your HC1.

lmdecroos, 2020-01-12:
@proddy : thanks for fixing

sorry for my complex heating setup :)

I believe that the details pump of my HC1 is part of the boiler data

selected flow temperature: 74 C
  Current flow temperature: 81.7 C
  Return temperature: 79.5 C
  Gas: on
  Boiler pump: on
  Fan: on
  Ignition: off
  Circulation pump: off
  Burner selected max power: 100 %
  Burner current power: 17 %

boiler_data Payload:{"wWComfort":"Hot","wWSelTemp":56,"wWDesinfectionTemp":70,"selFlowTemp":74,"selBurnPow":100,"curBurnPow":0,"pumpMod":30,"wWCircPump":0,"wWCurTmp":54.3,"wWCurFlow":0,"curFlowTemp":79.9,"retTemp":79,"switchTemp":75.6,"sysPress":1.4,"boilTemp":80.2,"wWActivated":"on","wWOnetime":"off","burnGas":"off","flameCurr":0,"heatPmp":"on","fanWork":"off","ignWork":"off","wWCirc":"off","heating_temp":76,"pump_mod_max":91,"pump_mod_min":30,"wWHeat":"off","wWStarts":5677,"wWWorkM":40827,"UBAuptime":3493002,"burnStarts":27634,"burnWorkMin":437011,"heatWorkMin":396184,"ServiceCode":"0Y","ServiceCodeNumber":204}

thermostat_data Payload:{"hc1":{"seltemp":19,"currtemp":20,"daytemp":19,"nighttemp":15,"holidayttemp":17,"heatingtype":1,"circuitcalctemp":5,"mode":"auto"},"hc2":{"seltemp":18,"currtemp":16.5,"daytemp":18,"nighttemp":13.5,"holidayttemp":17,"heatingtype":1,"circuitcalctemp":74,"mode":"auto”}}

and that the pump details of my HC2 are part of the mixing module data

Mixing module stats:
  Switch temperature: 74.8 C
  Mixing Circuit 1
   Current flow temperature: ? C
   Setpoint flow temperature: 0 C
   Current pump modulation: 100 %
   
mixing_data Payload:{"hc1":{"setflowTemp":0,"pumpMod":100}}

as in the above my HC1 is not active and my HC2 is active

proddy, 2020-01-12:
no worries @lmdecroos . You're a real power user of EMS-ESP and your input has helped me understand better heating devices and improve the code.

MichaelDvP, 2020-01-12:
@lmdecroos, yes, i think i also understand better. Please correct me if i'm wrong.
The picture shows that you have 2 unmixed circuits, in the open heating circuit right is no mixingvalve. 
The WM10 switches the pump in first circuit, but in EMS is only one WM10 (id 0x11,telegram 0x1E) allowed, so MM10 is used to switch the pump in hc2 and acts as switcher not as mixer.
As far as i read in code the WM10 is in devices, but the values for flowtemp and pump are not published (am i right?). 

@proddy, maybe it's a idea to rename "Mixing module stats:" to "Mixing/Switching module stats:" and
publish values for WM10 to hc1 and MM10 to hc2.

lmdecroos, 2020-01-12:
Yes, both circuits are the same
They have each only a circulation pump http://users.telenet.be/lmd/photos/ems/GrundFos_Alpha2L.JPG <http://users.telenet.be/lmd/photos/ems/GrundFos_Alpha2L.JPG>

I removed the the temperature sensor and cooled on some metal to validate that the read-out of “log w 0x1E” was coming from this sensor.
Herby i confirmed that the ( in my system) the 
Mixing module stats:
  Switch temperature: 41.9 C
  Mixing Circuit 1
   Current flow temperature: ? C
   Setpoint flow temperature: 0 C
   Current pump modulation: 0 %
Is coming from the sensor connected to the WM10

should in my system the flow temperature and switch temperature not be the same ?
The only other place where the temperatures are measured is in the boiler itself

The current pump modulation is the value of the pump modulation of my HC2 read out from the MM10,

MichaelDvP, 2020-01-12:
The Switch temperature is comming from the boiler in message type 19 the last two bytes. Check if they are identical to the WM10 temperature in message 1E. The temperature in 1E is actual not read by EMS-ESP. Or check if Current flow temperature from boiler is identical (message18, 6+7.byte).

The values under Mixing Circuit 1 are coming from MM10 in message AB

lmdecroos, 2020-01-12:
log w 0x1E
(17:44:13) 11 08 1E 00 01 66 

log w 0x19
(17:44:41) 08 00 19 00 80 00 02 82 80 00 00 00 00 00 00 6B F9 06 AB 40 00 00 00 06 0B A1 00 55 C7 01 66 
(17:44:47) 0B 88 19 00 20 
(17:44:47) 08 0B 19 00 80 00 02 82 80 00 00 00 00 00 00 6B F9 06 AB 40 00 00 00 06 0B A1 00 55 C7 01 66 

log w 0xAB
(17:46:44) 21 00 AB 00 00 83 00 00 00 01 00 

log w 0x18
(17:47:00) 08 00 18 00 07 02 3A 00 00 00 00 00 60 00 F3 02 66 02 53 00 00 0C 30 48 00 CB 00 00 00

MichaelDvP, 2020-01-12:
Ok, the from boiler published Switch temperature mentioned in Mixing module stats is same that WM10 reported. There is no need to read the 1E message. Only the MM10 values should default to hc2 (imho). For you the only relevant value from MM10 is the pump modulation, which can be 0 or 100%, because MM10 only switches a relais.

proddy, 2020-01-12:
is MM10 always HC2 then? I can change the code but thinking about others that may have a Switch or Mixer. Or should they just be one?

MichaelDvP, 2020-01-12:
@proddy, i think yes. In RC35 hc1 is always unmixed, mixing can only configured for hc2 upwards. 
For only switching WM10 is intended to use in hc1, so MM10 for switching is also hc2. 
As far as i understand for now, and all people in this thread have MM10 in hc2, so we have no vote against ;-)

PhillyGilly, 2020-01-12:
@proddy @MichaelDvP I have an MM200 which I am planning to deploy for my two HW storage cylinders with my HC1 being controlled directly from my WB Greenstar boiler under the overall control of my WB Sense II.
At the moment the json (in testing) is coming back:
{"hc1":{"flowTemp":23,"pumpMod":0,"valveStatus":0},"hc2":{"flowTemp":37.5,"pumpMod":0}}

In my test installation I am monitoring TC1 and TC2 which give the middle temperatures of Cyl#1 and Cyl#2, and controlling VC1 and VC2  for the inlet valves as shown in this figure.  
[image]

I don't have the MM10 or MM100 documentation but I imagine that they are as flexible as the MM200 which has 8 different applications/configurations shown. This doesn't even include my proposed deployment!

I'm nearly out of my depth but seems to me that there is a philosophical design point which we I discussed with myself (!) when considering the SM200 module in issue #267. I'll try to explain. There are a number of different places where the same things are identified and known about:

1. Physical Devices (temperature probes, valves., pumps, etc)
2. Boiler/Mixer Module/Solar Module/ect connection terminals 
3. Software/Configuration inside the Boiler/MM/SM/etc
4. EMS telegrams sent/received by Boiler/MM/SM/etc
5. Json messages contain in MQTT

As I only have my own installation I don't know what rules are about the mapping between these layers.  However, there must be a one-to-one mapping between the Software/Configuration and the EMS telegrams (even though we may struggle to reverse engineer it!) and there is definitely a one-to-one mapping between the EMS and MQTT (as @Proddy writes it). It would make sense if there is a one-to-one mapping between the connection terminals and the EMS telegrams(??) and I did some testing on my SM200 to support this idea.  But for the MM200 figure above we can see that Terminals PC1 can either be PC1 or PW1, so we can't state definitively that there is always a one-to-one map between a physical device and a Json MQTT.  

Therefore mapping device names to Json runs the risk that the device names might be wrong.
In my case data labelled "flow temperature" actually relates to "middle temperature".

[For the SM200 there is temperature data in the form 80 00 in an 0x0262 and that could have brought through to the Json just-in-case to give a solution for any of 22 different application/configuration shown in the manual.  However it was agreed that this would mean a lot of work on a zero priority issue so it  has been backlogged.]

In the same way, I'm not suggesting that you don't map hc2 to MM10 but to answer @proddy 's question above my MMx will be on dhw1.

I'm not saying that you shouldn't do that and it doesn't make too much difference to me as I will read the data from the Json and map it to sensors in HA which I can name appropriately. But I just wanted to share my thinking on some of the risks involved in the way in which devices names are used.

(Sorry for the long post.)
 

of software design in hand here and it

MichaelDvP, 2020-01-13:
@PhillyGilly 
MM200 are two MM100 in one case. It is EMS+ and can be configured to every hc. This is independed from the code for MM10 as EMS-device with fixed configuration, and setting MM10 to hc2 default will not affect MM200.

PhillyGilly, 2020-01-13:
👍 👍

lmdecroos, 2020-01-31:
I'm now running 1.9.5b20 and the display of MM10 seems to be ok

Mixing module data:
  Switch temperature: 53.6 C
  Mixing Circuit 2
   Setpoint flow temperature: 0 C
   Current pump modulation: 100 %

proddy, 2020-01-31:
Can we close it? I wang to get 1.9.5 out

MichaelDvP, 2020-01-31:
Generally, yes, the only thing is that MM10 should default to hc2. RC35 manual says that hc1 is unmixed and only a switching module (WM10) is possible. In EMS+ the mixer-modules (MM50, 100, 200) have a switch to select the hc, but MM10 is fixed to device-ID 0x21 and Message 0xAB. 
I'm a bit confused, that @lmdecroos last post says Mixing Circuit 2 for 1.9.5b20, but your code fixes message 0xAB to hc1. 
I have only hc2 working because i only have one mixed circuit and no unmixed circuits.  @lmdecroos has also MM10 on hc2 and WM10 on hc1.

proddy, 2020-01-31:
ok, so fix MM to HC2 (0xAB)?

MichaelDvP, 2020-01-31:
Yes, and than close this issue.

commit cb35277b (2019-12-30): Merge pull request #276 from MichaelDvP/MM10

MM10 mixer dat added, fixes #270
closed 2020-01-31 by proddy