emsesp/EMS-ESP #271: unable to change heating temperature without thermostat connected

arturasi, 2019-12-27:
My system is boiler Junkers CerapurSmart ZWB series (with built-in water heater), no any thermostats connected. I am unable to change boiler settings, values are left unchanged. My goal is to change heating temperature, changing "Heating temperature setting on the boiler" would be an ideal solution.

I think such a behaviour is because no thermostat is connected. Boiler see that and does not allow any changes.

I want to ask maybe somebody has similar standalone system and succeed to change boiler settings? If somebody own Junkers boiler + thermostat, can you disconnect thermostat & check for example "boiler wwtemp" command, does it work or not?

This is details on configuration I have:

> devices

> These device IDs are on the EMS Bus: 0x08 0x09
and 2 were recognized by EMS-ESP as:
 Bosch Condens 2500/Buderus Logamax GB062/Junkers Cerapur Top/Worcester Greenstar i/Generic HT3 (DeviceID:0x08 ProductID:95 Version:23.11)
 Bosch Condens 2500/Buderus Logamax GB062/Junkers Cerapur Top/Worcester Greenstar i/Generic HT3 (DeviceID:0x09 ProductID:95 Version:23.11)

> EMS Bus stats:
  Bus is connected, protocol: HT3
  Rx: # successful read requests=529, # CRC errors=0
  Tx: Last poll=0.779 seconds ago, # successful write requests=4

proddy, 2019-12-30:
It should work without a thermostat as its just EMS messages thrown on the bus. What happens when you do a `boiler wwtemp 50` from telnet with `log v` enabled? What is not working?

arturasi, 2019-12-30:
I have these warm water settings set from the boiler panel:

>  Warm Water selected temperature: 50 C
  Warm Water desired temperature: 75 C
  Warm Water current temperature: 35.4 C

after "boiler wwtemp 40" I see "write failed" error, setting remains the same (50=0x32)

> Setting boiler warm water temperature to 40 C
(23:47:09) Sending write of type 0x33 to 0x09, telegram: 8B 09 33 02 28 (CRC=D0)
(23:47:09) Sending validate of type 0x33 to 0x09, telegram: 8B 89 33 02 01 (CRC=9D)
-> Write failed. Compared set value 0x28 with received value of 0x32
-> Write didn't work, retrying (1/2)...

> (00:06:03) Sending raw: 8B 89 33 02 01
(00:06:03) Controller -> Me, type 0x33, telegram: 89 0B 33 02 32 (CRC=FA) #data=1

proddy, 2019-12-30:
I think I see the problem. From your earlier post:

```
These device IDs are on the EMS Bus: 0x08 0x09
and 2 were recognized by EMS-ESP as:
Bosch Condens 2500/Buderus Logamax GB062/Junkers Cerapur Top/Worcester Greenstar i/Generic HT3 (DeviceID:0x08 ProductID:95 Version:23.11)
Bosch Condens 2500/Buderus Logamax GB062/Junkers Cerapur Top/Worcester Greenstar i/Generic HT3 (DeviceID:0x09 ProductID:95 Version:23.11)
```

It's wrongly detecting the boiler (0x08) and the controller (0x09). Probably because they are both sending back the same version data, So when you set the boiler temp you are sending it the controller and not the boiler. The correct telegram should have been `8B 08 33 02 28`. If you follow me.

Anyway we need first fix the duplication. Can you do a `log v` and then run these 2 commands 👍 
`send 0B 88 02 00 20`
and
`send 0B 89 02 00 20`

proddy, 2019-12-30:
oh and which version are you on?

arturasi, 2019-12-30:
I use EMS-ESP version 1.9.5b4

both "send 0B 88 02 00 20" and "send 0B 89 02 00 20" returns nothing (echo only). but if I use 8B instead of 0B I get this:

> (00:39:53) Sending raw: 8B 88 02 00 20
(00:39:53) UBAMaster -> Me, type 0x02, telegram: 88 0B 02 00 5F 17 0B 00 00 00 00 00 00 00 (CRC=89) #data=10

>(00:40:10) Sending raw: 8B 89 02 00 20
(00:40:10) Controller -> Me, type 0x02, telegram: 89 0B 02 00 5F 17 0B 00 00 00 00 00 00 00 (CRC=82) #data=10

arturasi, 2019-12-30:
I tried to change destination from 09 to 08 for wwtemp write but with the same result:

> (00:45:12) Sending raw: 8B 08 33 02 28
(00:45:13) UBAMaster -> All, type 0x33, telegram: 88 00 33 00 08 FF 32 00 00 00 00 02 4B 00 FF FF (CRC=B3) #data=12

proddy, 2019-12-31:
1.9.7b8 should fix the controller and master trying to both be boilers.

arturasi, 2019-12-31:
Have installed 1.9.5b8 - nothing changed, same behaviour.

proddy, 2019-12-31:
type `devices` what does it show?

proddy, 2019-12-31:
> Have installed 1.9.5b8 - nothing changed, same behaviour.

that build doesn't fix wwtemp. Looks like the HT3 uses different telegrams for changing WW. In your case I would use the 1A telegram. See https://github.com/proddy/EMS-ESP/issues/39#issuecomment-569261593

arturasi, 2019-12-31:
> devices

> These device IDs are on the EMS Bus: 0x08 0x09
and 2 were recognized by EMS-ESP as:
 UBAMaster: Bosch Condens 2500/Buderus Logamax GB062/Junkers Cerapur Top/Worcester Greenstar i/Generic HT3 (DeviceID:0x08 ProductID:95 Version:23.11)
 Controller: Bosch Condens 2500/Buderus Logamax GB062/Junkers Cerapur Top/Worcester Greenstar i/Generic HT3 (DeviceID:0x09 ProductID:95 Version:23.11)

arturasi, 2019-12-31:
"boiler flowtemp 40" or "send 8B 08 1A 00 28" works but only in some strange way (can set value less than heating setting on the boiler). 

Anyway, it controls "selected flow temperature". My goal is to control heating temperature, I can't use this command because when hot water is in use, "selected flow temperature" switches to "warm water selected temperature". So when warm water is not used only heating is working and I have this:

>   Warm Water selected temperature: 50 C
  Warm Water current tap water flow: 0.0 l/min
...
  Warm Water 3-way valve: off
  Selected flow temperature: **42 C**
...
  Heating temperature setting on the boiler: **42 C**

when warm water in use boiler switches from heating to water:

>   Warm Water selected temperature: **50 C**
  Warm Water current tap water flow: 8.0 l/min
...
  Warm Water 3-way valve: on
  Selected flow temperature: **50 C**
...
  Heating temperature setting on the boiler: 42 C
 
I tried to change "Heating temperature setting on the boiler" manually but this parameter leaves the same:

> send 8b 08 16 00 FF **2C**

> (15:25:06) Sending raw: 8B 08 16 00 FF **2C**
(15:25:06) 8B 08 16 00 FF **2C**
(15:25:07) 88 00 16 00 FF **2A** 28 00 00 FB 03 01 03 64 0A 02

So therefore I have an idea maybe boiler monitors the bus and if it detects thermostat connected it allows these "unchangeable" in my case settings to be changed...

glitter-ball, 2019-12-31:
If you change the temperature settings manually on the boiler control panel and capture the telegrams, what do these show? I'm guessing you should see some telegrams coming from 0x09 to 0x08, if it behaves like my boiler.

Just thinking that if you can change the temperature from the control panel OK, then copying these telegrams from EMS-ESP might also allow you to change them as the boiler controller thinks the changes are coming from the control panel...

arturasi, 2019-12-31:
I tried that method but no messages on bus from controller to boiler appears during change, boiler only reports new temperature back periodically to all, that's it:

> (17:01:58) UBAMaster -> All, type 0x16, telegram: 88 00 16 00 FF **27** 28 00 00 FB 03 01 03 64 0A 02 (CRC=70) #data=12
(17:01:58) UBAMaster -> All, type 0x16, telegram: 88 00 16 00 FF **25** 28 00 00 FB 03 01 03 64 0A 02 (CRC=B8) #data=12
(17:01:59) UBAMaster -> All, type 0x16, telegram: 88 00 16 00 FF **23** 28 00 00 FB 03 01 03 64 0A 02 (CRC=F9) #data=12

glitter-ball, 2020-01-01:
Does the boiler have an outside temperature sensor or another way of knowing outside temperature (Wave / Easy controllers get temperature data from the internet)?

arturasi, 2020-01-01:
Nope, nothing, no external temperature sensors etc

glitter-ball, 2020-01-01:
I’m wondering, then, how the boiler decides on an appropriate flow temperature for heating. Does the value change from day to day or is it always fixed?

arturasi, 2020-01-01:
I think it just trying to keep the current flow temperature. This is the only way because it does not have return flow temperature sensor too...

glitter-ball, 2020-01-01:
I guess you have underfloor heating? So is the 42 deg C fixed to stop you damaging the floor by having too high a flow temp? Maybe there’s an installer setting that is restricting the flow temp?

arturasi, 2020-01-01:
Yes, correct, I have underfloor heating. 42C is just current temperature set, I can change it from the boiler panel up to 60C.

glitter-ball, 2020-01-02:
Hmm... maybe it is the absence of a thermostat device, then.

When I converted my system from a time switch to EMS bus control, the boiler did a certain amount of reconfiguring when it first started. I also had to disable the boiler’s internal weather compensation so that it used the room-influenced weather compensation in the thermostat.

Now, the thermostat dictates the flow temperature to the boiler. I ought to snoop the thermostat telegrams, perhaps, and see how it does that.

glitter-ball, 2020-01-02:
OK. I've captured some telegrams from the thermostat.

`(11:15:22) Thermostat -> Me, type 0x01A5, telegram: 90 0B FF 00 01 A5 00 B8 21 25 2A 00 25 1E 02 C1 03 03 01 02 C1 00 A5 00 00 11 01 03 07 34 00 (CRC=3D) #data=25`

Here, the 11th byte, set as 2A, contains the boiler setpoint flow temperature. 2A = 42ºC, which corresponds with the value shown in HA and on the thermostat. That packet was in response to an EMS-ESP read request, so let's look for one the thermostat proactively sends to the boiler...

```
(10:56:19) Thermostat -> All, type 0x01A5, telegram: 90 00 FF 00 01 A5 00 B8 21 25 2B 00 25 1E 02 D4 03 03 01 02 D4 00 92 00 00 11 01 03 07 34 00 (CRC=66) #data=25
(10:56:19) Thermostat -> All, type 0x01A5, telegram: 90 00 FF 19 01 A5 06 04 00 08 00 00 FF 64 4B 00 6D 03 FF 01 02 (CRC=72) #data=15
```
Here, the flow temperature is 2B in the 11th byte of the first telegram, equating to 43ºC. In the second telegram above, the value 4B in the 15th byte sets the maximum heating flow temperature to 75ºC from the weather compensation heating curve configured in the thermostat.

Does any of this provide useful clues?

arturasi, 2020-01-02:
I tried to repeat same telegrams but with source address 90 changed to 0B as it is sent from EMS-ESP device to all devices on bus. First telegram (data=25 bytes) failed to send, after "send 0B 00 FF 00 01 A5...."  ESP device just kicked me out from telnet (reboot or smth, not sure). Sending of second telegram was ok but after it "info" returned the same boiler settings with all temperatures unchanged.

I think when termostat is connected, on startup it sending some initial telegrams to reconfigure the boiler.

Thinking to buy cheapest Junkers CR10 thermostat to be able to sniff what's going on the bus when it is connected and find out how temperature control is working...

VooZ2, 2020-01-04:
I have thermostat but also can't change boiler temperature:

```
> boiler wwtemp 50

(09:20:59) UBAMaster -> All, type 0x18, telegram: 88 00 18 00 26 01 72 48 00 01 03 20 C0 80 00 80 00 80 00 FF FF FF 00 00 00 00 00 00 00 (CRC=29) #data=25

Setting boiler warm water temperature to 50 C
(09:21:00) UBAMaster -> All, type 0x34, telegram: 88 00 34 00 37 02 4C 02 4C A1 00 04 03 00 00 94 64 00 0C F9 00 (CRC=C3) #data=17
(09:21:00) Sending write of type 0x33 to 0x08, telegram: 8B 08 33 02 32 (CRC=C2)
(09:21:01) UBAMaster -> All, type 0x07, telegram: 88 00 07 00 0B 01 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=5D) #data=15
(09:21:01) UBAMaster -> All, type 0x33, telegram: 88 00 33 00 08 FF 37 00 00 1E 00 01 4B 00 FF FF (CRC=B6) #data=12
(09:21:01) Sending validate of type 0x33 to 0x08, telegram: 8B 88 33 02 01 (CRC=95)
-> Write failed. Compared set value 0x32 with received value of 0x37
-> Write didn't work, retrying (1/2)...
(09:21:01) Sending write of type 0x33 to 0x08, telegram: 8B 08 33 02 32 (CRC=C2)
(09:21:01) UBAMaster -> All, type 0x33, telegram: 88 00 33 00 08 FF 37 00 00 1E 00 01 4B 00 FF FF (CRC=B6) #data=12
(09:21:02) Sending validate of type 0x33 to 0x08, telegram: 8B 88 33 02 01 (CRC=95)
-> Write failed. Compared set value 0x32 with received value of 0x37
-> Write didn't work, retrying (2/2)...
(09:21:02) Sending write of type 0x33 to 0x08, telegram: 8B 08 33 02 32 (CRC=C2)
(09:21:02) UBAMaster -> All, type 0x33, telegram: 88 00 33 00 08 FF 37 00 00 1E 00 01 4B 00 FF FF (CRC=B6) #data=12
(09:21:02) Sending validate of type 0x33 to 0x08, telegram: 8B 88 33 02 01 (CRC=95)
-> Write failed. Compared set value 0x32 with received value of 0x37
-> Write failed. Giving up, removing from queue
```

My devices:
```
devices

These device IDs are on the EMS Bus: 0x08 0x09 0x10
and 3 were recognized by EMS-ESP as:
 UBAMaster: Bosch Condens 2500/Buderus Logamax GB062/Junkers Cerapur Top/Worcester Greenstar i/Generic HT3 (DeviceID:0x08 ProductID:95 Version:23.12)
 Controller: HT3 Controller (DeviceID:0x09 ProductID:95 Version:23.12)
 Thermostat: Junkers FW120 (DeviceID:0x10 ProductID:192 Version:53.03)
```

arturasi, 2020-01-04:
Have got Junkers CW100 thermostat, so now I can experiment with my system & sniff bus traffic to find out what's going on & how thermostat changes temperature on the boiler. Will report here if I find something useful.

VooZ2, 2020-01-10:
@arturasi any news?

arturasi, 2020-01-10:
I checked the bus traffic with CW100 connected, the algo of controlling heating temperature seems to be the following:

- Thermostat periodically sends "selected flow temperature" (type 1A) to the boiler + 2 other messages of type 35 & 23. Message 35 is always the same, 1A and 23 has desired temperature inside (1C=28C in my case):

> (21:59:54) Thermostat -> UBAMaster, type 0x1A, telegram: 98 08 1A 00 **1C** 64 64 (CRC=21) #data=3
...
(22:00:00) Thermostat -> UBAMaster, type 0x35, telegram: 98 08 35 00 11 01 (CRC=B0) #data=2
...
(22:00:23) Thermostat -> UBAMaster, type 0x23, telegram: 98 08 23 00 **1C** FF 64 (CRC=B5) #data=3

- Heating temperature set on the boiler's panel work as a limit, so if you set 45C on the boiler then thermostat cannot set value larger.

proddy, 2020-01-18:
I was looking through https://github.com/norberts1/hometop_HT3/blob/master/HT3/sw/etc/html/HT3-Bus_Telegramme.html today (which is amazing) and found a telegram type 0x1B which I hadn't come across before that supposedly sets the Warm Water on a Junkers HT3. You could test it, but be careful!

`send 8B 00 1B 00 XX`

where XX is the temp.

wharfedale01, 2020-02-08:
More news?

proddy, 2020-09-12:
closing this due to no activity. I'm unable to test.

closed 2020-09-12 by proddy