emsesp/EMS-ESP #272: Topic boiler_data is not submitted after upgrading to 1.9.4

kkberg, 2019-12-28:
*Before creating a new issue please check that you have:*

* *searched the existing [issues](https://github.com/proddy/EMS-ESP/issues) (both open and closed)*
* *searched the [wiki help pages](https://github.com/proddy/EMS-ESP/wiki/Troubleshooting)*

*Completing this template will help developers and contributors to address the issue. Try to be as specific and extensive as possible. If the information provided is not enough the issue will likely be closed.*

*You can now remove this line and the above ones. Text in italic is meant to be replaced by your own words. If any of the sections below are not relevant to the issue (for instance, the screenshots) then you can delete them.*

**Topic boiler_data is not submitted after upgrading to 1.9.4**
*in 1.9.3 topic boiler_data is directly submitted to MQTT IOBroker*

**update firmware from 1.9.3 to 1.9.4**
*start module*

**topic boiler_data should be submitted soon**
*I wamt to see boiler_data topic after starting WEMOS D1 Mini*

**Screenshots**
[image]
[image]

**Device information**
*Copy-paste here the information as it is outputted by the device. You can get this information by from the telnet session using the `system` command and `info`.*

**Additional context**
*Add any other context about the problem here.*

proddy, 2019-12-28:
1.9.5 has some improvements but it’s still not ideal. Can you try that and use publish_time of 10

kkberg, 2019-12-28:
Need to find the bin for 1.9.5 :)

kkberg, 2019-12-28:
Changed to dev 1.9.5, but no change. Even with 10 seconds publish time. I go back to 1.9.3. If it helps, I'm going to test coming betas to solve problem.

proddy, 2019-12-30:
I can't reproduce using the latest 1.9.5. `mqttlog` shows that the boiler_data is sent and in MQTT Explorer I see the inbound messages.

kkberg, 2020-01-02:
Okay, I give up.

kkberg, 2020-01-05:
Only for information. Time between topics is too long. Don't know what it is. 1.9.4 never sent boiler_data to iobroker.

[image]

proddy, 2020-01-05:
Ok, so you are getting the boiler data sent to the MQTT broker. But it just takes a long time? In 1.9.4 what happens when you use these commands in telnet with logging enabled:

`refresh` - does it fetch that data and do a publish?
`publish` - does it publish the data?

also check your `publish_time` parameter.

kkberg, 2020-01-05:
only heartbeat and shower_data is published
[image]

[image]

proddy, 2020-01-05:
go to telnet
type `log v` BEFORE doing the refresh or publish so you see what is happening
and type `set` to see your settings, check publish_time

https://github.com/proddy/EMS-ESP/wiki/Troubleshooting

kkberg, 2020-01-05:
[image]
[image]
[image]

proddy, 2020-01-05:
you have listen_mode set to on. Which means no Tx so probably the boiler is not detected. Do you see it in `info`.

kkberg, 2020-01-05:
I don't want to us Tx and wire is for Tx is also not connected. It works with 1.9.3 and same settings. I only want to see values.

proddy, 2020-01-05:
and `info` shows.....?

kkberg, 2020-01-05:
Boiler not enabled

proddy, 2020-01-05:
well, there's your answer. no boiler = no data. Since 1.9.4 Tx is needed to send requests to each of the connected devices to identify them. Otherwise it has no idea of the model and type. The project supports many brands now. Just hook up Tx and it should work.

closed 2020-01-05 by proddy

kkberg, 2020-01-05:
okay, good to know and I've to connect Tx. Wasn't aware off that and wanted to make sure that nothe can happen accidently. But thank you very much and I'm sorry for my mistake.

proddy, 2020-01-05:
You don't to worry. Tx is used only to send read "query" requests, not write requests so its harmless.

kkberg, 2020-01-05:
I now have a lot of detailed information more than in the earlier version. I'm really very happy with it. Thank you again.

proddy, 2020-01-05:
I will also look at the code and give a warning if Tx is not 'active'. I'm sure others will run into the same problem.