emsesp/EMS-ESP #273: Question: connection to MQTT Broker

giovanne123, 2019-12-29:
I started to use EMS-ESP version 1.9.4 on Wemos D1 mini and the bbqkees interface board.
The Wemos itself with the firmeware is working.

First I want to test the MQTT setup, therefore I configured the ems-esp within the MQTT Settings by activating and setting the broker IP and port.

The broker is a mosquitto mqtt broker in the same network (without user/pwd so far for testing).
Mosquitto is up and running an can be used (sub, pub) by different clients (windows, linux, android) in the network like, mosquitto_pub, mqtt-spy, ...

Within EMS-ESP I get:
```
...
Error! Unable to read the EMS bus.
**[MQTT] Connecting to MQTT...
[MQTT] TCP Disconnected
[MQTT] Disconnected!** (count 208)
Error! Unable to read the EMS bus.
...
```

System info:
```
ESP8266 System stats:

 [APP] EMS-ESP version: 1.9.4
 [APP] MyESP version: 1.2.22
 [APP] Build timestamp: 2019-12-18 11:26:06
 [APP] Uptime: 0 days 23 hours 35 minutes 9 seconds
 [APP] System Load: 1%
 [WIFI] WiFi Hostname: xx-xx-xx
 [WIFI] WiFi IP: 192.168.0.xx
 [WIFI] WiFi signal strength: 100%
 [WIFI] WiFi MAC: xx:xx:xx:xx:xx:xx
 [MQTT] is disconnected
 [NTP] Local Time is 12:28:58 CET  (1577622538)
 [SYSTEM] System is Stable
 [SYSTEM] Board: PLATFORMIO_D1_MINI
 [SYSTEM] CPU frequency: 80 MHz
 [SYSTEM] SDK version: 2.2.2-dev(38a443e)
 [SYSTEM] CPU chip ID: 0xXXXXXX
 [SYSTEM] Core version: 2_6_2
 [SYSTEM] Boot version: 31
 [SYSTEM] Boot mode: 1
 [SYSTEM] Last reset reason: Restart from terminal
 [SYSTEM] Restart count: 0
 [SYSTEM] # TCP disconnects: 214
...
```

Info: EMS bus isn't connected so far, I want to configure MQTT first.

**How can I analyse why EMS-ESP isn't connecting to the MQTT broker?** 
(or does it only work with connected EMS bus? I don't think so)
(also I think that for first MQTT tests the mosquitto broker is sufficient and Home-assistant isn't required, or?)

proddy, 2019-12-29:
Check your settings, using `set`

giovanne123, 2019-12-29:
the settings looking good using set, identical like in the web GUI:

```
Current settings:

  wifi_mode=client
  wifi_ssid=xyz_WLAN
  wifi_password=*****************************
  mqtt_enabled=on
  mqtt_ip=192.168.0.32
  mqtt_username=
  mqtt_password=
  mqtt_base=home
  mqtt_port=1883
  mqtt_keepalive=60
  mqtt_retain=off
  mqtt_qos=0
  mqtt_heartbeat=off
...
```

still :-(
```
[MQTT] Connecting to MQTT...
[MQTT] TCP Disconnected
[MQTT] Disconnected! (count 124)

```
with the above info/config (ip, port) I can connect to mosquitto with different other client.

giovanne123, 2019-12-30:
Now I have connected the EMS Bus lines also and the EMS is working fine :-)

**But I can't get the MQTT connection running, how can I debug where the problem is?** (see infos in below comments)

proddy, 2019-12-30:
ok, this is what I do when MQTT is failing:
- install mosquitto locally (on laptop/PC). Go to https://mosquitto.org/download/ , You only need the single executable and save this in a directory. You can delete it later. Or use the installer and it'll place it in C:\Program Files\mosquitto
- open a shell/dos prompt, go to the directory where you downloaded mosquitto and run 'mosquitto -v' to turn on verbose mode
- change EMS-ESP to point to the IP of the machine you just installed the broker
- see the logs from mosquitto. This will tell if there is a connection at all and what is causing the discconnects.

should take no more than 5 minutes...report back here or use Gitter

giovanne123, 2019-12-30:
ok, thanks.
Result - on EMS-ESP everything is ok :-) EMS-ESP is connecting fine to the mosquitto broker on windows laptop:
```
E:\Portable_Apps\mosquitto>mosquitto -v
1577705413: mosquitto version 1.6.8 starting
1577705413: Using default config.
1577705413: Opening ipv6 listen socket on port 1883.
1577705413: Opening ipv4 listen socket on port 1883.
1577705610: New connection from 192.168.0.129 on port 1883.
1577705610: New client connected from 192.168.0.129 as HHH-EMS-ESP (p2, c0, k60).
1577705610: Will message specified (7 bytes) (r1, q1).
1577705610:     home/HHH-EMS-ESP/status
1577705610: Sending CONNACK to HHH-EMS-ESP (0, 0)
1577705610: Received PUBLISH from HHH-EMS-ESP (d0, q0, r1, m0, 'home/HHH-EMS-ESP/status', ... (6 bytes))
1577705610: Received SUBSCRIBE from HHH-EMS-ESP
1577705610:     home/HHH-EMS-ESP/restart (QoS 0)
...
1577705610: Received PUBLISH from HHH-EMS-ESP (d0, q0, r0, m0, 'home/HHH-EMS-ESP/shower_data', ... (25 bytes))
...
```
Now I have to check why the EMS-ESP isn't connecting to the mosquitto on the Raspi...

giovanne123, 2019-12-30:
ok, solved. 
On my testing Pi (for some reason it is still running wheezy and therefor older mosquitto) the mosquitto version was to old :-(
`1577708219: Invalid protocol "MQTT" in CONNECT from 192.168.0.129`
(I know now that also mosquitto on wheezy isn't supported anymore)

Switched to other Rpi with newer sw and mosquitto and it is running well :-)

closed 2019-12-30 by giovanne123