emsesp/EMS-ESP #274: Enhancement: New Device - Buderus Logamatic Easycom (Adresse 13)

giovanne123, 2019-12-30:
Is only a 'Modem' device, but maybe it can be added to the configurations to be dicovered on the bus.
The Buderus Logamatic Easycom with address 0x13: [https://ibb.co/k1F0nzZ](https://ibb.co/k1F0nzZ)

For reference currently detected bus devices: [https://ibb.co/9qjRL0Y](https://ibb.co/9qjRL0Y)

proddy, 2019-12-30:
if you type `devices` from the telnet what does it show? I'm surprised it hasn't picked up the Modem

giovanne123, 2019-12-30:
These device IDs are on the EMS Bus: 0x08 0x09 0x0D 0x10 0x11 0x21
and 5 were recognized by EMS-ESP as:
 Sieger BK13,BK15/Nefit Smartline/Buderus GB1x2 (DeviceID:0x08 ProductID:xx Version:02.07)
 BC10/RFM20 Receiver (DeviceID:0x09 ProductID:xx Version:02.00)
 RC30 (DeviceID:0x10 ProductID:xx Version:02.08)
 WM10 Switch Module (DeviceID:0x11 ProductID:xx Version:02.00)
 MM10 Mixer Module (DeviceID:0x21 ProductID:xx Version:02.00)

proddy, 2019-12-30:
strange that the 0x13 is not broadcasting as a device on the bus. Are you sure its a real device and connected? Also whats with the 'xx' as product IDs. Is that a bug?

giovanne123, 2019-12-30:
product IDs aren't a bug, commented out from my side.

Yes it is a real device connected to the bus and I do live monitoring with Buderus ECO-SOFT 4000/EMS software which uses this device/interface (the Easycom is an older gateway and modem)

proddy, 2019-12-30:
do you see telegrams coming in from 0x13? If not, then its not supporting the EMS protocol and there's nothing we can do.

proddy, 2020-02-17:
closing this due to no activity.

closed 2020-02-17 by proddy