emsesp/EMS-ESP #275: Enhancement: Support of mini OLED display attached to Wemos D1

giovanne123, 2019-12-30:
Maybe nice to have, but support of mini OLED display attached to Wemos D1 would be fine. (e.g. [oled display](https://randomnerdtutorials.com/esp8266-0-96-inch-oled-display-with-arduino-ide/)
If I have attached a display to the Wemos it would be fine to display some information. Information maybe configurable (bus staus, heating status, ...).

bbqkees, 2019-12-30:
You can use an OLED display on a second Wemos with f.i. ESP-Easy to send the parameters from your home automation to that display.

proddy, 2020-07-31:
EMS-ESP on the ESP8622 is almost out of memory (96%) so adding this feature would most probably break it. It would be a nice addition to the ESP32 though.

closed 2020-09-12 by proddy