emsesp/EMS-ESP #277: Firmware updates

glitter-ball, 2019-12-31:
Question : is there any guidance on how to carry out firmware updates please?

I have a bbqkees gateway running 1.9.3. He, quite rightly, discourages f/w updates unless there’s a good reason, but I should probably be thinking about doing one as there seem to be some useful updates in later versions.
A few questions...

1) where/how would I find the latest stable production code? Still a GitHub newbie...
2) how do I do the update? I’ve seen the gateway listens on a separate port for updates but don’t know much more.

Conscious that I could probably work it all out by trial and error but don’t want to brick my gateway. So, a few tips would be really handy. Thanks.

proddy, 2019-12-31:
Yes, on my wiki I explain how to upgrade the firmware. It's a little technical though. You'll find it at https://github.com/proddy/EMS-ESP/wiki/Building-and-Uploading-the-Firmware. There's a lot of other good content on there if you're interested. Failing that I would also check out BBQKees' comprehensive installation guide from his new web site at https://bbqkees-electronics.nl/wiki/

You don't need to worry about bricking the device. The circuits sits on top of a Wemos D1 Mini ESP8266 so worst case is that you'll have to use a mini USB cable and plug it into laptop, erase the flash and upload a new one. This can be done with a few simple command lines.

The firmware images can be found on https://github.com/proddy/EMS-ESP/releases. The latest dev build is there too. Look under `Assets`.

But...easiest way is to use the Web Interface which automatically downloads and uploads the latest release. You can switch between the stable master releases (currently 1.9.4) or the latest dev build (1.9.5b8). 

Make sure you read the [ChangeLog](https://github.com/proddy/EMS-ESP/blob/dev/CHANGELOG.md) to see all the new things I'm fixing or adding.

good luck!

glitter-ball, 2019-12-31:
Thanks for all this info. Used the web interface which handled it all fine and made all the right choices for me. I should have been braver and tried that first!

The enhanced MQTT publishing is good  *8)

closed 2019-12-31 by glitter-ball