emsesp/EMS-ESP #278: New EMS device recognized as a ?: RC20RF (DeviceID:0x19 ProductID:93 Version:02.00)

lmdecroos, 2019-12-31:
When using 1.9.5b7 the name of the RC20RF thermostat is missing
New EMS device recognized as a ?: RC20RF (DeviceID:0x19 ProductID:93 Version:02.00)

when using version 1.9.3 the device is correctly recognized
Thermostat: RC20F (ProductID:93 Version:02.00)

proddy, 2019-12-31:
checked the code and can't possibly see why its not working. Will need to dig deeper. I guess you skipped 1.9.4 ? 1.9.3 was a loooong time ago for me!

lmdecroos, 2019-12-31:
Same under 1.9.4

New EMS device recognized as a UBAMaster: Nefit Topline/Buderus GB162 (DeviceID:0x08 ProductID:115 Version:05.05)
New EMS device recognized as a Controller: BC10/RFM20 Receiver (DeviceID:0x09 ProductID:68 Version:02.03)
New EMS device recognized as a Thermostat: RC35 (DeviceID:0x10 ProductID:86 Version:21.08)
New EMS device recognized as a Switching Module: WM10 Switch Module (DeviceID:0x11 ProductID:71 Version:02.00)
New EMS device recognized as a Thermostat (DeviceID:0x18 ProductID:94 Version:02.01)
New EMS device recognized as a ?: RC20RF (DeviceID:0x19 ProductID:93 Version:02.00)
New EMS device recognized as a Mixing Module: MM10 Mixer Module (DeviceID:0x21 ProductID:69 Version:02.01)

proddy, 2020-01-02:
can you send the response to a `send 0B 99 00 02 20`. This will ask the thermostat for its version information. Also while you're at it do a `devices`. All with 1.9.5 please.

lmdecroos, 2020-01-02:
output from 1.9.5b8

These device IDs are on the EMS Bus: 0x08 0x09 0x10 0x11 0x18 0x19 0x21
and 7 were recognized by EMS-ESP as:
 UBAMaster: Nefit Topline/Buderus GB162 (DeviceID:0x08 ProductID:115 Version:05.05)
 Controller: BC10/RFM20 Receiver (DeviceID:0x09 ProductID:68 Version:02.03)
 Thermostat: RC35 (DeviceID:0x10 ProductID:86 Version:21.08)
 Switching Module: WM10 Switch Module (DeviceID:0x11 ProductID:71 Version:02.00)
 Thermostat: unknown? (DeviceID:0x18 ProductID:94 Version:02.01)
 0x19: RC20RF (DeviceID:0x19 ProductID:93 Version:02.00)
 Mixing Module: MM10 Mixer Module (DeviceID:0x21 ProductID:69 Version:02.01)

send 0B 99 00 02 20  doesn’t return anything

proddy, 2020-01-02:
I may have the product ids mixed up. Are you sure the RC20RF is on device id 0x19? And what is on 0x18? Perhaps you can double check in the console or something. 

Does send 0B 98 00 02 20 give anything either?

lmdecroos, 2020-01-02:
not sure if this is the correct filtering :

log w 0x18

System Logging set to Watch mode
send 0B (21:37:49) 08 00 18 00 07 01 CA 00 00 00 00 00 60 00 D5 02 5D 01 F2 00 00 0C 30 48 00 CB 00 00 00 
98 00 02 20

(21:37:55) Sending raw: 0B 98 00 02 20 
(21:37:59) 08 00 18 00 07 01 CA 00 00 00 00 00 60 00 D5 02 5E 01 F2 00 00 0C 30 48 00 CB 00 00 00 
Fetching data from EMS devices
Starting scheduled MQTT publish...
Publishing boiler data via MQTT
Publishing thermostat data via MQTT
Publishing mixing device data via MQTT
(21:38:09) 08 00 18 00 07 01 CB 00 00 00 00 00 60 00 D4 02 5D 01 F1 00 00 0C 30 48 00 CB 00 00 00 
(21:38:13) 0B 88 18 00 20 
(21:38:13) 08 0B 18 00 07 01 CA 00 00 00 00 00 60 00 D5 02 5D 01 F2 00 00 0C 30 48 00 CB 00 00 00 
(21:38:19) 08 00 18 00 07 01 CA 00 00 00 00 00 60 00 D5 02 5E 01 F2 00 00 0C 30 48 00 CB 00 00 00 

log w 0x19

System Logging set to Watch mode
send 0B 99 00 02 20

(21:38:41) Sending raw: 0B 99 00 02 20 
(21:38:59) 08 00 19 00 80 00 02 18 80 00 00 00 00 00 00 6B 6D 06 9A E4 00 00 00 05 FB CC 00 55 52 00 C6 
Fetching data from EMS devices
(21:39:14) 0B 88 19 00 20 
(21:39:14) 08 0B 19 00 80 00 02 18 80 00 00 00 00 00 00 6B 6D 06 9A E4 00 00 00 05 FB CC 00 55 52 00 C6 

hope this helps

proddy, 2020-01-02:
no, you can't use the Log watch functionality as this is only for telegram types and not device types. I'm interested in what kind of data is being sent from device id's 0x18 and 0x19 (this is 1st byte of the telegram). Use `log v` to capture some when you have time. And try that send again with verbose logging. thanks!

but do you know what these two devices could be? any other guesses?

lmdecroos, 2020-01-02:
21:52:25) Sending raw: 0B 98 00 02 20 
(21:52:25) Thermostat -> Me, telegram: 18 0B 00 02 (CRC=EE)
(21:52:25) UBAMaster -> All, type 0x07, telegram: 08 00 07 00 0B 03 03 02 00 00 00 00 00 00 00 00 00 (CRC=B7) #data=13
(21:52:26) UBAMaster -> All, type 0x18, telegram: 08 00 18 00 07 01 B9 00 00 00 00 00 60 00 D4 02 5C 01 E2 00 00 0C 30 48 00 CB 00 00 00 (CRC=16) #data=25

(21:52:36) UBAMaster -> All, type 0x18, telegram: 08 00 18 00 07 01 B9 00 00 00 00 00 60 00 D4 02 5D 01 E2 00 00 0C 30 48 00 CB 00 00 00 (CRC=9F) #data=25

(21:52:46) UBAMaster -> All, type 0x18, telegram: 08 00 18 00 07 01 B8 00 00 00 00 00 60 00 D4 02 5D 01 E2 00 00 0C 30 48 00 CB 00 00 00 (CRC=40) #data=25

(21:52:56) UBAMaster -> All, type 0x18, telegram: 08 00 18 00 07 01 B8 00 00 00 00 00 60 00 D4 02 5C 01 E1 00 00 0C 30 48 00 CB 00 00 00 (CRC=65) #data=25

(21:53:06) UBAMaster -> All, type 0x18, telegram: 08 00 18 00 07 01 B8 00 00 00 00 00 60 00 D4 02 5D 01 E2 00 00 0C 30 48 00 CB 00 00 00 (CRC=40) #data=25
(21:53:06) UBAMaster -> All, type 0x34, telegram: 08 00 34 00 3B 02 5D 02 5D 21 00 00 03 00 00 9F 18 00 16 1B 00 80 00 (CRC=13) #data=19

(21:53:14) Sending read of type 0x18 to 0x08, telegram: 0B 88 18 00 20 (CRC=D4)
(21:53:14) UBAMaster -> Me, type 0x18, telegram: 08 0B 18 00 07 01 B8 00 00 00 00 00 60 00 D4 02 5C 01 E1 00 00 0C 30 48 00 CB 00 00 00 (CRC=D6) #data=25

(21:53:16) UBAMaster -> All, type 0x18, telegram: 08 00 18 00 07 01 B8 00 00 00 00 00 60 00 D4 02 5C 01 E1 00 00 0C 30 48 00 CB 00 00 00 (CRC=65) #data=25

(21:53:19) UBAMaster -> All, type 0x18, telegram: 08 00 18 00 07 01 B7 00 00 00 00 00 60 00 D4 02 5D 01 E2 00 00 0C 30 48 00 CB 00 00 00 (CRC=C1) #data=25
(21:53:20) UBAMaster -> All, type 0x18, telegram: 08 00 18 00 07 01 B8 00 00 00 00 00 60 00 D4 02 5C 01 E1 00 00 0C 30 48 00 CB 00 00 00 (CRC=65) #data=25

(21:53:26) UBAMaster -> All, type 0x18, telegram: 08 00 18 00 07 01 B8 00 00 00 00 00 60 00 D4 02 5C 01 E1 00 00 0C 30 48 00 CB 00 00 00 (CRC=65) #data=25
(21:53:27) UBAMaster -> All, type 0x18, telegram: 08 00 18 00 07 01 B8 00 00 00 00 00 60 00 D4 02 5C 01 E1 00 00 0C 30 48 00 CB 00 00 00 (CRC=65) #data=25
(21:53:30) UBAMaster -> All, type 0x18, telegram: 08 00 18 00 07 01 B8 00 00 00 00 00 60 00 D4 02 5D 01 E1 00 00 0C 30 48 00 CB 00 00 00 (CRC=EC) #data=25

(21:53:32) UBAMaster -> All, type 0x18, telegram: 08 00 18 00 07 01 B7 00 00 00 00 00 60 00 D4 02 5D 01 E1 00 00 0C 30 48 00 CB 00 00 00 (CRC=6D) #data=25

(21:53:36) UBAMaster -> All, type 0x18, telegram: 08 00 18 00 07 01 B7 00 00 00 00 00 60 00 D4 02 5D 01 E1 00 00 0C 30 48 00 CB 00 00 00 (CRC=6D) #data=25

(21:53:46) UBAMaster -> All, type 0x18, telegram: 08 00 18 00 07 01 B7 00 00 00 00 00 60 00 D4 02 5C 01 DF 00 00 0C 30 48 00 CB 00 00 00 (CRC=94) #data=25

(21:53:57) UBAMaster -> All, type 0x18, telegram: 08 00 18 00 07 01 B7 00 00 00 00 00 60 00 D4 02 5D 01 E1 00 00 0C 30 48 00 CB 00 00 00 (CRC=6D) #data=25

(21:54:06) UBAMaster -> All, type 0x18, telegram: 08 00 18 00 07 01 B7 00 00 00 00 00 60 00 D4 02 5C 01 DF 00 00 0C 30 48 00 CB 00 00 00 (CRC=94) #data=25

(21:54:15) Sending read of type 0x18 to 0x08, telegram: 0B 88 18 00 20 (CRC=D4)
(21:54:15) UBAMaster -> Me, type 0x18, telegram: 08 0B 18 00 07 01 B7 00 00 00 00 00 60 00 D4 02 5D 01 E1 00 00 0C 30 48 00 CB 00 00 00 (CRC=DE) #data=25
(21:54:15) UBAMaster -> All, type 0x18, telegram: 08 00 18 00 07 01 B7 00 00 00 00 00 60 00 D4 02 5C 01 E1 00 00 0C 30 48 00 CB 00 00 00 (CRC=E4) #data=25

(22:11:47) Sending raw: 0B 99 00 02 20 
(22:11:48) 0x19 -> Me, telegram: 19 0B 00 02 (CRC=E6)

(22:11:52) UBAMaster -> All, type 0x19, telegram: 08 00 19 00 80 00 01 F2 80 00 00 00 00 00 00 6B 6D 06 9A E4 00 00 00 05 FB CC 00 55 52 00 BF (CRC=0B) #data=27
(22:12:14) Sending read of type 0x19 to 0x08, telegram: 0B 88 19 00 20 (CRC=D0)
(22:12:14) UBAMaster -> Me, type 0x19, telegram: 08 0B 19 00 80 00 01 F1 80 00 00 00 00 00 00 6B 6D 06 9A E4 00 00 00 05 FB CC 00 55 52 00 BF (CRC=05) #data=27

(22:12:42) 0x19 -> All, type 0xAF, telegram: 19 00 AF 00 00 86 00 06 06 (CRC=2E) #data=5

(22:12:52) UBAMaster -> All, type 0x19, telegram: 08 00 19 00 80 00 01 F0 80 00 00 00 00 00 00 6B 6D 06 9A E4 00 00 00 05 FB CC 00 55 52 00 BF (CRC=5C) #data=27
(22:13:13) Sending read of type 0x19 to 0x08, telegram: 0B 88 19 00 20 (CRC=D0)
(22:13:13) UBAMaster -> Me, type 0x19, telegram: 08 0B 19 00 80 00 01 F0 80 00 00 00 00 00 00 6B 6D 06 9A E4 00 00 00 05 FB CC 00 55 52 00 BF (CRC=A2) #data=27

(22:13:42) 0x19 -> All, type 0xAF, telegram: 19 00 AF 00 00 86 00 06 06 (CRC=2E) #data=5

(22:13:52) UBAMaster -> All, type 0x19, telegram: 08 00 19 00 80 00 01 EF 80 00 00 00 00 00 00 6B 6D 06 9A E4 00 00 00 05 FB CC 00 55 52 00 BF (CRC=CD) #data=27

(22:14:13) Sending read of type 0x19 to 0x08, telegram: 0B 88 19 00 20 (CRC=D0)
(22:14:13) UBAMaster -> Me, type 0x19, telegram: 08 0B 19 00 80 00 01 EF 80 00 00 00 00 00 00 6B 6D 06 9A E4 00 00 00 05 FB CC 00 55 52 00 BF (CRC=33) #data=27

(22:14:42) 0x19 -> All, type 0xAF, telegram: 19 00 AF 00 00 86 00 06 06 (CRC=2E) #data=5

Thermostat: unknown? (DeviceID:0x18 ProductID:94 Version:02.01)
 0x19: RC20RF (DeviceID:0x19 ProductID:93 Version:02.00)

I would assume that 0x18 is the first channel on my “Controller: BC10/RFM20 Receiver (DeviceID:0x09 ProductID:68 Version:02.03)” this channel is not in use but is always polling for a remote thermostat ( the led blinks every few seconds) 0x19: RC20RF (DeviceID:0x19 ProductID:93 Version:02.00) 0x19 is the actieve remote RC20RF thermostat who is controlling my HC2 . this RC20RF is communicating on channel 2 of my RFM20

proddy, 2020-01-02:
could be. I don't see any messages from 0x18 and 0x19 only sends out a poll on telegram type 0xAF. 

Still they should be replying with something when sending the 0x02 request for their version.

lmdecroos, 2020-01-03:
when removing the batteries from my RC20FM remote thermostat and doing a power cycle of my boiler

These device IDs are on the EMS Bus: 0x08 0x09 0x10 0x11 0x18 0x19 0x21
and 7 were recognized by EMS-ESP as:
 UBAMaster: Nefit Topline/Buderus GB162 (DeviceID:0x08 ProductID:115 Version:05.05)
 Controller: BC10/RFM20 Receiver (DeviceID:0x09 ProductID:68 Version:02.03)
 Switching Module: WM10 Switch Module (DeviceID:0x11 ProductID:71 Version:02.00)
 Thermostat: unknown? (DeviceID:0x18 ProductID:94 Version:02.01)
 Thermostat: RC35 (DeviceID:0x10 ProductID:86 Version:21.08)
 0x19: unknown? (DeviceID:0x19 ProductID:94 Version:02.01)
 Mixing Module: MM10 Mixer Module (DeviceID:0x21 ProductID:69 Version:02.01)

so the 0x19 is the RC20RF thermostat that can’t find by the boiler as it’s powered off

When inserting the batteries in my RC20RF then i have the following devices :

 These device IDs are on the EMS Bus: 0x08 0x09 0x10 0x11 0x18 0x19 0x21
and 8 were recognized by EMS-ESP as:
 UBAMaster: Nefit Topline/Buderus GB162 (DeviceID:0x08 ProductID:115 Version:05.05)
 Controller: BC10/RFM20 Receiver (DeviceID:0x09 ProductID:68 Version:02.03)
 Switching Module: WM10 Switch Module (DeviceID:0x11 ProductID:71 Version:02.00)
 Thermostat: unknown? (DeviceID:0x18 ProductID:94 Version:02.01)
 Thermostat: RC35 (DeviceID:0x10 ProductID:86 Version:21.08)
 0x19: unknown? (DeviceID:0x19 ProductID:94 Version:02.01)
 Mixing Module: MM10 Mixer Module (DeviceID:0x21 ProductID:69 Version:02.01)
 0x19: RC20RF (DeviceID:0x19 ProductID:93 Version:02.00)

as the RC20RF is connected to the second channel on my RFM20, it has ID 0x19
the first channel on my RFM20, who is not in use but is polling for a device, has ID  to my knowledge 0x18

proddy, 2020-01-03:
Ok, I'm starting to understand this now. Two more questions
 - I classify the RC29RF (0x19) as a Thermostat, See this [line](https://github.com/proddy/EMS-ESP/blob/18c6f7a7b72d51faee69157b39384e68c4dadb6a/src/ems_devices.h#L286). This means you it shows up as a Thermostat and you can change it values. Perhaps that is wrong and it should be re-classified as a controller. What do you think?
- There is still an unknown device with product_id 94 which is not in my database. It shows up on both channels 0x18 and 0x19. Maybe this is a modem and just a channel its listening too. How do you think I should classify those?

moustic999, 2020-01-03:
The rc20rf is nothing more than a rc20, but wireless. There is a receiver in the heater. Rc35 is installed as primary and the rc20 is secondary. It is installed in channel 2 because it controls HC2. Rc20 35 controls HC1

I had a doc showing practical installation examples from buderus. I will try to find it back as it explains most of the use cases.

-------- Message d'origine --------

lmdecroos, 2020-01-03:
the controller is the Controller: BC10/RFM20 Receiver (DeviceID:0x09 ProductID:68 Version:02.03)  is the interface between the remote thermostats and the EMS bus
the thermostat is the 0x19: RC20RF (DeviceID:0x19 ProductID:93 Version:02.00)

i would classify product_id 94 as remote device not connected ( a channel who is listing for communication with a remote device)

lmdecroos, 2020-01-03:
https://buderus-be-fl.boschtt-documents.com/download/pdf?file=6720619902&token=jd9f2jcpl5v3patqdqa4vckqs7 <https://buderus-be-fl.boschtt-documents.com/download/pdf?file=6720619902&token=jd9f2jcpl5v3patqdqa4vckqs7>

and https://buderus-be-fl.boschtt-documents.com/download/pdf?file=7747009376&token=jd9f2jcpl5v3patqdqa4vckqs7 <https://buderus-be-fl.boschtt-documents.com/download/pdf?file=7747009376&token=jd9f2jcpl5v3patqdqa4vckqs7>  ( in dutch)

proddy, 2020-01-03:
I made some change. Can you see what happens on your system with the latest dev build. Also double check with `devices`.

lmdecroos, 2020-01-03:
it seems you re-introduced an old bug from issue #162
when sending 0B 98 02 00 20 my gateway crashes after 20 seconds  :(

System Logging set to Verbose
(21:19:59) UBAMaster -> Me, type 0x16, telegram: 08 0B 16 00 FF 4C 64 00 06 FA 0A 01 05 5B 1E 00 (CRC=2F) #data=12
(21:19:59) Sending read of type 0x14 to 0x08, telegram: 0B 88 14 00 20 (CRC=E4)
(21:19:59) UBAMaster -> Me, type 0x14, telegram: 08 0B 14 00 35 1B 60 (CRC=90) #data=3
(21:19:59) UBAMaster -> All, type 0x18, telegram: 08 00 18 00 1F 01 54 64 11 09 01 25 60 00 B8 02 12 01 2E 00 28 0D 2D 48 00 C8 00 02 00 (CRC=B4) #data=25
(21:20:00) UBAMaster -> All, type 0x07, telegram: 08 00 07 00 03 03 00 00 00 00 00 00 00 00 00 00 00 (CRC=53) #data=13
(21:20:03) UBAMaster -> All, type 0x07, telegram: 08 00 07 00 03 03 01 00 00 00 00 00 00 00 00 00 00 (CRC=37) #data=13
Requesting type Version(0x02) from dest 0x18
(21:20:03) UBAMaster -> All, type 0x18, telegram: 08 00 18 00 1F 01 54 64 11 09 01 25 60 00 B7 02 12 01 30 00 28 0D 2D 48 00 C8 00 02 00 (CRC=3A) #data=25
(21:20:04) UBAMaster -> All, type 0x34, telegram: 08 00 34 00 3B 02 12 02 12 21 00 00 03 00 00 9E F0 00 16 1C 00 80 00 (CRC=40) #data=19
(21:20:04) Sending read of type 0x02 to 0x18, telegram: 0B 98 02 00 20 (CRC=3C)

Jan  3 21:19:10 ems-esp - 000+00:00:04.049 I 2: [eventlog] System booted
Jan  3 21:19:33 ems-esp - 000+00:00:00.191 I 0: [syslog] Log level set to DEBUG
Jan  3 21:19:33 ems-esp - 000+00:00:00.191 I 1: [setup] Application started
Jan  3 21:19:33 ems-esp - 000+00:00:04.059 I 2: [eventlog] System booted
Jan  3 21:19:51 ems-esp - 000+00:00:00.192 I 0: [syslog] Log level set to DEBUG
Jan  3 21:19:51 ems-esp - 000+00:00:00.192 I 1: [setup] Application started
Jan  3 21:19:51 ems-esp - 000+00:00:04.053 I 2: [eventlog] System booted
Jan  3 21:20:18 ems-esp - 000+00:00:00.192 I 0: [syslog] Log level set to DEBUG
Jan  3 21:20:18 ems-esp - 000+00:00:00.192 I 1: [setup] Application started
Jan  3 21:20:18 ems-esp - 000+00:00:04.046 I 2: [eventlog] System booted

proddy, 2020-01-03:
there was a bug in the process version code. I think its fixed now. You can see the last command was a version request (0x02) to 0x18 which if your weird unknown device we decided to call a 'remote device' :-)

lmdecroos, 2020-01-03:
Now i have :

These device IDs are on the EMS Bus: 0x08 0x09 0x10 0x11 0x18 0x19 0x21
and 7 were recognized by EMS-ESP as:
 UBAMaster: Nefit Topline/Buderus GB162 (DeviceID:0x08 ProductID:115 Version:05.05)
 Controller: BC10/RFM20 Receiver (DeviceID:0x09 ProductID:68 Version:02.03)
 Thermostat: RC35 (DeviceID:0x10 ProductID:86 Version:21.08)
 Switching Module: WM10 Switch Module (DeviceID:0x11 ProductID:71 Version:02.00)
 Thermostat: unknown? (DeviceID:0x18 ProductID:94 Version:02.01)
 Modem: unknown? (DeviceID:0x19 ProductID:93 Version:02.00)
 Mixing Module: MM10 Mixer Module (DeviceID:0x21 ProductID:69 Version:02.01)

should product ID 93 ( device 0x19) not be the thermostat : RC20RF and product ID 94( device 0x18) Modem : unknown ?

proddy, 2020-01-03:
those unknowns are not correct. I'll fix it.

proddy, 2020-01-04:
can you try again with 1.9.5b12?

lmdecroos, 2020-01-04:
tested 1.9.5.b13

These device IDs are on the EMS Bus: 0x08 0x09 0x10 0x11 0x18 0x19 0x21
and 7 were recognized by EMS-ESP as:
 UBAMaster: Nefit Topline/Buderus GB162 (DeviceID:0x08 ProductID:115 Version:05.05)
 Controller: BC10/RFM20 Receiver (DeviceID:0x09 ProductID:68 Version:02.03)
 Thermostat: RC35 (DeviceID:0x10 ProductID:86 Version:21.08)
 Switching Module: WM10 Switch Module (DeviceID:0x11 ProductID:71 Version:02.00)
 Thermostat: RC Remote Device (DeviceID:0x18 ProductID:94 Version:02.01)
 0x19: RC20RF (DeviceID:0x19 ProductID:93 Version:02.00)
 Mixing Module: MM10 Mixer Module (DeviceID:0x21 ProductID:69 Version:02.01)

proddy, 2020-01-04:
ok, re-wrote all the logic again in 1.9.5b14. Can you see if it made a difference?

lmdecroos, 2020-01-05:
b14 looks ok :)

   These device IDs are on the EMS Bus: 0x08 0x09 0x10 0x11 0x18 0x19 0x21
and 7 were recognized by EMS-ESP as:
 UBAMaster: Nefit Topline/Buderus GB162 (DeviceID:0x08 ProductID:115 Version:05.05)
 Controller: BC10/RFM20 Receiver (DeviceID:0x09 ProductID:68 Version:02.03)
 Thermostat: RC35 (DeviceID:0x10 ProductID:86 Version:21.08)
 Switching Module: WM10 Switch Module (DeviceID:0x11 ProductID:71 Version:02.00)
 Gateway: RC Remote Device (DeviceID:0x18 ProductID:94 Version:02.01)
 Thermostat: RC20RF (DeviceID:0x19 ProductID:93 Version:02.00)
 Mixing Module: MM10 Mixer Module (DeviceID:0x21 ProductID:69 Version:02.01)

SpaceTeddy, 2020-01-05:
dear all,
I would like to mention, that with my "Sieger BK15" Boiler + es73 Controller, the RC20RF is recognized as 0x18 and it works since 2018 w/o any trouble. 
Please think about, that there may any difference in adressing, if a RC35 or es73 is used.

thx, 
Chris

```
devices

These device IDs are on the EMS Bus: 0x08 0x09 0x10 0x18
and 4 were recognized by EMS-ESP as:
 Sieger BK13,BK15/Nefit Smartline/Buderus GB1x2 (DeviceID:0x08 ProductID:64 Version:03.06)
 BC10/RFM20 Receiver (DeviceID:0x09 ProductID:68 Version:02.03)
 Sieger ES73 (DeviceID:0x10 ProductID:76 Version:02.08)
 RC20RF (DeviceID:0x18 ProductID:93 Version:02.00)
```

proddy, 2020-01-05:
Hi Chris. Is there something not working in the latest build? Not really
sure what to fix. I’m guessing you don’t have the rc20rf but then what
should it be?

lmdecroos, 2020-01-05:
@SpaceTeddy

I would assume that you have only one Heat Circuit and that the first led on you RFM20 is blinking and the 2 others are off

SpaceTeddy, 2020-01-05:
@lmdecroos ,
correct, I have only one hc in my house and the RC20rf is setup to HC1 at the RF control system

SpaceTeddy, 2020-01-05:
@proddy ,

currently I stay with the 1.9.4b4 because it work for me fine and works stable for months, But I can test the latest beta for sure. Hope I can do this in the next days.
kr,
Chris

proddy, 2020-01-05:
ok, but still don't know what the problem is?!

SpaceTeddy, 2020-01-06:
well, there is currently no problem, if the EMS-ESP can handle the RC20RF as 0x18 or 0x19. My concern is, that if I do an update to the latest version, the RC20RF is not recognized anymore. :)

proddy, 2020-01-06:
it should work, and detect the RC20RF as a thermostat. You may need to use the new command `set master_thermostat <n>` to identify which one you want to use as the main one.

closed 2020-01-26 by proddy