emsesp/EMS-ESP #279: Bosch two channel devices

PhillyGilly, 2020-01-01:
I have two twin channel Bosch devices: an SM200 solar module and a MM200 mixer module. These appear to be multiple channel versions of the SM100 and the MM100 devices which I can see other people using successfully.
In the case of the SM200 (thank you Proddy and S-Przybylski) I've got it working like an SM100, that is reporting on reporting Collector Temperature, Solar Pump, Energy and **one** Tank Bottom Temperature.
My installation looks like Option B here:
[image]

[MS200 Wiring Guide Aug 16 Extra Info.pdf](https://github.com/proddy/EMS-ESP/files/4013999/MS200.Wiring.Guide.Aug.16.Extra.Info.pdf)
So I can see PS1, TS1 and TS2 but not TS5 or VS2.

Obviously I have some work to do to find this data and my job will be easier when I know two things:
a. What the string for a three way valve looks like in other telegrams?
b. Do Bosch/Junkers have a general approach to multiple channel devices? 

Thermostats have additional telegrams e.g
    Topic:home/ems-esp/thermostat_cmd_temp1
    Topic:home/ems-esp/thermostat_cmd_mode1
    Topic:home/ems-esp/thermostat_cmd_temp2
    Topic:home/ems-esp/thermostat_cmd_mode2

Any thoughts or ideas are appreciated. TIA.

proddy, 2020-01-05:
I've seen dual device id's being used before. It would be interesting to do a 2 min trace to see which is broadcasting what.

Is that 0x10/207 the CS200 ?

PhillyGilly, 2020-01-06:
I managed to get an installation manual for the CS200 and I think that there may have been some configuration errors by the installers of my system.  It appears that there are displays for the Option B above which my controller isn't showing, so which I need sort out before running this trace. 

The "CS200" has Sense II written on the back which is why I to it in issue #267 as a Sense II working in passive mode.  I spoke to Bosch UK who confirmed that the CS200 is the same physical device sold as Sense II for domestic installation and CS200 for commercial solar installations.

proddy, 2020-01-06:
Ok, I'll add product ID 207 as a CS200/Sense II

PhillyGilly, 2020-01-06:
I have reconfigured the installation and it appears as though all the data missing is arriving at the Sense II which self identifies as a CS200 (see images).
[image]
[image]
[image]
I am going to close this Issue and post the trace in #267

closed 2020-01-06 by PhillyGilly