emsesp/EMS-ESP #280: EMS-ESP web interface fails to load with MS Edge or Firefox

glitter-ball, 2020-01-02:
**Bug description**
1.9.4 web interface fails when using Microsoft Edge and Firefox 71.0 on 64-bit Windows 10. (This may be a duplicate of another issue - if so, apologies!)

**Steps to reproduce**
Connect to EMS-ESP and enter password. Click login button.

**Expected behavior**
Should proceed to dashboard screen with menu.

**Screenshots**
As password is typed in the, reveal icon appears at the RH end of the password entry field and works OK to reveal the password. Hovering mouse pointer over Login button causes it to become highlighted. When the button is clicked nothing happens and the highlight remains active even when the mouse pointer is moved away. Symptoms identical for MS Edge and Firefox.

```
system
**Device information**
 [APP] EMS-ESP version: 1.9.4
 [APP] MyESP version: 1.2.22
 [APP] Build timestamp: 2019-12-15 23:13:29
 [APP] Uptime: 1 day 23 hours 10 minutes 5 seconds
 [APP] System Load: 1%
 [WIFI] WiFi Hostname: ems-esp
 [WIFI] WiFi IP: 192.168.1.66
 [WIFI] WiFi signal strength: 74%
 [WIFI] WiFi MAC: 2C:F4:32:4A:A0:98
 [MQTT] is connected (heartbeat enabled)
 [NTP] Local Time is 11:58:26 GMT (1577966306)
 [SYSTEM] System is Stable
 [SYSTEM] Board: PLATFORMIO_D1_MINI
 [SYSTEM] CPU frequency: 80 MHz
 [SYSTEM] SDK version: 2.2.2-dev(38a443e)
 [SYSTEM] CPU chip ID: 0x4AA098
 [SYSTEM] Core version: 2_6_2
 [SYSTEM] Boot version: 31
 [SYSTEM] Boot mode: 1
 [SYSTEM] Last reset reason: Restart from terminal
 [SYSTEM] Restart count: 0
 [SYSTEM] # TCP disconnects: 0
 [SYSTEM] rtcmem status: blocks:2 addr:0x60001280
 [SYSTEM] rtcmem 00: 1163087990
 [SYSTEM] rtcmem 01: 65536
 [FLASH] Flash chip ID: 0x1640EF
 [FLASH] Flash speed: 40000000 Hz
 [FLASH] Flash mode: DIO
 [FLASH] Flash size (CHIP): 4194304
 [FLASH] Flash size (SDK): 4194304
 [FLASH] Flash Reserved: 4096
 [MEM] Firmware size: 614912
 [MEM] Max OTA size: 2523136
 [MEM] OTA Reserved: 16384
 [MEM] Free Heap: 24128 bytes initially | 15224 bytes used (63%) | 8904 bytes free (36%)

info
EMS-ESP system stats:
  System logging set to None
  LED: on, Listen mode: off
  Boiler: enabled, Thermostat: enabled, Solar Module: disabled, Mixing Module: disabled
  Shower Timer: disabled, Shower Alert: disabled

EMS Bus stats:
  Bus is connected, protocol: HT3
  Rx: # successful read requests=28309, # CRC errors=2
  Tx: Last poll=2.974 seconds ago, # successful write requests=0

Boiler stats:
  Boiler: Buderus Logamax plus/GB192/Bosch Condens GC9000 (DeviceID:0x08 ProductID:208 Version:01.01)
  Hot tap water: off
  Central heating: active
  Warm Water activated: on
  Warm Water circulation pump available: off
  Warm Water comfort setting: Hot
  Warm Water selected temperature: 60 C
  Warm Water desired temperature: 70 C
  Warm Water current temperature: 55.7 C
  Warm Water current tap water flow: 0.0 l/min
  Warm Water # starts: 2521 times
  Warm Water active time: 17 days 21 hours 28 minutes
  Warm Water 3-way valve: off
  Selected flow temperature: 40 C
  Current flow temperature: 40.5 C
  Return temperature: ? C
  Gas: on
  Boiler pump: on
  Fan: on
  Ignition: off
  Circulation pump: off
  Burner selected max power: 100 %
  Burner current power: 30 %
  Flame current: 17.2 uA
  System pressure: ? bar
  System service code:  (200)
  Heating temperature setting on the boiler: 75 C
  Boiler circuit pump modulation max power: 100 %
  Boiler circuit pump modulation min power: 60 %
  Outside temperature: 8.6 C
  Boiler temperature: 55.7 C
  Pump modulation: 96 %
  Burner # starts: 9259 times
  Total burner operating time: 75 days 1 hours 48 minutes
  Total heat operating time: 57 days 4 hours 20 minutes
  Total UBA working time: 580 days 23 hours 2 minutes

Thermostat stats:
  Thermostat: RC300/RC310/Moduline 3000/Bosch CW400/W-B Sense II (DeviceID:0x10 ProductID:158 Version:33.03)
  Thermostat time is 11:58:46 02/01/2020
  Heating Circuit 1
   Current room temperature: 18.5 C
   Setpoint room temperature: 18.5 C
   Mode is set to auto
   Day Mode is set to day

```

**Additional context**
No issues with Google Chrome. MS Edge would not connect at all until local IP range added to Trusted Sites. Now it connects but the web interface fails. Symptoms identical for MS Edge and Firefox.

glitter-ball, 2020-01-02:
The web interface works OK with Google Chrome. But... interestingly, Chrome does not display the reveal password icon in the login dialog. Curious...

proddy, 2020-01-02:
there are many (known) issues with the web interface which is caused by how the web socket authentication works. Hitting refresh usually causes this and quick remedy is to click the Logout button. Sadly its too much effort to easily fix it so I decided to completely re-write it using another approach. I'm half-way through the new implementation now so expect to see it coming in version 2 soon. It will support all browsers and be a lot more secure.

Having said that I tested on Win10-64 with Firefox (version 71.0) and MS Edge (Version 79.0.309.56) and both work fine with the latest 1.9.5 dev firmware.

closed 2020-01-02 by proddy

glitter-ball, 2020-01-02:
Thanks - sounds a mare to fix, especially given constraints of the hardware! I'll dig out a Chrome laptop on the rare occasion that I need the web interface over telnet.