emsesp/EMS-ESP #281: Change sensor state texts in HA

glitter-ball, 2020-01-02:
**Question**
How do I change the texts HA displays for sensors from EMS-ESP?

Conscious that this might be better asked on an HA forum, but it involves MQTT data from EMS-ESP, so hoping someone here can help.

I pick up the 3-way valve position from my boiler which HA shows as ON (hot water position) or OFF (heating position). I'd like to change the widget so it shows 'Heating' or 'Hot Water' instead.

The relevant bit of my sensor.yaml file looks like this:

```
- platform: mqtt
  state_topic: 'home/ems-esp/boiler_data'
  name: '3-way valve'
  icon: 'mdi:valve'
  value_template: '{{ value_json.wWHeat }}'
#  value_template: >-
#         {% if is_state('value_json.wWHeat', 'off') %}
#          HW
#        {% else %}
#          CH
#        {% endif %}

```
I tried the commented out lines in place of the original value_template line but that failed with an error. I then tried to use this example:

```
- platform: mqtt
  name: DoorSensor1
  friendly_name: My Door
  state_topic: "house/door"
  state_open: "OPENED"
  state_closed: "CLOSED"
  optimistic: false
  retain: false
  value_template: '{{ value }}'
```
So I added state_xxx definitions and that, too, barfed with an error when I checked the config file. So, any ideas on this would be appreciated, assuming that it's actually possible. Thanks.

proddy, 2020-01-02:
Haven't tested it but your first try looks ok. Something like:

```yaml
- platform: mqtt
  state_topic: 'home/ems-esp/boiler_data'
  name: '3-way valve'
  icon: 'mdi:valve'
  value_template: "{% if value_json.wWHeat == 'off' %} HW {% else %} CH {% endif %}"
```

its yaml so watch out for indentation. Every space counts!

glitter-ball, 2020-01-02:
Fantastic - thanks. Your code was much neater than my first attempt and actually worked *8).

Next plan... DS18B20 to pick up boiler return temp.

glitter-ball, 2020-01-02:
Experimented and built a string of IFs to decode ServiceCodeNumber...

```
- platform: mqtt
  state_topic: 'home/ems-esp/boiler_data'
  name: 'Status'
  unit_of_measurement: ''
  icon: 'mdi:power-cycle'
#  value_template: '{{ value_json.ServiceCodeNumber }}'
  value_template: "{% if value_json.ServiceCodeNumber == 200 %} CH active 
    {% elif value_json.ServiceCodeNumber == 201 %} HW active
    {% elif value_json.ServiceCodeNumber == 202 %} CH anti cycle
    {% elif value_json.ServiceCodeNumber == 203 %} Stand by
    {% elif value_json.ServiceCodeNumber == 204 %} CH cooling
    {% elif value_json.ServiceCodeNumber == 208 %} Service test
    {% elif value_json.ServiceCodeNumber == 265 %} Low CH load
    {% elif value_json.ServiceCodeNumber == 268 %} Component test
    {% elif value_json.ServiceCodeNumber == 270 %} Power up
    {% elif value_json.ServiceCodeNumber == 283 %} Burner start
    {% elif value_json.ServiceCodeNumber == 284 %} Ignition
    {% elif value_json.ServiceCodeNumber == 305 %} HW anti cycle
    {% elif value_json.ServiceCodeNumber == 357 %} Air purge
    {% elif value_json.ServiceCodeNumber == 358 %} Valve kick
    {% else %} {{ value_json.ServiceCodeNumber }} {% endif %}"
  
```

closed 2020-01-02 by glitter-ball