emsesp/EMS-ESP #282: Compiled 1.9.x dev branch on Linux doesn't boot

lmdecroos, 2020-01-02:
When compiling a release branch all is working fine and my binary is running on my EMS-EPS gateway.

When compiling a DEV branch, my binary file is more than 100kbyte larger, than a similar release branch and my EMS-ESP doesn't boot.
I need to re-flash, over serial, with a known working binary, to get my gateway back to live

Linking .pio/build/release/EMS-ESP-1_9_4-d1_mini.elf
Building .pio/build/release/EMS-ESP-1_9_4-d1_mini.bin
Retrieving maximum program size .pio/build/release/EMS-ESP-1_9_4-d1_mini.elf
Checking size .pio/build/release/EMS-ESP-1_9_4-d1_mini.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
DATA:    [=======   ]  66.8% (used 54740 bytes from 81920 bytes)
PROGRAM: [======    ]  58.6% (used 611612 bytes from 1044464 bytes)
Creating BIN file ".pio/build/release/EMS-ESP-1_9_4-d1_mini.bin" using ".pio/build/release/EMS-ESP-1_9_4-d1_mini.elf"

Linking .pio/build/debug/EMS-ESP-1_9_5b9-d1_mini.elf
Retrieving maximum program size .pio/build/debug/EMS-ESP-1_9_5b9-d1_mini.elf
Checking size .pio/build/debug/EMS-ESP-1_9_5b9-d1_mini.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
DATA:    [=======   ]  68.5% (used 56120 bytes from 81920 bytes)
PROGRAM: [=======   ]  69.1% (used 721780 bytes from 1044464 bytes)
Building .pio/build/debug/EMS-ESP-1_9_5b9-d1_mini.bin
Creating BIN file ".pio/build/debug/EMS-ESP-1_9_5b9-d1_mini.bin" using ".pio/build/debug/EMS-ESP-1_9_5b9-d1_mini.elf" 

The DATA size looks ok, but the PROGRAM size is huge

I'm always starting with a clean clone of the git source code

Tested on Debian amd64 and raspbian  both buster up to date releases
Python 2.7.16
PlatformIO 4.1.0
node v13.5.0  / npm 6.13.4 on Raspberry 
node v12.14 on amd64

proddy, 2020-01-02:
I see you're using the `build.sh` script. I wrote that for the automatic travis CI builds, not to be used as standalone.

The easiest and normal way to build the firmware is to use platformio. So

```sh
pio run -e release
```

which gives me on the latest dev build:

```
DATA:    [=======   ]  67.1% (used 54976 bytes from 81920 bytes)
PROGRAM: [======    ]  58.7% (used 612648 bytes from 1044464 bytes)
Creating BIN file ".pio/build/release/firmware.bin" using ".pio/build/release/firmware.elf"
```

The whole of the web site is compressed and stored as raw data in flash (PROGRAM) which is why you see around 600kb  used.

If you get anything different, delete the .pio directory which will force all the 3rd party libraries to be fetched.

lmdecroos, 2020-01-02:
the only way to get a bootable binary is :

2039  rm -rf EMS-ESP
 2040  git clone --single-branch --branch dev https://github.com/proddy/EMS-ESP.git
 2041  cd EMS-ESP/tools/webfilesbuilder/
 2042  npm install
 2043  cd ../..
 2044  pio run -e debug
 2045  rm -rf .pio
 2046  pio run -e release

strange that i have first to compile a debug version and then a release version.
in case i start immediately with building the release version then the compiling fails:

Building in release mode
LINKFLAGS = -Wl,-Teagle.flash.4m1m.ld -Os -nostdlib -Wl,--no-check-sections -Wl,-static -Wl,--gc-sections -Wl,-wrap,system_restart_local -Wl,-wrap,spi_flash_read -u app_entry -u _printf_float -u _scanf_float -u _DebugExceptionVector -u _DoubleExceptionVector -u _KernelExceptionVector -u _NMIExceptionVector -u _UserExceptionVectors
Compiling .pio/build/release/src/MyESP.cpp.o
Compiling .pio/build/release/src/Ntp.cpp.o
Compiling .pio/build/release/src/TelnetSpy.cpp.o
Compiling .pio/build/release/src/TimeLib.cpp.o
In file included from src/MyESP.cpp:10:0:
src/MyESP.h:53:55: fatal error: webh/glyphicons-halflings-regular.woff.gz.h: No such file or directory
 #include "webh/glyphicons-halflings-regular.woff.gz.h"
                                                       ^
compilation terminated.
*** [.pio/build/release/src/MyESP.cpp.o] Error 1
In file included from src/Ntp.cpp:6:0:
src/MyESP.h:53:55: fatal error: webh/glyphicons-halflings-regular.woff.gz.h: No such file or directory
 #include "webh/glyphicons-halflings-regular.woff.gz.h"
                                                       ^
compilation terminated.
*** [.pio/build/release/src/Ntp.cpp.o] Error 1
===================================================================== [FAILED] Took 21.90 seconds =====================================================================

Environment    Status    Duration
-------------  --------  ------------
travis         IGNORED
esp12e         IGNORED
d1_mini        IGNORED
nodemcuv2      IGNORED
nodemcu        IGNORED
debug          IGNORED
release        FAILED    00:00:21.900

proddy, 2020-01-02:
that's normal. You'll see in the platformio.ini the `scripts/pre_script.py` is commented out for the release. The idea is to build the web files only once as they don't typically change very often and I use 'release' as the default target. I'll do a build on an ubutnu box using "pio run -e release" and see if it boots

lmdecroos, 2020-01-03:
ok thanks this helps

proddy, 2020-01-07:
Tested on a Ubuntu 19.10 eoan using my instructions on the wiki (https://github.com/proddy/EMS-ESP/wiki/Building-and-Uploading-the-Firmware/_edit#on-ubuntu-using-command-line-interface-for-first-time) and it works.

closed 2020-01-07 by proddy

lmdecroos, 2020-01-10:
many thanks for the detailed instructions

Validated on debian buster

URL for the instruction on the new Wiki https://proddy.github.io/EMS-ESP/#/Building-firmware <https://proddy.github.io/EMS-ESP/#/Building-firmware>