emsesp/EMS-ESP #283: Commit of 1.9.5b11 crashes (boot loop)

tbr, 2020-01-03:
The commit to version 1.9.5b11 breaks my system. The mini-d1 gets stuck in a boot loop. 1.9.5b10 still works.

I tried to search for differences, and I believe the trigger for the crash is in ems.cpp on line 1714 (the while loop searching for a type) might not end correctly, overflowing i and continuing the loop. This will read out-of-bounds in EMS_Devices array... anyway, b10 works, b11 crashes and the only code change is in the ems.cpp while loop.

tbr, 2020-01-03:
Changing the `||` to `&&` in the while condition (and optionally adding a break when type is found to early break the for loop) will fix the code (and I think intended behavior).

proddy, 2020-01-03:
Yup, shoddy coding on my part. Thanks for pointing out the fix.

tbr, 2020-01-03:
There also seems something wrong with the way the typeId is handled.

My home has two thermostats, one is an RC300 and the other an RC100. The RC100 has device ID 0x38, which is recognized as a heat pump.

Here is the "devices" command output (using 1.9.5b11):

> These device IDs are on the EMS Bus: 0x08 0x09 0x10 0x20 0x21 0x30 0x38
and 7 were recognized by EMS-ESP as:
 UBAMaster: Nefit Topline/Buderus GB162 (DeviceID:0x08 ProductID:115 Version:05.09)
 Controller: BC10/RFM20 Receiver (DeviceID:0x09 ProductID:68 Version:02.03)
 Thermostat: RC300/RC310/Moduline 3000/Bosch CW400/W-B Sense II (DeviceID:0x10 ProductID:158 Version:11.09)
 Mixing Module: MM100 Mixing Module (DeviceID:0x20 ProductID:160 Version:24.03)
 Mixing Module: MM100 Mixing Module (DeviceID:0x21 ProductID:160 Version:24.03)
 Solar Module: SM100 Solar Module (DeviceID:0x30 ProductID:163 Version:27.05)
 Heat Pump: unknown? (DeviceID:0x38 ProductID:165 Version:12.04)

> You have a device is that is not yet known by EMS-ESP. Please report this as a GitHub issue so we can expand the EMS device library.

tbr, 2020-01-03:
(Stupid close and comment button... sorry)

proddy, 2020-01-03:
I used to think the product IDs are unique, then heard from other users that they can differ depending on the device ID. I could be wrong though. Some examples are product IDs 78 and 95 (see `ems_devices.h`).

And I also thought, until now, that deviceID 0x38 was specifically for Heat Pumps. It's strange to see it being used for an RC100.  In issue #249 another user supposedly also with an RC100 reported that it was broadcasting on deviceid 0x18??

So, what to do? Go back to making productIDs unique and ignore duplicates or increase the mapping between deviceIDs and known productIDs?

The bigger this project gets the more complicated it becomes.

tbr, 2020-01-03:
Yes, I see.

I would think that the productId's can overlap, depending on the "Brand" (it's all Bosch inside, but still there's Buderus, Junkers, ...). Now, even with the overlap between brands, this still gives the conflict for 95, although that one seems to be both referring to an HT3 (the boiler and the controller). It might be one physical product acting as two devices?

The 78 seems less of a problem if you know the bus type (or deduct the brand from the boiler).

That said, I'm not sure when I first noticed that the RC100 was detected as Heat Pump. On 1.9.4 it wasn't, but looking at the code that seems correct, because the productId-deviceId mapping wasn't added yet. And, with the 1.9.5bxx firmwares I was focussing on only on the master thermostat problem, but somehow didn't notice the new bug. I also assume that my RC100 didn't suddenly change it's deviceId from 0x18 to 0x38, because that would worry me (for my CV installation). Can this happen?

But, you are right that the project becomes a mess in terms of complexity and with the current input of issues, it seems to get worse. I really would like to help, but it would be good to coordinate better (instead of me pushing PR's that go against what you people are planning).

proddy, 2020-01-04:
In 1.9.5b12 I went back to making the product_id unique and will deal with the special cases as they get reported.

On contributing, it's only been me writing the code. I own an old boiler and a very old Buderus RC20 and started off with just basic EMS 1.0 support. Then requests came in for EMS 2.0 (EMS+) which took months to get right, all by trial and error followed by supporting HT3 (Junkers/Bosch/Worcester) which was a whole new kettle of fish. I don't personally benefit from any of the time I put into the project except making people happy, but it is time consuming and any help is more than welcome. If a few more people are comfortable hacking away at the code then I agree there should be a backlog of features and some more structure.

Right now I'm re-writing the core (new web UI, better MQTT support and logging).

tbr, 2020-01-04:
Well, just let me know if I can help you. I'll definitely continue to hunt for bugs, since I'm using the project form my HA integration :) Again, thanks a lot for your effort. I think that many people agree with me on this one.

tbr, 2020-01-04:
Btw: I found another wiki site that confirms 0x38 to be the RC100. See https://emswiki.thefischer.net/doku.php?id=wiki:ems:ems-telegramme

proddy, 2020-01-04:
yup, it must be 0x38. In the latest dev build (1.9.5b14) I ripped out all the logic pinning product ids to device ids. So assuming product ids are unique and don't really care which device ID they are broadcasting on. I probably broke a zillion other things in the process. Let me know if you find any strange bugs.

tbr, 2020-01-04:
thanks... will test :)

proddy, 2020-01-18:
let me know if this still happens and we'll reopen it

closed 2020-01-18 by proddy