emsesp/EMS-ESP #286: DS18B20s not detected - ideas, please.

glitter-ball, 2020-01-04:
**Question**
I've tried to connect a DS18B20 sensor to my bbqkees V1.3 board running 1.9.4 and it's not detected, either in web status page or telnet after info. 

- I've tried a second sensor (from a different supplier) with the same negative result.
- A scope shows a static 3.3 volts DC (approx) on the 3V3 and D5 wires from the board, so the power supply is OK and the D5 line is (presumably) being pulled up by the 4k7 onboard resistor.
- Tried parasitic mode as well as 3-wire.
- Have restarted or power cycled the board between attempts. Would be good to understand when the detection code runs - after a power cycle or on a warm restart?
- Checked in telnet and web interface that Dallas GPIO pin is D5.

Flummoxed and have run out of things to try. It's possible I have two duff sensors or maybe there's something else going on.

proddy, 2020-01-04:
quick way to test the Dallas sensor
 - pull out the Wemos D1 from the bbqkees board
 - looking at the sensor top down with the flat side facing south, connect the left pin to GND on the Wemos. Connect the middle pin to D5. Connect the right pin to +5V on the Wemos (which is 1st pin).
 - Power with USB
 - In Telnet type `info`, at the top you should see under "EMS-ESP system stats:" whether any external sensors are attached and further down the real data is shown

this works for me. Just tried it. Do make sure you're not using a Parasite B20S.

MichaelDvP, 2020-01-05:
I had the same problem 2 weeks ago. 
There are two board-versions with different pin sequence. 
The older with GND-data-Vcc, the newer with data-GND-Vcc. 
https://bbqkees-electronics.nl/wiki/gateway/adding-external-sensors.html

glitter-ball, 2020-01-05:
I’ve got a V1.3 board and have checked the wiring with a scope. Most likely cause at the moment is that the D5 GPIO line from the Wemos isn’t getting to the output socket because all I can see is +3.3V from the pull-up resistor with no logic zeroes.

Does anyone know whether the 1-wire output is polled beyond startup if it doesn’t find any devices initially?

MichaelDvP, 2020-01-05:
Just to be sure: You know that port numbering is a bit confusing and Wemos D5 is GPIO14?

proddy, 2020-01-05:
> I’ve got a V1.3 board and have checked the wiring with a scope. Most likely cause at the moment is that the D5 GPIO line from the Wemos isn’t getting to the output socket because all I can see is [...]

The Dallas device detection is only done once after startup

glitter-ball, 2020-01-05:
Thanks, all for your input - great to have the help.

I think I've cracked it - I started to wonder why I was seeing the LED pulse train on D4 of the Wemos when it's set to GPIO2!

With Dallas GPIO set to 14, I now have a pulse train on the yellow wire and the sensor now appears in the telnet 'info' dump:

```
External temperature sensors:
  Sensor #1 DS18B20: 18.68 C
```
So it was down to me mapping the wrong GPIO in the settings page *8( !! Maybe this is worth a mention in the wikis (here and bbqkees) when they next get an update?

So, big progress. But... the sensor isn't appearing in the MQTT list on the status web page, the 'mqttlog' dump in telnet, MQTT Explorer or HA's MQTT listener, despite restarting EMS-ESP and the MQTT broker.

Do I need to do something extra to get the external sensor published via MQTT?

Thanks again!

proddy, 2020-01-05:
Glad you got that sorted, The mapping between ESP pin numbers and their GPIOs are confusing as they differ also per manufacturer.  I did explictly point this out in the telnet console after a `set` command:

```
set led_gpio <gpio>          set the LED pin. Default is the onboard LED 2. For external D1 use 5
set dallas_gpio <gpio>       set the external Dallas temperature sensors pin. Default is 14 for D5
```

Perhaps its just best not to even mention the D* naming and refer only to GPIOs.

So for MQTT publishing the sensors, like everything, it should work ! But chances are that I broke something ;-) Do you see anything in `mqttlog` after you use the command `publish` ?

glitter-ball, 2020-01-05:
It's appeared in mqttlog and elsewhere after I did the publish...

glitter-ball, 2020-01-05:
I've now experimented a bit more. I had auto publishing on (publish_time=0). Setting this to a fixed value didn't make the sensor appear. With a fixed value and a 'publish' it did appear, but only the once and didn't refresh until I did another 'publish'.

glitter-ball, 2020-01-07:
I'll close this question thread and open a new bug thread for the external sensor values not being published automatically.

closed 2020-01-07 by glitter-ball