emsesp/EMS-ESP #287: External sensor values not published automatically via MQTT in 1.9.4

glitter-ball, 2020-01-07:
**Bug description**
In 1.9.4, data from external sensors is not published automatically via MQTT.

**Steps to reproduce**
Connect an external DS sensor and restart the EMS-ESP. Although the sensor appears in the `info` dump, no MQTT data is published for the sensor. The `publish` command does cause external sensor data to be published once. Changing `publish_time`  from automatic (zero) has no impact.

**Expected behavior**
Once an external sensor is detected at startup, MQTT data from it should be published automatically.

proddy, 2020-01-07:
Could you check if this is working as expected in 1.9.5 dev ?

glitter-ball, 2020-01-08:
Sorry - news is not good.

I've loaded the dev build and, when I restart, the sensor appears in `info` but there's no sensor data in `mqttlog`:

```
MQTT publish log:
  Timestamp:19:39:01 Topic:status Payload:online
  Timestamp:19:39:01 Topic:start Payload:start
  Timestamp:19:39:01 Topic:shower_data Payload:{"timer":"0","alert":"0"}
  Timestamp:19:39:13 Topic:tapwater_active Payload:0
  Timestamp:19:39:13 Topic:heating_active Payload:1
  Timestamp:19:39:51 Topic:thermostat_data Payload:{"hc1":{"seltemp":18.5,"currtemp":18.5,"mode":"auto"}}
  Timestamp:19:39:54 Topic:boiler_data Payload:{"wWComfort":"Hot","wWSelTemp":60,"wWDesinfectionTemp":70,"selFlowTemp":35,"selBurnPow":100,"curBurnPow":30,"pumpMod":100,"wWCircPump":0,"outdoorTemp":10,"wWCurTmp":57.1,"wWCurFlow":0,"curFlowTemp":37.6,"switchTemp":0,"wWActivated":"on","wWOnetime":"off","burnGas":"on","flameCurr":17.4,"heatPmp":"on","fanWork":"on","ignWork":"off","wWCirc":"off","heating_temp":75,"pump_mod_max":100,"pump_mod_min":60,"wWHeat":"off","wWStarts":2544,"wWWorkM":26032,"UBAuptime":845526,"burnStarts":9476,"burnWorkMin":110567,"heatWorkMin":84535,"ServiceCode":"","ServiceCodeNumber":200}
```
Sensor MQTT is also missing from the web interface status page.

When I do a `publish`, sensor data appears in `mqttlog` and in MQTT section on web interface status page. Thereafter, even if the sensor value changes, no automatic MQTT updates are broadcast - each time, a manual `publish` command is required to get the sensor data out.

Setting `publish_time` to non-zero doesn't cause the data to be published either. :crying_cat_face:

proddy, 2020-01-08:
ok, thanks for testing. I'm on it....

glitter-ball, 2020-01-08:
No problem - it'll get there. I wonder how many other people are using DS sensors? Maybe only the crazy hardened few (like me). It's a great function to have though and my first plan is to fit a permanent sensor to get boiler return temp via it. Thereafter, I might try...

- Temporary one at top of hot water tank, to see how temperature varies between outlet to taps and bottom of tank where sensor is.

- Temporary one outside next to my weather compensation sensor. I have a nasty suspicion that, with the wind in the wrong direction, flue products are hitting the weather comp sensor, resulting in sudden temperature spikes - oops... <sigh>

- Temporary one at far end of heating circuit to see how much flow temp drops off by the time the water gets there. Maybe I should upgrade my pipe insulation...

- etc.

proddy, 2020-01-11:
@glitter-ball can't reproduce on 1.9.4 and 1.9.5. Took a new EMS gateway board, installed EMS-ESP fresh after formatting the flash, set the settings, set publish_time to 10 and I see the external temps been published via MQTT regularly. Perhaps there is something in your configuration? I check with bbqkees and he's confirmed he checks this too on each board he sends out.

[image]

glitter-ball, 2020-01-11:
I was using publish_time=0. Is this affecting it?

When I tried publish_time=10 it still didn’t work.

Maybe I do a factory reset and try again...

proddy, 2020-01-11:
check
 - that you're definitely seeing values in telnet after a `info`
 - doing a `publish` in telnet shows ""Publishing external sensor data via MQTT"

if so there may be one clause in the code that is preventing it from working. And I'll need to send you a debug version to try out. let me know...

glitter-ball, 2020-01-11:
OK. Real progress this time.

If I restart with `publish_time=0`, no sensor data is published until I do a manual publish. Thereafter, it won't update it either without a manual publish. If I restart with`publish_time=10`, sensor data appears automatically and updates every 10 seconds.

So, the sensor publishing appears to break with `publish_time=0`.

With publish_time at either 0 or 10, I don't get the "Publishing external sensor data via MQTT" message. All I get is...

```
publish

Shower alert has been set to disabled
Shower timer has been set to disabled
```

So, I can get it working with publish_time=10, but not if I set publishing to be automatic so it looks like something around the automatic publishing and the sensors. 

Not sure if lack of automatic publishing is a big deal for me or not!

proddy, 2020-01-11:
publish_time=0 is evil, it will attempt to send MQTT every time it receives something from the EMS bus which could cause flooding. I suggest to throttle it down to 5 or 10 seconds. Anyway, it doesn't explain the strange behaviour you're seeing. I made some improvements to the code. Let's see if that helps.

proddy, 2020-01-11:
fixed the problem with 0 as automatic mode, and just removed it. The minimal value for publish_time is now 1.

glitter-ball, 2020-01-12:
OK - thanks. Can probably close this now.

closed 2020-01-14 by proddy