emsesp/EMS-ESP #288: HomeAssistant Discovery

Xeyame, 2020-01-07:
Hello,

Right now i am using home-assistant for controlling ems-esp over mqtt. Playing around with tasmota made me find out about a pretty neat thing, which is mqtt discovery. Basically you put the device on the same mqtt server and all settings get automatically added into home assistant, and it gets added to the "devices" section. This would be pretty neat to have added :D

Here some more info: [``](https://www.home-assistant.io/docs/mqtt/discovery/). Hope you can look into this, and thanks for making this project!!

proddy, 2020-01-08:
That would be nice to have, I agree. It would mean changing the form of the MQTT topics again though.

bbqkees, 2020-01-08:
It could be nice for HA, but changing the MQTT topics will break compatibility with other MQTT systems like Domoticz etc.
Also it would mean you have to listen to 100+ additional topics.

As it is now you just have to copy the preconfigured YAML files into the HA config and your (almost) done.

Perhaps the better option for HA would be to make a HA integration for EMS-ESP.

Xeyame, 2020-01-08:
Just like with tasmota you could maybe make it an configurable setting

moustic999, 2020-01-08:
Many others project have it as a configurable item 👍 

zigbee2mqtt : https://www.zigbee2mqtt.io/integration/home_assistant.html
> In your Zigbee2mqtt configuration.yaml set homeassistant: true

espHome : https://esphome.io/components/api.html and https://esphome.io/components/mqtt.html

proddy, 2020-01-08:
yes, the correct way would be to have these all configurable. Most of the elements are already there (state, commands, status base). It would work for the thermostat temperatures and the binary switches (heating on/off) but 'm worried how all the boiler data will be transmitted. According to the HA [doc](https://www.home-assistant.io/docs/mqtt/discovery/) each boiler value will need to be sent individually as a sensor, as opposed to one large payload.

Anyway a nice project for someone to experiment with!

Xeyame, 2020-01-08:
Yes I could at one point look into that :p, but personally I only use the thermostat from HA and log the rest into influx/grafana so eh

TimoFriedri, 2020-01-20:
Hi,

I'd rather prefer to have some mqtt free solution for home assistant like I mentioned in issue 87.
[Issue 87](https://github.com/proddy/EMS-ESP/issues/87)

I know MQTT is the universal solution but being spoiled by the esphome integration I really miss that for  ems-esp.

proddy, 2020-01-20:
@TimoFriedri ESPhome wrote their own API using RPC/TCP including the Home Assistance component.  That's a lot of engineering work and doable with their hundreds of contributors but sadly EMS-ESP is just me and I simply don't have the time or the need for this feature. It would be easier to just port the EMS-ESP logic to ESPhome.

TimoFriedri, 2020-01-20:
Ok, I get it. Still, thanks a lot for the nice work you have done.

In that case I would also strongly vote for the optionally MQTT topics for HA auto discovery.

proddy, 2020-05-13:
I added MQTT auto discovery in EMS-ESP version 2.0, only for the Thermostat now. I'm still "on the fence" to what other data from EMS-ESP is actually useful for people to control.

proddy, 2020-09-01:
todo: add Boiler

proddy, 2020-09-01:
as requested by @TimoFriedri , include the boiler as sensors in the MQTT Discovery with Home Assistant.

stravinci, 2020-09-02:
Hi,
I upgrade to 2.0 version yesterday and I see that system auto discovery 4 climates instances - hc1, hc2, hc3, hc4. But only first have any values (eg. 'set 21,5 degree' and 'current 31,5 degree'). And another case, why it is humidity climate?

proddy, 2020-09-02:
@stravinci are you saying the other heating circuits (hc2,3,4) are also published but with empty values? That was not the intention so looks like a bug. What do you see in the console (`show`) for the thermostats and also what is shown in the WebU when looking the Devices? Also which Thermostat are you using?

stravinci, 2020-09-02:
Yes, I see 4 HC's.
Console:
https://alioth.workupload.com/image/n8kUMx7r8Pg

Web:
https://alioth.workupload.com/image/VtCdsY7jE8p

HA:
https://botein.workupload.com/image/Q3bPKFGRxfb

My FW120 is located at boiler so I don't want to set boiler room temperature - boiler know that thermostat is located at his front so it change menu where room temperature does not exist.

proddy, 2020-09-02:
Ok, I'll look into fixing this in 2.0.1 (which is in the dev branch).

(BTW those images didn't upload. just drag&drop into the github issue next time)

stravinci, 2020-09-02:
Ok, one more time screens.
Console:
[image]
Web:
[image]
HA:
[image]

proddy, 2020-09-02:
I see now, the MQTT discovery config topics are sent, not the payloads. I can fix that.
BTW you can also click on the device in the Web UI to show it's details.

stravinci, 2020-09-02:
Nice UI :)
[image]
[image]

TimoFriedri, 2020-09-02:
@proddy  Am I right with the assumption that your TODO means that you planto publish each boiler sensor value as HA discovery entity? 
Do you also plan that for all thermostat value?

I know these are some dozen entities in HA, but I think with the new integrations/devices/entities and enable/disable this is no problem to organize on HA side.

proddy, 2020-09-02:
@TimoFriedri yes, I started working on this last night and quickly discovered why I avoided it in the beginning. There are so many values, 56 in fact. Options to consider:

A) Create a MQTT sensor value for each of the Boiler's values. This means sending 56 topics prefixed with 'homeassistant/config'. Not very elegant. It does mean its easy to create a dashboard using the entities card.
B) Create a single topic for HA's Discovery and have all the 56 boiler values as attributes of the sensor. It's nice and tidy, but it will be hard to show all the values on a lovelace dashboard because each value will need to separately added in the form `sensor.boiler.attribute.xxx` (or use `state_attr()`).  
C) Same as option A but in the Web UI the user defines which of the values they want published. This is lot of programming.

I'm tending to go with B. And use the same with the Thermostat.

Xeyame, 2020-09-02:
Hey, just tested the latest release (was busy for a long time and kinda forgot about it.)
Everything works great, thank you so much! New interface is so much better :)
I do have an interesting issue where i can't access the web gui from my desktop, but it's fine from my phone and laptop (and i think the issue is related to the MTU or tcp clamping settings... somewhere in my network xd)

But anyways, more values shown via autodiscovery would be nice, i'm sorta missing a few now (like outside temperature sensor).
You can make the MQTT integration autodiscover a device (being ems-esp), and then add all sensors to that device, if you don't want a sensor to show up, you can disable it from the device page in HASS. Having it be a full device (and not just an autodiscovered entity) also eases making automation scripts in home assistant, and it's basically option B but you add an device entry to each sensor and register it to the device registry (if i understand correctly).

proddy, 2020-09-02:
@Xeyame yes, that's a great suggestion. I didn't know adding a HA 'device' is so powerful (https://developers.home-assistant.io/docs/device_registry_index/). I would need some help here though to get it right, if you're able to. Starting with the manually adding the topics and testing in HA and when we know the mapping I can code it into EMS-ESP.

Xeyame, 2020-09-02:
Right now i'm looking how tasmota registers itself, as i couldn't find much information on how the registering of an actual device happens over mqtt, hope i can get some more info soon.

stravinci, 2020-09-02:
Did you see this:
https://www.home-assistant.io/docs/mqtt/discovery/#sensors-with-multiple-values

TimoFriedri, 2020-09-02:
Here is the schema for the device config.
Does the esp8266/esp32 has some kind of serial/uniqure identifier?

https://www.home-assistant.io/integrations/sensor.mqtt/

Xeyame, 2020-09-02:
Rough sketch, probably quite a few things just plain wrong, but got this far now

Basically, you don't acutally "register" a  device, you just add the device's data to some sensor and it gets stored. The most neat way to this is add an status sensor, ill call it `EMS-ESP status`. We register it just like a normal sensor, and in this case we can set it to take something like the RSSI measurement. Tasmota also adds an availiability lastwill, kept that in. We send this over MQTT to set the sensor's config:
```
{
  "name": "EMS-ESP status",
  
  "availability_topic": "ems-esp/AVTY",
  "payload_available": "Online",
  "payload_not_available": "Offline",
  "json_attributes_topic": "ems-esp/HASS_STATE",
  "state_topic": "ems-esp/HASS_STATE",
  "unit_of_measurement": "%",
  "value_template": "{{value_json['RSSI']}}",
  "icon": "mdi:thermostat",
  "uniq_id": "ems_esp_status",
  "dev": {
    "ids": [
      "esp_mac_address"
    ],
    "name": "EMS-ESP",
    "sw_version": "2.0.0",
    "manufacturer": "EMS-ESP"
  }
}
```

The ems-esp/HASS_STATE topic gets publishings about the status of the board like any regular sensor (example):
```
{
  "Version": "2.0",
  "Module": "EMS-ESP",
  "RSSI": "76",
}
```

and the ems-esp/AVTY topic gets availability messages, things like "Online" and "Offline" with retain flags set or maybe like `payload_turn_off` when the connection to the boiler is lost

For each sensor (one being boiler, and one being thermostat, or maybe more ?) you again send over the config to `homeassistant/sensor/ems_thermostat_temp/config` (ems_thermostat_temp being the sensor here):
```
{"device_class": "temperature", "name": "Set Room Temperature", "state_topic": "homeassistant/sensor/ems_thermostat_temp/state", "unit_of_measurement": "°C", "value_template": "{{ value_json.temperature}}" }
``` and repeat like this for something like `ems_hotwater_temp` :
```
{"device_class": "humidity", "name": "Set Hot water temperature", "state_topic": "homeassistant/sensor/ems_hotwater_temp/state", "unit_of_measurement": "%", "value_template": "{{ value_json.temperature}}" }
```
And then send over like usual, somwething like `{"temperature":"20"} to the state_topic

Basically the unique id (in this case the MAC of the esp) makes HASS bind it to the device.

The abbreviations found on https://www.home-assistant.io/docs/mqtt/discovery/ can be used instead too (tasmota does this, may save some network traffic i guess), didn't use them here because it's way more understandable for me this way :)

TimoFriedri, 2020-09-02:
@Xeyame 

I think that's basically it. 
But would you do 3 devices for thermostat, controller and boiler?
My favorite atm would be that the esp-ems is the device and it has all the entities, like climate.hc1, sensor.sysPress, sensor.wW..., sensor.designtemp, ...

BTW: 
  "payload_available": "Online",
  "payload_not_available": "Offline",
This can be left out, since it is default.

Xeyame, 2020-09-02:
@TimoFriedri Well, tought that made the most sense, also for when you have multiple thermostats, and also from the HASS website:

https://developers.home-assistant.io/docs/device_registry_index/#what-is-a-device
> A device in Home Assistant represents a physical device that has its own control unit. The control unit itself does not have to be smart, but it should be in control of what happens. For example, an Ecobee thermostat with 4 room sensors equals 5 devices in Home Assistant, one for the thermostat including all sensors inside it, and one for each sensor. Each device exists in a specific geographical area, and may have more than one input or output within that area.

So well reading this, i kind of confused sensors and devices i guess. I think they are saying that we should have 3 devices and add sensors with single values? Or maybe have a sensor with multiple values when they are a single "thing"? (Like group current and set temp, the pump percentages, the working/uptimes?) I don't really know what fits best here.

TimoFriedri, 2020-09-02:
@Xeyame 
Hmm, for organizational reasons I would also prefer one ems-esp devices with roughly 60 entities.  

But thinking about areas. Maybe it might be a better idea to have the actual boiler as an device and the thermostat, since they can be in different areas.

I think the controller is usually in the heating unit (e.g. the BC10 in a GB162) so it might be considered 1 device.

Not sure what would be the most reasonable.

proddy, 2020-09-02:
@stravinci that last change should fix the issue empty heating circuits poping up in HA

stravinci, 2020-09-02:
@proddy I'm not sure what changes are made?
[image]
[image]
I reboot HA, then I wait for all EMS updates - this screen was create ~17-18 minutes after restart HA.

proddy, 2020-09-02:
the retain flag is on so the MQTT broker is remembering the last topic. You can use [MQTTExplorer](http://mqtt-explorer.com/) to delete them. Remove all the topics `homeassistant/climate/ems-esp/hc*/config`. I set the retain flag on purpose so if the MQTT server drops out it will stay be able to understand the state data being sent from EMS-ESP.

stravinci, 2020-09-02:
Ok, I removed all climate->ems-esp topics and reboot EMS-ESP. All HC's come back.
[image]

proddy, 2020-09-02:
damn. I'll need to write some more test code to figure it out.

proddy, 2020-09-02:
I checked, and tested and it really shouldn't be happening. Can you send some logs from `watch 170` so I can see what is coming in for hc2 ?

stravinci, 2020-09-02:
You mean `watch on 170`?

I execute this command but in few minutes there was no message.
IN MQTT also I get only hc1 messsages, but hc2-4 are created after reboot.

proddy, 2020-09-02:
yes, sorry, typo. Actually its best to watch all traffic from the thermostat so a `watch raw 10`. There is something that is creating the heatingcircuits but I don't know what it is.

stravinci, 2020-09-03:
Ok,let me know if you need longer log.

`
┌──────────────────────────────────────────┐
│ EMS-ESP version 2.0.1b2                  │
│ https://github.com/proddy/EMS-ESP        │
│                                          │
│ type help to show available commands     │
└──────────────────────────────────────────┘

ems-esp:/$ watch raw 10
Watching incoming telegrams, displayed as raw bytes
Filtering only telegrams that match a device ID or telegram type of 0x10
000+10:48:17.954 N 0: [telegram] Rx: 90 08 35 00 11 00 A8
000+10:48:17.984 N 1: [telegram] Rx: 90 08 1A 00 00 F9
000+10:48:18.017 N 2: [telegram] Rx: 90 08 1A 02 00 FD
000+10:48:18.054 N 3: [telegram] Rx: 90 08 23 00 00 00 00 74
000+10:48:18.083 N 4: [telegram] Rx: 90 08 1A 04 03 F2
000+10:48:26.855 N 5: [telegram] Rx: 90 00 06 00 14 09 08 03 2D 14 04 00 71
000+10:48:30.221 N 6: [telegram] Rx: 90 88 02 00 0A F7
000+10:48:30.252 N 7: [telegram] Rx: 88 10 02 00 5F 17 0C 00 00 00 00 00 00 00 D                                                                                                                                                                                                                                             4
000+10:48:42.312 N 8: [telegram] Rx: 90 88 16 00 02 AF
000+10:48:42.328 N 9: [telegram] Rx: 88 10 16 00 00 00 39
000+10:48:42.491 N 10: [telegram] Rx: 90 00 FF 00 00 6F 03 02 00 D7 01 50 F3 34                                                                                                                                                                                                                                              00 E0
000+10:49:01.446 N 11: [telegram] Rx: 90 0B FF 00 00 6F 03 02 00 D7 01 50 F3 34                                                                                                                                                                                                                                              00 A5
000+10:49:01.697 N 12: [telegram] Rx: 90 0B FF 00 00 65 01 00 01 23 3C 41 03 01                                                                                                                                                                                                                                              00 03 FF 28 FA 01 04 28 2B 2B 01 01 00 25
000+10:49:26.775 N 13: [telegram] Rx: 90 00 06 00 14 09 08 03 2E 14 04 00 69
000+10:49:31.206 N 14: [telegram] Rx: 90 88 02 00 0A F7
000+10:49:31.236 N 15: [telegram] Rx: 88 10 02 00 5F 17 0C 00 00 00 00 00 00 00                                                                                                                                                                                                                                              D4
000+10:49:43.297 N 16: [telegram] Rx: 90 88 16 00 02 AF
000+10:49:43.317 N 17: [telegram] Rx: 88 10 16 00 00 00 39
000+10:49:43.476 N 18: [telegram] Rx: 90 00 FF 00 00 6F 03 02 00 D7 01 50 F3 34                                                                                                                                                                                                                                              00 E0
000+10:50:01.340 N 19: [telegram] Rx: 90 0B FF 00 00 6F 03 02 00 D7 01 50 F3 34                                                                                                                                                                                                                                              00 A5
000+10:50:01.593 N 20: [telegram] Rx: 90 0B FF 00 00 65 01 00 01 23 3C 41 03 01                                                                                                                                                                                                                                              00 03 FF 28 FA 01 04 28 2B 2B 01 01 00 25
000+10:50:19.964 N 21: [telegram] Rx: 90 08 35 00 11 00 A8
000+10:50:20.018 N 22: [telegram] Rx: 90 08 1A 00 00 F9
000+10:50:20.044 N 23: [telegram] Rx: 90 08 1A 02 00 FD
000+10:50:20.080 N 24: [telegram] Rx: 90 08 23 00 00 00 00 74
000+10:50:20.109 N 25: [telegram] Rx: 90 08 1A 04 03 F2
000+10:50:26.850 N 26: [telegram] Rx: 90 00 06 00 14 09 08 03 2F 14 04 00 61
000+10:50:32.101 N 27: [telegram] Rx: 90 88 02 00 0A F7
000+10:50:32.127 N 28: [telegram] Rx: 88 10 02 00 5F 17 0C 00 00 00 00 00 00 00                                                                                                                                                                                                                                              D4
000+10:50:44.306 N 29: [telegram] Rx: 90 88 16 00 02 AF
000+10:50:44.327 N 30: [telegram] Rx: 88 10 16 00 00 00 39
000+10:50:44.484 N 31: [telegram] Rx: 90 00 FF 00 00 6F 03 02 00 D7 01 50 F3 34                                                                                                                                                                                                                                              00 E0
000+10:51:01.449 N 32: [telegram] Rx: 90 0B FF 00 00 6F 03 02 00 D7 01 50 F3 34                                                                                                                                                                                                                                              00 A5
[... cut ...]
000+10:54:01.446 N 64: [telegram] Rx: 90 0B FF 00 00 6F 03 02 00 D7 01 50 F3 34                                                                                                                                                                                                                                              00 A5
000+10:54:01.664 N 65: [telegram] Rx: 90 0B FF 00 00 65 01 00 01 23 3C 41 03 01                                                                                                                                                                                                                                              00 03 FF 28 FA 01 04 28 2B 2B 01 01 00 25 `

proddy, 2020-09-03:
I'm only seeing HC1 (FF 00 00 6F). I guess we need some other users to see if they're experiencing the same thing. My thermostat only has a single heating circuit so its hard to simulate.

Xeyame, 2020-09-03:
Are you sure it's acutually ems-esp adding the values, i've seen home assitant randomly add things that were deleted. I also only have a single heating circuit so can't test aswell (and on v2.0.0 it doesn't happen for me ;P) Might be obvious, but did you try to test with the ems-esp not connected?

stravinci, 2020-09-03:
@Xeyame when I remove HC2-4 using MQTT Exporter they disappear from HA. They come back only after reboot EMS-ESP.
Maybe it is option to collect logs from start EMS-ESP?

proddy, 2020-09-03:
@stravinci what I do in cases like this is download a local copy of mosquitto.exe and run it on my desktop (Win10) from the command line with verbose logging `-v`, and have EMS-ESP connect to it. Then check the logs what is being published by EMS-ESP. You'll see if EMS-ESP is sending multiple topics for each heating circuit quickly.

stravinci, 2020-09-03:
@proddy I switched to local MQTT broker, then EMS-ESP autodiscover only 1 HC. But after reboot EMS apear new ones.
[image]

stravinci, 2020-09-03:
And here are maybe interesting logs.
[image]
[image]

proddy, 2020-09-03:
thanks, those logs are helpful as I see its also subscribing to ht3 and ht4 when you don't have them active. back to the drawing board then.

proddy, 2020-09-03:
Think I may have found the problem, but need to test it with live data tonight.
@MichaelDvP, this:
https://github.com/proddy/EMS-ESP/blob/66d105f99f73c878e09129d1df6932aa6eeb5f64/src/helpers.cpp#L345
will always return true. I'll change it to check for a valid range.

stravinci, 2020-09-04:
@proddy tested new dev build - I see only HC1 in HA :)

proddy, 2020-09-05:
thanks @stravinci . So its working? You know, I'm still not 100% sure why it failed in the first place. I made some minor adjustments but nothing that would effect the behaviour

proddy, 2020-09-16:
> Found this out so far, hope it helps :)
> ....
> Then you want to set up a few sensors (probably one per device, so usually 3; thermostat, controller and boiler), and make it have multiple values like @stravinci said. You also add the device part like this to the sensor's config
> 
> ```
>     "ids": [
>       "esp_mac_address"
>     ]
> ```
> 
> Basically the unique id (in this case the MAC of the esp) makes HASS bind it to the device.

Thanks @Xeyame . I'm back to looking at MQTT Discovery again with the aim to have EMS-ESP as a single HA device, each having a device with multiple sensors as you suggested. I can bulld the EMS-ESP device and add a few sensors, but the climate doesn't get attached to the HA device. I'm using:

```
{
  "name": "hc1",
  "uniq_id": "hc1",
  "~": "homeassistant/climate/ems-esp/hc1",
  "mode_cmd_t": "~/cmd_mode",
  "mode_stat_t": "~/state",
  "temp_cmd_t": "~/cmd_temp",
  "temp_stat_t": "~/state",
  "curr_temp_t": "~/state",
  "mode_stat_tpl": "{{value_json.hc1.mode}}",
  "temp_stat_tpl": "{{value_json.hc1.seltemp}}",
  "curr_temp_tpl": "{{value_json.hc1.currtemp}}",
  "min_temp": "5",
  "max_temp": "40",
  "temp_step": "0.5",
  "modes": [
    "night",
    "day",
    "auto"
  ],
  "dev": {
    "ids": [
      "esp_mac_address"
    ]
  }
}
```

I'll have another try later this week.

proddy, 2020-09-28:
Think I got the HA Discovery working now. EMS-ESP will add a new HA device and attach all the device data as entities to it. See https://emsesp.github.io/docs/#/Home-Assistant. If there are issues let me know, it seems to work on my system. Remember to remove all previous .yaml files except any custom automations.

TimoFriedri, 2020-09-28:
First of all,

thanks, awesome work.

Now my question: Why don't you expose all values? E.g. "System Pressure", which would be really nice to have for leak alarms.

stravinci, 2020-09-28:
Nice work.
[image]
i want to note:
retain - after restart HA, entities has no values
As @TimoFriedri I used more sensors, eg. outdoor temperature sensor

proddy, 2020-09-28:
@TimoFriedri @stravinci thanks for the feedback. I only added a subset of the boiler value to equal the ones in displayed in the WebUI but can easily add the rest (it's just one line). I'll also make sure the topics are retained, thanks

proddy, 2020-09-29:
@stravinci I checked and all topics are already retained. When I restart HA the topics are still there. So you may want to check your system to double check.

proddy, 2020-09-29:
@TimoFriedri added System Pressure

stravinci, 2020-09-29:
@proddy sorry for the confusion. I change FW at EMS-ESP, then I saw device and new entities from EMS-ESP in HA, so I remove all sensors from my configuration.yaml file and restart HA, after that all sensors was unavailable. But today I test it once again and all it is ok, so it must be some dependency with old entities (probably names).

proddy, 2020-09-29:
no worries. I had the same thing. I still have old states and entities that I can't seem to get rid off in HA, from previous version of EMS-ESP when I added them manually.

proddy, 2020-09-30:
@TimoFriedri made a point that it would be better to have the boiler and thermostat as separate HA Devices as they may be in different areas. Also request is to add all the boiler parameters. I was going to make it flexible where a user can choose via the WebUI but probably easier to blast them all out. I will need to modify the MQTT library to increase the queue size, which will get big.

TimoFriedri, 2020-10-01:
It seems this is still open for discussion/improvement, so I'd like to give my input
My suggestion would be as follows:

1 device in HA for each physical device in the central heating setup + ems-esp itself.
ATM (beta3) there is ems-esp and ems-esp_boiler. 
(in case of my setup in the basement)
--> keep ems-esp device (I am ok with single sensor for status and some values as attributes here)
--> keep boiler device and add ALL sensors (as in telnet show cmd)
--> add thermostat device (RC35) and add the hc1 entity to this device
(--> add controller device (BC10) and add entities to this one, maybe like "service code"-sensor) (not sure about this one, in the webui you have "service code" under the boiler, but I think it is rather a property of the controller)

--> rename devices according to their real physical names.
e.g.: EMS-ESP_boiler --> Topline/GB162

This would give a nice mapping of the real world setup to the HA setup.
Don't worry about to many devices or entities. This has become easily manageable in the HA interface. E.g. it is a piece of cake for me to simply disable the "Heat pump circulation"-sensor since I don't have this.

Thanks again for your work (source code and community response).

[13 more replies]

closed 2020-10-09 by proddy