emsesp/EMS-ESP #289: TLS support web interface and tls on mqtt (fingerprint)

kroon040, 2020-01-08:
Hi,

I's currently trying to secure all my traffic. Tasmota is working fine with tls fingerprinting and can disable the web interface.

Now thinking how to secure the EMS ESP.

Is tls on mqtt possible (tls fingerprinting)?
Is the web interface possbile with tls?
Option to disable telnet interface (with mqtt command or in web)

proddy, 2020-01-08:
Version 2.0 which I'm still developing has secure TLS in the web and much tighter authentication using tokens and different levels of administration. The same is in the telnet interface. And there is an wifi Captive Portal for those that want to run it standalone.

I haven't though about disabling the web though - can you explain how Tasmota does this?

SSL in MQTT is possible. Username/Password is already included. Not sure that is high priority though.

bbqkees, 2020-01-08:
As an additional security measure we could have the firmware check a specific spare IO pin at boot. If its f.i. connected to ground, it disables the web interface and Telnet.
I could add a jumper on the Gateway board for this purpose.
If all settings are done you install the jumper and you would need physical access to the device in order to change it again.

proddy, 2020-01-08:
I looked at adding TLS to MQTT, it's a bit a work but doable. My only concern is the huge amount of CPU and Memory it will chew up. I do agree that we need to take security seriously with IoT devices. Tasmota did a good write up on why.

kroon040, 2020-01-08:
I haven't though about disabling the web though - can you explain how Tasmota does this?

In tasmota you can send en cmnd/name/WebServer 0 to disable and 2 for admin interface enable.

There are more options, but if the webserver uses a certificate, than the webinterface can be enabled.

Also wat kees mentioned is also  a good idea to do it with a jumper. You can also do is that after a boot of the esp that you have 30 seconds to enable the webserver in the properties again.

bbqkees, 2020-01-09:
I checked the hardware design of the Gateway and there is currently no really good option for a jumper. I do have a few spare IO available on existing headers, but the only usable combination would be a jumper between A0 and D0. This would mean something like using D0 as an output HIGH and measuring 1V on the ADC of the ESP.
I also could add a jumper between D0 and GND, but D0 is GPIO16 which only has an internal pull down and not pull up, so that's not useful for detecting LOW.
Last option is a jumper between D2 and D3, but D3 is GPIO0 which is used for selecting the boot mode so not an ideal option. 

I could add a dedicated jumper in a new hardware revision but I just ordered new boards so that would have to wait.

proddy, 2020-01-09:
we can also just add a feature to disable the Telnet and MQTT via an MQTT topic like Tasmota does. First thing is getting TLS/SSL on MQTT working.

bbqkees, 2020-01-09:
That's perhaps the easiest method.

proddy, 2020-08-19:
in v2 the telnet is open but commands secure behind a password. TLS is in the web but not MQTT as it's too heavy for the ESP8266. Disabling the web is a feature I can add, also the Telnet. But it's not high priority.

proddy, 2020-09-23:
sticking to basic username/password authentication. Supporting TLS in web and mqtt is a lot of work and requires considerable memory.

closed 2020-09-23 by proddy