emsesp/EMS-ESP #291: Does the Buderus Logamax plus GB062-24 is supported?

Mark-81, 2020-01-09:
I saw my device is mentioned in [this](https://github.com/proddy/EMS-ESP/issues/205) issue, so I guess the user has successfully connected it. But in the official [complete list](https://github.com/proddy/EMS-ESP/wiki/Supported-EMS-Devices) of supported product is not present.

Are the docs not updated yet, or it's not officially supported (and we are on our own) ?

proddy, 2020-01-09:
yup, should work. Have you tried it?

Mark-81, 2020-01-09:
Not yet. 
I've just discovered this great repo (after realized that commercial brands like Netatmo support only GB072) and I'm still reading and learning to understand if it may fit at my home.

proddy, 2020-01-09:
ah ok. You could build a simple test circuit for $10 and try it out. Let me know if you need help

Mark-81, 2020-01-09:
Yep, I see. No problem at all to build it, thanks. I'll try it!

closed 2020-01-09 by Mark-81