emsesp/EMS-ESP #292: Getting started: use case

Mark-81, 2020-01-09:
I read the wiki and some issues, but I didn't find answers to my questions.
If I didn't search well, please help me to find the correct link!

Scenario
--
I have a Buderus Logamax plus GB062-24 and a Logamatic TC100.
My home has two floors and I'm very disappointed about the TC100 control. It cannot handles the two areas so it's very hard to keep the temperature I wish.

I'm going to replace the thermostatic valves with "smart" ones, throw the TC100 out of my window and use a more suitable system (like this) to control the boiler.

Questions
--
1. I read the Disclaimer, but I'm not sure if using a different device than the original one (i.e. TC100) may invalidate the warranty. 

2. I have Home Assistant already run at home. So I'm going to control EMS-ESP from it. What about the thermostatic WiFi valves? Given that I choose them compatible with HA, how they work with EMS-ESP? I mean: the heater on/off and the flow temperature depend on the requested temperature of all the valves. May I create my own automation in HA to set this behavior?

bbqkees, 2020-01-09:
EMS-ESP does not replace the thermostat. EMS-ESP can change the temperature settings of the existing thermostat (TC100 read only though) and then its up to the thermostat to control the boiler.

Using another thermostat than the TC100 will not void any warranty.

In HA you need to create your own automation to control the valves.
EMS-ESP by itself will not turn on those valves.

Mark-81, 2020-01-09:
> EMS-ESP does not replace the thermostat. EMS-ESP can change the temperature settings of the existing thermostat (TC100 read only though) and then its up to the thermostat to control the boiler.

I didn't understand this, thanks for pointing it out.
Unfortunately, this is not what I'm looking for.

I'm searching for a way to get remove TC100 and control the boiler directly (like TC100 does).

bbqkees, 2020-01-09:
Most boilers have an on/off terminal next to the EMS terminal.
My own Nefit boiler has it as well. The on/off pins are connected to a zone control system with several thermostats.
If one of the thermostats turns on, this means there is a demand for heat so the zone control will shortcut the on/off pins and the boiler will start heating.

I now have a Moduline 400 thermostat connected to the EMS bus for testing and both systems work fine while being simultaneously connected.
If the boiler starts heating due to the on/off control, I can still see the heating icon on the Moduline thermostat.

So perhaps not the best method but you can connect your own controlled relay to the on/off pins to turn on the heating. the boiler will still modulate based on f.i. return temperature.

Mark-81, 2020-01-09:
> So perhaps not the best method but you can connect your own controlled relay to the on/off pins to turn on the heating. the boiler will still modulate based on f.i. return temperature.

If I understand correctly, I might:

- leave the TC100 there (in order to let it set the flow temperature based on the external temperature)
- add a relay on the on/off pins (todo: check if my boiler has them) in order to by-pass the TC100 control
- this relay will be controlled by HA automation, based upon the temperatures requested in the rooms (read by the valves)

I should set the TC100 to a very low temperature, so I can decide when turn on the heater. But I'm afraid whether it will update the flow temperature in function of the outside temperature.

But then... I don't see any need for EMS-ESP. I just need to control the WiFi valves and a relay.
Perhaps I'm still missing the point.

bbqkees, 2020-01-09:
I'm not suggesting you need EMS-ESP in your case.
But is is still useful because it can read out all boiler parameters so you can finetune based on f.i. boiler modulation etc.

Some boilers have a combined EMS on/off terminal. Depends on the brand.

One thing you need to keep in mind is that an EMS thermostat in the end will be better at saving gas than using your own on/off system.
And at some point HA will fail so you need to have a failsafe built in (like keeping the TC100).

proddy, 2020-01-09:
Although I would not recommend it, in theory you could use EMS-ESP and the EMS Gateway board to build a dumb thermostat, picking up ambient room temperatures via the external sensors and using Home Assistant to change the boiler's flow temperature based on rules and time periods. A few users have experimented with this I recall.

Mark-81, 2020-01-09:
> I'm not suggesting you need EMS-ESP in your case.
> But is is still useful because it can read out all boiler parameters so you can finetune based on f.i. boiler modulation etc.

I apologize for my dumb questions, but I understood the EMS bus is the one that connects the TC100 to the boiler. So why it's "read-only"? Why it's not possible to set the parameters and not read them only?

> One thing you need to keep in mind is that an EMS thermostat in the end will be better at saving gas than using your own on/off system.

But I don't want to make an of/off system... for this reason I'm trying to understand how to act _like_ the TC100 to set the flow temperature, instead of just enable/disable the heater like common thermostats.

Mark-81, 2020-01-09:
> Although I would not recommend it, in theory you could use EMS-ESP and the EMS Gateway board to build a dumb thermostat, picking up ambient room temperatures via the external sensors and using Home [...]

The EMS Gateway is [this](https://github.com/proddy/EMS-ESP/blob/master/doc/schematics/Schematic_EMS-ESP.png) circuit, connected on the same EMS bus of the original thermostat, right? 

By the way, I don't think this term is still correct in a "smart" system. A thermostat (like TC100) just picks the room temperature and if it's below the current setpoint turn on the heater (optionally setting the flow temperature to save gas). Using thermostatic valves in such a scenario is pretty useless... they can just limit the temperature in some rooms, but they cannot turn on the heater if more heat if required.

Example
--
In my home, the thermostat is placed in the living room at first floor. During the night I would low the temperature here, and heat up a little bit more the rooms at the second floor.
But with a thermostat it's impossible! 

To heat the second floor I must set an higher temperature on the thermostat so it will turn on the heater. But to avoid to heat the living room I should set the thermostatic valves... hence it makes no sense at all to have a thermostat using thermostatic valves! They are enough.

Don't you recommend this approach for the reasons expressed by bbqkees? 
A fail-safe system is easily accomplished keeping the thermostat there and routing the bus to it when needed - i.e. using a N.C. relay. About saving gas: I'afraid I waste a lot of gas heating up my first floor when I just need to heat the second one...

bbqkees, 2020-01-09:
That is the circuit yes. You can also get a complete board in my web shop.

But the TC100 already has the ability to add radiator control valves.
So I guess if you use those you can solve your problem while keeping your existing thermostat.
[Buderus RF control valves](https://www.buderus-logamatictc100.com)

Mark-81, 2020-01-09:
> But the TC100 already has the ability to add radiator control valves.
> So I guess if you use those you can solve your problem while keeping your existing thermostat.
> [Buderus RF control valves](https://www.buderus-logamatictc100.com)

Unfortunately it seems mine is not compatible: it's a "Thermostat Logamatic TC100", not a "Logamatic TC100" (they are genius... two different products with almost the same name):

https://www.buderus.com/it/it/ocs/residenziale/termostato-logamatic-tc100-757785-p/

Sorry I didn't find an English link.

iladi4466, 2020-01-10:
Hi,

My house is in floor heated. The warm water comes from a Buderus boiler by a 11 way distributor to the pipes in the floor. 

Since recently I had a Nest reading ambient temperature in one room and control ON/OFF the boiler, all the house a single heating zone.

Now I have a Xiaomi Aqara thermistor and a self made ESP8266 - relay electronics that replace Nest, ON/OFF controlling boiler. Next I plan to use another ESP+relay electronics to control the 11 way distributor and control the rooms independently. 

So far the ESP+relay solution seems better than the Nest solution since via HA I have better control of hysteresis/heat inertia.

I use the EMS-ESP module to read various info from the boiler, sadly not all the data is available, maybe in a future firmware. I had also wished to control the heating agent temperature via the EMS-ESP module.

proddy, 2020-01-10:
@iladi4466 please create a new "enhancement" issue for the data you need that is missing so we can add to the next release. Also one for controlling the heating agent.

proddy, 2020-07-31:
There is also some Python code (for home assistant for example) that can send values to the Thermostat TC100 so you could use this to most of the logic.

closed 2020-07-31 by proddy