emsesp/EMS-ESP #294: Function WarmWaterOneTime?

MichaelDvP, 2020-01-12:
I have tried to activate WW-OneTime over mqtt, but it does not work as expected.

My idea is: I have a bathroom with showers far away from boiler and warm water circulation every 30 min. All other rooms with warm water are near the boiler and need no circulation. I want to install a aqara-door-contact and activate circulation if someone uses this bathroom.

I have activated OneTime in mqtt and nothing changes, circulation stays off. I checked tho protokoll.

RC35 sends 0x35 every minute with two bytes:
(16:40:12) Thermostat -> UBAMaster, type 0x35, telegram: 10 08 35 00 11 11 (CRC=30) #data=2

Activating oneTime by mqtt sends one byte: 
Setting boiler warm water OneTime loading on
(16:44:59) Sending write of type 0x35 to 0x08, telegram: 0B 08 35 00 22 (CRC=06)
deactivating:
Setting boiler warm water OneTime loading off
(16:51:17) Sending write of type 0x35 to 0x08, telegram: 0B 08 35 00 02 (CRC=26)

and in code i found: 
Set WWOneTime0x35:
EMS_TxTelegram.dataValue     = (activated ? 0x22 : 0x02); // 0x22 is on, 0x02 is off for RC20RF

Then i pushed WW Button on RC35
(17:39:04) Thermostat -> UBAMaster, type 0x35, telegram: 10 08 35 00 11 33 (CRC=12) #data=2
(17:39:06) Thermostat -> UBAMaster, type 0x35, telegram: 10 08 35 00 11 11 (CRC=30) #data=2
and RC35 display says "circulation started" and circulation is active for 3 min.
but OneTime-Flag isn't set. 

Now my questions: 
Does WWOneTime only work in ECO-Mode?
Or only if RC20RF is installed? 
Has anyone worked with OneTime?

For me i have added a mqtt-subscription "boiler_cmd_wwcirculation" to start the circulation with 
Setting boiler warm water circulation on
(11:57:49) Sending write of type 0x35 to 0x08, telegram: 0B 08 35 00 11 33 (CRC=59)
this works, but i think normaly OneTime should do this.

SpaceTeddy, 2020-01-12:
@MichaelDvP ,
I made this pull request for my Sieger boiler and es73 Thermostat, which is more/less compatible with the Buderus RC30 controller. I already noticed, that the original RC30/RC35 may work with 0B 08 35 00 33 -> ON  and 0B 08 35 00 03 -> abort that WWoneTime warm water buffer loading. you can test this with the "send" command, but please note, that the loading only works, if the WW temp of your buffer is less that 5°C less then your target temperature.

MichaelDvP, 2020-01-13:
@SpaceTeddy, thank you, i understand now. In wwmode hot the automatic loading starts 5 deg. below setpoint, so onetime will never do. And i think the command only starts the loading, for circulation the second byte is needed. 
It seems that the 0x35 command is a bit field with set/clear in the high nibble and mask in the low nibble. In status 0x34 databytes 5(10) and 7(12) giving the status of  loading and circulation. The bit fields are maybe similar.
So bit1 is daymode, bit2 is onetime, and circulation-byte also bit1 daymode and bit2 circulation_on. 
I started the circulation now with 0B 08 35 01 22  and that works. 
So the command for onetime is also correct, it leaves daymode untouched. Sending 03 will clear onetime and daymode bits. 
I close this question and test a bit what the other bits do.

closed 2020-01-13 by MichaelDvP