emsesp/EMS-ESP #296: Posibility to increase or decrease heat agent temperature from Hass.io

iladi4466, 2020-01-13:
It would be a nice feature if you can make possible to increase/decrease the temperature of the heating agent. Since the interface gives the possibility to attach an external temperature sensor it can be a great add to control remotely the heating agent. A simple one as the one that controls the hot water is OK.

As far as reading various data from the boiler I understand that data depends on boiler. Is the list of telegrams that appears in the status web page of the interface the only one I can get from the boiler, or are other that I cannot see?

proddy, 2020-01-13:
you could use the MQTT topic `boiler_cmd_wwtemp` to set the temperature of the hot water. The automation you would need to build yourself in HA.

I'm not sure what you mean by the list of telegrams? Most boiler data is generic. Is there something that you are missing?

iladi4466, 2020-01-13:
Hi,

Am not talking about the hot water, there you have create a control already, I am talking about the setting the temperature of the heating agent of the house, floor in my case.
On the boiler there are some curves that determine the temperature of the heating agent according to an external temperature sensor. And this is what I want to try to control. Now I can read external sensor thru your gateway, the temp of the heating agent, again thru you gateway, but I don’t know the command to set the heating agent (nice if you implement this like you implement the control of the hot water.

About telegrams.

I can see a list of messages coming from the boiler. Some of them seems not to work with my boiler. Are there other messages I can search that are compatible with my boiler?

Sorry for any inconvenience, there was my firs 2 post on github, and also I am not a native English speaker.

Regards,
Adrian

bbqkees, 2020-01-13:
What do you mean by 'heating agent'?
You mean you want to turn on the boiler itself or f.i. an external relay? Or something else?

bbqkees, 2020-01-13:
Or do you want to simulate an outdoor temperature sensor?

iladi4466, 2020-01-13:
Yes, that’s it. I want to put a Dallas thermistor on your box, make it read external temperatures and set the heating agent according to that.

bbqkees, 2020-01-13:
Why not connect a temperature sensor to the boiler itself?
The connector is either on the boiler or on the thermostat.

And don't mistake an 'external' sensor with 'outdoor' sensor. Some of the boilers can use an outside temperature sensor for weather controlled heating.
(If that is what you want to achieve).

iladi4466, 2020-01-13:
Because the curves that controls that in the boiler are mainly for radiator heating, not fot in floor heating. Radiators use a much higher temperature. I want to stay as low as I can with the water in the floor while I still have the desired temp and comfort in the house.

bbqkees, 2020-01-13:
So in summary you want to set selFlowTemp?

iladi4466, 2020-01-13:
Yes

proddy, 2020-01-13:
ok, thats possible.

glitter-ball, 2020-01-16:
> Because the curves that controls that in the boiler are mainly for radiator heating, not fot in floor heating. Radiators use a much higher temperature. I want to stay as low as I can with the water in the floor while I still have the desired temp and comfort in the house.
> 
> Now I only have 30 C, minimum, but is relatively warm outside. If not automated I have to go to the boiler and raise it to 33-35 maybe 40 C if it gets colder.

No option on your boiler or thermostat to set the heat emitter type (radiator or underfloor) and change the heating curve? That would be the best option. If not, then I guess you're trying to build a weather compensation algorithm in HA? If you added an internal sensor you could add room influence to the curve as well.

proddy, 2020-09-23:
was added to 2.0.1 - `call boiler flowtemp` in console or via MQTT. See API

iladi4466, 2020-09-24:
Thanx very much.

I tried to update firmware and I have an issue.

I cannot see any data in the web interface, only the left-side menu. But any click on the menu does nothing.

I unplug the controller, plug it back and then a can insert password and navigate menus normally (except upgrade, error connecting to server).

But after some time If I try a new reconnect the same as above happens.

Also it says I cannot upgrade firmware no internet connection although in the router I can see the controller.

Regards,
Adrian

proddy, 2020-09-24:
which firmware bin do you use? From which version did you upgrade from? Does the Telnet console work?

iladi4466, 2020-09-24:
Currently on 1.95

Version 1.94 before

I don’t know about telnet, but in Home Assistannt and Grafana I can see variations of parametters of the sensors (mqtt) so the module is working,

proddy, 2020-09-24:
I would grab the latest 2.1 dev build and try that.

iladi4466, 2020-09-25:
Hi

I could not upgrade via normal method, I was forced to upgrade via cable method  don’t know why.

But it is all good now. Up & running.

closed 2020-09-25 by proddy