emsesp/EMS-ESP #297: Is my boiler compatible

Chuckame, 2020-01-13:
Hi,
I'm really interested about this project because of my boiler company support tells to me "you cannot connect your boiler to iot world because of ISM1 interface", but this interface is compatible with this project!
But I don't know what I do search and how to get more specified informations... 
I've got a "ZBS 22/210S-3 MA23 Solar S0200", this reference means something to someone?

Thanks

bbqkees, 2020-01-14:
Check the installation manual to see if it says something like 'EMS bus'.

proddy, 2020-01-14:
You do need an ISM1 interface (http://documents.junkers.com/sitemap/document/id/6720615095)

Chuckame, 2020-01-14:
The technical support said to me that I've got an ISM1 interface, it's enough to get this library working?

Chuckame, 2020-01-15:
Found there is a Bosch Heatronic 3 bus compatible, but no EMS compatible... Maybe the same.
Someone has this Heatronic module?

bbqkees, 2020-01-15:
What thermostat do you have?

proddy, 2020-01-15:
Your Junkers works with HT3 (a variation of EMS+) so in theory it should work. I would just try it out and see for yourself and if not help contribute to the code to enhance it. You just need an interface board and they're cheap to make or available to buy from bbqkees

glitter-ball, 2020-01-16:
> Found there is a Bosch Heatronic 3 bus compatible, but no EMS compatible... Maybe the same.
> Someone has this Heatronic module?

My Worcester-Bosch GreenStar 24i is using Junkers HT3 / EMS+ and works OK with EMS-ESP.

Chuckame, 2020-01-19:
I saw on https://hotgoodies.nl/shop/ems/ems-bus-gateway-instructions.html, apparently from bbqkees website :
Boiler compatibility
The interface is electrically compatible with all Bosch brand boilers, heat pumps and devices that have an EMS, EMS+, EMS2, EMS Plus, Heatronic or Heatronic 3 interface.

Chuckame, 2020-01-19:
> What thermostat do you have?

I have this one : https://www.destockage-habitat.com/saint_ouen-93400/regulation_a_programmation_hebdomadaire_avec_sonde_exterieure_un_circuit_elm_leblanc/648855

Chuckame, 2020-01-20:
No it's an FR100, apparently compatible !

proddy, 2020-01-24:
cool. it'll work. if it doesn't we'll make it work.

closed 2020-01-24 by proddy