emsesp/EMS-ESP #298: Gate restart when trying to change temperatures - 1.9.5b17

majdzik84, 2020-01-14:
When I change the room / water temperature in HA, the gate restarts. There is no information in telnet - it simply disconnects. The temperature remains unchanged.

proddy, 2020-01-14:
Thanks for reporting. I'll look into it. I've also heard reports of the latest 1.9.5 of not always detecting EMS devices.

proddy, 2020-01-14:
I tested this on 1.9.5b17 and can't reproduce. Does it always happen? Does it happen if you publish MQTT commands? Or if you try the same from Telnet commands?

majdzik84, 2020-01-14:
I don't think it happens every time. (I tested it before and it seemed that every time) I just raised the temperature and it was set correctly.

-The problem (restart) usually occurs when I set the temperature
- I also noticed restart in telnet but not so often.

After restart, the situation returned and I noticed that it does not give set temp. And current temp.

[image]

sometimes I also see
[image]

proddy, 2020-01-19:
@majdzik84 lets focus on the thermostat readings issue. Are you saying after a restart HC2 is shown correctly and then you get wrong values? Is it just the Mode?

majdzik84, 2020-01-19:
At this point, I see what below:
[image]

At this point, I see what below:

Note that this is night (off) mode.
How do I change the mode to e.g. DAY (heat)

[image]

Unfortunately, when the temperature changes, the gate restarts

proddy, 2020-01-19:
Could you create a new issue for the HC2 mode issue, and lets leave this to the crashes.

On the crashes: I just can't reproduce so need to work with you on a test case that always fails
On the modes: I need some example telegrams for 0x47 and 0x48 (this is HC2 on RC35) to see why the data is missing.

proddy, 2020-01-28:
@majdzik84 are you still experiencing restarts when changing temperatures via MQTT and the Telnet client?

majdzik84, 2020-01-28:
@proddy Unfortunately, restarts still occur at every temperature change by mqtt.

Telnet did not cause any restart during 30 attempts to change temperatures and modes

proddy, 2020-01-31:
@majdzik84 can you post the MQTT command you are sending? We need to trace where in the code its crashing

majdzik84, 2020-01-31:
@proddy mqtt command is sent by Home assistant 0.104.3. how can i get them ??

proddy, 2020-01-31:
try and reproduce with Telnet. Then I know if the bug is in the core code or the mqtt code. Open a telnet and do something like `thermostat temp 22`  or `thermostat temp 2 22` for HC2. We're just trying to make it crash

majdzik84, 2020-01-31:
Change in telenet did not cause restart (10 changes - 0 restarts)
A change in HA always causes a restart. (10 changes - 10 restarts)

proddy, 2020-02-01:
can you use something like MQTT explorer to capture the MQTT that is being sent from HA?

majdzik84, 2020-02-01:
I don't know where I can get the specific mqtt command. Will this screen help? I just changed the HC2 temperature to 21.5C

[image]

MichaelDvP, 2020-02-01:
use **thermostat_cmd_temp2 21.5** to change the temperature, only insert the temperature in °C 
I don't know why mqtt-explorer does not show the other cmd-subscriptions, but also writing a wrong, but valid, json to writeable commands like thermostat_cmd affects a restart. 
**thermostat_cmd {"cmd":"temp2","data":21}** 
works as expected.
**thermostat_cmd {"hc2":{"temp": 21}}**
or something else restarts the ems-esp.
double-check the wiki to send valid commands.

proddy, 2020-02-02:
thanks @MichaelDvP for pointing out the error (again). I'm fixing it now

proddy, 2020-02-02:
fixed. @majdzik84 you shouldn't see any crashes anymore. But I think the problem is your Home Assistant configuration as it looks like you're sending the wrong topics. See https://emsesp.github.io/docs/#/Home-Assistant

majdzik84, 2020-02-02:
@proddy   unfortunately the patch doesn't work for me :(
My HA configuration is exactly like in the wiki. In version 1.9.4b10, these settings worked correctly.

```
  - platform: mqtt
    name: Parter
    modes:
      - "auto"
      - "heat"
      - "off"

    mode_state_topic: "home/ems-esp/thermostat_data"
    current_temperature_topic: "home/ems-esp/thermostat_data"
    temperature_state_topic: "home/ems-esp/thermostat_data"

    temperature_command_topic: "home/ems-esp/thermostat_cmd_temp2"
    mode_command_topic: "home/ems-esp/thermostat_cmd_mode2"

    mode_state_template: "{% if value_json.hc2.mode == 'manual' %} heat {% else %} {{value_json.hc2.mode}} {% endif %}"
    current_temperature_template: "{{ value_json.hc2.currtemp }}"
    temperature_state_template: "{{ value_json.hc2.seltemp }}"

    temp_step: 0.5

```

proddy, 2020-02-02:
oh that's not good. Ok, one more try.

- Telnet to EMS-ESP. You don't logging on for this.
- Use your MQTT Explorer and Publish the Topic `home/ems-esp/thermostat_cmd_temp2` with a value of 20 (use raw or json)

You should see `"Setting new thermostat temperature to 20.0 for heating circuit 2 type 0 (0=auto,1=night,2=day,3=holiday)"` in the Telnet window, and no crashes.

does that work?

majdzik84, 2020-02-02:
Did not work.
The temperature change information is not displayed (gate restart)

proddy, 2020-02-02:
very strange. I'll keep investigating.

proddy, 2020-02-02:
can't figure out what it's failing. Can you try other commands directly from MQTT Explorer, like
topic: `home/ems-esp/thermostat_cmd_mode2` payload:`auto` as JSON or raw
or
topic: `home/ems-esp/thermostat_cmd` payload:`{"cmd":"temp2", "data": 20}` as JSON

if any of those cause a crash I'm out of ideas and will need to send a special firmware version with additional debug logs to trace.

majdzik84, 2020-02-03:
After entering the above commands, the changes worked as they should and there was no restart

majdzik84, 2020-02-03:
@proddy. I also tried now  `home/ems-esp/thermostat_cmd_temp2 with a value of 20 ` and it worked !
I started looking for differences and I think I found situations when the command did not work properly.

### Look: **There is no Mixing module data in telnet:**

[image]

When I reset the gate in telnet, I see the **Mixing module data: and now I can't change the temperature** because the gate restart appears.
[image]

proddy, 2020-02-03:
I can't remember - do you have a mixing module at all?

majdzik84, 2020-02-03:
Yes,I have MM10 module

proddy, 2020-02-03:
ok, so without the MM10 showing up you can set the temp. But with the MM10 recognized it crashes the system? Very strange!

majdzik84, 2020-02-03:
Yes. This is how it looks. In the earlier case, when MM10 could not be detected, I could change the temperature successfully. (In 1.9.4b10 - it has never detected MM10 yet, that's why I haven't experienced it before).

I just tried to change the temperature of hot water and despite the detection of MM10 it turned into a telenet one time, an error popped up and the gate reset (I don't know if it will help)

[image]

proddy, 2020-02-03:
ok, I'll look into the MM10. The piece of code was written by someone else so I need to check what could be causing it.

proddy, 2020-02-03:
I'll need to create a dev build for you with some extra debug. Are you ok to compile and upload yourself or would you prefer receiving the firmware bin files?

majdzik84, 2020-02-03:
if I can, please .bin

proddy, 2020-02-08:
ok, try this [bin](https://github.com/proddy/EMS-ESP/releases/download/travis-dev-build/EMS-ESP-1_9_5b29-d1_mini-noMM.bin) file. I've removed the Mixer code. Please try both telnet and MQTT commands and see if it crashes. If it does, great, we know where to look...

thanks for helping out

majdzik84, 2020-02-08:
@proddy , It looks like the mixer is causing the problem. Now everything works fine (mqtt and telenet did not cause any restart)

proddy, 2020-02-08:
great! I'll start debugging...

proddy, 2020-02-08:
new version available, same filename. this time it will capture the mixing data but not publish to MQTT. see if this one crashes...

majdzik84, 2020-02-09:
Works correctly. No resets

proddy, 2020-02-09:
cool. and if you type `info` does it show Mixer information?

majdzik84, 2020-02-09:
Yes. information is displayed.
[image]

I noticed that there was a bug in boiler temperature (jump temp.)
[image]
[image]

proddy, 2020-02-09:
what happened with the boiler temp exactly?

proddy, 2020-02-09:
ok, new version - this time with MQTT. let me know if it crashes!

majdzik84, 2020-02-09:
Unfortunately there are restarts again. 
Sometimes, after restarting, it is possible to change the temperature once and an error is displayed
```
Setting new thermostat temperature to 21.50 for heating circuit 2 type 0 (0=auto                 ,1=night,2=day,3=holiday)
[MQTT] Error publishing to home/ems-esp/mixing_data with payload {"hc2":{"flowTe                 mp":25.8,"setflowTemp":22,"pumpMod":100}} [error 0]
```

Boiler temperature suddenly changes values ​​from e.g. 40C to 15C.
[image]

proddy, 2020-02-09:
Ok, I think I have a fix for the boiler temp. It's a clash with some Bosch logic I added a while back.

As for the MM10, I need some example telegrams to replicate the error. Are you able to capture 1 or 2 telegrams of type 0x01D8 and 0xAB for me?

proddy, 2020-02-09:
fix for boiler temp in https://github.com/proddy/EMS-ESP/releases/download/travis-dev-build/EMS-ESP-1_9_5b30-d1_mini-noMM.bin

majdzik84, 2020-02-09:
I was only able to capture the 0xAB telegram by entering log v (should I use another command for ?)
```
(20:14:31) Mixing Module -> All, type 0xAB, telegram: 21 00 AB 00 1C 01 17 64 FC 01 00 (CRC=58) #data=7
(20:15:30) Mixing Module -> All, type 0xAB, telegram: 21 00 AB 00 1C 01 1C 64 F9 01 00 (CRC=FC) #data=7
(20:16:30) Mixing Module -> All, type 0xAB, telegram: 21 00 AB 00 1C 01 1D 64 FA 01 00 (CRC=E0) #data=7
(20:17:30) Mixing Module -> All, type 0xAB, telegram: 21 00 AB 00 1C 01 1F 64 F9 01 00 (CRC=CC) #data=7
```

proddy, 2020-02-09:
that's fine. I can simulate the MM10 data with your 0xAB example. But I don't get any crashes so the problem is elsewhere.

MichaelDvP, 2020-02-10:
I don't know if it helps. In my system there is also RC35 and MM10, but a different boiler and i do not use HA, i have ioBroker running on a raspberry with ioBroker-mqtt-server.
I can set temperatures, circulation, etc, via mqtt without restart and the system seems stable, but today i realized a unexpected restart 2 days ago.
```
[APP] EMS-ESP version: 1.9.5b28
[APP] MyESP version: 1.2.26
[APP] Build timestamp: 2020-02-04 08:58:56
[APP] Uptime: 1 day 20 hours 15 minutes 15 seconds
[SYSTEM] Last reset reason: Hardware Watchdog
[SYSTEM] Last reset info: Fatal exception:4 flag:1 (WDT) epc1:0x40000f68 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000
```
At the time of the restart (08.02.2020-12:25) i was updating the raspberry and some ioBroker adapters, and rebooted the raspberry afterwards. 
I tried to trigger a restart by disable MQTT-Server, but i get:
```
[MQTT] TCP Disconnected
[MQTT] Disconnected! (count 1)
[MQTT] Connecting to MQTT...
[MQTT] TCP Disconnected
[MQTT] Disconnected! (count 2)
...
[MQTT] Disconnected! (count 15)
[MQTT] Connecting to MQTT...
[MQTT] MQTT connected
Setting new thermostat temperature to 21.0 for heating circuit 2 type 0 (0=auto,1=night,2=day,3=holiday)
Setting boiler warm water circulation off
```
a restart of the raspberry gives the same result, no restart of ems-esp. 
A full protocol is attached (for mqtt-settings etc. btw: i missed the ntp-server-setting in terminal)

A hour ago i updated and rebooted my wifi-router and ems-esp crashes to.
```
 [SYSTEM] Last reset reason: Exception
 [SYSTEM] Last reset info: Fatal exception:9 flag:2 (EXCEPTION) epc1:0x401053a0 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000003 depc:0x00000000
```
But i can't say what exactly triggers the ems-esp-restart.

[20200210_084047_EMS.log](https://github.com/proddy/EMS-ESP/files/4180374/20200210_084047_EMS.log)

proddy, 2020-02-11:
I still don't know what is causing the crashes. I'm hoping that someone else will bump into this problem so I can triangulate. Thanks @MichaelDvP for providing the feedback

majdzik84, 2020-02-11:
ok, in that case i'm coming back to the version you shared earlier without mixer data, hoping that in the future it will be resolved.

proddy, 2020-02-11:
@majdzik84 one thing we could try is if you just leave logging on for 2 minutes (full verbose) and I use that to simulate your setup and watch for the crash. I've looked through the Mixing code and can't see why it would cause a crash, but then it always ends up being something really obvious

majdzik84, 2020-02-11:
@proddy  I'm sending logs :)

[logs.txt](https://github.com/proddy/EMS-ESP/files/4188127/logs.txt)

proddy, 2020-02-11:
oh, well there goes my evening ! :-)

MichaelDvP, 2020-02-12:
@majdzik84 you are publishing very often. Maybe it's not the mixer, but the more payload in short time which overloads the server and causes the reboots. Have you tried to set publish_time to a larger value (i use 60 sec).

proddy, 2020-02-12:
well spotted ! Yes looks like publish time is 5?

majdzik84, 2020-02-12:
yes, the time was set to 5sec. I changed to 60 sec but there are no changes. Still restarts

proddy, 2020-02-12:
@majdzik84 I used your data and still couldn't crash EMS-ESP. Two more things you could try:
- leave verbose logging on, send a command via MQTT and when it crashes send me the log so I can see the last known operation

majdzik84, 2020-02-12:
that's what i did last time.

```
Starting scheduled MQTT publish...
Publishing boiler data via MQTT
Publishing thermostat data via MQTT
Publishing external sensor data via MQTT
(21:23:44) UBAMaster -> All, type 0x34, telegram: 08 00 34 00 0A 00 75 00 75 00 00 00 03 00 00 00 78 00 00 3C 00 (CRC=CA) #data=17
(21:23:44) UBAMaster -> All, type 0x18, telegram: 08 00 18 00 2C 01 6D 64 00 41 01 24 62 80 00 00 75 01 65 00 00 10 36 41 00 E3 00 02 00 (CRC=BF) #data=25

RESTART
```

proddy, 2020-02-12:
@majdzik84 made a debug version [here](https://github.com/proddy/EMS-ESP/releases/download/travis-dev-build/EMS-ESP-1_9_5b30-d1_mini-debug.bin)

majdzik84, 2020-02-12:
I have uploaded. Should I send a new log v? restart still occurs

proddy, 2020-02-12:
yes please. Just want to know what happens just before it crashes. Probably an MQTT publish?

majdzik84, 2020-02-12:
Logs [logs.txt](https://github.com/proddy/EMS-ESP/files/4194270/logs.txt)

[49 more replies]

closed 2020-02-25 by proddy