emsesp/EMS-ESP #301: Nefit Proline HRC30 CW5 - Tap Water Mode

TimmermanTim, 2020-01-17:
**Bug description**
My boiler (Nefit Proline HRC30) has four settings for warm water that I can choose in the menu on the boiler: 

– Comf (EMS-ESP: hot)
– Norm (ESM-ESP: intelligent)
– Eco (ESM-ESP: eco)
– OFF (warm water disabled)

There is no separate setting to switch off warm water.

From EMS-ESP I can send the "boiler comfort" command but it doesn't support the setting "off".
If i send the command  "boiler tapwater off", nothing happens on the boiler and the warm water setting isn't switched to "OFF".
Subsequently warm water still functions. This also means that the shower timer and one-time shot of cold water don't work.

**Steps to reproduce**
Send command "boiler tapwater off" to a Nefit HRC30.

**Expected behavior**
By sending the command  "boiler tapwater off" the boiler should set the warm water setting to "OFF".

**Screenshots**

**Normal, intelligent mode:**

[image]

Boiler stats:
  Boiler: Nefit Proline (DeviceID:0x08 ProductID:122 Version:03.09)
  Hot tap water: off
  Central heating: off
  Warm Water activated: **on**
  Warm Water circulation pump available: on
  Warm Water comfort setting: **Intelligent**

**Setting "OFF" in the boiler menu:**

[image]

Boiler stats:
  Boiler: Nefit Proline (DeviceID:0x08 ProductID:122 Version:03.09)
  Hot tap water: off
  Central heating: off
  Warm Water activated: **off**
  Warm Water circulation pump available: on
  Warm Water comfort setting: **Hot**

**Setting boiler tapwater off from EMS-ESP:**

Boiler stats:
  Boiler: Nefit Proline (DeviceID:0x08 ProductID:122 Version:03.09)
  Hot tap water: off
  Central heating: off
  Warm Water activated: **on**
  Warm Water circulation pump available: on
  Warm Water comfort setting: Intelligent

**Device information**
ESP8266 System stats:

 [APP] EMS-ESP version: 1.9.4
 [APP] MyESP version: 1.2.22
 [APP] Build timestamp: 2019-12-15 23:13:29
 [APP] Uptime: 0 days 0 hours 42 minutes 50 seconds
 [APP] System Load: 1%
 [WIFI] WiFi Hostname: ems-esp
 [WIFI] WiFi IP: 192.168.10.112
 [WIFI] WiFi signal strength: 92%
 [WIFI] WiFi MAC: 98:F4:AB:CD:65:C5
 [MQTT] is connected (heartbeat enabled)
 [NTP] Local Time is 11:57:38 CET  (1579262258)
 [SYSTEM] System is Stable
 [SYSTEM] Board: PLATFORMIO_D1_MINI
 [SYSTEM] CPU frequency: 80 MHz
 [SYSTEM] SDK version: 2.2.2-dev(38a443e)
 [SYSTEM] CPU chip ID: 0xCD65C5
 [SYSTEM] Core version: 2_6_2
 [SYSTEM] Boot version: 31
 [SYSTEM] Boot mode: 1
 [SYSTEM] Last reset reason: Power on
 [SYSTEM] Last reset info: flag: 0
 [SYSTEM] Restart count: 0
 [SYSTEM] # TCP disconnects: 0
 [SYSTEM] rtcmem status: blocks:2 addr:0x60001280
 [SYSTEM] rtcmem 00: 1163087990
 [SYSTEM] rtcmem 01: 65536
 [FLASH] Flash chip ID: 0x1640EF
 [FLASH] Flash speed: 40000000 Hz
 [FLASH] Flash mode: DIO
 [FLASH] Flash size (CHIP): 4194304
 [FLASH] Flash size (SDK): 4194304
 [FLASH] Flash Reserved: 4096
 [MEM] Firmware size: 614912
 [MEM] Max OTA size: 2523136
 [MEM] OTA Reserved: 16384
 [MEM] Free Heap: 24128 bytes initially | 13800 bytes used (57%) | 10328 bytes free (42%)

**Info:**

EMS-ESP system stats:
  System logging set to None
  LED: on, Listen mode: off
  Boiler: enabled, Thermostat: disabled, Solar Module: disabled, Mixing Module: disabled
  Shower Timer: enabled, Shower Alert: enabled

EMS Bus stats:
  Bus is connected, protocol: Buderus
  Rx: # successful read requests=223, # CRC errors=26
  Tx: Last poll=0.148 seconds ago, # successful write requests=52

Boiler stats:
  Boiler: Nefit Proline (DeviceID:0x08 ProductID:122 Version:03.09)
  Hot tap water: off
  Central heating: off
  Warm Water activated: on
  Warm Water circulation pump available: on
  Warm Water comfort setting: Intelligent
  Warm Water selected temperature: 60 C
  Warm Water desired temperature: 70 C
  Warm Water current temperature: ? C
  Warm Water current tap water flow: 0.0 l/min
  Warm Water # starts: 40824 times
  Warm Water active time: 31 days 20 hours 3 minutes
  Warm Water 3-way valve: on
  Selected flow temperature: 5 C
  Current flow temperature: 59.8 C
  Return temperature: 58.7 C
  Gas: off
  Boiler pump: off
  Fan: off
  Ignition: off
  Circulation pump: off
  Burner selected max power: 0 %
  Burner current power: 0 %
  Flame current: 0.0 uA
  System pressure: 1.4 bar
  System service code: 0H (203)
  Heating temperature setting on the boiler: 65 C
  Boiler circuit pump modulation max power: 100 %
  Boiler circuit pump modulation min power: 60 %
  Boiler temperature: ? C
  Pump modulation: 0 %
  Burner # starts: 58940 times
  Total burner operating time: 168 days 3 hours 3 minutes
  Total heat operating time: 136 days 7 hours 0 minutes
  Total UBA working time: 1890 days 9 hours 43 minutes

MichaelDvP, 2020-01-17:
There is a mqtt command  "ems.boiler_cmd_wwactivated" for this. 
Or in terminal try 
send 0b 08 33 01 00
send 0b 08 33 01 FF
for off/on

TimmermanTim, 2020-01-18:
> There is a mqtt command "ems.boiler_cmd_wwactivated" for this.
> Or in terminal try
> send 0b 08 33 01 00
> send 0b 08 33 01 FF
> for off/on

Unfortunately the commands do not work. Tapwater keeps active.
I recently got a Moduline 400 and if I switch off the tapwater there, it is not sensed by EMS-ESP.

proddy, 2020-01-18:
@TimmermanTim correct, I hadn't implemented 'off'. The telegram logic is indeed

`0B 08 33 09 XX`

where XX is 0x00 for Hot, 0xD8 for Eco and 0xEC for Intelligent.

To get the value of Off I'll need an example 0x33 UBAParameterWW telegram when in off mode. You can do this with `read boiler 33` with verbose logging. Post it back here and we'll figure out the rest together.

TimmermanTim, 2020-01-18:
@proddy 
My boiler reacts differently, the messages for Eco and Intelligent are the same, but for Hot and Off they are different:

Off:
Requesting type UBAParameterWW(0x33) from dest 0x08
(03:56:20.758) Sending read of type 0x33 to 0x08, telegram: 0B 88 33 00 20 (CRC=78)
(03:56:20.799) Boiler -> me, type 0x33, telegram: 08 0B 33 00 08 00 3C FB 00 28 FF 02 46 00 FF (CRC=FF) #data=11

On (intelligent):
Requesting type UBAParameterWW(0x33) from dest 0x08
(03:56:55.434) Sending read of type 0x33 to 0x08, telegram: 0B 88 33 00 20 (CRC=78)
(03:56:55.476) Boiler -> me, type 0x33, telegram: 08 0B 33 00 08 FF 3C FB 00 28 FF 02 46 EC FF (CRC=40) #data=11

Comfort (Hot):
Requesting type UBAParameterWW(0x33) from dest 0x08
(00:02:48.098) Sending read of type 0x33 to 0x08, telegram: 0B 88 33 00 20 (CRC=78)
(00:02:48.139) Boiler -> me, type 0x33, telegram: 08 0B 33 00 08 FF 3C FB 00 28 FF 02 46 00 FF (CRC=81) #data=11

Eco:
Requesting type UBAParameterWW(0x33) from dest 0x08
(00:06:08.589) Sending read of type 0x33 to 0x08, telegram: 0B 88 33 00 20 (CRC=78)
(00:06:08.630) Boiler -> me, type 0x33, telegram: 08 0B 33 00 08 FF 3C FB 00 28 FF 02 46 D8 FF (CRC=28) #data=11

proddy, 2020-01-19:
ok so the changes I suggest when showing the warm water comfort
- rename Hot to Comfort/Hot
- show Off if the Warm Water activated is 'off', taking from the boiler stats and not 0x33

?

MichaelDvP, 2020-01-19:
@proddy I thought boiler stats "Warm Water activated"  is 33 byte1,  and comfort setting is 33 byte9.
Also the setting from TimmermanTim  shows off als byte1=0, byte9=0, and the comfort settings as byte1=FF, byte9= 00,EC,D8. So maybe the off-command should also set comfort to hot to give a correct display. 
@TimmermanTim what's the display if you send
send 0b 08 33 01 00
send 0b 08 33 09 00

TimmermanTim, 2020-01-19:
@MichaelDvP 
send 0b 08 33 01 00 - Turns warm water to OFF
send 0b 08 33 09 00 - Doesn't seem to do anything

TimmermanTim, 2020-01-19:
I believe the "boiler comfort" command works for hot, intelligent and eco. Only the "off" setting was missing. I will double check this tonight.

proddy, 2020-01-19:
> @proddy I thought boiler stats "Warm Water activated" is 33 byte1, and comfort setting is 33 byte9.
> Also the setting from TimmermanTim shows off als byte1=0, byte9=0, and the comfort settings as [...]

that's correct. Unless @TimmermanTim wants to look like the front-panel of the boiler which says 'Off'

proddy, 2020-01-19:
the boiler comfort command only changes byte (comfort setting). Off is a special case. Do you really want to turn off the WW? If so its probably better to make a dedicated command.

TimmermanTim, 2020-01-19:
@proddy I was under the impression that the command `boiler tapwater off` would set `Warm Water activated: on` to `Warm Water activated: off` but it doesn't on my boiler.
Sending `send 0b 08 33 01 00` does turn the tapwater off on my boiler. And under `info` i see `Warm Water activated: off`.
But if I then issue the command `boiler comfort intelligent` tapwater is still switched off.

The strange thing is the shower timer seems to work. How is the stream of cold water achieved in EMS-ESP? Does it work differently than issuing `boiler tapwater off`?

proddy, 2020-01-19:
the command `boiler tapwater` is bit of untested territory. Basically it puts the whole boiler in what is known as Function Test mode (this is what the repairman does) and then turns off the 3-way valve. This was the only way I could simulate sending abruptly stopping the hot water from streaming. I'll remove the function in the next version.

I was wondering why do you want to turn off the warm tap water anyway? What are you trying to achieve?

TimmermanTim, 2020-01-19:
In the end I want to be able to turn off the tapwater when there's no one home for a longer period of time like on holiday to save energy. I can already put the boiler in power save mode by putting the thermostat in low/night mode. 
I know now I can achieve this by using the MQTT command `home/ems-esp/boiler_cmd_wwactivated 0`. So this issue can be closed. Thanks to all of you for your help!

closed 2020-01-19 by TimmermanTim