emsesp/EMS-ESP #302: Status / Error Code to Useful Message

TimmermanTim, 2020-01-17:
**Is your feature request related to a problem? Please describe.**
No

**Describe the solution you'd like**
Make "System service code" user friendly. So, create a "database" of status and error codes for specific devices and show the user a user friendly message instead of the code.
In the service manual for my boiler the error codes are explained. So I could fill a table with status code and the meaning.

**Describe alternatives you've considered**
Linking the code to text in HASS.

IGastel, 2020-01-17:
I am working on a something for a user friendly status message. I have a
python nested dictionary (seperate file) with the Nefit error codes and
(dutch) descriptions.
I am trying to make an appdaemon app for home assistant. But i have to
learn python and appdaemon first. So it will take a while. I can share my
dictionary. I have combined the status code and error number to create a
single line search entry. I got the information from the installation
manual.

PhillyGilly, 2020-01-17:
Did you see the discussion in #281 ? (it's a closed issue so you my have to sort by closed.)
It worked for me.

glitter-ball, 2020-01-18:
The code I used in #281 is an easy way to go, albeit it's a bit crude. You can then customise the message completely for your local language and specific to your boiler. Short-lived statuses (e.g. ignition) may well be missed because it won't be up long enough for MQTT to pick it up.
[image]

proddy, 2020-01-18:
@TimmermanTim it would be easier to implement the logic on the receiving end (i.e. domoticz, node-red, home assistance) for the simple reason that there are so many variations of boilers from the different brands and the ESP8266 simply doesn't have enough available memory to store all the messages. With telnet and web running the heap left is only around 8k.

TimmermanTim, 2020-01-18:
> @TimmermanTim it would be easier to implement the logic on the receiving end (i.e. domoticz, node-red, home assistance) for the simple reason that there are so many variations of boilers from the [...]

Thanks for all your replies and ideas. I will implement the solution given by @glitter-ball.

closed 2020-01-18 by TimmermanTim