emsesp/EMS-ESP #305: HC2 mode issue - RC35

majdzik84, 2020-01-19:
https://github.com/proddy/EMS-ESP/issues/298#issuecomment-576006762
The HC2 circuit values ​​do not appear. When the temperature is set, the gate restarts

```
(17:42:58) Sending read of type 0x48 to 0x10, telegram: 0B 90 48 00 20 (CRC=4D)
(17:42:58) Thermostat -> Me, type 0x48, telegram: 10 0B 48 00 00 00 2A 00 E0 00 00 00 00 00 00 00 00 11 1B 00 (CRC=41) #data=16
(17:42:58) Sending read of type 0x47 to 0x10, telegram: 0B 90 47 00 20 (CRC=71)
(17:42:58) Thermostat -> Me, type 0x47, telegram: 10 0B 47 00 03 2A 2C 26 08 28 00 00 00 05 05 2D 01 01 04 4B 05 4B 01 00 3C FF 0D 05 05 02 02 (CRC=E0) #data=27

```

When temperatures appear, the reading is as follows

```
(17:47:59) Thermostat -> Me, type 0x48, telegram: 10 0B 48 00 40 00 2A 00 E0 00 00 00 00 00 00 00 00 11 05 00 (CRC=9E) #data=16
(17:47:59) Thermostat -> Me, type 0x47, telegram: 10 0B 47 00 03 2A 2C 26 08 28 00 00 00 05 05 2D 01 01 04 4B 05 4B 01 00 3C FF 0D 05 05 02 02 (CRC=E0) #data=27
```

proddy, 2020-01-19:
thanks, looking into it...

lmdecroos, 2020-01-19:
@majdzik84  Do you have a separate thermostat for your HC2? I have a similar behavior. 
Since version 1.9.5bx the temperatures of my HC2 are only available in the gateway after the mode has been changed ( Auto / Day / Night). It seems the new logic needs a trigger to filter the temperature and setpoint from the telegram messages from HC2

majdzik84, 2020-01-19:
@lmdecroos I have only one thermostat - RC35 ( HC2 control)

proddy, 2020-01-28:
Can someone explain the problem again? Is it the HC2 is not displayed and only after the mode has changed? Does it get displayed when EMS-ESP boots up? Also when the temp is changed?

I don't have the RC35 so difficult to reproduce at home.

lmdecroos, 2020-01-28:
@proddy see my posts in #238

At boot it’s not detecting the room and setpoint temperature
A temperature change is not triggering the readout
Only when after a switch from mode it’s start detecting these temperatures

I’m using b17 with an uptime of 2 weeks, I can test a newer build by end of this week

majdzik84, 2020-01-29:
Unfortunately, this problem still occurs in the 1.9.5b20 version

lmdecroos, 2020-01-30:
yes same here
only after a change from night to day the temperatures for HC2 become ( within a few seconds) available

proddy, 2020-01-31:
does `thermostat read 48` refresh the data correctly? What is the output from the 0x48 telegram?

majdzik84, 2020-01-31:
No, the data does not appear. I can only see ...
[image]

`(00:32:33) Thermostat -> Me, type 0x48, telegram: 10 0B 48 00 00 00 2A 00 DE 00 00 00 00 00 00 00 64 00 16 00 (CRC=F4) #data=16`

lmdecroos, 2020-01-31:
same for me

(00:32:39) Thermostat -> Me, type 0x48, telegram: 10 0B 48 00 00 00 1B 00 A2 00 00 00 00 00 00 00 00 00 05 00 (CRC=C2) #data=16
(00:32:40) Thermostat -> Me, type 0x47, telegram: 10 0B 47 00 01 1B 24 22 00 28 00 00 00 05 05 2D 01 01 04 4B 05 4B 01 00 3C FF 11 05 05 02 01 (CRC=BA) #data=27

(00:33:41) Thermostat -> Me, type 0x48, telegram: 10 0B 48 00 00 00 1B 00 A2 00 00 00 00 00 00 00 00 00 05 00 (CRC=C2) #data=16
(00:33:42) Thermostat -> Me, type 0x47, telegram: 10 0B 47 00 01 1B 24 22 00 28 00 00 00 05 05 2D 01 01 04 4B 05 4B 01 00 3C FF 11 05 05 02 01 (CRC=BA) #data=27

Heating Circuit 2
 Current room temperature: ? C
 Setpoint room temperature: ? C
 Program is set to Summer mode
 Day temperature: 18.0 C
 Night temperature: 13.5 C
 Vacation temperature: 17.0 C
 Calculatet flow temperature: ? C
 Mode is set to night

proddy, 2020-02-01:
thanks @lmdecroos @majdzik84  for the additional info. I helped me find the root cause of this problem. It's related to an earlier change I made in https://github.com/proddy/EMS-ESP/issues/238#issuecomment-570269941 which said:

> I see the problem now. It's this following telegram from when HC3 is queried: 
`Thermostat -> me, type 0x52, telegram: 10 0B 52 00 00 00 00 00 00 00 00 00 00 00 00 00 00 11 00 00 (CRC=B6) #data=16`
 because when a 0x52 comes in I just assume there is a HC3 active. Which now is not always the case. I've changed the code to look at the first data byte and if it's zero I will ignore. I'm basing this off your 0x3E telegram (HC1) who's first byte
 is 0x04 meaning in automatic mode. Hope that helps, thanks for working with me on fixing this.

In the example 0x48's you sent, the first data byte (i.e. the 5th in the whole telegram) is always zero so the complete telegram is ignored. e.g. 

type 0x48, telegram: 10 0B 48 00 **00** 00 1B 00 A2 00 00 00 00 00 00 00 00 00 05 00

I'll revert the change. Let me know if it causes any other issues with HC1

lmdecroos, 2020-02-01:
> thanks @lmdecroos <https://github.com/lmdecroos> @majdzik84 <https://github.com/majdzik84> for the additional info. I helped me find the root cause of this problem. It's related to an earlier change I made in #238 (comment) <https://github.com/proddy/EMS-ESP/issues/238#issuecomment-570269941> which said:

just tested b24 and was immediately  ok after boot (and a reboot)  :)

@proddy : well done !

Thermostat data:
  Thermostat: RC35 (DeviceID:0x10 ProductID:86 Version:21.08)
  Thermostat time is 13:37:40 01/02/2020
  Heating Circuit 1
   Current room temperature: 19.5 C
   Setpoint room temperature: 19.0 C
   Day temperature: 19.0 C
   Night temperature: 15.0 C
   Vacation temperature: 17.0 C
   Calculated flow temperature: 5 C
   Mode is set to auto
   Day Mode is set to day
  Heating Circuit 2
   Current room temperature: 20.2 C
   Setpoint room temperature: 13.5 C
   Day temperature: 18.0 C
   Night temperature: 13.5 C
   Vacation temperature: 17.0 C
   Calculated flow temperature: 5 C
   Mode is set to night
   Day Mode is set to night

Mixing module data:
  Switch temperature: 76.4 C
  Mixing Circuit 2
   Setpoint flow temperature: 0 C
   Current pump modulation: 0 %

proddy, 2020-02-02:
ok, closing this for now and will re-open if we see problems occurring

closed 2020-02-02 by proddy