emsesp/EMS-ESP #306: Error Reset - button

majdzik84, 2020-01-20:
Is it possible to add the option to activate the RESET button located on the gas furnace (it is used to reset the error)?
[image]

giovanne123, 2020-01-23:
I'm also interested in this enhancement. Would be fine if it is possible to send the "Reset" command via EMS-Bus/EMS-ESP.
But maybe the Buderus controller doesn't send the pressed button to the bus because it is only relevant for the controller itself to reset and not for the connected devices !?

bbqkees, 2020-01-24:
I don't think the boiler sends out a kind of a reset telegram.

proddy, 2020-01-24:
I saw telegram type 0x05 looks like it resets error messages on the boiler. You can try sending some read/query requests to see what it comes back with and then send a write to set the numbers. Bit of trial and error.

proddy, 2020-03-08:
I haven't been able to find an EMS command that does this.

proddy, 2020-03-10:
@norberts1 wrote when I asked:

@proddy regarding that #306 I found also that message:0x05 with many parameters set-/select-able.
But it seems also a bit dangerous telegram using it, if somebody don't know what he is doing.
At least to reset that error-memory there is one/two placeholder-byte(s) for it in that message.
What I found is:
```
payload-byte   MsgID  Meaning
0 & 1          5_0_0  Reset-Mode (Manuel, others)
.
.
12 & 13        5_12_0  Reset that Error-memory
...
```
So that resulting (untested) command could be:
`8B 00 05 0C <reset-flag>`
I don't know regarding that value of reset-flag, but it should be: 1, or 0x00FF.

proddy, 2020-11-15:
@susisstrolc found:

> Pushing the "Reset" Button on MC10 shows the following telegram:
[telegram] Rx: 09 08 05 08 FF 2B
So it might be possible to do a Reset on minor (non-locking errors) via the EMS bus.

worth trying out?

susisstrolch, 2020-11-17:
Some notes from the wild...
Today our boiler also shows the "H6" message. At first I tried to fake in as MC10:
```
ems-esp:/# send telegram "09 08 05 08 FF"
002+16:05:26.356 D 11: [telegram] Sending write Tx [#103], telegram: 09 08 05 08 FF 2B
002+16:05:26.666 D 12: [telegram] Sending write Tx [#104], telegram: 09 08 05 08 FF 2B
002+16:05:26.980 D 13: [telegram] Sending write Tx [#105], telegram: 09 08 05 08 FF 2B
002+16:05:27.305 D 14: [telegram] Sending write Tx [#106], telegram: 09 08 05 08 FF 2B
002+16:05:27.523 E 15: [telegram] Last Tx Write operation failed after 3 retries. Ignoring request.
```
Result: boiler immediatelly showed "6L"...
So I tried the "reguar" way:
```
ems-esp:/# send telegram "0B 08 05 08 FF"
002+16:08:16.851 N 50: [telegram] Rx: 08 0B 05 00 66
002+16:08:16.857 N 51: [emsesp] Boiler(0x08) -> Me(0x0B), ?(0x05), data: <empty>
002+16:08:17.880 N 52: [telegram] Rx: 08 00 20 00 94 0B 08 11 1C 1C 02 57 B9 00 94 0B 07 11 38 1B 02 57 B5 00 94 0B 07 11 14 1E 0C
002+16:08:18.174 N 53: [telegram] Rx: 08 00 20 1A 02 57 B2 00 94 0B 06 11 2B 1A 02 57 AF 00 94 0B 06 11 1D 1D 02 57 AE 00 89
002+16:08:18.579 N 54: [telegram] Rx: 08 00 18 00 2E 00 F7 64 00 01 01 30 01 83 00 7D 00 80 00 00 00 FF 30 43 00 00 FF 00 10 83 00 01
002+16:08:18.859 N 55: [telegram] Rx: 08 00 18 1B 00 00 00 00 00 C6 6F 2E 00 00 F4
002+16:08:19.277 N 56: [telegram] Rx: 08 00 1F 00 8F 04 0F 15 1B 3A 01 30 59 00 8D 0C 07 16 05 3F 00 D8 8A 00 8D 0C 16 15 03 3A 8C
002+16:08:19.574 N 57: [telegram] Rx: 08 00 1F 1A 00 D8 84 00 8D 0C 14 15 14 38 00 D8 7C 00 8D 0C 03 15 30 34 00 D8 2B 00 39
002+16:08:19.978 N 58: [telegram] Rx: 08 00 18 00 2E 00 F7 64 00 01 01 30 01 83 00 7D 00 80 00 00 00 FF 30 43 00 00 FF 00 10 83 00 01
002+16:08:20.258 N 59: [telegram] Rx: 08 00 18 1B 00 00 00 00 00 C8 34 2E 00 00 FE
002+16:08:20.656 N 60: [telegram] Rx: 08 00 15 00 00 3C 11 0B 15 4B
002+16:08:20.871 N 61: [telegram] Rx: 08 00 1C 00 94 0B 0D 0F 10 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 D2
002+16:08:26.458 N 62: [telegram] Rx: 08 00 07 00 0B 01 00 00 00 00 00 00 00 00 00 00 00 00 00 F3
002+16:08:29.070 N 63: [telegram] Rx: 08 00 18 00 2E 00 F7 64 00 01 01 30 01 83 00 7D 00 80 00 00 00 FF 30 43 00 00 FF 00 10 83 00 01
002+16:08:29.350 N 64: [telegram] Rx: 08 00 18 1B 00 00 00 00 00 D9 E6 2E 00 00 31
```
MC10 now shows the regular temperature - no warning anymore.

proddy, 2020-11-17:
interesting, thanks for sharing @susisstrolch . Is the H6 the same as reported in the thermostat on https://github.com/proddy/EMS-ESP/issues/614 ?

susisstrolch, 2020-11-17:
Jep, that's the same one.
H6 may be triggered by various events. In my case I have multiple occurances of ServiceCode 512 - Flammenabriss.
With my boiler, if I have 6 consecutive 512 w/o any successfull start between, it'll go to 6L / 548.
Otherwise, if you have sporadic reoccuring 512 it will show the H6.
By the way, H4 will be shown if the ignition kathodes will go bad (and may be other minor support cases).

susisstrolch, 2020-11-17:
> Some notes from the wild...
Today our boiler also shows the "H6" message. At first I tried to fake in as MC10:
> Result: boiler immediatelly showed "6L"...

The "6L" wasn't a result of the faked "MC10 / BC10" telegram. Instead the error simply occured before my test.
So I also found out that you cannot reset locking errors via telegram.

MichaelDvP, 2020-11-17:
I think the 6L code you have also received in serviceCode and it is stored in 0x10 or 0x11 telegram?
Have you seen the H6 code somewhere in telegrams, we are searching for it in #614?

Interresting the catched telegrams: 0x20 seems to be a log, last event on 17.11.2020 08:28 
0x15 is the next maintenance on 17.11.2021, maybe updated by the reset-code?
0x1C is also unknown and contains a date 15.11.2020 13:16 Do you know what happend to this time?

susisstrolch, 2020-11-17:
> 0x1C is also unknown and contains a date 15.11.2020 13:16 Do you know what happend to this time?

Hmm, I could try to retreive it from the FHEM DB...

proddy, 2020-11-17:
> Nothing peculiar (at least by servicecode). syslog doesn't help, because it doesn't show the raw telegrams.

`log watch` will send raw telegrams to the syslog

susisstrolch, 2020-11-17:
@proddy remember my comments @gitter about decoupling syslog priority from "log watch" or "watch"...
syslog priority is settable via web - log watch isn't...

A compile time switch to set the default log level would be an alternative, so syslog would send raw at the first minute...

MichaelDvP, 2020-11-17:
There is the unused loglevel `trace` / `all` above debug. Use this to add the telegrams to the debug output?
Then we only need to add ALL to the web selectbox (EMSESPSettingsController.tsx, line233).

i.e. emsesp.cpp line 605:
```
        } else {
            LOG_TRACE(pretty_telegram(telegram).c_str());
        }
    } else {
        LOG_TRACE(pretty_telegram(telegram).c_str());
    }
```

susisstrolch, 2020-11-17:
@MichaelDvP Sounds good!

susisstrolch, 2020-11-18:
I'll investigate during the day, because I currently (exactly: since the last maintenance) have the problem with the reoccuring H6 warning and Service code 513...

risen, 2021-03-22:
A remote reset would be quite useful for me too.

I'm using a Buderus GB132, BC10 and RC35. I regularly get a "4E" service code, which also requires me to press the reset button on the BC10.

When I press the button, I see the following:
```
[emsesp] Controller(0x09) -> Boiler(0x08), ?(0x05), data: 5A
```

Or again in "raw" mode:
```
[telegram] Rx: 09 08 05 00 5A 9E
```

Do you think it's safe to replay this telegram?

proddy, 2021-03-22:
give it a go! `send telegram "09 08 05 00 5A"`

MichaelDvP, 2021-03-22:
`call system send "0B 08 05 00 5A"`

risen, 2021-03-22:
```
ems-esp:/# call system send "0B 08 05 00 5A"
000+00:01:30.570 N 44: [emsesp] Boiler(0x08) -> Me(0x0B), ?(0x05), data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
```
It works!

Then a whole bunch of messages like this:
```
000+00:01:31.370 N 45: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 00 02 D5 00 00 00 02 24 E0 80 00 02 5E 02 D0 00 00 10 30 55 01 0E 00 00 00
000+00:01:31.894 N 46: [emsesp] Thermostat(0x10) <- Boiler(0x08), UBAErrorMessage1(0x10), data: 1B
000+00:01:31.928 N 47: [emsesp] Boiler(0x08) -> Thermostat(0x10), UBAErrorMessage1(0x10), data: 34 45 00 E1 95 03 0E 16 2B 00 00 00
000+00:01:32.022 N 48: [emsesp] Thermostat(0x10) <- Boiler(0x08), UBAErrorMessage1(0x10), data: 1B (offset 12)
000+00:01:32.052 N 49: [emsesp] Boiler(0x08) -> Thermostat(0x10), UBAErrorMessage1(0x10), data: 34 45 00 E1 95 03 0C 16 1F 00 00 00 (offset 12)
000+00:01:32.148 N 50: [emsesp] Thermostat(0x10) <- Boiler(0x08), UBAErrorMessage1(0x10), data: 1B (offset 24)
000+00:01:32.180 N 51: [emsesp] Boiler(0x08) -> Thermostat(0x10), UBAErrorMessage1(0x10), data: 34 45 00 E1 95 03 17 15 0E 00 00 00 (offset 24)
000+00:01:32.272 N 52: [emsesp] Thermostat(0x10) <- Boiler(0x08), UBAErrorMessage1(0x10), data: 1B (offset 36)
000+00:01:32.300 N 53: [emsesp] Boiler(0x08) -> Thermostat(0x10), UBAErrorMessage1(0x10), data: 34 45 00 E1 95 03 0E 14 2F 00 00 00 (offset 36)
000+00:01:32.469 N 54: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 00 02 D5 00 00 00 02 24 E0 80 00 02 5F 02 CF 00 00 12 30 55 01 0E 00 00 00
000+00:01:32.894 N 55: [emsesp] Thermostat(0x10) <- Boiler(0x08), UBAErrorMessage1(0x10), data: 1B (offset 48)
000+00:01:32.922 N 56: [emsesp] Boiler(0x08) -> Thermostat(0x10), UBAErrorMessage1(0x10), data: 34 45 00 E1 95 03 14 13 32 00 00 00 (offset 48)
000+00:01:33.023 N 57: [emsesp] Thermostat(0x10) <- Boiler(0x08), UBAErrorMessage1(0x10), data: 1B (offset 60)
000+00:01:33.046 N 58: [emsesp] Boiler(0x08) -> Thermostat(0x10), UBAErrorMessage1(0x10), data: 34 46 00 DB 95 03 13 13 39 00 00 00 (offset 60)
000+00:01:33.148 N 59: [emsesp] Thermostat(0x10) <- Boiler(0x08), UBAErrorMessage1(0x10), data: 18 (offset 72)
000+00:01:33.174 N 60: [emsesp] Boiler(0x08) -> Thermostat(0x10), UBAErrorMessage1(0x10), data: 34 46 00 DB 95 03 0A 13 0C 00 00 00 (offset 72)
000+00:01:33.273 N 61: [emsesp] Thermostat(0x10) <- Boiler(0x08), UBAErrorMessage1(0x10), data: 0C (offset 84)
000+00:01:33.294 N 62: [emsesp] Boiler(0x08) -> Thermostat(0x10), UBAErrorMessage1(0x10), data: 34 45 00 E1 95 03 14 12 16 00 00 00 (offset 84)
000+00:01:33.433 N 63: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 00 02 D5 00 00 00 02 24 E0 80 00 02 5E 02 CF 00 00 11 30 55 01 0E 00 00 00
000+00:01:33.894 N 64: [emsesp] Thermostat(0x10) <- Boiler(0x08), UBAErrorMessage2(0x11), data: 1B
000+00:01:33.915 N 65: [emsesp] Boiler(0x08) -> Thermostat(0x10), UBAErrorMessage2(0x11), data: 32 46 01 04 95 03 0E 16 21 00 00 00
000+00:01:34.022 N 66: [emsesp] Thermostat(0x10) <- Boiler(0x08), UBAErrorMessage2(0x11), data: 1B (offset 12)
000+00:01:34.069 N 67: [emsesp] Boiler(0x08) -> Thermostat(0x10), UBAErrorMessage2(0x11), data: 32 45 00 CF 00 52 A6 AC 00 00 00 00 (offset 12)
000+00:01:34.147 N 68: [emsesp] Thermostat(0x10) <- Boiler(0x08), UBAErrorMessage2(0x11), data: 1B (offset 24)
000+00:01:34.168 N 69: [emsesp] Boiler(0x08) -> Thermostat(0x10), UBAErrorMessage2(0x11), data: 32 45 00 CF 00 52 A6 AC 00 00 00 00 (offset 24)
000+00:01:34.273 N 70: [emsesp] Thermostat(0x10) <- Boiler(0x08), UBAErrorMessage2(0x11), data: 18 (offset 36)
000+00:01:34.313 N 71: [emsesp] Boiler(0x08) -> Thermostat(0x10), UBAErrorMessage2(0x11), data: 32 55 00 D5 95 03 0D 16 3B 00 00 00 (offset 36)
000+00:01:34.644 N 72: [emsesp] Thermostat(0x10) <- Boiler(0x08), UBAErrorMessage2(0x11), data: 0C (offset 48)
000+00:01:34.665 N 73: [emsesp] Boiler(0x08) -> Thermostat(0x10), UBAErrorMessage2(0x11), data: 30 59 01 15 95 03 0D 16 33 00 00 00 (offset 48)
```

But anyway, it works - great! I'll be able to hook up a reset button in home assistant.

MichaelDvP, 2021-03-22:
> Then a whole bunch of messages like this:

This is only the thermostat queries the error-logs from boiler to show on demand. There are no change in settings.

MichaelDvP, 2021-03-22:
Afair @susisstrolch has checked and logged this before and discovered two commands, one for reset the maintenance message and one for the error, but sending reset to often will cause other errors. First we used the reset command as `call boiler reset error` and `call boiler reset maintenance`. Than we implemented in maintenance a reset subcommand `call boiler maintenance reset` and skip the reset command. The `set_reset()` is in the code, but the command is not registered. We can add it again, but i think a we should add a warning to wiki not use it if there is no error.

giovanne123, 2021-05-18:
> ```
> ems-esp:/# call system send "0B 08 05 00 5A"
> 000+00:01:30.570 N 44: [emsesp] Boiler(0x08) -> Me(0x0B), ?(0x05), data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
> ```
>
> It works! [...]

Thanks, today I also tested again (in ESP32/V3.1.1b0) when there is the need to press the Reset Button at the Boiler and it is working form me.  No need to go to the Boiler anymore :-)
Tested in console:
```
ems-esp32:$ su
Password:
006+08:27:00.042 N 0: [shell] su session opened on console
ems-esp32:# call system send "0B 08 05 00 5A"
006+08:27:24.737 N 1: [emsesp] Boiler(0x08) -> Me(0x0B), ?(0x05), data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

```

proddy, 2021-07-25:
@MichaelDvP is this a function we should bring over to ESP32 and make officially, just for completeness?

MichaelDvP, 2021-07-26:
It is already in emsesp32, [see here](https://github.com/emsesp/EMS-ESP32/blob/f97cdcb4d6132e3298e90613913d2a8c3184192a/src/devices/boiler.cpp#L95) [and here](https://github.com/emsesp/EMS-ESP32/blob/f97cdcb4d6132e3298e90613913d2a8c3184192a/src/devices/boiler.cpp#L1249-L1269)., also shown on the webpage without value. Also in [documentation](https://emsesp.github.io/docs/#/API?id=boiler).

proddy, 2021-07-26:
you're right, thanks! Closing this.

closed 2021-07-26 by proddy