emsesp/EMS-ESP #310: RC35: "temp" command changes day/night temperature instead of seltemp

kmarhold, 2020-01-26:
What happens:

When sending a `temp` command to the thermostat, it changes the day/night temperature depending on which mode is currently active. This change is then propagated to `seltemp`. Therefore it is not possible to set a temporarily different temperature as is possible with the physical thermostat.

What should happen:

`daytemp` and `nighttemp` should change the respective temperatures and `temp` should change the selected temperature directly via `seltemp`, not by changing the day/night temperature first.

Equipment:
RC35 Thermostat
Buderus Logamax GB132-24

proddy, 2020-01-26:
That makes sense. but I'm not sure how to directly set the selected temperature though. Any ideas?

MichaelDvP, 2020-01-26:
@proddy I think its in data-pos 37 (42 absolut) in the same telegram per hc (3D,47,51,5B).
see: https://emswiki.thefischer.net/doku.php?id=wiki:ems:telegramme

proddy, 2020-01-26:
@MichaelDvP nice find. So @kmarhold try 'send 0B 10 3D 25 1E`

for HC1, x25 is the offset 37 and 1E is 15 degrees

kmarhold, 2020-01-26:
I'm using bbqkees's board and have only been using MQTT so far, could you point me to some more info on how to send this command ?

proddy, 2020-01-26:
Go to the wiki. It's mentioned in the front page of this repo. https://github.com/proddy/EMS-ESP

MichaelDvP, 2020-01-26:
@proddy 
I have tried on RC35 hc2 (0x47), works as expected and changes only the seltemp.

proddy, 2020-01-26:
> @proddy
> I have tried on RC35 hc2 (0x47), works as expected and changes only the seltemp.

great, I'll modify the code.

proddy, 2020-02-02:
> @proddy
> I have tried on RC35 hc2 (0x47), works as expected and changes only the seltemp.

Hi @MichaelDvP,

I ran into a problem with the RC30 when setting the seltemp as reported in #318.  That thermostat doesn't support offset 37 to change the flow temp, but the rest is the same. Could you see if it works if you use offset 18 instead? Looking at the doc I think they are similar. So both these commands:

'send 0B 10 3D 25 1E`
and
'send 0B 10 3D 12 1E`

should do the same thing and set the temp to 15 degrees. Could you check if you have time?

MichaelDvP, 2020-02-03:
@proddy 0x18 seems to be the upper temperature of the heating curve (in my case 72°C). Writing to this parameter does not change it.

proddy, 2020-02-03:
it's 0x12 (18 decimal) and according to the [wiki](https://emswiki.thefischer.net/doku.php?id=wiki:ems:telegramme#wwbetriebsart) its something called the "design temperature". Is this what you tried?

MichaelDvP, 2020-02-05:
Ups, yes i tried 0x12 (double checked the log).

proddy, 2020-02-05:
ok, I merged the old tech from 1.9.4 so it should work correctly now.

closed 2020-02-05 by proddy