emsesp/EMS-ESP #311: Add support for Buderus Logamax Plus GB022-24/24K (DeviceID:0x08 ProductID:84 Version:01.09)

lendy007, 2020-01-28:
Hello,

I would like to ask to add support for Buderus Logamax Plus GB022-24/24K, see below. If I can help somehow to include it in product I am happy to help!

This unknown boiler device is also the reason why I am only receiving hc1 information on EMS-ESP? (no boiler data)

New EMS device recognized as a UBAMaster (DeviceID:0x08 ProductID:84 Version:01.                                                                      09)
New EMS device recognized as a Controller (DeviceID:0x09 ProductID:84 Version:01                                                                      .09)

Thank you

proddy, 2020-01-28:
yes, I'll add to the 1.9.5 dev version and you can upgrade via the web UI or download the firmware yourself.

lendy007, 2020-01-28:
Thank you very much, looking forward! :)

lendy007, 2020-01-28:
I have one additional question - is it possible if boiler is not "known" that setting thermostat isn't working properly? My RC30 is detected correctly (just boiler is unknown), but when I set thermostat temperature regardless if via telnet or via MQTT - nothing will happen on thermostat...

Thank you

**before:**
```
{"hc1":{"seltemp":10,"currtemp":10,"daytemp":16,"nighttemp":10,"holidayttemp":17,"heatingtype":1,"circuitcalctemp":31,"mode":"night"}}
```

**telnet command:**
```
thermostat temp 1 17.0
Setting new thermostat temperature to 17.0 for heating circuit 1 type 0 (0=auto,1=night,2=day,3=holiday)
```

**after:**
```
{"hc1":{"seltemp":10,"currtemp":10,"daytemp":16,"nighttemp":10,"holidayttemp":17,"heatingtype":1,"circuitcalctemp":31,"mode":"night"}}
```

proddy, 2020-01-28:
I tested on my RC20 and it works (I don't have an RC30). It can take a few seconds for the thermostat to respond and adjust to the new temperature. Try
`log v`
`thermostat temp 17`
and see if there are any errors. It should say something like "Write successful". Then do a `info` and see if the "Setpoint room temperature" has changed.

I've added your boiler. It's in 1.9.5b20

lendy007, 2020-01-28:
Thank you, for new build, I am able to see now boiler data and also control the temperature.

Unfortunately in this build I lost completely RC30 thermostat also for view. In version 1.9.4 after boot RC30 isn't recognized, but I am able to perform autodetect scan in telnet and then RC30 will appear. In 1.9.5b20 this isn't working...

**Autodetect in 1.9.5b20:**
```
Scanning for new EMS devices and attached external sensors...
Requesting type UBADevices(0x07) from dest 0x08
(21:54:31) Sending read of type 0x07 to 0x08, telegram: 0B 88 07 00 20 (CRC=A8)
(21:54:31) UBAMaster -> Me, type 0x07, telegram: 08 0B 07 00 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=7C) #data=15
Requesting type Version(0x02) from dest 0x08
Requesting type Version(0x02) from dest 0x09
(21:54:32) Sending read of type 0x02 to 0x08, telegram: 0B 88 02 00 20 (CRC=BC)
(21:54:32) UBAMaster -> Me, type 0x02, telegram: 08 0B 02 00 54 01 09 (CRC=29) #data=3
New EMS device recognized as a UBAMaster: Buderus Logamax Plus GB022 (DeviceID:0x08 ProductID:84 Version:01.09)
Requesting type UBAMonitorFast(0x18) from dest 0x08
Requesting type UBAMonitorSlow(0x19) from dest 0x08
Requesting type UBAParameterWW(0x33) from dest 0x08
Requesting type UBAParametersMessage(0x16) from dest 0x08
Requesting type UBATotalUptimeMessage(0x14) from dest 0x08
(21:54:32) Sending read of type 0x02 to 0x09, telegram: 0B 89 02 00 20 (CRC=B4)
(21:54:32) 0x09 -> Me, type 0x02, telegram: 09 0B 02 00 54 01 09 (CRC=69) #data=3
New EMS device recognized as a UBAMaster: Buderus Logamax Plus GB022 (DeviceID:0x09 ProductID:84 Version:01.09)
(21:54:32) Sending read of type 0x18 to 0x08, telegram: 0B 88 18 00 20 (CRC=D4)
(21:54:32) UBAMaster -> Me, type 0x18, telegram: 08 0B 18 00 1F 01 6B 64 19 09 11 25 00 00 BE 00 BE 80 00 FF FF FF 2D 48 00 00 00 01 00 (CRC=17) #data=25
(21:54:32) Sending read of type 0x19 to 0x08, telegram: 0B 88 19 00 20 (CRC=D0)
(21:54:32) UBAMaster -> Me, type 0x19, telegram: 08 0B 19 00 FF F6 01 7C 80 00 FF FF 04 64 (CRC=9F) #data=10
(21:54:32) Sending read of type 0x33 to 0x08, telegram: 0B 88 33 00 20 (CRC=78)
(21:54:32) UBAMaster -> Me, type 0x33, telegram: 08 0B 33 00 08 FF 28 00 00 00 00 00 00 00 FF (CRC=D6) #data=11
(21:54:33) Sending read of type 0x16 to 0x08, telegram: 0B 88 16 00 20 (CRC=EC)
(21:54:33) UBAMaster -> Me, type 0x16, telegram: 08 0B 16 00 FF 28 64 00 06 FA 0A 01 05 (CRC=76) #data=9
(21:54:33) Sending read of type 0x14 to 0x08, telegram: 0B 88 14 00 20 (CRC=E4)
(21:54:33) UBAMaster -> Me, type 0x14, telegram: 08 0B 14 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=5F) #data=28
(21:54:34) UBAMaster -> All, type 0x18, telegram: 08 00 18 00 1F 01 6B 64 19 09 11 25 00 00 BE 00 BE 80 00 FF FF FF 2D 48 00 00 00 01 00 (CRC=A4) #data=25
(21:54:34) UBAMaster -> All, type 0x34, telegram: 08 00 34 00 0A 00 BC 00 BC 00 00 00 01 00 (CRC=8F) #data=10
log nFetching data from EMS devices
Requesting type UBAMonitorFast(0x18) from dest 0x08
Requesting type UBAMonitorSlow(0x19) from dest 0x08
Requesting type UBAParameterWW(0x33) from dest 0x08
Requesting type UBAParametersMessage(0x16) from dest 0x08
Requesting type UBATotalUptimeMessage(0x14) from dest 0x08
(21:54:36) Sending read of type 0x18 to 0x08, telegram: 0B 88 18 00 20 (CRC=D4)

(21:54:36) UBAMaster -> Me, type 0x18, telegram: 08 0B 18 00 1F 01 6B 64 19 09 11 25 00 00 BC 00 BC 80 00 FF FF FF 2D 48 00 00 00 01 00 (CRC=30) #data=25
```

**Autodetect in 1.9.4:**
```
log v

System Logging set to Verbose
autodetect scan(21:57:55) UBAMaster -> All, type 0x18, telegram: 08 00 18 00 20 01 23 64 00 01 11 20 00 00 BC 00 BC 80 00 FF FF FF 30 59 00 00 00 00 00 (CRC=8E) #data=25
(21:57:55) UBAMaster -> All, type 0x34, telegram: 08 00 34 00 0A 00 BC 00 BC 00 00 00 01 00 (CRC=8F) #data=10

Started scanning the EMS bus for known devices
Requesting type Version(0x02) from dest 0x02
Requesting type Version(0x02) from dest 0x08
Requesting type Version(0x02) from dest 0x09
Requesting type Version(0x02) from dest 0x10
Requesting type Version(0x02) from dest 0x17
Requesting type Version(0x02) from dest 0x18
Requesting type Version(0x02) from dest 0x20
Requesting type Version(0x02) from dest 0x21
Requesting type Version(0x02) from dest 0x30
Requesting type Version(0x02) from dest 0x38
Requesting type Version(0x02) from dest 0x48
(21:57:55) Sending read of type 0x02 to 0x02, telegram: 0B 82 02 00 20 (CRC=EC)
(21:58:05) UBAMaster -> All, type 0x18, telegram: 08 00 18 00 20 01 23 64 00 01 11 20 00 00 BC 00 BC 80 00 FF FF FF 30 59 00 00 00 00 00 (CRC=8E) #data=25
(21:58:05) UBAMaster -> All, type 0x19, telegram: 08 00 19 00 FF F6 01 32 80 00 FF FF 01 64 (CRC=19) #data=10
(21:58:05) UBAMaster -> All, type 0x34, telegram: 08 00 34 00 0A 00 BC 00 BC 00 00 00 01 00 (CRC=8F) #data=10
[... 9 lines cut ...]
Requesting type RCTime(0x06) from dest 0x10
Publishing thermostat data via MQTT
(21:58:06) Sending read of type 0x02 to 0x17, telegram: 0B 97 02 00 20 (CRC=44)
```

**Try to set temperature on 1.9.4 with enabled logging:** (it failed, but don't know what can I do with it)
```
thermostat temp 11

Setting new thermostat temperature to 11.0 for heating circuit 1 type 0 (0=auto,1=night,2=day,3=holiday)
(22:03:51) Sending write of type 0xA7 to 0x10, telegram: 0B 10 A7 1C 16 (CRC=B0)
(22:03:51) Sending validate of type 0xA7 to 0x10, telegram: 0B 90 A7 1C 01 (CRC=C3)
-> Write failed. Compared set value 0x16 with received value of 0xE7
-> Write didn't work, retrying (1/2)...
[... cut ...]
(22:03:52) Sending write of type 0xA7 to 0x10, telegram: 0B 10 A7 1C 16 (CRC=B0)
(22:03:52) Sending validate of type 0xA7 to 0x10, telegram: 0B 90 A7 1C 01 (CRC=C3)
-> Write failed. Compared set value 0x16 with received value of 0xE7
-> Write failed. Giving up, removing from queue
(22:03:54) UBAMaster -> All, type 0x18, telegram: 08 00 18 00 20 01 0C 64 00 01 11 20 00 00 BC 00 BC 80 00 FF FF FF 30 59 00 00 00 00 00 (CRC=85) #data=25
(22:03:54) UBAMaster -> All, type 0x34, telegram: 08 00 34 00 0A 00 BC 00 BC 00 00 00 01 00 (CRC=8F) #data=10
```

proddy, 2020-01-31:
Can you try the latest 1.9.5b22 and report back?

lendy007, 2020-01-31:
Hey,

unfortunately still the same on 1.9.5b22 - boiler detected, but RC30 not :(

```
autodetect

Scanning for new EMS devices and attached external sensors...
Requesting type UBADevices(0x07) from dest 0x08
(11:44:10) Sending read of type 0x07 to 0x08, telegram: 0B 88 07 00 20 (CRC=A8)
(11:44:10) UBAMaster -> Me, type 0x07, telegram: 08 0B 07 00 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=7C) #data=15
Requesting type Version(0x02) from dest 0x08
Requesting type Version(0x02) from dest 0x09
(11:44:10) Sending read of type 0x02 to 0x08, telegram: 0B 88 02 00 20 (CRC=BC)
(11:44:10) UBAMaster -> Me, type 0x02, telegram: 08 0B 02 00 54 01 09 (CRC=29) #data=3
New EMS device recognized as a UBAMaster: Buderus Logamax Plus GB022 (DeviceID:0x08 ProductID:84 Version:01.09)
Requesting type UBAMonitorFast(0x18) from dest 0x08
Requesting type UBAMonitorSlow(0x19) from dest 0x08
Requesting type UBAParameterWW(0x33) from dest 0x08
Requesting type UBAParametersMessage(0x16) from dest 0x08
Requesting type UBATotalUptimeMessage(0x14) from dest 0x08
(11:44:10) Sending read of type 0x02 to 0x09, telegram: 0B 89 02 00 20 (CRC=B4)
(11:44:10) 0x09 -> Me, type 0x02, telegram: 09 0B 02 00 54 01 09 (CRC=69) #data=3
New EMS device recognized as a UBAMaster: Buderus Logamax Plus GB022 (DeviceID:0x09 ProductID:84 Version:01.09)
(11:44:10) Sending read of type 0x18 to 0x08, telegram: 0B 88 18 00 20 (CRC=D4)
(11:44:10) UBAMaster -> Me, type 0x18, telegram: 08 0B 18 00 21 01 76 64 19 09 11 25 00 00 8A 00 8A 80 00 FF FF FF 2D 48 00 00 00 01 00 (CRC=2F) #data=25
(11:44:11) Sending read of type 0x19 to 0x08, telegram: 0B 88 19 00 20 (CRC=D0)
(11:44:11) UBAMaster -> Me, type 0x19, telegram: 08 0B 19 00 00 32 01 89 80 00 FF FF 05 64 (CRC=CA) #data=10
(11:44:11) Sending read of type 0x33 to 0x08, telegram: 0B 88 33 00 20 (CRC=78)
(11:44:11) UBAMaster -> Me, type 0x33, telegram: 08 0B 33 00 08 FF 28 00 00 00 00 00 00 00 FF (CRC=D6) #data=11
(11:44:11) Sending read of type 0x16 to 0x08, telegram: 0B 88 16 00 20 (CRC=EC)
(11:44:11) UBAMaster -> Me, type 0x16, telegram: 08 0B 16 00 FF 28 64 00 06 FA 0A 01 05 (CRC=76) #data=9
(11:44:11) Sending read of type 0x14 to 0x08, telegram: 0B 88 14 00 20 (CRC=E4)
(11:44:11) UBAMaster -> Me, type 0x14, telegram: 08 0B 14 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=5F) #data=28
(11:44:12) UBAMaster -> All, type 0x18, telegram: 08 00 18 00 21 01 78 64 19 09 11 25 00 00 88 00 88 80 00 FF FF FF 2D 48 00 00 00 01 00 (CRC=E5) #data=25
(11:44:12) UBAMaster -> All, type 0x19, telegram: 08 00 19 00 00 32 01 87 80 00 FF FF 05 64 (CRC=CF) #data=10
(11:44:12) UBAMaster -> All, type 0x34, telegram: 08 00 34 00 0A 00 88 00 88 00 00 00 01 00 (CRC=2A) #data=10

```

proddy, 2020-01-31:
the duplicate product id was fixed in the latest build you tried. Could you do a `devices` too - I want to see if the RC30 pops up.

proddy, 2020-01-31:
I see the issue. When the Master is queried to show all connected EMS devices it comes back with
`telegram: 08 0B 07 00 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00`
The 0x03 is a bitmap saying 0x08 and 0x09 (00000011) are present. 0x08 is the boiler and 0x09 the onboard controller. There is nothing else on the bus, no thermostats...?

bbqkees, 2020-01-31:
maybe you can try to temporarily unplug the boiler and see if the thermostat is detected again after a few minutes.

lendy007, 2020-01-31:
That's strange, because when I downgrade to version 1.9.4, although I need to perform `autodetect scan` after each reboot, but it will detect RC30 after it... So this points me to some changes in code, not problem on a bus...

proddy, 2020-01-31:
In 1.9.4 I used a different technique to discover the EMS devices, basically a brute-force attack pinging all known devices from the EMS-ESP library and hoping one would respond. That's before I discovered this technique which is cleaner. You're the first to report back an issue so defintely something we should look into. When you perform a `devices` command it will ask the Master boiler for all registered clients on the bus. You should be seeing `0x08 0x09 0x010`. Can you try that? Mine looks like:

[image]

lendy007, 2020-01-31:
Mine like this:

[image]

lendy007, 2020-01-31:
I have partialy good news! 🥇 

I power cycled my boiler as @bbqkees recommended and now on version **1.9.5b22** I was able to see also RC30 in the list until I tried to set a temperature on thermostat via telnet... It wasn't working and after that system rebooted and now I don't see RC30 again (maybe once I power-cycle boiler again it will appear again).

proddy, 2020-01-31:
awesome! I'm actually happy it crashed - I've been trying to reproduce that bug (see #298). Now I have two reports. Still, I don't know the root cause but you can help me. Let's focus on that first. What was the exact command you entered into the terminal? Can you try again with log v and show when it crashes?

lendy007, 2020-01-31:
It took about 45 seconds after command executed to crash... Unfortunately after crash I don't see RC30... I always need to powercycle boiler to appear it...

```
log v

System Logging set to Verbose
thermostat temp (22:07:39) UBAMaster -> All, type 0x18, telegram: 08 00 18 00 1C 00 E0 64 00 01 11 20 00 00 8B 00 8B 80 00 FF FF FF 30 59 00 00 00 00 00 (CRC=0B) #data=25
(22:07:39) UBAMaster -> All, type 0x34, telegram: 08 00 34 00 0A 00 8D 00 8D 00 00 00 01 00 (CRC=38) #data=10
11.0

Setting new thermostat temperature to 11.0 for heating circuit 1 type 0 (0=auto,1=night,2=day,3=holiday)
(22:07:41) Sending write of type 0xA7 to 0x10, telegram: 0B 10 A7 1C 16 (CRC=B0)
(22:07:42) Sending validate of type 0xA7 to 0x10, telegram: 0B 90 A7 1C 01 (CRC=C3)
-> Write failed. Compared set value 0x16 with received value of 0xE7
-> Write didn't work, retrying (1/2)...
(22:07:42) Sending write of type 0xA7 to 0x10, telegram: 0B 10 A7 1C 16 (CRC=B0)
(22:07:42) Sending validate of type 0xA7 to 0x10, telegram: 0B 90 A7 1C 01 (CRC=C3)
-> Write failed. Compared set value 0x16 with received value of 0xE7
-> Write didn't work, retrying (2/2)...
(22:07:42) Sending write of type 0xA7 to 0x10, telegram: 0B 10 A7 1C 16 (CRC=B0)
(22:07:43) Sending validate of type 0xA7 to 0x10, telegram: 0B 90 A7 1C 01 (CRC=C3)
-> Write failed. Compared set value 0x16 with received value of 0xE7
-> Write failed. Giving up, removing from queue
(22:07:49) UBAMaster -> All, type 0x18, telegram: 08 00 18 00 1C 00 E0 64 00 01 11 20 00 00 8D 00 8D 80 00 FF FF FF 30 59 00 00 00 00 00 (CRC=62) #data=25
(22:07:49) UBAMaster -> All, type 0x34, telegram: 08 00 34 00 0A 00 8D 00 8D 00 00 00 01 00 (CRC=38) #data=10
thermostat temp 1 11.0

Setting new thermostat temperature to 11.0 for heating circuit 1 type 0 (0=auto,1=night,2=day,3=holiday)
(22:07:57) Sending write of type 0xA7 to 0x10, telegram: 0B 10 A7 1C 16 (CRC=B0)
(22:07:57) Sending validate of type 0xA7 to 0x10, telegram: 0B 90 A7 1C 01 (CRC=C3)
-> Write failed. Compared set value 0x16 with received value of 0xE7
-> Write didn't work, retrying (1/2)...
(22:07:58) Sending write of type 0xA7 to 0x10, telegram: 0B 10 A7 1C 16 (CRC=B0)
(22:07:58) Sending validate of type 0xA7 to 0x10, telegram: 0B 90 A7 1C 01 (CRC=C3)
-> Write failed. Compared set value 0x16 with received value of 0xE7
-> Write didn't work, retrying (2/2)...
(22:07:58) Sending write of type 0xA7 to 0x10, telegram: 0B 10 A7 1C 16 (CRC=B0)
(22:07:59) UBAMaster -> All, type 0x18, telegram: 08 00 18 00 1C 00 E0 64 00 01 11 20 00 00 8D 00 8D 80 00 FF FF FF 30 59 00 00 00 00 00 (CRC=62) #data=25
(22:07:59) UBAMaster -> All, type 0x19, telegram: 08 00 19 00 00 32 00 F0 80 00 FF FF 00 64 (CRC=B9) #data=10
(22:07:59) UBAMaster -> All, type 0x34, telegram: 08 00 34 00 0A 00 8D 00 8D 00 00 00 01 00 (CRC=38) #data=10
(22:07:59) Sending validate of type 0xA7 to 0x10, telegram: 0B 90 A7 1C 01 (CRC=C3)
-> Write failed. Compared set value 0x16 with received value of 0xE7
-> Write failed. Giving up, removing from queue
Fetching data from EMS devices
Requesting type RC30StatusMessage(0x41) from dest 0x10
Requesting type RC30Set(0xA7) from dest 0x10
Requesting type RCTime(0x06) from dest 0x10
Requesting type UBAMonitorFast(0x18) from dest 0x08
Requesting type UBAMonitorSlow(0x19) from dest 0x08
Requesting type UBAParameterWW(0x33) from dest 0x08
Requesting type UBAParametersMessage(0x16) from dest 0x08
Requesting type UBATotalUptimeMessage(0x14) from dest 0x08
(22:08:05) Sending read of type 0x41 to 0x10, telegram: 0B 90 41 00 20 (CRC=69)
(22:08:05) Thermostat -> Me, telegram: 10 0B 41 00 (CRC=2E)
(22:08:06) Sending read of type 0xA7 to 0x10, telegram: 0B 90 A7 00 20 (CRC=DA)
(22:08:06) Thermostat -> Me, telegram: 10 0B A7 00 (CRC=FB)
(22:08:06) Sending read of type 0x06 to 0x10, telegram: 0B 90 06 00 20 (CRC=6C)
(22:08:06) Thermostat -> Me, type 0x06, telegram: 10 0B 06 00 14 01 17 1F 06 20 04 00 (CRC=1E) #data=8
(22:08:06) Sending read of type 0x18 to 0x08, telegram: 0B 88 18 00 20 (CRC=D4)
(22:08:06) UBAMaster -> Me, type 0x18, telegram: 08 0B 18 00 1C 00 E0 64 00 01 11 20 00 00 8B 00 8B 80 00 FF FF FF 30 59 00 00 00 00 00 (CRC=B8) #data=25
(22:08:06) Sending read of type 0x19 to 0x08, telegram: 0B 88 19 00 20 (CRC=D0)
(22:08:06) UBAMaster -> Me, type 0x19, telegram: 08 0B 19 00 00 32 00 F0 80 00 FF FF 00 64 (CRC=17) #data=10
(22:08:07) Sending read of type 0x33 to 0x08, telegram: 0B 88 33 00 20 (CRC=78)
(22:08:07) UBAMaster -> Me, type 0x33, telegram: 08 0B 33 00 08 FF 28 00 00 00 00 00 00 00 FF (CRC=D6) #data=11
(22:08:07) Sending read of type 0x16 to 0x08, telegram: 0B 88 16 00 20 (CRC=EC)
(22:08:07) UBAMaster -> Me, type 0x16, telegram: 08 0B 16 00 FF 28 64 00 06 FA 0A 01 05 (CRC=76) #data=9
(22:08:07) Sending read of type 0x14 to 0x08, telegram: 0B 88 14 00 20 (CRC=E4)
(22:08:07) UBAMaster -> Me, type 0x14, telegram: 08 0B 14 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=5F) #data=28
(22:08:09) UBAMaster -> All, type 0x18, telegram: 08 00 18 00 1C 00 DD 64 00 01 11 20 00 00 8D 00 8D 80 00 FF FF FF 30 59 00 00 00 00 00 (CRC=8B) #data=25
(22:08:09) UBAMaster -> All, type 0x34, telegram: 08 00 34 00 0A 00 8D 00 8D 00 00 00 01 00 (CRC=38) #data=10
(22:08:09) Thermostat -> All, type 0x3E, telegram: 10 00 3E 00 00 00 14 00 5F 00 00 11 23 31 00 00 64 00 1C (CRC=77) #data=15
(22:08:10) Thermostat -> UBAMaster, type 0x35, telegram: 10 08 35 00 01 01 (CRC=00) #data=2

```

proddy, 2020-01-31:
hmm, something fishy is going on. I'll run some tests tomorrow. I'll see if I can borrow an RC30 from bbqkees too.

proddy, 2020-02-01:
From the telegram log it looks like your thermostat is not behaving like an RC30 as I would expect. The telegrams command to read the mode (0xA7) and read the data (0x41) come back blank when they are queried:

>(22:08:05) Sending read of type 0x41 to 0x10, telegram: 0B 90 41 00 20 (CRC=69)
(22:08:05) Thermostat -> Me, telegram: 10 0B 41 00 (CRC=2E)
(22:08:06) Sending read of type 0xA7 to 0x10, telegram: 0B 90 A7 00 20 (CRC=DA)
(22:08:06) Thermostat -> Me, telegram: 10 0B A7 00 (CRC=FB)

and this is why the writing is not working.

Can you find out exactly which make and model thermostat you have?
Also the output of this command: `send 0B 07 02 00 20`

lendy007, 2020-02-01:
This is exact model I have

[image]

proddy, 2020-02-01:
thanks @lendy007 . working on a fix...

proddy, 2020-02-02:
Can you see if the latest build 1.9.5b25 helps?

lendy007, 2020-02-02:
I have tried to update, but thermostat not recognized. When I will visit the house, I'll try to power cycle the boiler and try to control thermostat...

```
devices

These device IDs are on the EMS Bus: 0x08 0x09
and 2 were recognized by EMS-ESP as:
 UBAMaster: Buderus Logamax Plus GB022 (DeviceID:0x08 ProductID:84 Version:01.09)
 unknown?: unknown? (DeviceID:0x09 ProductID:84 Version:01.09)
```

lendy007, 2020-02-03:
Hello,

after power cycling boiler, I got all devices, but still not able to set thermostat temperature :-( Also detecting RC30 only on power-cycle of boiler also isn't very stable solution...

```
These device IDs are on the EMS Bus: 0x08 0x09
and 3 were recognized by EMS-ESP as:
 UBAMaster: Buderus Logamax Plus GB022 (DeviceID:0x08 ProductID:84 Version:01.09)
 unknown?: unknown? (DeviceID:0x09 ProductID:84 Version:01.09)
 Thermostat: RC30 (DeviceID:0x10 ProductID:67 Version:02.08)
```

```
thermostat temp 1 11

Setting new thermostat temperature to 11.0 for heating circuit 1 type 0 (0=auto,1=night,2=day,3=holiday)
(13:49:01) Sending write of type 0x3D to 0x10, telegram: 0B 10 3D 25 16 (CRC=98)
(13:49:02) Sending validate of type 0x3D to 0x10, telegram: 0B 90 3D 25 01 (CRC=EB)
-> Write failed. Compared set value 0x16 with received value of 0xF3
-> Write didn't work, retrying (1/2)...
(13:49:02) Sending write of type 0x3D to 0x10, telegram: 0B 10 3D 25 16 (CRC=98)
(13:49:02) Sending validate of type 0x3D to 0x10, telegram: 0B 90 3D 25 01 (CRC=EB)
-> Write failed. Compared set value 0x16 with received value of 0xF3
-> Write didn't work, retrying (2/2)...
(13:49:02) Sending write of type 0x3D to 0x10, telegram: 0B 10 3D 25 16 (CRC=98)
(13:49:03) Sending validate of type 0x3D to 0x10, telegram: 0B 90 3D 25 01 (CRC=EB)
-> Write failed. Compared set value 0x16 with received value of 0xF3
-> Write failed. Giving up, removing from queue
(13:49:05) UBAMaster -> All, type 0x18, telegram: 08 00 18 00 28 01 B1 64 19 09 11 25 00 00 83 00 83 80 00 FF FF FF 2D 48 00 00 00 01 00 (CRC=7A) #data=25
(13:49:05) UBAMaster -> All, type 0x34, telegram: 08 00 34 00 0A 00 83 00 83 00 00 00 01 00 (CRC=AE) #data=10
log(13:49:12) Thermostat -> All, type 0x3D, telegram: 10 00 3D 00 01 14 20 22 06 28 00 00 00 05 05 2D 01 01 04 4B 05 4B 01 00 3C FF 11 F6 05 02 02 (CRC=39) #data=27
 n(13:49:13) Thermostat -> All, type 0x3D, telegram: 10 00 3D 1B 00 02 0F 32 0A (CRC=47) #data=5

(13:49:14) Thermostat -> UBAMaster, type 0x07, telegram: 10 88 07 00 0F (CRC=2E) #data=1
```

proddy, 2020-02-03:
ok, I know the problem with the RC30 and have a fix coming up soon. Same as #318 

Also need to find out why your 0x09 controller is not appearing anymore, as a ? ?. Looking into that too.

lendy007, 2020-02-03:
Great! Much appreciated, you should have some "Donate button" on your page as it seems that you are the only one working on this open-source project and really in professional way like just very few of paid solutions do!!!

bbqkees, 2020-02-03:
There is a 'sponsor' button on the top of the page. That will lead you to PayPal.

lendy007, 2020-02-03:
Thank you @bbqkees! Wasn't aware there is some :)

lendy007, 2020-02-03:
Would it be possible, at least for the time testing builds, to add back also `autodetect scan` option in telnet so I don't need to drive to house every time I need to power cycle boiler?

proddy, 2020-02-03:
thanks @lendy007 for the kind donation! Yes, its been a long journey and its people like you that keep me going. Happy end-users! I've added back`autodetect scan` and also hopefully fixed the RC30 issue, in 1.9.5b26

lendy007, 2020-02-04:
Hello, I have tried 1.9.5b27 and it seems that via telnet also control of RC30 is working fine, need to verify if also physically on thermostat values are changed or mode switched, but anyway, thank you very much @proddy ! :)

Is there also some way how to avoid doing `autodetect scan` after every reboot of EMS gateway? Unfortunately RC30 isn't detected automatically and after reboot of EMS gateway I all the time need either to perform this command or power cycle boiler. Or is there some kind of "startup" command where I can put this command to be executed after each reboot of gateway?

Thank you

proddy, 2020-02-04:
It's very strange that your Master EMS device (the boiler) is not recognizing the thermostat. An normal `autodetect` should ask the Boiler which devices it knows about and this should include the thermostat. I haven't yet seen a case where this doesn't work. Need to think of how best to approach this.

lendy007, 2020-02-04:
I am very sorry to bother again and again, but I passed to next step and I am now trying to send MQTT message to be able to integrate EMS-ESP with OpenHAB smarthome system...

But when I publish MQTT with topic `ems-esp/thermostat_cmd_temp1` and some value, it will not perform anything on thermostat, it will remain on the same value. (Command `thermostat temp 1 11` via telnet is working fine).

[image]

proddy, 2020-02-04:
when you publish `ems-esp/thermostat_cmd_temp1` you should see a message pop up in the Telnet window.

But I think your problem is that you're missing the mqtt base in the topic. It's default to "home".

proddy, 2020-02-04:
> Hello, I have tried 1.9.5b27 and it seems that via telnet also control of RC30 is working fine, need to verify if also physically on thermostat values are changed or mode switched, but anyway, thank [...]

Ok, discussed with master KeesBBQ. We thought:
* `devices` shows the known devices recognized by EMS-ESP, and which are on/off e.g. the heating circuits
* `devices detect` will ask the Boiler/Master which devices he has connected and will also try and connect to them
* `devices detect force` will do a force check on all known devices in the EMS-ESP library and see which ones acknowledge. Then add them to the device list.
* `devices save` will save the known devices to EMS-ESP so on next reboot it can quickly connect to them.

will that work for you?

lendy007, 2020-02-04:
This sounds like an ideal solution... Great idea!

Regarding setting temperature via MQTT - I have changed base to home, so I am receiving data now at `home/ems-esp/thermostat_data`, but when I publish topic, no pop up in telnet and thermostat not changed :-( Am I doing something wrong according to pic below?

[image]

lendy007, 2020-02-04:
Yes, that MQTT problem was caused by using anonymous authentication to MQTT. I will update wiki about this.

lendy007, 2020-02-04:
Unfortunately after clicking on "Improve this article" I am getting 404 error :( Feels like I am really a trouble man :-D

proddy, 2020-02-04:
you're just the best testerI have! That was another bug in the wiki and shame on me for not testing the link. I think should work now.

proddy, 2020-02-05:
can we close this bug now? It was related to supporting the Logamax Plus. A new enhancement issue can be created for the device check as we discussed earlier.

lendy007, 2020-02-05:
Sure, I think a lot more than just pure support was fixed within this bug :) 
I will create enhancement issue for the device check.

Thank you once again for a big help!

closed 2020-02-05 by lendy007