emsesp/EMS-ESP #312: No Access Point called ems-esp

sand3rk, 2020-01-29:
**Question**
Is it correct that after flashing EMS-ESP the software does **_not_** go in Access Point mode with a SSID called ems-esp, if I don't have a interface board connected?

 I'm still waiting for my interface board to arrive, but thought I test out the software.

**Screenshots**
https://imgur.com/xWkJ1bT

**Device information**
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz

**Additional context**
I tried version 1.9.4 and 1.9.3

bbqkees, 2020-01-29:
You are flashing a bin file for the ESP8266 on an ESP32.
EMS-ESP runs on a ESP8266.

sand3rk, 2020-01-29:
Ok, I thought a ESP32 works too. I will wait for the ESP8266 to arrive. Thanks

closed 2020-01-29 by sand3rk