emsesp/EMS-ESP #313: boiler flowtemp and boiler wwtemp can't be set at Junkers Cerapur ZSB 24-5 C 23

grappa24, 2020-01-29:
flowtemp and wwtemp seem to be determined by boiler control and can't be overridden via ems. 
The commands are accepted, values were set for 1-2 sec but then the original hardware settings are recovered.

wwtemp is determined by a boiler control:
[image]

flowtemp seems to be determined by the "heating curve parameters" ...

here my info log from telnet:
[telnet log ems-esp - info.log](https://github.com/proddy/EMS-ESP/files/4126996/telnet.log.ems-esp.-.info.log)

**not really an issue for me, but I wan't to understand if its a bug or a feature :-)**

regards grappa24
(Dieter)

grappa24, 2020-01-29:
forgot the log of "boiler flowtemp xx":
[boiler flowtemp 66.log](https://github.com/proddy/EMS-ESP/files/4127474/boiler.flowtemp.66.log)

proddy, 2020-01-29:
So there must be a manual override switch somewhere for that type of boiler. If you manually change the flowtemp can you sniff what telegrams are being sent? Then it's just a case of reverse engineering and mimicking that behavior.

grappa24, 2020-01-29:
just changed the flowtemp first to 56 and then to 55 at teh boiler; as I see it there are telegrams in line 49-51 and 86-88 of the log:

`(03:10:15.285) Controller -> UBAMaster, type 0x33, telegram: 89 08 33 01 FF (CRC=29) #data=1
(03:10:15.310) Controller -> UBAMaster, type 0x33, telegram: 89 08 33 09 00 (CRC=C6) #data=1
(03:10:15.335) Controller -> UBAMaster, type 0x33, telegram: 89 08 33 02 38 (CRC=E8) #data=1`

`(03:10:40.609) Controller -> UBAMaster, type 0x33, telegram: 89 08 33 01 FF (CRC=29) #data=1
(03:10:40.634) Controller -> UBAMaster, type 0x33, telegram: 89 08 33 09 00 (CRC=C6) #data=1
(03:10:40.659) Controller -> UBAMaster, type 0x33, telegram: 89 08 33 02 37 (CRC=E7) #data=1`
[telnet log set wwtemp.log](https://github.com/proddy/EMS-ESP/files/4127799/telnet.log.set.wwtemp.log)
or do you need a raw log?

bbqkees, 2020-01-29:
So its likely a 'feature' because of the manual control knobs.

proddy, 2020-01-31:
@grappa24 it looks like you're changing the boilers DHW (Hot Water) temperature and not the flow temperature. Really depends on what you're trying to do. If you want to mimic that warm water manual dial then use `boiler wwtemp`. This is the 0x33 telegram examples you posted above.

grappa24, 2020-01-31:
@proddy you are right.  I tried to change the boilers DHW, but even with boiler wwtemp it does not work. But it's o.k. for me since I understand why (as bbqkees said ... manual control knobs).
Same with flowtemp; flowtemp is the product of a "complex" algorith (e.g. dependant of outside temp, ... ) and it makes no sense to set it directly. I use setting thermostat temperature instead.
So feel free to close this issue -:)

proddy, 2020-01-31:
ok, at least now we're all wiser! thanks for using the software and providing feedback.

closed 2020-01-31 by proddy