emsesp/EMS-ESP #316: second FR120 thermostat not fully recognized

tboschman, 2020-01-29:
Hi,

I have 2 FR120 thermostats on a junkers system. Both are discovered but only one shows up as a thermostat.
This one is listed as thermostat and reports the settings:
`Junkers FR120 (Version:55.01 ProductID:191 DeviceID:0x10)`

This one does not report any values :
`Junkers FR120 (Version:55.01 ProductID:191 DeviceID:0x19)`

Is it possible that the DeviceID:0x19 is missing as thermostat type?

proddy, 2020-01-29:
Currently the code only supports 1 single thermostat for reading and writing. You can set which one using the command `set master_thermostat <device id>` which is in release 1.9.5. There is already an enhancement to natively support more than one. It's not difficult to add, just on a long list of other feature requests.

proddy, 2020-05-16:
fixed in v2

closed 2020-05-16 by proddy