emsesp/EMS-ESP #317: Bosch CT200 not discovered in FW 1.9.4 (using BBQkees D1 mini)

martynhr, 2020-01-29:
Worcester/Bosch Greenstar 37CDI (2012) Boiler
Worcester/Bosch Easy Control CT200 Thermostat

Guy, relatively new to this so please bear with me.
Recently purchased a BBQKees Gateway Premium II for a bit more boiler info.

Home Assistant running prior to purchase and was using a Node MCU with 4 dallas Temp Sensors to monitor Boiler Temperatures. (Programmed with Arduino IDE

Initial setup was dead easy and EMS-ESP recognised the boiler and thermostat although I could not get and reading from the Thermostat. (FW version 1.9.2.)

Updated to 1.9.4 (to enable static ip config) and have now lost discovery of the Thermostat completely.
I have tried 1 , 2 and 3 (EMS, EMS+ and Junkers) Tx modes to no avail.
Listen mode is disabled

EMS Dashboard
[image]

System Status:
[image]

I have also tried FW 1.9.5b20 without success.

Any ideas?

**Question**
*A clear and concise description of what the problem/doubt is.*

**Screenshots**
*If applicable, add screenshots to help explain your problem.*

**Device information**
*Copy-paste here the information as it is outputted by the device. You can get this information by from the telnet session with the logging set to Verbose mode.*

**Additional context**
*Add any other context about the problem here.*

proddy, 2020-01-29:
Hi @martynhr and welcome. Could you try the latest 1.9.5 again and type `devices` and post the output and then the command `autodetect`. Let's see what is happening.

martynhr, 2020-01-29:
Hi, Thanks for the reply, (fast)
I'm trying not to sound think, but, where do I type devices in the web interface? I don't see any command console. Sorry. Like i said, this is new to me.

martynhr, 2020-01-29:
Ah, Telnet using Putty via USB I presume?

proddy, 2020-01-29:
haha, no worries. Check out the Wiki https://emsesp.github.io/docs/#/

martynhr, 2020-01-29:
Done:

[image]

martynhr, 2020-01-29:
Are these any use to you:

[image]

[image]

proddy, 2020-01-29:
I found the problem. It's happened before. I used to think all product IDs are unique but it turns out the Thermostat CT200 has the same ID (203) as the Buderus Logamax U122 and Junkers Cerapur. So its detecting your thermostat as another boiler!

I'll need to think on the best approach around this.

Also note writing doesn't work yet on the CT200. See #239

martynhr, 2020-01-29:
Well done, OK well i'll leave that with you for now.
Just an idea, would it be worth somehow linking the CT200 ID to a selected country somehow?? As we dont have the Buderus and Junkers in the UK.

bbqkees, 2020-01-30:
Bosch sells the same hardware in various countries under different brand names.

Perhaps it should be a check of device ID && product ID.

proddy, 2020-01-30:
exactly, this is what I've started coding. The UBA Master is always on DeviceID 0x08 so I've separated this logic out to scan for boilers first and then the other devices (thermostats, solar modules etc)

martynhr, 2020-01-30:
Good luck solving the problem. Let me know if you want me to test anything.

proddy, 2020-01-30:
@martynhr see if 1.9.5b21 does the trick. https://github.com/proddy/EMS-ESP/releases/tag/travis-dev-build

martynhr, 2020-01-31:
OK FW Updated to 1.9.5b21
Results: Using Tx1
[image]

proddy, 2020-01-31:
Can you just stick to Telnet and send the output of devices & autodetect with verbose logging.

rskallies, 2020-01-31:
As I own also a CT200 I'll provide my logs.  
[autodetect.txt](https://github.com/proddy/EMS-ESP/files/4141385/autodetect.txt)
[devices.txt](https://github.com/proddy/EMS-ESP/files/4141387/devices.txt)

proddy, 2020-01-31:
thanks @rskallies . With those dumps I was able to find the error. Hopefully this b23 release is better.

rskallies, 2020-02-01:
@proddy . Yes, it works now with b23. Great!

These device IDs are on the EMS Bus: 0x08 0x09 0x18
and 3 were recognized by EMS-ESP as:
 UBAMaster: Buderus GBx72/Nefit Trendline/Junkers Cerapur/Worcester Greenstar Si/27i (DeviceID:0x08 ProductID:123 Version:06.03)
 Controller: Controller (DeviceID:0x09 ProductID:152 Version:03.01)
 Thermostat: Bosch EasyControl CT200 (DeviceID:0x18 ProductID:203 Version:02.08)

proddy, 2020-02-01:
great! if you find any other problems please reopen this issue.

closed 2020-02-01 by proddy