emsesp/EMS-ESP #318: RC30 issue

jerome992, 2020-02-01:
Hello,

I have the following devices:

Sieger BK13,BK15/Nefit Smartline/Buderus GB1x2 (DeviceID:0x08 ProductID:64 Version:03.05)
BC10/RFM20 Receiver (DeviceID:0x09 ProductID:68 Version:02.02)
RC30 (DeviceID:0x10 ProductID:67 Version:02.08)

My RC30 is not working correctly, it is impossible to set the mode or temperature. The day and night temperature and the mode are not reported correctly.

My RC30 has the following features:
* can have multiple heating circuits,
* has different temperature registers for day, night and probably vacation,
* is responding with no data to  "send 0b 90 41 00 20 "
(10:21:56) Sending raw: 0B 90 41 00 20 
(10:21:56) Thermostat -> Me, telegram: 10 0B 41 00 (CRC=2E)
* is responding with no data to  "send 0b 90 A7 00 20 "
(10:26:17) Sending raw: 0B 90 A7 00 20 
(10:26:17) Thermostat -> Me, telegram: 10 0B A7 00 (CRC=FB)
* is responding to "send 0b 90 3d 00 20 "
(10:22:45) Sending raw: 0B 90 3D 00 20 
(10:22:45) Thermostat -> Me, type 0x3D, telegram: 10 0B 3D 00 01 22 26 22 02 28 01 02 00 05 05 2D 01 01 04 4B 05 41 01 00 3C FF 11 05 05 02 02 (CRC=A0) #data=27
* is responding to "send 0b 90 3e 00 20 " with only 15 bytes instead of more as the RC35.
(10:27:14) Sending raw: 0B 90 3E 00 20 
(10:27:14) Thermostat -> Me, type 0x3E, telegram: 10 0B 3E 00 04 02 26 00 CF 00 00 21 2F 3B 00 00 64 11 20 (CRC=65) #data=15
 

The RC30 behaves almost like a RC35, with an exception of 15 bytes for the message type 3e. After applying the the joined patch [patch_rc30.txt](https://github.com/proddy/EMS-ESP/files/4142392/patch_rc30.txt), it is working correctly:

Thermostat stats:
  Thermostat: RC30 (DeviceID:0x10 ProductID:67 Version:02.08)
  Thermostat time is 10:30:16 01/02/2020
  Heating Circuit 1
   Current room temperature: 20.7 C
   Setpoint room temperature: 19.0 C
   Day temperature: 19.0 C
   Night temperature: 17.0 C
   Vacation temperature: 17.0 C
   Mode is set to auto
   Day Mode is set to day

   Sincerely,
   Jerome

proddy, 2020-02-01:
Thanks for explaining that. I didn't realize the RC30 supported multiple thermostats. I've been puzzling why it wouldn't respond to 0x41 and 0x47 (See my earlier post from this morning https://github.com/proddy/EMS-ESP/issues/311#issuecomment-581006939).

Do you want to do a PR or shall I merge for you?

jerome992, 2020-02-01:
No problem,

you can merge it.

proddy, 2020-02-02:
There's an easier way to fix this without changing the code. I just changed one line in `ems_devices.h` from
    `{67, EMS_DEVICE_TYPE_THERMOSTAT, "RC30", EMS_DEVICE_FLAG_RC30}`
to 
    `{67, EMS_DEVICE_TYPE_THERMOSTAT, "RC30", EMS_DEVICE_FLAG_RC35}`

which will make thermostats with productID 67, like your Buderus RC30 behave like an RC35. What confused me before is the Nefit Moduline 400 which says its an RC30 but uses an older protocol as reported in #262

proddy, 2020-02-02:
@jerome992 could you check the latest 1.9.5b25 release?

jerome992, 2020-02-02:
The version 1.9.5b25 is not working because the RC30 has only 15 values in message type 0x3d:

System Logging set to Verbose
Setting new thermostat temperature to 20.50 for heating circuit 1 type 0 (0=auto,1=night,2=day,3=holiday)
Publishing boiler data via MQTT
Publishing hot water and heating states via MQTT
Publishing thermostat data via MQTT
(17:17:18) Sending write of type 0x3D to 0x10, telegram: 0B 10 3D 25 29 (CRC=A7)
(17:17:18) UBAMaster -> All, type 0x07, telegram: 08 00 07 00 0B 01 00 00 00 00 00 00 00 00 00 00 00 (CRC=F6) #data=13
(17:17:18) Sending validate of type 0x3D to 0x10, telegram: 0B 90 3D 25 01 (CRC=EB)
-> Write failed. Compared set value 0x29 with received value of 0xF3
-> Write didn't work, retrying (1/2)...
(17:17:19) UBAMaster -> All, type 0x18, telegram: 08 00 18 00 27 01 6E 64 14 09 01 25 60 80 00 02 29 01 45 00 4C 12 2D 48 00 C8 00 02 00 (CRC=21) #data=25
(17:17:19) UBAMaster -> All, type 0x34, telegram: 08 00 34 00 3C 02 29 02 29 21 00 00 03 00 02 35 AE 00 29 DC 00 (CRC=6A) #data=17
(17:17:19) Sending write of type 0x3D to 0x10, telegram: 0B 10 3D 25 29 (CRC=A7)
(17:17:19) Sending validate of type 0x3D to 0x10, telegram: 0B 90 3D 25 01 (CRC=EB)
-> Write failed. Compared set value 0x29 with received value of 0xF3
-> Write didn't work, retrying (2/2)...
(17:17:20) Sending write of type 0x3D to 0x10, telegram: 0B 10 3D 25 29 (CRC=A7)
(17:17:20) Sending validate of type 0x3D to 0x10, telegram: 0B 90 3D 25 01 (CRC=EB)
-> Write failed. Compared set value 0x29 with received value of 0xF3
-> Write failed. Giving up, removing from queue

proddy, 2020-02-02:
Not sure I understand. Yes the 0x3E is shorter by 1 byte than the RC35 but this shouldn't affect any of the working. 

In the latest build are you seeing all the values (in `info`)? Even if the write fails the validate is it working anyway, i.e. does it adjust the temperature?

I don't see what you changed in the code to make it work

jerome992, 2020-02-02:
between 1.9.4 and 1.9.5b25 you have add in ems.cpp:
EMS_TxTelegram.offset = EMS_OFFSET_RC35Set_seltemp;

EMS_OFFSET_RC35Set_seltemp is defined as 37 and not supported with the RC30.

jerome992, 2020-02-02:
I have made the patch against 1.9.4 and not the latest dev branch.

proddy, 2020-02-02:
trying one more thing, if that doesn't work I'll create a special flag.

bbqkees, 2020-02-03:
Is the RFM20 linking the RC30 to the bus? Perhaps that's why the messages are shorter.

proddy, 2020-02-03:
see if 1.9.5b26 works any better

jerome992, 2020-02-03:
* set  temperature => yes;
* set mode => no because;
* missing info for RC30 in info command.

you have forgotten to add EMS_DEVICE_FLAG_RC30N in the following lines:

diff --git a/src/ems-esp.cpp b/src/ems-esp.cpp
index 9a6c234..0793c38 100644
--- a/src/ems-esp.cpp
+++ b/src/ems-esp.cpp
@@ -196,7 +196,7 @@ _EMS_THERMOSTAT_MODE _getThermostatDayMode(uint8_t hc_num) {
         } else if (mode == 2) {
             thermoMode = EMS_THERMOSTAT_MODE_NIGHT;
         }
-    } else if (model == EMS_DEVICE_FLAG_RC35) {
+    } else if ((model == EMS_DEVICE_FLAG_RC35) || (model == EMS_DEVICE_FLAG_RC30N)) {
         if (mode == 0) {
             thermoMode = EMS_THERMOSTAT_MODE_NIGHT;
         } else if (mode == 1) {
@@ -435,7 +435,7 @@ void showInfo() {
 
                 // Render Day/Night/Holiday Temperature on RC35s
                 // there is no single setpoint temp, but one for day, night and vacation
-                if (model == EMS_DEVICE_FLAG_RC35) {
+                if ((model == EMS_DEVICE_FLAG_RC35) || (model == EMS_DEVICE_FLAG_RC30N)) {
                     if (EMS_Thermostat.hc[hc_num - 1].summer_mode) {
                         myDebug_P(PSTR("   Program is set to Summer mode"));
                     } else if (EMS_Thermostat.hc[hc_num - 1].holiday_mode) {
diff --git a/src/ems.cpp b/src/ems.cpp
index f0f2e6f..b1b5bcd 100644
--- a/src/ems.cpp
+++ b/src/ems.cpp
@@ -1995,6 +1995,7 @@ void ems_getThermostatValues() {
         ems_doReadCommand(EMS_TYPE_EasyStatusMessage, device_id);
         break;
     case EMS_DEVICE_FLAG_RC35:
+    case EMS_DEVICE_FLAG_RC30N:
         for (uint8_t hc_num = 1; hc_num <= EMS_THERMOSTAT_MAXHC; hc_num++) {
             if (hc_num == 1) {
                 statusMsg = EMS_TYPE_RC35StatusMessage_HC1;
@@ -2550,7 +2551,7 @@ void ems_setThermostatMode(uint8_t mode, uint8_t hc_num) {
         EMS_TxTelegram.type_validate      = EMS_TYPE_RC30Set;
         EMS_TxTelegram.comparisonPostRead = EMS_TYPE_RC30StatusMessage;
 
-    } else if (model == EMS_DEVICE_FLAG_RC35) {
+    } else if ((model == EMS_DEVICE_FLAG_RC35)|| (model == EMS_DEVICE_FLAG_RC30N)) {
         if (hc_num == 1) {
             EMS_TxTelegram.type               = EMS_TYPE_RC35Set_HC1;
             EMS_TxTelegram.comparisonPostRead = EMS_TYPE_RC35StatusMessage_HC1;
@@ -3207,4 +3208,4 @@ void ems_scanDevices() {
     for (uint8_t device_id : Device_Ids) {
         ems_doReadCommand(EMS_TYPE_Version, device_id);
     }
-}
\ No newline at end of file
+}

jerome992, 2020-02-03:
set temperature => yes;
set mode => no because;
missing info for RC30 in info command.

you have forgotten to add EMS_DEVICE_FLAG_RC30N in the following lines:

```
diff --git a/src/ems-esp.cpp b/src/ems-esp.cpp
index 9a6c234..0793c38 100644
--- a/src/ems-esp.cpp
+++ b/src/ems-esp.cpp
@@ -196,7 +196,7 @@ _EMS_THERMOSTAT_MODE _getThermostatDayMode(uint8_t hc_num) {
         } else if (mode == 2) {
             thermoMode = EMS_THERMOSTAT_MODE_NIGHT;
         }
-    } else if (model == EMS_DEVICE_FLAG_RC35) {
+    } else if ((model == EMS_DEVICE_FLAG_RC35) || (model == EMS_DEVICE_FLAG_RC30N)) {
         if (mode == 0) {
             thermoMode = EMS_THERMOSTAT_MODE_NIGHT;
         } else if (mode == 1) {
@@ -435,7 +435,7 @@ void showInfo() {
 
                 // Render Day/Night/Holiday Temperature on RC35s
                 // there is no single setpoint temp, but one for day, night and vacation
-                if (model == EMS_DEVICE_FLAG_RC35) {
+                if ((model == EMS_DEVICE_FLAG_RC35) || (model == EMS_DEVICE_FLAG_RC30N)) {
                     if (EMS_Thermostat.hc[hc_num - 1].summer_mode) {
                         myDebug_P(PSTR("   Program is set to Summer mode"));
                     } else if (EMS_Thermostat.hc[hc_num - 1].holiday_mode) {
diff --git a/src/ems.cpp b/src/ems.cpp
index f0f2e6f..b1b5bcd 100644
--- a/src/ems.cpp
+++ b/src/ems.cpp
@@ -1995,6 +1995,7 @@ void ems_getThermostatValues() {
         ems_doReadCommand(EMS_TYPE_EasyStatusMessage, device_id);
         break;
     case EMS_DEVICE_FLAG_RC35:
+    case EMS_DEVICE_FLAG_RC30N:
         for (uint8_t hc_num = 1; hc_num <= EMS_THERMOSTAT_MAXHC; hc_num++) {
             if (hc_num == 1) {
                 statusMsg = EMS_TYPE_RC35StatusMessage_HC1;
@@ -2550,7 +2551,7 @@ void ems_setThermostatMode(uint8_t mode, uint8_t hc_num) {
         EMS_TxTelegram.type_validate      = EMS_TYPE_RC30Set;
         EMS_TxTelegram.comparisonPostRead = EMS_TYPE_RC30StatusMessage;
 
-    } else if (model == EMS_DEVICE_FLAG_RC35) {
+    } else if ((model == EMS_DEVICE_FLAG_RC35)|| (model == EMS_DEVICE_FLAG_RC30N)) {
         if (hc_num == 1) {
             EMS_TxTelegram.type               = EMS_TYPE_RC35Set_HC1;
             EMS_TxTelegram.comparisonPostRead = EMS_TYPE_RC35StatusMessage_HC1;
@@ -3207,4 +3208,4 @@ void ems_scanDevices() {
     for (uint8_t device_id : Device_Ids) {
         ems_doReadCommand(EMS_TYPE_Version, device_id);
     }
-}
\ No newline at end of file
+}
```

jerome992, 2020-02-03:
the RC30 does not have the notion of seltemp. You can either set the temperature for the the day or for the night but you cannot set a temporary temperature until the next switching point.

proddy, 2020-02-03:
oops, pretty sloppy of me. Thanks for tidying up the code.

closed 2020-02-23 by proddy