emsesp/EMS-ESP #321: Missing Parameters

martynhr, 2020-02-03:
Hi,

I am missing a few parameters from my EMS Dashboard

[image]

Boiler Temperature showing as undefined and also Return Temperature showing the same.
(I have no retTemp in boiler_data)

Also I have no data from the Thermostat?

Any Ideas?

(Well done on fixing the CT200 discovery issue) :-)

proddy, 2020-02-03:
oh, that's not good. Can you go to telnet and see if Tx is working? Tx is needed to fetch the values as it makes requests to the EMS bus. Do a `info` and you should see something like
```
EMS Bus status:
  Bus is connected, protocol: Buderus
  Rx: # successful read requests=13078, # CRC errors=37
  Tx: Last poll=2.389 seconds ago, # successful write requests=6
```

See alsohttps://emsesp.github.io/docs/#/Troubleshooting?id=tx-is-just-not-working

martynhr, 2020-02-03:
[image]

proddy, 2020-02-03:
what boiler do you have ?

martynhr, 2020-02-03:
Worcester Greenstar 37cdi

proddy, 2020-02-03:
ok, we'll look into the boiler first, then the CT200. So for Junkers boiler temp and return temp haven't been implemented. Do you know what the expected values should be? For example are they available on an app, or in the thermostat or in the boiler display itself so we can compare?

What would help is to capture some of the telegrams. Via telnet, `log v` and leave it for 30-60 seconds.

martynhr, 2020-02-03:
Flow temp I can see from the boiler display but return temp I can only get from the Dallas temp sensor I have connected to the return pipe of the boiler. 
When I captured this screen shot the flow temp was 55.2C and the return 39.6C.

[image]

martynhr, 2020-02-04:
Hi,

Tried the info from the link you sent above. (https://emsesp.github.io/docs/#/Troubleshooting?id=tx-is-just-not-working)

Tried sending send 0B 88 02 00 20 and 8B 88 02 00 20 on all 3 tx modes still don't see any successful writes when typing info.

Any ideas?

proddy, 2020-02-04:
your tx_mode is fine, I can see the Tx working fine in the screenshot above with the command `send 8B 88 14 00 20`. You will also see this in the `info` command at the top.

I also see the telegram 0x0A which I use to get the thermostat values from a CT100 which has values 0 and 0 for the temps. Then I realized I never got around to adding support for the CT200 as I don't know where the values are! See https://github.com/proddy/EMS-ESP/issues/239#issuecomment-553857183

So you can help me. I just need the telegrams and `log t` will just log these thermostat telegrams to the console.

martynhr, 2020-02-04:
telegrams??

martynhr, 2020-02-04:
[image]

Also FYI: I updated to Version 1.9.5b27 earlier and lost the CT200.

proddy, 2020-02-04:
60 seconds will do. make a change on the thermostat, let me know what temp you set and then we'll see if we can find where the value is stored in the telegram.

martynhr, 2020-02-04:
Initially room temp 22.5 and setpoint 22.0
@ 20:29:58 setpoint changed to 19.5
20:30:51 setpoint changed to 25.0

[image]

martynhr, 2020-02-04:
FYI i installed the latest FW earlier Version 1.9.5b27 and lost the CT200 like before.
https://github.com/proddy/EMS-ESP/issues/317

proddy, 2020-02-04:
`devices` and `autodetect` don't show the CT200 anymore?

martynhr, 2020-02-04:
> `devices` and `autodetect` don't show the CT200 anymore?

didnt show correctly in the Dashboard as CT200, it came up like this

[image]

martynhr, 2020-02-04:
Here it is with FW version 1.9.5b27

[image]

proddy, 2020-02-04:
1.9.5 looks fine, it's finding the CT200. Although I couldn't see the temps in the log you sent. Could you leave it running again for 30 seconds, changing the temp but this time use verbose logging 'log v'. And just capture all the output into a text file and post it here as an attachment, thanks

proddy, 2020-02-04:
oh just saw the error in the CT200 detected with the latest 1.9.5. I'm making a quick fix now.

martynhr, 2020-02-04:
Initial temps Room 22.5C setpoint 20.0C
20:52:16 setpoint changed to 25.0C
20:52:44 setpoint changed to 20.0C
[ESM Log.txt](https://github.com/proddy/EMS-ESP/files/4155983/ESM.Log.txt)

martynhr, 2020-02-04:
> oh just saw the error in the CT200 detected with the latest 1.9.5. I'm making a quick fix now.

Well spotted! :-)

proddy, 2020-02-04:
not good news I'm afraid. The CT200 is not sending any data on the EMS bus accept the date/time and some flags for the boiler, and nothing on the temperatures. It's what I remembered from other Easy type devices. Since it's a closed system and connected to an App and the Internet there is no need to send its data along the EMS bus as the boiler doesn't really care. As with the other Easy devices the only way to extract the temps is to use XMPP and http requests. See https://emsesp.github.io/docs/#/wish-list

sorry can't help you with the thermostat. 

but we should find a solution to the return temp

martynhr, 2020-02-04:
How does the boiler know when to change heating temps if not sent by the EMS bus?

I've just been watching the log and noticed a corrupt telegram every now and again, could this be anything to do with it?

I presume you looking for something with the address 0x3d or 0x3e?

martynhr, 2020-02-04:
Try this:
Initially room temp 22.5 and setpoint 5C
21:34:39 setpoint 30C
21:35:09 setpoint 5C
21:35:46 setpoint 30C
21:35:31 setpoint 5C
[CT200.txt](https://github.com/proddy/EMS-ESP/files/4156283/CT200.txt)

proddy, 2020-02-04:
yes, the telegram with type 0x1A tells the boiler to start the burner, so with:

`(20:52:16) Thermostat -> UBAMaster, type 0x1A, telegram: 98 08 1A 00 50 54 (CRC=06) #data=2`

the performance of the heating circuit is set to 80% and the warm water to 84%. I suspect the heating curve is built into the thermostat. What we need are the actual setpoint temps.

proddy, 2020-02-04:
have a look at the telegram details on https://emswiki.thefischer.net/doku.php?id=wiki:ems:telegramme#wwbetriebsart

martynhr, 2020-02-04:
I'm sure the heating curve is in the thermostat. I had a Worcester Wave before, very similar design and operation, and the heating curve could be changed via various setting if you selected outdoor temperature control.

I'll have look at the telegram details tomorrow. My brain is fried now!!

proddy, 2020-02-08:
boiler temp and return temp are not in the EMS messages from W-B Boilers. Flowtemp is shown correctly I think. Support for CT200 is covered in a separate issue as a new feature. 

I'm closing this, unless there is still a problem I need to look into.

closed 2020-02-08 by proddy

martynhr, 2020-02-08:
Yes no probs mate

VBP8501, 2020-02-09:
@proddy : could you perhaps adapt the "supported devices" page in the wiki https://emsesp.github.io/docs/#/Supported-EMS-Devices to explain more precisely what "only read" includes for the Easycontrol thermostat. I've bought a bbqkees gateway to try to avoid http requests but seems that I'll need to go along that route anyhow. 

FYI, I did a "thermostat read" on all possible 1 byte hex values, only the already known messages gave a meaningful response. The rest was a disappointing Thermostat -> Me, telegram: 98 0B XX 00 (CRC=YY)

On the plus side, before I found this thread, I've spend some time trying to troubleshoot what happens and am impressed with the work you and the rest of the team has put into this application. Good job!

proddy, 2020-02-09:
thanks @VBP8501 . It is indeed a shame the 'easy' variants choose a closed system and we're unable to send EMS messages too it. I suspect there must be a back-door on these Thermostats to allow EMS in so the Bosch/Buderus/Nefit service guys can run diagnostics. I just haven't figured out how. Sending XMPP from the EMS-ESP would be a nice addition, but quite a bit of work.

I will improve the wiki to make it clearer on what I mean by "read only".

bbqkees, 2020-02-10:
I think its possible there is no backdoor at all.
The web KM200 only supports the RC30, RC35, RC300 and RC310. So although very popular, the web-enabled thermostats from Bosch are not on the list.
This might have to do with the fact that these thermostats are not writeable/sending parameters via the bus.
Its a strange design decision though.